espressif/esp-hosted #333: Getting MAC times out with esp_hosted_ng  &  ESP32-S3

sergey-suloev, 2024-02-10:
Hello, I am trying esp_hosted_ng  with a ESP32-S3 module. 
I made some minor changes to your code in order to load the linux SPI driver via device tree. 
See https://github.com/sergey-suloev/esp-hosted/tree/master/esp_hosted_ng/host.
But I can only receive a bootup event , the "get MAC" command gets timed out. 
Can you give any advice on how to proceed ?
Here is the log I am getting:

[    8.655700] esp32_spi:esp_init: esp_init: enter
[    8.660724] esp32_spi spi0.0: Probing ESP32 SPI-driver...
[    8.666350] esp32_spi spi0.0: Hardware reset
[    8.882552] esp32_spi spi0.0: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [30], SPI mode [0x02]
[    8.894346] sun50i-a64-pinctrl 1c20800.pinctrl: supply vcc-ph not found, using dummy regulator
[    9.110220] esp32_spi spi0.0: Probe success.
[    9.114789] esp32_spi:esp_init: esp_init: ret=0
[   10.346525] esp32_spi:process_esp_bootup_event: Received ESP bootup event
[   10.353395] esp32_spi:process_event_esp_bootup: Bootup Event tag: 3
[   10.359721] esp32_spi:esp_validate_chipset: Chipset=ESP32-S3 ID=09 detected over SPI
[   10.367700] esp32_spi:process_event_esp_bootup: Bootup Event tag: 2
[   10.374018] esp32_spi:process_event_esp_bootup: Bootup Event tag: 0
[   10.380408] esp32_spi:process_event_esp_bootup: Bootup Event tag: 1
[   10.386793] esp32_spi:process_fw_data: ESP chipset's last reset cause:
[   10.395922] esp32_spi:print_reset_reason: POWERON_RESET
[   10.402828] esp32_spi:check_esp_version: ESP Firmware version: 1.0.3
[   10.419129] esp32_spi:esp_reg_notifier: Driver init is ongoing
[   11.647084] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   15.582193] esp32_spi:wait_and_decode_cmd_resp: Command[3] timed out
[   15.588626] esp32_spi:cmd_get_mac: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   15.597219] esp32_spi:esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[   15.605108] esp32_spi:process_event_esp_bootup: network iterface init failed

mantriyogesh, 2024-02-13:
What is the git commit you use at both, ESP and host?
Also, can you please attach the ESP side log?

sergey-suloev, 2024-02-13:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbb8
load:0x403cc700,len:0x2e28
entry 0x403c98f0
I (27) boot: ESP-IDF d3c99ed3-dirty 2nd stage bootloader
I (27) boot: compile time Feb 13 2024 13:13:44
W (27) boot: Unicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (80) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 factory          factory app      00 00 00010000 00100000
I (95) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (102) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (110) boot: End of partition table
I (114) boot: Defaulting to factory image
I (119) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=22f10h (143120) map
I (153) esp_image: segment 1: paddr=00032f38 vaddr=3fc98f00 size=042b0h ( 17072) load
I (157) esp_image: segment 2: paddr=000371f0 vaddr=40374000 size=08e28h ( 36392) load
I (167) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=7ad90h (503184) map
I (257) esp_image: segment 4: paddr=000badb8 vaddr=4037ce28 size=0c080h ( 49280) load
I (277) boot: Loaded app from partition at offset 0x10000
I (277) boot: Disabling RNG early entropy source...
I (289) cpu_start: Unicore app
D (289) flash HPM: HPM with dummy, status is 3
I (289) cpu_start: Pro cpu up.
I (290) cpu_start: Single core mode
D (302) clk: RTC_SLOW_CLK calibration value: 4014566
I (311) cpu_start: Pro cpu start user code
I (311) cpu_start: cpu freq: 240000000 Hz
I (311) cpu_start: Application information:
I (314) cpu_start: Project name:     network_adapter
I (319) cpu_start: App version:      release/ng-v1.0.2-186-g240c10e-
I (326) cpu_start: Compile time:     Feb 13 2024 13:13:41
I (333) cpu_start: ELF file SHA256:  d96af30d5cac8f25...
I (339) cpu_start: ESP-IDF:          d3c99ed3-dirty
I (344) cpu_start: Min chip rev:     v0.0
I (349) cpu_start: Max chip rev:     v0.99 
I (354) cpu_start: Chip rev:         v0.1
D (358) memory_layout: Checking 6 reserved memory ranges:
D (364) memory_layout: Reserved memory range 0x3fc84000 - 0x3fc98f00
D (370) memory_layout: Reserved memory range 0x3fc98f00 - 0x3fca0998
D (377) memory_layout: Reserved memory range 0x3fceee34 - 0x3fcf0000
D (383) memory_layout: Reserved memory range 0x40374000 - 0x40388f00
D (389) memory_layout: Reserved memory range 0x600fe000 - 0x600fe010
D (396) memory_layout: Reserved memory range 0x600fffe8 - 0x60100000
D (402) memory_layout: Building list of available memory regions:
D (409) memory_layout: Available memory region 0x3fca0998 - 0x3fcb0000
D (415) memory_layout: Available memory region 0x3fcb0000 - 0x3fcc0000
D (422) memory_layout: Available memory region 0x3fcc0000 - 0x3fcd0000
D (428) memory_layout: Available memory region 0x3fcd0000 - 0x3fce0000
D (435) memory_layout: Available memory region 0x3fce0000 - 0x3fce9710
D (441) memory_layout: Available memory region 0x3fce9710 - 0x3fceee34
D (448) memory_layout: Available memory region 0x3fcf0000 - 0x3fcf8000
D (455) memory_layout: Available memory region 0x600fe010 - 0x600fffe8
I (461) heap_init: Initializing. RAM available for dynamic allocation:
D (469) heap_init: New heap initialised at 0x3fca0998
I (474) heap_init: At 3FCA0998 len 00048D78 (291 KiB): DRAM
[... 95 lines cut ...]
I (983) FW_SPI: Using SPI interface
I (987) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (996) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (1006) gdma: new group (0) at 0x3fcb6b14
D (1009) gdma: new pair (0,0) at 0x3fcb6b58
D (1014) gdma: new tx channel (0,0) at 0x3fcb6ae0
D (1018) gdma: new rx channel (0,0) at 0x3fcb6b78
D (1023) spi: SPI2 use iomux pins.
D (1027) intr_alloc: Connected src 21 to int 17 (cpu 0)
I (2033) FW_MAIN: Initial set up done
I (2033) main_task: Returned from app_main()
I (2156) FW_MAIN: INIT Interface command

D (2156) event: created task for loop 0x3fca5610
D (2156) event: created event loop 0x3fca5610
D (2159) wifi:filter: set rx policy=0
I (2162) wifi:mode : sta (34:85:18:96:6a:f4)
I (2166) wifi:enable tsf
D (2168) wifi:filter: set rx policy=1
D (2171) wifi:connect status 0 -> 0
D (2175) event: running task for loop 0x3fca5610
D (2179) event: running post WIFI_EVENT:2 with handler 0x4200a320 and context 0x3fca71c4 on loop 0x3fca5610
I (2189) FW_CMD: Wifi Sta mode set

I (2266) FW_MAIN: Get MAC command

sergey-suloev, 2024-02-13:
[   13.077390] esp32_spi:esp_init: esp_init: enter
[   13.082180] esp32_spi spi0.0: Probing ESP32 SPI-driver...
[   13.087688] esp32_spi spi0.0: Hardware reset
[   13.311364] esp32_spi spi0.0: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [30], SPI mode [0x02]
[   13.541040] esp32_spi spi0.0: Probe success.
[   13.545596] esp32_spi:esp_init: esp_init: ret=0
[   14.881558] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   15.140231] esp32_spi:process_esp_bootup_event: Received ESP bootup event
[   15.147159] esp32_spi:process_event_esp_bootup: Bootup Event tag: 3
[   15.153486] esp32_spi:esp_validate_chipset: Chipset=ESP32-S3 ID=09 detected over SPI
[   15.153498] esp32_spi:process_event_esp_bootup: Bootup Event tag: 2
[   15.153505] esp32_spi:process_event_esp_bootup: Bootup Event tag: 0
[   15.153510] esp32_spi:process_event_esp_bootup: Bootup Event tag: 1
[   15.153515] esp32_spi:process_fw_data: ESP chipset's last reset cause:
[   15.153520] esp32_spi:print_reset_reason: POWERON_RESET
[   15.153525] esp32_spi:check_esp_version: ESP Firmware version: 1.0.3
[   15.153899] esp32_spi:esp_reg_notifier: Driver init is ongoing
[   20.287113] esp32_spi:wait_and_decode_cmd_resp: Command[3] timed out
[   20.293534] esp32_spi:cmd_get_mac: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   20.302189] esp32_spi:esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[   20.309987] esp32_spi:process_event_esp_bootup: network iterface init failed

sergey-suloev, 2024-02-13:
@mantriyogesh  I am using your latest code for both ESP & host with minimal changes . Here is my fork https://github.com/sergey-suloev/esp-hosted/.

sergey-suloev, 2024-02-13:
Here is my hardware part that I'm xperimenting on
[image]

mantriyogesh, 2024-02-13:
Thanks for sharing the image.

The jumper wires are too big. Can you please use shorter wires, possibly <=10cm for **SPI** ?

Porting guide: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md

I quick visited, there is a lot of code change in your branch.

sergey-suloev, 2024-02-13:
@mantriyogesh  I understand that shorter cables are better, but lower freqs didn't work at all. I tried 20M, 10M, 4M, 2M but I wasn't even able to get bootup event

sergey-suloev, 2024-02-13:
> Thanks for sharing the image.
> 
> The jumper wires are too big. Can you please use shorter wires, possibly <=10cm for **SPI** ?
> 
> Porting guide: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md
> 
> I quick visited, there is a lot of code change in your branch.

It may seem a lot of changes but there are no principle changes. The problem with your master branch is that the driver isn't usable with  DT. In my environment everything is loaded via DT.

sergey-suloev, 2024-02-14:
@mantriyogesh  If you look at ESP log you will see that ESP has received the "get mac" command but it didn't respond to it. So MOSI is fine too.

sergey-suloev, 2024-02-14:
[   12.895943] esp32_spi:esp_init: esp_init: enter
[   12.900693] esp32_spi spi0.0: Probing ESP32 SPI-driver...
[   12.906293] esp32_spi spi0.0: Issue hardware reset...
[   13.141329] esp32_spi spi0.0: Done.
[   13.145083] esp32_spi spi0.0: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [30], SPI mode [0x01]
[   13.371343] esp32_spi spi0.0: Probe success.
[   13.375902] esp32_spi:esp_init: esp_init: ret=0
[   14.586536] esp32_spi spi0.0: data ready interrupt: irq=47
[   14.592131] esp32_spi spi0.0: handshake interrupt: irq=46
[   14.597605] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   14.605049] esp32_spi spi0.0: SPI transfer done.
[   14.609707] esp32_spi spi0.0: process_rx_buf
[   14.614117] esp32_spi spi0.0: process_rx_buf: new packet received
[   14.620291] esp32_spi spi0.0: read packet: len=38
[   14.625130] esp32_spi:process_esp_bootup_event: Received ESP bootup event
[   14.631999] esp32_spi:process_event_esp_bootup: Bootup Event tag: 3
[   14.638296] esp32_spi:esp_validate_chipset: Chipset=ESP32-S3 ID=09 detected over SPI
[   14.638305] esp32_spi:process_event_esp_bootup: Bootup Event tag: 2
[   14.638310] esp32_spi:process_event_esp_bootup: Bootup Event tag: 0
[   14.651832] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   14.652392] esp32_spi:process_event_esp_bootup: Bootup Event tag: 1
[   14.673486] esp32_spi:process_fw_data: ESP chipset's last reset cause:
[   14.673500] esp32_spi:print_reset_reason: POWERON_RESET
[   14.673505] esp32_spi:check_esp_version: ESP Firmware version: 1.0.3
[   14.673839] esp32_spi:esp_reg_notifier: Driver init is ongoing
[   14.751325] esp32_spi spi0.0: write packet: len=20
[   14.756384] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=0
[   14.763919] esp32_spi spi0.0: SPI transfer done.
[   14.768591] esp32_spi spi0.0: data ready interrupt: irq=47
[   14.774283] esp32_spi spi0.0: process_rx_buf
[   14.778619] esp32_spi spi0.0: process_rx_buf 252
[   14.783407] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   14.790747] esp32_spi spi0.0: SPI transfer done.
[   14.795506] esp32_spi spi0.0: handshake interrupt: irq=46
[   14.801033] esp32_spi spi0.0: process_rx_buf
[   14.805380] esp32_spi spi0.0: process_rx_buf 252
[   14.810092] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   14.817417] esp32_spi spi0.0: SPI transfer done.
[   14.822163] esp32_spi spi0.0: process_rx_buf
[   14.826597] esp32_spi spi0.0: process_rx_buf: new packet received
[   14.832812] esp32_spi spi0.0: read packet: len=20
[   14.891332] esp32_spi spi0.0: write packet: len=20
[   14.896174] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=0
[   14.903442] esp32_spi spi0.0: SPI transfer done.
[   14.908091] esp32_spi spi0.0: data ready interrupt: irq=47
[   14.913636] esp32_spi spi0.0: process_rx_buf
[   14.917920] esp32_spi spi0.0: process_rx_buf 252
[   14.922599] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   14.929782] esp32_spi spi0.0: SPI transfer done.
[   14.934468] esp32_spi spi0.0: process_rx_buf
[   14.938751] esp32_spi spi0.0: process_rx_buf 252
[   19.841337] esp32_spi:wait_and_decode_cmd_resp: Command[3] timed out
[   19.847763] esp32_spi:cmd_get_mac: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   19.856354] esp32_spi:esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[   19.864190] esp32_spi:process_event_esp_bootup: network iterface init failed

sergey-suloev, 2024-02-14:
@mantriyogesh  I added more logs to the driver and it's obvious that timeout happens because handshake interrupt doesn't happen.

sergey-suloev, 2024-02-14:
The following log happens when I set a SPI  freq < 30M:

[   12.819285] esp32_spi:esp_init: esp_init: enter
[   12.824103] esp32_spi spi0.0: Probing ESP32 SPI-driver...
[   12.829593] esp32_spi spi0.0: Issue hardware reset...
[   13.051357] esp32_spi spi0.0: Done.
[   13.055184] esp32_spi spi0.0: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [80], SPI mode [0x01]
[   13.281308] esp32_spi spi0.0: Probe success.
[   13.285864] esp32_spi:esp_init: esp_init: ret=0
[   14.502511] esp32_spi spi0.0: data ready interrupt: irq=47
[   14.508063] esp32_spi spi0.0: handshake interrupt: irq=46
[   14.513674] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   14.520753] esp32_spi spi0.0: SPI transfer done.
[   14.525442] esp32_spi spi0.0: process_rx_buf
[   14.529726] esp32_spi spi0.0: process_rx_buf 260

mantriyogesh, 2024-02-15:
https://github.com/espressif/esp-hosted/issues/333#issuecomment-1944447631

Something like wrong in setting the smaller freq.
From log,
```
[ 13.055184] esp32_spi spi0.0: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [80], SPI mode [0x01]
```
It appears that current frequency been set is 80MHz.

sergey-suloev, 2024-02-15:
Yes, you are right. A mistype in that freq value.
Here is the correct log for 10M

[   12.490743] esp32_spi:esp_init: esp_init: enter
[   12.495654] esp32_spi spi0.0: Probing ESP32 SPI-driver...
[   12.501180] esp32_spi spi0.0: Issue hardware reset...
[   12.542097] dwmac-sun8i 1c30000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   12.553235] dwmac-sun8i 1c30000.ethernet eth0: PHY [mdio_mux-0.1:01] driver [Generic PHY] (irq=POLL)
[   12.562546] dwmac-sun8i 1c30000.ethernet eth0: No Safety Features support found
[   12.569898] dwmac-sun8i 1c30000.ethernet eth0: No MAC Management Counters available
[   12.577614] dwmac-sun8i 1c30000.ethernet eth0: PTP not supported by HW
[   12.587446] dwmac-sun8i 1c30000.ethernet eth0: configuring for phy/mii link mode
[   12.721556] esp32_spi spi0.0: Done.
[   12.726270] esp32_spi spi0.0: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [10], SPI mode [0x01]
[   12.961522] esp32_spi spi0.0: Probe success.
[   12.966070] esp32_spi:esp_init: esp_init: ret=0
[   14.174986] esp32_spi spi0.0: data ready interrupt: irq=47
[   14.180553] esp32_spi spi0.0: handshake interrupt: irq=46
[   14.186056] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   14.195145] esp32_spi spi0.0: SPI transfer done.
[   14.199824] esp32_spi spi0.0: process_rx_buf
[   14.204198] esp32_spi spi0.0: process_rx_buf: new packet received
[   14.210340] esp32_spi spi0.0: read packet: len=38
[   14.215156] esp32_spi:process_esp_bootup_event: Received ESP bootup event
[   14.222035] esp32_spi:process_event_esp_bootup: Bootup Event tag: 3
[   14.228356] esp32_spi:esp_validate_chipset: Chipset=ESP32-S3 ID=09 detected over SPI
[   14.236172] esp32_spi:process_event_esp_bootup: Bootup Event tag: 2
[   14.242482] esp32_spi:process_event_esp_bootup: Bootup Event tag: 0
[   14.248763] esp32_spi:process_event_esp_bootup: Bootup Event tag: 1
[   14.255056] esp32_spi:process_fw_data: ESP chipset's last reset cause:
[   14.261612] esp32_spi:print_reset_reason: POWERON_RESET
[   14.266856] esp32_spi:check_esp_version: ESP Firmware version: 1.0.3
[   14.273622] esp32_spi:esp_reg_notifier: Driver init is ongoing
[   14.331860] esp32_spi spi0.0: write packet: len=20
[   14.339453] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=0
[   14.349411] esp32_spi spi0.0: SPI transfer done.
[   14.354149] esp32_spi spi0.0: data ready interrupt: irq=47
[   14.359721] esp32_spi spi0.0: process_rx_buf
[   14.364063] esp32_spi spi0.0: process_rx_buf 252
[   14.368715] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   14.377715] esp32_spi spi0.0: SPI transfer done.
[   14.382429] esp32_spi spi0.0: handshake interrupt: irq=46
[   14.387905] esp32_spi spi0.0: process_rx_buf
[   14.387919] esp32_spi spi0.0: process_rx_buf 252
[   14.396938] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   14.407357] esp32_spi spi0.0: SPI transfer done.
[   14.412127] esp32_spi spi0.0: process_rx_buf
[   14.416433] esp32_spi spi0.0: process_rx_buf: new packet received
[   14.422650] esp32_spi spi0.0: read packet: len=20
[   14.481520] esp32_spi spi0.0: write packet: len=20
[   14.486363] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=0
[   14.495335] esp32_spi spi0.0: SPI transfer done.
[   14.499990] esp32_spi spi0.0: data ready interrupt: irq=47
[   14.505528] esp32_spi spi0.0: process_rx_buf
[   14.509823] esp32_spi spi0.0: process_rx_buf 252
[   14.514492] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   14.523389] esp32_spi spi0.0: SPI transfer done.
[   14.528027] esp32_spi spi0.0: process_rx_buf
[   14.532329] esp32_spi spi0.0: process_rx_buf 252
[   14.642005] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   19.441512] esp32_spi:wait_and_decode_cmd_resp: Command[3] timed out
[   19.447917] esp32_spi:cmd_get_mac: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   19.456498] esp32_spi:esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[   19.464270] esp32_spi:process_event_esp_bootup: network iterface init failed

sergey-suloev, 2024-02-15:
I think ESP side must have more extensive logging

mantriyogesh, 2024-02-15:
I can send you to enable the logging, specially at the transport level (both ESP and host),
but the most imp thing is to trim the wire length.

Also do you happen to have logic analyzer at your side?

sergey-suloev, 2024-02-15:
@mantriyogesh I already changed wires to 9cm, there is no change in behavior.

mantriyogesh, 2024-02-15:
Do you use
1. correct input power rated adapter?
Insufficient power can lead to incorrect peripheral behaviour.
2. is your Linux allowed to power save? if so, can you please disable it?

As I understand, the SPI is unreliably working, with some messages reaching and some not, which could be classic issue of interoperability of spi timing. but to confirm, logic analyzer dump can be useful.

Let me provide some debug logs to introduce in both esp and host.

sergey-suloev, 2024-02-15:
#
# CPU Power Management
#

#
# CPU Frequency scaling
#
CONFIG_CPU_FREQ=y
CONFIG_CPU_FREQ_GOV_ATTR_SET=y
CONFIG_CPU_FREQ_GOV_COMMON=y
CONFIG_CPU_FREQ_STAT=y
CONFIG_CPU_FREQ_DEFAULT_GOV_PERFORMANCE=y
# CONFIG_CPU_FREQ_DEFAULT_GOV_POWERSAVE is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_USERSPACE is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_ONDEMAND is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_CONSERVATIVE is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_SCHEDUTIL is not set
CONFIG_CPU_FREQ_GOV_PERFORMANCE=y
CONFIG_CPU_FREQ_GOV_POWERSAVE=m
CONFIG_CPU_FREQ_GOV_USERSPACE=m
CONFIG_CPU_FREQ_GOV_ONDEMAND=y
CONFIG_CPU_FREQ_GOV_CONSERVATIVE=m
CONFIG_CPU_FREQ_GOV_SCHEDUTIL=y

#
# CPU frequency scaling drivers
#
CONFIG_CPUFREQ_DT=y
CONFIG_CPUFREQ_DT_PLATDEV=y
# CONFIG_ARM_ALLWINNER_SUN50I_CPUFREQ_NVMEM is not set
# end of CPU Frequency scaling

#
# CPU Idle
#
# CONFIG_CPU_IDLE is not set
# end of CPU Idle
# end of CPU Power Management

sergey-suloev, 2024-02-15:
> Do you use
>
> 1. correct input power rated adapter?
>    Insufficient power can lead to incorrect peripheral behaviour.
> 2. is your Linux allowed to power save? if so, can you please disable it?
>
> As I [...]

Why do you think so ?  Look at this post again https://github.com/espressif/esp-hosted/issues/333#issuecomment-1944431087

mantriyogesh, 2024-02-15:
base: 4f75cfcaf22831c9ebd0800bb30cb3a62d7ba6d2
[0001-Debug-logs-uncompiled-added.patch](https://github.com/espressif/esp-hosted/files/14294250/0001-Debug-logs-uncompiled-added.patch)

mantriyogesh, 2024-02-15:
I thought, 

```
[ 14.592131] esp32_spi spi0.0: handshake interrupt: irq=46
```

is handshake interrupt?

mantriyogesh, 2024-02-15:
above debug patch is uncompiled one, so you might have to fix and build, if any build issue.
base is your branch commit latest.

esp and host both has changes.
basically, whatever at transport (rx/tx) is expected to print.

the esp is sending the msg, but it is not understood by the host, due to spi timings.
most likely this will confirm, that the spi incoming message at host is bit shifted, but to confirm nicely, you would need logic analyzer.

If the logs have shown the output is bit shifted, you might have to resolve your porting for spi timings.
check porting guide again, as these are scenarios already covered there.

We work with different kind of hosts and these interop spi timing issues are common.
Once you fix, it will always work.

sergey-suloev, 2024-02-15:
> I thought,
> 
> ```
> [ 14.592131] esp32_spi spi0.0: handshake interrupt: irq=46
> ```
> 
> is handshake interrupt?

yes, it is an interrupt

mantriyogesh, 2024-02-15:
if you get into 

```
if (offset != sizeof(struct esp_payload_header)) {
            dev_dbg(&spi_ctx->spi->dev, "%s %u\n", __func__, __LINE__);
            return -EINVAL;
    }
```
Anyway, it means that your timings are not correct.

What is offset received? it will 1 (or 2?) bit left or right shifted.
anyway, logic analyzer?

sergey-suloev, 2024-02-15:
> if you get into
>
> ```
> if (offset != sizeof(struct esp_payload_header)) {
>             dev_dbg(&spi_ctx->spi->dev, "%s %u\n", __func__, __LINE__);
>             return -EINVAL;
>     }
> ```
>
> Anyway, it [...]

offset isn't a problem anymore, it happened only on 80MHz (I set this freq by mistake)

mantriyogesh, 2024-02-16:
Have you tried https://github.com/espressif/esp-hosted/issues/333#issuecomment-1945701242?

sergey-suloev, 2024-02-17:
here is my capture with logic analyzer at SPI freq 1MHz
https://drive.google.com/file/d/1Isq1rs5Kpysx5t9fVQevwqLsxdciqiHc/view?usp=sharing

mantriyogesh, 2024-02-17:
Please check https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L545

Logic.
Alternatively, try FG once with correct freq.

sergey-suloev, 2024-02-17:
> Please check https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L545
> 
> Logic. Alternatively, plz try FG once with correct freq.

do you mean I can try  FG firmware with NG host driver ??

sergey-suloev, 2024-02-17:
[   13.134574] esp32_spi spi0.0: Probing ESP32 SPI-driver...
[   13.151091] esp32_spi spi0.0: Issue hardware reset...
[   13.391377] esp32_spi spi0.0: Done.
[   13.395161] esp32_spi spi0.0: ESP32 peripheral is registered to SPI bus [0], chip select [0], SPI Clock [10], SPI mode [0x02]
[   13.406798] esp32_spi spi0.0: handshake interrupt: irq=46
[   13.412271] esp32_spi spi0.0: data ready interrupt: irq=47
[   13.417870] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   13.426877] esp32_spi spi0.0: SPI transfer done.
[   13.431580] esp32_spi spi0.0: process_rx_buf
[   13.435880] esp32_spi spi0.0: process_rx_buf: header->if_type(15) >= ESP_MAX_IF(5)
[   13.631367] esp32_spi spi0.0: Probe success.
[   13.635928] esp32_spi:esp_init: ret=0
[   13.896556] esp32_spi spi0.0: data ready interrupt: irq=47
[   13.902117] esp32_spi spi0.0: handshake interrupt: irq=46
[   13.907624] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   13.916593] esp32_spi spi0.0: SPI transfer done.
[   13.921240] esp32_spi spi0.0: handshake interrupt: irq=46
[   13.926688] esp32_spi spi0.0: process_rx_buf
[   13.930972] esp32_spi spi0.0: process_rx_buf: header->if_type(5) >= ESP_MAX_IF(5)
[   13.938513] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=1
[   13.947434] esp32_spi spi0.0: SPI transfer done.
[   13.952102] esp32_spi spi0.0: handshake interrupt: irq=46
[   13.957529] esp32_spi spi0.0: process_rx_buf
[   13.961839] esp32_spi spi0.0: process_rx_buf: wrong offset 1361 != 12
[   13.968311] esp32_spi spi0.0: work handler: trans_ready=1, rx_pending=0

sergey-suloev, 2024-02-17:
@mantriyogesh  I added "hs disable" to NG and it worked. Thanks a lot!

closed 2024-02-17 by sergey-suloev

mantriyogesh, 2024-02-18:
Btw, you may optionally optimise your freq of operation, by increasing spi clock in 1M steps.
Higher the freq, better the performance.
You could reach till 40m, by the time your setup respects interoperability and spi timings.
Proper PCB will be much robust than jumper cables setup.

sergey-suloev, 2024-02-18:
> Btw, you may optionally optimise your freq of operation, by increasing spi clock in 1M steps. Higher the freq, better the performance. You could reach till 40m, by the time your setup respects interoperability and spi timings. Proper PCB will be much robust than jumper cables setup.

Ok, I will try increase the freq later. At the moment it 100% stable at 30M and any lower freq too.