espressif/esp-hosted #336: I need to use ESP32-C3 or ESP32 as the wifi module for other Linux hosts

qinyunliu, 2024-02-18:
I need to use ESP32-C3 or ESP32 as the wifi module for other Linux hosts!
<esp-hosted-ng>  Use method on other ARM development board
[image]
[image]

mantriyogesh, 2024-02-18:
You can choose one from FG or NG,  whichever suits your development workflow.
Please let us know if you have followed the complete documentation.

qinyunliu, 2024-02-18:
I have chosen the NG solution，I modified Makefiel to generate esp32_spi.ko，
insmod esp32_spi.ko   Only one line of information is displayed
"ep32_spi:spi__dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI clock [10]"
I don't know if this is loading successfully,"ifconfig -a " The port number is not shown  wpa_supplicant  and bluez Can't call it

kapilkedawat, 2024-02-19:
Hi @qinyunliu Please recheck the connections and provide full host and slave logs.

qinyunliu, 2024-02-20:
[root@buildroot]:/home/drivers$:insmod esp32_spi.ko 
esp32_spi: loading out-of-tree module taints kernel.
Division by zero in kernel.
CPU: 0 PID: 1200 Comm: insmod Tainted: G           O    4.4.289 #4
Hardware name: NUC980
Backtrace: 
[<c0012de0>] (dump_backtrace) from [<c0012fcc>] (show_stack+0x18/0x1c)
 r6:00000000 r5:c3b5a400 r4:c3ace210 r3:00000000
[<c0012fb4>] (show_stack) from [<c0487b6c>] (dump_stack+0x20/0x28)
[<c0487b4c>] (dump_stack) from [<c0012f34>] (__div0+0x18/0x20)
[<c0012f1c>] (__div0) from [<c01a01b8>] (Ldiv0+0x8/0x10)
[<c023624c>] (nuc980_qspi0_update_state) from [<c0236394>] (nuc980_qspi0_setup+0x20/0xfc)
 r6:00000000 r5:c3ace210 r4:c3b5a400 r3:c3ace000
[<c0236374>] (nuc980_qspi0_setup) from [<c02330a4>] (spi_setup+0xdc/0x160)
 r5:00000000 r4:c3b5a400
[<c0232fc8>] (spi_setup) from [<c02331a4>] (spi_add_device+0x7c/0x134)
 r6:00000000 r5:c3ace000 r4:c3b5a400 r3:00000000
[<c0233128>] (spi_add_device) from [<c023344c>] (spi_new_device+0x7c/0xb8)
 r7:bf00824c r6:00000000 r5:c385bd50 r4:c3b5a400
[<c02333d0>] (spi_new_device) from [<bf005924>] (esp_init_interface_layer+0x15c/0x47c [esp32_spi])
 r6:00000000 r5:bf00824c r4:00000000 r3:00000009
[<bf0057c8>] (esp_init_interface_layer [esp32_spi]) from [<bf00b198>] (esp_init+0x198/0x1ec [esp32_spi])
 r7:ffffffe0 r6:bf008198 r5:00000000 r4:bf008160
[<bf00b000>] (esp_init [esp32_spi]) from [<c00095c8>] (do_one_initcall+0x88/0x1f0)
 r8:bf00b000 r7:c3ba7064 r6:c3ba3f00 r5:c05f7b80 r4:c05f7b80
[<c0009540>] (do_one_initcall) from [<c04868b0>] (do_init_module+0x60/0x1c0)
 r10:c3ba7048 r9:bf008000 r8:00000001 r7:c3ba7064 r6:c3ba36a0 r5:00000001
 r4:bf008000
[<c0486850>] (do_init_module) from [<c006f698>] (load_module+0x1a24/0x1d48)
 r6:c3ba7040 r5:00000001 r4:c385bf3c
[<c006dc74>] (load_module) from [<c006fa90>] (SyS_init_module+0xd4/0x130)
 r10:00000000 r9:004bb008 r8:c385a000 r7:c4924204 r6:00000000 r5:b6eb420c
 r4:0000e204
[<c006f9bc>] (SyS_init_module) from [<c000fce0>] (ret_fast_syscall+0x0/0x3c)
 r9:c385a000 r8:c000fec8 r7:00000080 r6:004bb008 r5:b6ea6008 r4:beef6f23
esp32_spi:spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [0]

qinyunliu, 2024-02-20:
sorry,Apart from this log, there is no other information

mantriyogesh, 2024-02-20:
1. Have you checked the [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) ?
2. setup picture?
3. ESP side logs?
4. From log,
```
esp32_spi:spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [0]
```
your current clock doesn't seem to be in effect.
Please follow porting guide, to first get your SPI and GPIOs verified to be working.

kangpei1, 2024-04-28:
@mantriyogesh hi, i used esp32 as the wifi module for  Linux, i  choice ng  spi driver,but ,i get this log for linux, "add_network_iface() failure",can you hele me or give me some advice for this question, Looking forward to your reply。
[image]

mantriyogesh, 2024-04-29:
From your current set of logs looks, your SPI pins are configured fine and bootup event is received, but failure reason in ESP log might be the key.

Also could you please first help to assess your transport (Tx and Rx both directions)?
For this, you can run raw [transport test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md)

It is worthwhile to cross-check your porting against [Porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) for any frequently faced issues.

kangpei1, 2024-04-29:
> From your current set of logs looks, your SPI pins are configured fine and bootup event is received, but failure reason in ESP log might be the key.
>
> Also could you please first help to assess your [...]

Thank you very much for your reply. Through log analysis, it was found that my Linux host did not receive an interrupt, but there was an interrupt generated through the oscilloscope. My Linux host did not enter the interrupt function.

kangpei1, 2024-04-29:
[image]
This is my esp32 log

mantriyogesh, 2024-05-10:
Hello @kangpei1 ,

Anything to be done from our side further?
If the driver works for you, please feel free to close this issue.

kangpei1, 2024-05-10:
> Hello @kangpei1 ,
> 
> Anything to be done from our side further? If the driver works for you, please feel free to close this issue.

thank you for you help,the drivers works

kangpei1, 2024-05-10:
> Hello @kangpei1 ,
> 
> Anything to be done from our side further? If the driver works for you, please feel free to close this issue.
I cant  close this issue,Not created by me

mantriyogesh, 2024-05-10:
Oh! I got confused on this.
@qinyunliu , Are you still blocked on this ?

QuiGonRazor, 2024-07-19:
Hello,
I'm experiencing the same issue.

I'm trying to use the ESP32C3 as wifi/bt module. I've downloaded the last commit and compiled both kernel driver and ESP's firmware. 

I've added a lot of comments to understand what's happening and it seems it is able to receive only bootup events. 
After it receives only zeroes

Below an output of the kernel driver

`[ 2384.918061] esp32_spi: esp_reset: Triggering ESP reset.
[ 2385.176641] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [2],chip select [0], SPI Clock [30]
[ 2385.332717] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.338331] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.348134] esp32_spi: esp_spi_work: SPI Transaction ENTRY
[ 2385.348176] esp32_spi: esp_spi_work: SPI Transaction GPIOS DONE HANDSHAKE_PIN:0 SPI_DATA_READY_PIN:0
[ 2385.352372] esp32_spi: esp_spi_work: SPI Transaction DONE
[ 2385.361428] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.361441] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.373767] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.373777] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.394695] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.394704] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.412170] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.417769] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.424928] esp32_spi: spi_init: SPI. DONE 0
[ 2385.432601] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.438200] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.451877] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.457479] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.471916] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.477515] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.492102] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.497704] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.511885] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.517485] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.531941] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.537538] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.551620] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.557215] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.571781] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.577377] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.591662] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.597261] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.611555] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.617154] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.629917] esp32_spi: esp_spi_work: SPI Transaction ENTRY
[ 2385.629961] esp32_spi: esp_spi_work: SPI Transaction GPIOS DONE HANDSHAKE_PIN:0 SPI_DATA_READY_PIN:0
[ 2385.633993] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.634007] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.657130] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.657140] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.671605] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.678135] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.691558] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.697154] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.711535] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.717136] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.731615] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.737215] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.751679] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.757275] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.771808] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.777410] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.791551] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.797151] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.811660] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.817254] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[ 2385.831890] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY IRQ
[ 2385.837485] esp32_spi: spi_data_ready_interrupt_handler: SPI. DATA READY OK
[... 80 lines cut ...]
[ 2387.528329] esp32_spi: esp_cfg80211_add_iface: STEP4
[ 2387.533078] esp32_spi: esp_cfg80211_add_iface: Updated priv[0] to c5ff6d80
[ 2387.540086] esp32_spi: cmd_init_interface: INIT INTERFACE QUEUEING
[ 2387.546144] esp32_spi: wait_and_decode_cmd_resp: Adapter type:2 bus_type:spi
[ 2387.576770] esp32_spi: esp_cmd_work: Processing Command [0x1]
[... cut ...]
[ 2387.757656] esp32_spi: process_rx_buf: SPI PROCESS RX
[ 2387.761526] esp32_spi: esp_spi_work: SPI Transaction DONE
[ 2397.926479] esp32_spi: wait_and_decode_cmd_resp: Command[0x1] timed out
[ 2397.937087] esp32_spi: cmd_init_interface: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[ 2397.946299] esp32_spi: esp_cfg80211_add_iface: FAIL INIT_INTERFACE
[ 2397.957587] esp32_spi: esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[ 2397.963967] esp32_spi: process_event_esp_bootup: network iterface init failed`

mantriyogesh, 2024-07-19:
1. Can you please let us know the host SoC, CPU, and kernel version details?
2. Does it work if you add
https://github.com/espressif/esp-hosted/blob/da21e362d5bced2e8ed9bc4eadb9cffb1d6c1be1/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L619
and associate code from FG?

QuiGonRazor, 2024-07-19:
Hello,
below the details

Architecture:        armv7l
Byte Order:          Little Endian
CPU(s):              1
On-line CPU(s) list: 0
Thread(s) per core:  1
Core(s) per socket:  1
Socket(s):           1
Vendor ID:           ARM
Model:               5
Model name:          Cortex-A7
Stepping:            r0p5
CPU max MHz:         650.0000
CPU min MHz:         650.0000
BogoMIPS:            48.00
Flags:               half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm

Linux stm32mp151a-mx 5.10.10 #1 SMP PREEMPT Wed Mar 10 11:42:35 UTC 2021 armv7l armv7l armv7l GNU/Linux

I'm gonna try point 2

QuiGonRazor, 2024-07-19:
@mantriyogesh Which code should I try? Are you suggesting to copy FG code into NG driver?

mantriyogesh, 2024-07-19:
3. Can you also send us the picture, how you connect the C3 with STM32MP ?
4. Before doing (2) above, can you please comment: https://github.com/espressif/esp-hosted/blob/da21e362d5bced2e8ed9bc4eadb9cffb1d6c1be1/esp_hosted_ng/host/main.c#L311
this should be simple to try.

Regarding, your latest comment, I meant, try (4) first. It is safe to keep (4) in the code, unless you get stable solution.
Details of (2):
There are some Linux hosts, with which we observe race condition on SPI. Host runs back to back SPI (wrongly), because they observe the Handshake was high from earlier transaction, just due to some timing issue (only happens for some hosts).
So, you can try to add **selective** code to 'disable handshake when CS is asserted (enabled, which means generally set to 0)' from FG to NG.
Only  spi_slave_api.c register_hs_disable_pin() function to be added in NG and called similar place as that of FG.

Do not even attempt NG module with FG firmware or vice versa, it would not work.
Also use latest master of NG at both sides, ESP and host.

QuiGonRazor, 2024-07-19:
I confirm you I'm using NG for both kernel ESP driver and firmware. The repo has been cloned 2 days ago I think it is updated.

I've already tried to comment the spi clock adjusting with no luck

I'm gonna try the point 2. 

Not sure if I can share an image, it is a prototype. I'm gonna check and see what I can do

mantriyogesh, 2024-07-19:
> Not sure if I can share an image, it is a prototype. I'm gonna check and see what I can do

In case you are using the jumper cables to evaluate, Just ensure that the cable lengths are minimal <6cm, with good quality and points, and equal lengths.

> I've already tried to comment the spi clock adjusting with no luck

for clock adjustments, the module param, `clockspeed` is useful.
Irrespective of the module param, comment the function, `esp_adjust_spi_clock()` as above.

mantriyogesh, 2024-07-19:
Also, if you have logic analyser, would really be helpful to understand the case.
Is the Linux based on allwinner CPU?

QuiGonRazor, 2024-07-19:
I've found the issue using the logical analyzer. 
I had to add a further line into the device tree to manage correctly the chip select.

Now the wlan0 interface is up, it communicates and I can see the hci0 bluetooth device. Today and next week I'll proceed with a stress test for both connections.

Thank you for your support. For me this issue is solved

closed 2024-07-19 by mantriyogesh