espressif/esp-hosted #338: esp_hosted_fg ：the module crashed in ap mode

wky19971002, 2024-02-19:
hi，It has been some time since our last question ，we refer to your opinion to modify the program，and now esp32 can work!
But now we have a new problem，esp32 works properly without connecting to the Internet，once we connect to the Internet，
esp32 crashed quickly，it seems crashed in SPI，and cannot be restarted.

After some time the esp32 printed the following log:
[14:13:05.560] W (433760) wifi:pp q full: 7-3fcd5490
[14:13:05.560] E (433760) wifi:fail to post timer, type=12
[14:13:07.544] W (435760) wifi:pp q full: 7-3fcd5490
[14:13:07.544] E (435760) wifi:fail to post timer, type=12
[14:13:09.542] W (437760) wifi:pp q full: 7-3fcd5490
[14:13:09.542] E (437760) wifi:fail to post timer, type=12
[14:13:11.542] W (439760) wifi:pp q full: 7-3fcd5490
[14:13:11.542] E (439760) wifi:fail to post timer, type=12
[14:13:13.542] W (441760) wifi:pp q full: 7-3fcd5490
[14:13:13.542] E (441760) wifi:fail to post timer, type=12
[14:13:15.541] W (443760) wifi:pp q full: 7-3fcd5490
[14:13:15.541] E (443760) wifi:fail to post timer, type=12
[14:13:17.541] W (445760) wifi:pp q full: 7-3fcd5490
[14:13:17.557] E (445760) wifi:fail to post timer, type=12

This is the full log:
[14:05:51.612] ESP-ROM:esp8684-api2-20220127
[14:05:51.612] Build:Ja
[14:05:51.628] n 27 2022ESP-ROM:esp8684-api2-20220127
[14:05:51.787] Build:Jan 27 2022
[14:05:51.977] rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
[14:05:51.977] SPIWP:0xee
[14:05:51.977] mode:DIO, clock div:1
[14:05:51.977] load:0x3fcd5c80,len:0x16c4
[14:05:51.977] load:0x403acb70,len:0x974
[14:05:51.977] load:0x403aeb70,len:0x297c
[14:05:51.977] entry 0x403acb70
[14:05:51.977] [0;32mI (32) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader[0m
[14:05:51.977] [0;32mI (32) boot: compile time Feb 19 2024 14:03:16[0m
[14:05:51.977] [0;32mI (32) boot: chip revision: v1.0[0m
[14:05:51.977] [0;32mI (38) boot.esp32c2: MMU Page Size  : 32K[0m
[14:05:51.977] [0;32mI (45) boot.esp32c2: SPI Speed      : 60MHz[0m
[14:05:51.977] [0;32mI (52) boot.esp32c2: SPI Mode       : DIO[0m
[14:05:51.977] [0;32mI (59) boot.esp32c2: SPI Flash Size : 2MB[0m
[14:05:51.977] [0;32mI (67) boot: Enabling RNG early entropy source...[0m
[14:05:51.977] [0;32mI (75) boot: Partition Table:[0m
[14:05:51.977] [0;32mI (81) boot: ## Label            Usage          Type ST Offset   Length[0m
[14:05:51.977] [0;32mI (92) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[14:05:51.977] [0;32mI (103) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[14:05:51.977] [0;32mI (115) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[14:05:51.977] [0;32mI (126) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[14:05:51.977] [0;32mI (138) boot:  4 ota_1            OTA app          00 11 00100000 000f0000[0m
[14:05:51.977] [0;32mI (150) boot: End of partition table[0m
[14:05:51.977] [0;32mI (156) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=282d8h (164568) map[0m
[14:05:51.977] [0;32mI (206) esp_image: segment 1: paddr=00038300 vaddr=3fcae330 size=024c0h (  9408) load[0m
[14:05:52.042] [0;32mI (209) esp_image: segment 2: paddr=0003a7c8 vaddr=40380000 size=05850h ( 22608) load[0m
[14:05:52.042] [0;32mI (221) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=5f2dch (389852) map[0m
[14:05:52.042] 
[14:05:52.551] [0;32mI (617) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[14:05:52.551] [0;32mI (631) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[14:05:52.551] [0;32mI (645) SPI_DRIVER: SPI Ctrl:1 mode: 3, Freq:ConfigAtHost
[14:05:52.551] GPIOs: MOSI: 7, MISO: 2, CS: 10, CLK: 6 HS: 4 DR: 5
[14:05:52.551] [0m
[14:05:52.551] [0;32mI (662) SPI_DRIVER: Hosted SPI queue size: Tx:10 Rx:10[0m
[14:05:52.551] [0;32mI (673) NETWORK_ADAPTER: Initial set up done[0m
[14:05:52.551] [0;32mI (678) slave_rpc: event ESPInit[0m
[14:05:52.551] [0;32mI (685) slave_sleep: gpio wakeup source [GPIO 10] with wakeup level [low] is ready[0m
[14:05:52.551] [0;32mI (697) pm: Frequency switching config: CPU_MAX: 120, APB_MAX: 80, APB_MIN: 13, Light sleep: ENABLED[0m
[14:05:52.551] [0;32mI (712) slave_sleep: PM configured with Min_Freq 13MHz Max_Freq 120MHz
[14:05:52.551] [0m
[14:05:52.551] [0;32mI (723) slave_sleep: FreeRTOS tickless idle
[14:05:52.551] [0m
[14:05:52.551] [0;32mI (731) main_task: Returned from app_main()[0m
[14:05:52.551] 
[14:05:54.997] [0;32mI (3213) slave_rpc: Resp_MSGId for req[0x116] is [0x216][0m
[14:05:55.186] [0;32mI (3213) slave_rpc: Received Req [0x116][0m
[14:05:55.186] [0;32mI (3214) pp: pp rom version: de8c800[0m
[14:05:55.186] [0;32mI (3219) net80211: net80211 rom version: de8c800[0m
[14:05:55.186] I (3229) wifi:wifi driver task: 3fcbfb54, prio:23, stack:3584, core=0
[14:05:55.186] I (3239) wifi:wifi firmware version: 91b9630
[14:05:55.186] I (3243) wifi:wifi certification version: v7.0
[14:05:55.186] I (3249) wifi:config NVS flash: enabled
[14:05:55.186] I (3255) wifi:config nano formating: disabled
[14:05:55.186] I (3261) wifi:Init data frame dynamic rx buffer num: 32
[14:05:55.186] I (3269) wifi:Init static rx mgmt buffer num: 5
[14:05:55.186] I (3275) wifi:Init management short buffer num: 32
[14:05:55.186] I (3282) wifi:Init dynamic tx buffer num: 32
[14:05:55.186] I (3288) wifi:Init static tx FG buffer num: 2
[14:05:55.186] I (3294) wifi:Init static rx buffer size: 1600
[14:05:55.186] I (3301) wifi:Init static rx buffer num: 10
[14:05:55.186] I (3307) wifi:Init dynamic rx buffer num: 32
[14:05:55.186] [0;32mI (3314) wifi_init: rx ba win: 6[0m
[14:05:55.186] [0;32mI (3319) wifi_init: tcpip mbox: 32[0m
[14:05:55.186] [0;32mI (3325) wifi_init: udp mbox: 6[0m
[14:05:55.186] [0;32mI (3331) wifi_init: tcp mbox: 6[0m
[14:05:55.186] [0;32mI (3337) wifi_init: tcp tx win: 5744[0m
[14:05:55.186] [0;32mI (3344) wifi_init: tcp rx win: 5744[0m
[14:05:55.186] [0;32mI (3350) wifi_init: tcp mss: 1440[0m
[14:05:55.186] [0;32mI (3356) wifi_init: WiFi IRAM OP enabled[0m
[14:05:55.186] [0;32mI (3363) wifi_init: WiFi RX IRAM OP enabled[0m
[14:05:55.186] [0;32mI (3371) wifi_init: WiFi SLP IRAM OP enabled[0m
[14:05:55.186] [0;32mI (3417) slave_rpc: Resp_MSGId for req[0x104] is [0x204][0m
[14:05:55.266] [0;32mI (3417) slave_rpc: Received Req [0x104][0m
[14:05:55.266] [0;32mI (3418) slave_rpc: Set wifi mode 2 [0m
[14:05:55.266] [0;32mI (3447) slave_rpc: Resp_MSGId for req[0x118] is [0x218][0m
[14:05:55.266] [0;32mI (3447) slave_rpc: Received Req [0x118][0m
[14:05:55.266] [0;32mI (3447) phy_init: phy_version 270,39d177a,Oct 13 2023,16:18:10[0m
[14:05:55.266] I (3484) wifi:mode : softAP (08:3a:8d:45:2a:2d)
[14:05:55.323] I (3484) wifi:Total power save buffer number: 16
[14:05:55.323] I (3485) wifi:Init max length of beacon: 752/752
[14:05:55.323] I (3487) wifi:Init max length of beacon: 752/752
[14:05:55.323] I (3494) wifi:Set ps type: 1, coexist: 0
[14:05:55.323] [0;32mI (3500) slave_rpc: softap started[0m
[14:05:55.323] [0;32mI (3506) slave_rpc: Sending Wi-Fi event [12][0m
[14:05:55.323] [0;32mI (3541) slave_rpc: Resp_MSGId for req[0x101] is [0x201][0m
[14:05:55.413] [0;32mI (3541) slave_rpc: Received Req [0x101][0m
[14:05:55.413] [0;32mI (3541) slave_rpc: Get softap mac address[0m
[14:05:55.413] [0;32mI (3548) slave_rpc: mac [08:3a:8d:45:2a:2d] [0m
[14:05:55.413] 8 3a 8d 45 2a 2d
[14:05:55.413] [0;32mI (3582) slave_rpc: Resp_MSGId for req[0x101] is [0x201][0m
[14:05:55.413] [0;32mI (3582) slave_rpc: Received Req [0x101][0m
[14:05:55.413] [0;32mI (3582) slave_rpc: Get softap mac address[0m
[14:05:55.413] [0;32mI (3589) slave_rpc: mac [08:3a:8d:45:2a:2d] [0m
[14:05:55.413] 8 3a 8d 45 2a 2d
[14:06:17.567] I (25752) wifi:station: b0:dc:ef:11:6b:bf join, AID=1, bgn, 20
[14:06:17.567] [0;32mI (25753) slave_rpc: station b0:dc:ef:11:6b:bf join, AID=1[0m
[14:06:17.567] rpc_evt_ap_staconn_conn_disconn event:14
[14:06:17.567] W (25798) wifi:<ba-add>idx:2 (ifx:1, b0:dc:ef:11:6b:bf), tid:0, ssn:10, winSize:64

We hope to get your help!

wky19971002, 2024-02-20:
Can anyone please answer my question

mantriyogesh, 2024-02-20:
Sure. Apology for the delay..
I am trying to better understand the issue, first.

> once we connect to the Internet

1. Are you trying to give internet to softap? Not understood clearly.
2. Can you please share the all the steps you have used at Linux along with their output?
3. While testing, can you please enable extra ESP logging:
- A. ESP logging to 'Debug'
[image]
- B. Enable 'Print debug messages from WPA Supplicant'
[image]

Basically, can you please attach the full textual logs in (2) commands, outputs to lead/reproduce the problem
(3) ESP full log and dmesg full log from boot on Linux

wky19971002, 2024-02-20:
1.yes, I'm trying to give internet to softap.
2. ESP logging to 'Debug' : While testing, if ESP logging to 'Debug', It has some effect on the performance of esp32, So I had to lower the log level,that gets network traffic into esp32.

mantriyogesh, 2024-02-20:
1. How?
2. Still waiting for logs all logs. I am trying to get to know what state it fails, yet

wky19971002, 2024-02-21:
1.We used the 4-g module as the host,with esp32-c2 chip , with spi connections.After connecting to ap we can get the correct ip address,spi works properly.But after the 4g module connect to the Internet, the network traffic enters esp32 through spi,this is when esp32 will crash.
2.
[09:27:38.361] ESP-ROM:esp8684-api2-20220127
[09:27:38.378] Build:Jan 27 2022
[09:27:38.378] rst:0x1 (POWEESP-ROM:esp8684-api2-20220127
[09:27:38.735] Build:Jan 27 2022
[09:27:38.735] rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
[09:27:38.735] SPIWP:0xee
[09:27:38.735] mode:DIO, clock div:1
[09:27:38.735] load:0x3fcd5c80,len:0x16c4
[09:27:38.735] load:0x403acb70,len:0x974
[09:27:38.735] load:0x403aeb70,len:0x297c
[09:27:38.735] entry 0x403acb70
[09:27:38.735] [0;32mI (32) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader[0m
[09:27:38.735] [0;32mI (32) boot: compile time Feb 20 2024 17:03:27[0m
[09:27:38.735] [0;32mI (32) boot: chip revision: v1.0[0m
[09:27:38.735] [0;32mI (38) boot.esp32c2: MMU Page Size  : 32K[0m
[09:27:38.735] [0;32mI (45) boot.esp32c2: SPI Speed      : 60MHz[0m
[09:27:38.735] [0;32mI (52) boot.esp32c2: SPI Mode       : DIO[0m
[09:27:38.735] [0;32mI (59) boot.esp32c2: SPI Flash Size : 2MB[0m
[09:27:38.735] [0;32mI (67) boot: Enabling RNG early entropy source...[0m
[09:27:38.735] [0;32mI (75) boot: Partition Table:[0m
[09:27:38.735] [0;32mI (81) boot: ## Label            Usage          Type ST Offset   Length[0m
[09:27:38.735] [0;32mI (92) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[09:27:38.735] [0;32mI (103) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[09:27:38.735] [0;32mI (115) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[09:27:38.735] [0;32mI (126) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[09:27:38.735] [0;32mI (138) boot:  4 ota_1            OTA app          00 11 00100000 000f0000[0m
[09:27:38.735] [0;32mI (150) boot: End of partition table[0m
[09:27:38.735] [0;32mI (156) esp_image: segment 0: paddr=00010020 vaddr=3c068020 size=32b38h (207672) map[0m
[09:27:38.735] [0;32mI (216) esp_image: segment 1: paddr=00042b60 vaddr=3fcae3c0 size=02524h (  9508) load[0m
[09:27:38.802] [0;32mI (219) esp_image: segment 2: paddr=0004508c vaddr=40380000 size=02f8ch ( 12172) load[0m
[09:27:38.802] [0;32mI (228) esp_image: segment 3: paddr=00048020 vaddr=42000020 size=669f0h (420336) map[0m
[09:27:38.802] [0;32mI (332) esp_image: segment 4: paddr=000aea18 vaddr=40382f8c size=0b42ch ( 46124) load[0m
[09:27:39.403] [0;32mI (348) boot: Loaded app from partition at offset 0x10000[0m
[09:27:39.403] [0;32mI (348) boot: Disabling RNG early entropy source...[0m
[09:27:39.403] [0;32mI (366) cpu_start: Unicore app[0m
[09:27:39.403] [0;32mI (366) cpu_start: Pro cpu up.[0m
[09:27:39.403] D (375) clk: RTC_SLOW_CLK calibration value: 4239518[0m
[09:27:39.403] [0;32mI (385) cpu_start: Pro cpu start user code[0m
[09:27:39.403] [0;32mI (385) cpu_start: cpu freq: 120000000 Hz[0m
[09:27:39.403] [0;32mI (386) cpu_start: Application information:[0m
[09:27:39.403] [0;32mI (390) cpu_start: Project name:     network_adapter[0m
[09:27:39.403] [0;32mI (399) cpu_start: App version:      001-dirty[0m
[09:27:39.403] [0;32mI (407) cpu_start: Compile time:     Feb 20 2024 17:03:17[0m
[09:27:39.403] [0;32mI (416) cpu_start: ELF file SHA256:  0915f9d600372764...[0m
[09:27:39.403] [0;32mI (425) cpu_start: ESP-IDF:          v5.1.2-dirty[0m
[09:27:39.403] [0;32mI (433) cpu_start: Min chip rev:     v1.0[0m
[09:27:39.403] [0;32mI (441) cpu_start: Max chip rev:     v1.99 [0m
[09:27:39.403] [0;32mI (448) cpu_start: Chip rev:         v1.0[0m
[09:27:39.403] D (455) memory_layout: Checking 3 reserved memory ranges:[0m
[09:27:39.403] D (464) memory_layout: Reserved memory range 0x3fca0000 - 0x3fcae3c0[0m
[09:27:39.403] D (474) memory_layout: Reserved memory range 0x3fcae3c0 - 0x3fcb4f80[0m
[09:27:39.403] D (484) memory_layout: Reserved memory range 0x3fcdf4bc - 0x3fce0000[0m
[09:27:39.403] D (493) memory_layout: Building list of available memory regions:[0m
[09:27:39.403] D (503) memory_layout: Available memory region 0x3fcb4f80 - 0x3fcc0000[0m
[09:27:39.403] D (513) memory_layout: Available memory region 0x3fcc0000 - 0x3fcdcb70[0m
[09:27:39.403] D (523) memory_layout: Available memory region 0x3fcdcb70 - 0x3fcdf4bc[0m
[09:27:39.403] [0;32mI (533) heap_init: Initializing. RAM available for dynamic allocation:[0m
[09:27:39.403] D (544) heap_init: New heap initialised at 0x3fcb4f80[0m
[09:27:39.403] [0;32mI (552) heap_init: At 3FCB4F80 len 00027BF0 (158 KiB): D/IRAM[0m
[09:27:39.403] [0;32mI (562) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM[0m
[09:27:39.403] D (573) intr_alloc: Connected src 21 to int 2 (cpu 0)[0m
[09:27:39.403] D (581) spi_flash: trying chip: issi[0m
[09:27:39.403] D (586) spi_flash: trying chip: gd[0m
[09:27:39.403] D (591) spi_flash: trying chip: mxic[0m
[09:27:39.403] D (597) spi_flash: trying chip: winbond[0m
[09:27:39.403] D (603) spi_flash: trying chip: boya[0m
[09:27:39.403] D (609) spi_flash: trying chip: th[0m
[09:27:39.403] D (614) spi_flash: trying chip: generic[0m
[... 225 lines cut ...]
[09:27:43.335] [0;32mI (4727) slave_rpc: mac [08:3a:8d:45:2a:2d] [0m
[09:27:43.335] 8 3a 8d 45 2a 2d
[09:27:43.335] D (4738) protocomm: No encrypt ret 0[0m
[09:27:43.335] D (4750) SPI_DRIVER: Tx New[0m
[... cut ...]
[14:13:09.542] E (437760) wifi:fail to post timer, type=12
[14:13:11.542] W (439760) wifi:pp q full: 7-3fcd5490
[14:13:11.542] E (439760) wifi:fail to post timer, type=12
[14:13:13.542] W (441760) wifi:pp q full: 7-3fcd5490
[14:13:13.542] E (441760) wifi:fail to post timer, type=12
[14:13:15.541] W (443760) wifi:pp q full: 7-3fcd5490
[14:13:15.541] E (443760) wifi:fail to post timer, type=12
[14:13:17.541] W (445760) wifi:pp q full: 7-3fcd5490
[14:13:17.557] E (445760) wifi:fail to post timer, type=12

mantriyogesh, 2024-02-21:
With exact your test, it would be hard to replicate at our place.

Anyway, If you assign **static** IP addresses to self (softap) and station connecting softap and just send some ping packets etc, does it work first?

Also please make sure your code looks similar here:
https://github.com/espressif/esp-hosted/blob/61c9415c1eb619787cc151dadc708780a172a413/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c#L212-L233

wky19971002, 2024-02-21:
I have already modified this part of the code,it works,and I can send ping packets  successful,But soon the esp32 will crash.

mantriyogesh, 2024-02-21:
Can you please send the commands and output?
Assumption: 'esp32' here I refer to 'esp32c2'.
crashes 'soon': means? Do you have a backtrace?

You can enable gdb_stub to stop after crash.
[image]
[image]

wky19971002, 2024-02-21:
I don't think I integrated the gdbstub part. Let me try.
When I say crash, I mean the hot spot disappears and the esp32 has to be powered back on to work.