espressif/esp-hosted #341: ESP-Hosted-FG:about Bluetooth stack

1653323560, 2024-02-20:
I saw in the ESP-Hosted FG document that the Bluetooth stack for MCU will be released in the future, when will it be released?
I wanted to adapt the ESP-Hosted FG to the STM32H750, but I couldn't find a suitable Bluetooth stack for MCU.
[image]

mantriyogesh, 2024-02-21:
Hello @1653323560 ,

We had some work done on this, to port NimBLE on host.
Do you need classicBT or BLE?
There is no limitations from ESP (depends on ESP chip you select) for Classic BT or BLE, but NimBLE stack only supports BLE.

mantriyogesh, 2024-02-21:
We have tested this with ESP as host, not STM32.
So it could be even tested at your side, with connecting two ESPs.
Slave ESP will expose the Bluetooth controller and Host ESP will have BLE stack running.

Another factor while choosing NimBLE was it's memory requirements.
Technically, any other Bluetooth stack would also work.

Also, to add further, NimBLE as bluetooth stack evaluated in recently. Let me check if we can push it over feature/esp_as_mcu_host branch.

1653323560, 2024-02-21:
I need classicBT and BLE，so I need a dual BTstack.
Is it possible to port the ESP32 bluedroid to STM32?

gavin-hy, 2024-03-04:
Hi, @1653323560 ,I recommend you to refer to the demos of NXP's i.MXRT series microcontrollers. There are relevant Bluetooth protocol stack routines included.

mantriyogesh, 2024-03-04:
@1653323560 although it is technically possible, we might take some to provide out-of-the-box support over ESP-Hosted.

Technically, any Bluetooth stack (at host) should work. For getting Classic BT & BLE,  `ESP32` can be used.

1653323560, 2024-03-04:
@mantriyogesh So the ESP-Hosted will go on provide more support for the MCU based dual BTstack?

gavin-hy, 2024-03-05:
@mantriyogesh  In the near future, the ESP32-P4 is about to be launched. Considering that the P4 lacks WiFi and BLE functionalities, do we need to carry out relevant slave adaptation work? Additionally, I'm aware that the ESP32-C5 supports 5G and will also be launched soon. If we use it as a slave and employ the SDIO interface, do we have plans to develop its 5G-related features?

mantriyogesh, 2024-03-05:
Hello @gavin-hy,

> do we need to carry out relevant slave adaptation work?

ESP32-P4 is already supported with ESP-Hosted as host. You can use any of ESP32, ESP32-C2/C3/C6/S2/S3 as slave.

Out of these chipsets, ESP32 and ESP32-C6 supports the SDIO, which is already functioning with around 50Mbps Tx and 50Mbps Rx UDP iperf throughput.

> ESP32-C5 supports 5G

Yes, ESP32-C5 would be supported as slave chipset. But ESP32-C5 would not have SDIO peripheral. So we are thinking to support using SPI (the full-duplex driver already stable) & QSPI (need to create new)

> do we have plans to develop its 5G-related features?

Yes, if any more new changes for station / softap mode done as part of 2.4-5G, we will adapt in ESP-Hosted

gavin-hy, 2024-03-05:
Hi, @mantriyogesh  Thank you for your quick response. I learned from Espressif's official website that the ESP32-C5 comes with SDIO Slave functionality! Has this peripheral been removed from the mass-produced chips?
The following image was captured from the official website!
[image]

mantriyogesh, 2024-03-05:
I understand your confusion here.

I can't send you the documentation for this (yet not released), but most likely, the C5 MP chips would not have slave 'SDIO' or 'SDIO 2.0'.