espressif/esp-hosted #342: SDIO communication Issue with STM32MP135

pasco1995, 2024-02-23:
I'm using STM32MP135f-dk Development Board with ESP32 as Wifi Module. I removed the Murata Wifi/BT Module (sdmmc2 in Device Tree) and attached the ESP32 with Data[0-3], CMD, CLK, Reset, GND. ESP32 is powered via MicroUSB Cable,

Currently im using the master Branch of ESP-Hosted with deactivated bluetooth.
I tested with RaspberryPi (3 B+) with same ESP32 SW (Wifi over SDIO) and it worked correctly.

On the STM32 Board, when it comes to scanning request, the reading of data packets fail.
Du you have any advice?

Here are the logs:

<details close>
  <summary>ESP32 logs</summary>

```

I (30) boot: ESP-IDF d3c99ed3-dirty 2nd stage bootloader
I (30) boot: compile time Feb 23 2024 10:39:08
I (30) boot: Multicore bootloader
I (35) boot: chip revision: v3.0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 factory          factory app      00 00 00010000 00100000
I (98) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (106) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (113) boot: End of partition table
I (118) boot: Defaulting to factory image
I (122) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1e5d4h (124372) map
I (176) esp_image: segment 1: paddr=0002e5fc vaddr=3ffb0000 size=01a1ch (  6684) load
I (179) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=70460h (459872) map
I (348) esp_image: segment 3: paddr=000a0488 vaddr=3ffb1a1c size=01d70h (  7536) load
I (351) esp_image: segment 4: paddr=000a2200 vaddr=40080000 size=15d34h ( 89396) load
I (401) boot: Loaded app from partition at offset 0x10000
I (401) boot: Disabling RNG early entropy source...
I (413) cpu_start: Multicore app
I (413) cpu_start: Pro cpu up.
I (414) cpu_start: Starting app cpu, entry point is 0x400813f8
I (0) cpu_start: App cpu up.
I (433) cpu_start: Pro cpu start user code
I (433) cpu_start: cpu freq: 240000000 Hz
I (433) cpu_start: Application information:
I (438) cpu_start: Project name:     network_adapter
I (444) cpu_start: App version:      release/ng-v1.0.2-188-gcdd6cb6-
I (451) cpu_start: Compile time:     Feb 23 2024 10:39:16
I (457) cpu_start: ELF file SHA256:  8e92ab07016559e6...
I (463) cpu_start: ESP-IDF:          d3c99ed3-dirty
I (468) cpu_start: Min chip rev:     v0.0
I (473) cpu_start: Max chip rev:     v3.99
I (478) cpu_start: Chip rev:         v3.0
I (483) heap_init: Initializing. RAM available for dynamic allocation:
I (490) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (496) heap_init: At 3FFC1650 len 0001E9B0 (122 KiB): DRAM
I (502) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (509) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (515) heap_init: At 40095D34 len 0000A2CC (40 KiB): IRAM
I (523) spi_flash: detected chip: gd
I (525) spi_flash: flash io: dio
W (529) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (543) app_start: Starting scheduler on CPU0
I (548) app_start: Starting scheduler on CPU1
I (548) main_task: Started on CPU0
I (558) main_task: Calling app_main()
I (561) stats: *********************************************************************
I (569) stats:                 ESP-Hosted Firmware version :: 1.0.3
[... 1011 lines cut ...]
[ 2014.707802] esp32_sdio:read_packet: Len from Slave: 3445
[ 2014.707847] esp32_sdio:read_packet: Rx large packet: 3445
[ 2014.722665] esp32_sdio:read_packet: Failed to read data - -84 [6 - 3072]
[ 2014.730355] esp32_sdio:read_packet: Len from Slave: 3680
[ 2014.730396] esp32_sdio:read_packet: Rx large packet: 3680
[ 2014.743492] esp32_sdio:read_packet: Failed to read data - -84 [7 - 3584]
[ 2014.750785] esp32_sdio:read_packet: Len from Slave: 3917
[ 2014.750821] esp32_sdio:read_packet: Rx large packet: 3917
[ 2014.763379] esp32_sdio:read_packet: Failed to read data - -84 [7 - 3584]
[ 2014.770561] esp32_sdio:read_packet: Len from Slave: 4140
[ 2014.770594] esp32_sdio:read_packet: Rx large packet: 4140
[ 2014.782997] esp32_sdio:read_packet: Failed to read data - -84 [8 - 4096]
[ 2014.791446] esp32_sdio:read_packet: Len from Slave: 4350
[ 2014.791478] esp32_sdio:read_packet: Rx large packet: 4350
[ 2014.804194] esp32_sdio:read_packet: Failed to read data - -84 [8 - 4096]
[ 2014.827831] esp32_sdio:read_packet: Len from Slave: 4612
[ 2014.827874] esp32_sdio:read_packet: Rx large packet: 4612
[ 2014.840838] esp32_sdio:read_packet: Failed to read data - -84 [9 - 4608]
[ 2015.907793] esp32_sdio:read_packet: Len from Slave: 4628
[ 2015.907836] esp32_sdio:read_packet: Rx large packet: 4628
[ 2015.922677] esp32_sdio:read_packet: Failed to read data - -84 [9 - 4608]

```

</details>

pasco1995, 2024-02-23:
Additional Info:

<details close>
  <summary>Device Tree</summary>
  
```

// SPDX-License-Identifier: (GPL-2.0-or-later OR BSD-3-Clause)
/*
 * Copyright (C) 2024, STMicroelectronics - All Rights Reserved
 * Author: STM32CubeMX code generation for STMicroelectronics.
 */

/* For more information on Device Tree configuration, please refer to
 * https://wiki.st.com/stm32mpu/wiki/Category:Device_tree_configuration
 */

/dts-v1/;
#include <dt-bindings/pinctrl/stm32-pinfunc.h>

#include "stm32mp135.dtsi"
#include "stm32mp13xf.dtsi"

/* USER CODE BEGIN includes */
#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>
#include <dt-bindings/leds/common.h>
#include <dt-bindings/regulator/st,stm32mp13-regulator.h>
#include <dt-bindings/rtc/rtc-stm32.h>
/* USER CODE END includes */

/ {
	model = "STMicroelectronics STM32MP135F-DK STM32CubeMX board - openstlinux-6.1-yocto-mickledore-mp1-v23.06.21";
	compatible = "st,stm32mp135f-ngw-mx", "st,stm32mp135f-dk", "st,stm32mp135";

	memory@c0000000 {
		device_type = "memory";
		reg = <0xc0000000 0x20000000>;

		/* USER CODE BEGIN memory */
		/* USER CODE END memory */
	};

	reserved-memory {
		#address-cells = <1>;
		#size-cells = <1>;
		ranges;

		/* USER CODE BEGIN reserved-memory */

		optee@dd000000{
			reg = <0xdd000000 0x3000000>;
			no-map;
		};
		/* USER CODE END reserved-memory */
	};

	/* USER CODE BEGIN root */

	aliases{
		ethernet0 = &eth1;
		ethernet1 = &eth2;
		serial0 = &uart4;
		serial1 = &usart1;
		serial2 = &uart8;
		serial3 = &usart2;
	};

	chosen{
		stdout-path = "serial0:115200n8";
[... 2932 lines cut ...]
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

```

</details>

mantriyogesh, 2024-02-24:
Thanks a lot, for sharing all the detailed logs.
I will check these and get back to you.

mantriyogesh, 2024-02-26:
I had checked the logs and I could see that you face error in reading data from sdio , 'Rx large packet' with increasing length.

Bootup event (first host rx) and get mac seems passed (subsequent SDIO rx/tx passed)

1. Is this wpa_supplicant your only one instance?
2. How is your setup connected? With Jumper wires? How big is length? Setup picture?
3. External pull-ups connected? Please refer to [Porting_Guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) for general precautions to be taken while porting
4. most importantly, host is going to check how much data the slave has written. It will issue get_read_len before actual read, where you see changing, gradually increasing read length.
But has the slave actually written that much and that big packets?
host: https://github.com/espressif/esp-hosted/blob/043e95e2f6fa67b8e232f26a4ea93a2cef07190d/esp_hosted_ng/host/sdio/esp_sdio.c#L392

5. Once (2) is confirmed, Does the host tx reach to the ESP, I do not see that in esp logs. Can you please verify rx / tx both sides are fine?

slave: https://github.com/espressif/esp-hosted/blob/043e95e2f6fa67b8e232f26a4ea93a2cef07190d/esp_hosted_ng/esp/esp_driver/network_adapter/main/sdio_slave_api.c#L281

slave tx: 

https://github.com/espressif/esp-hosted/blob/043e95e2f6fa67b8e232f26a4ea93a2cef07190d/esp_hosted_ng/esp/esp_driver/network_adapter/main/sdio_slave_api.c#L287-L291

slave rx:

https://github.com/espressif/esp-hosted/blob/043e95e2f6fa67b8e232f26a4ea93a2cef07190d/esp_hosted_ng/esp/esp_driver/network_adapter/main/sdio_slave_api.c#L419-L422

host tx: 

https://github.com/espressif/esp-hosted/blob/043e95e2f6fa67b8e232f26a4ea93a2cef07190d/esp_hosted_ng/host/sdio/esp_sdio.c#L639
```
esp_info("new tx\n");
esp_hex_dump("host tx: ", tx_skb->data, tx_skb->len);
```

host rx: 

https://github.com/espressif/esp-hosted/blob/043e95e2f6fa67b8e232f26a4ea93a2cef07190d/esp_hosted_ng/host/main.c#L662

```
esp_info("new rx\n");
esp_hex_dump("host rx: ", skb->data, len+offset);
```

pasco1995, 2024-02-26:
Thanks for your reply!

(1) yes, the wpa_supplicant is the only instance.

(2) yes, jumper wires (~ 6 cm).

[image]

[image]

(3) yes, pullups are present with 47 kOhm each (Data 0 - 3, CMD).

(4) (5) I activated the logs for host and slave. It seems that slave really sends that much data (while scan request) :

<details close>
<summary>Host Logs</summary>

```

root@stm32mp1:~# insmod esp32_sdio.ko clockspeed=1

[  100.189164] esp32_sdio: loading out-of-tree module taints kernel.
[  100.420535] esp32_sdio: probe of mmc1:0001:1 failed with error -110
[  101.080192] esp32_sdio: probe of mmc1:0001:2 failed with error -16
[  101.085424] mmc1: card 0001 removed
[  110.048309] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[  110.063493] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[  110.073742] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[  110.085858] mmc1: new SDIO card at address 0001
[  110.100319] esp32_sdio:esp_probe: ESP network device detected
[  110.112452] esp32_sdio:esp_probe: Hz: 1000000, min: 24438, max: 50000000
[  110.112488] esp32_sdio:esp_probe: SDIO: 1 MHz
[  110.124990] esp32_sdio:get_firmware_data: Rx Pre ====== 0
[  110.139499] esp32_sdio:get_firmware_data: Rx Pos ======  0
[  110.144371] esp32_sdio:get_firmware_data: Tx Pre ======  0
[  110.149052] esp32_sdio:get_firmware_data: Tx Pos ======  10
[  110.158442] esp32_sdio:esp_probe: after init context
[  110.158980] esp32_sdio:esp_probe: ESP SDIO probe completed
[  110.169153] esp32_sdio: probe of mmc1:0001:2 failed with error -22
[  110.870138] esp32_sdio:read_packet: Len from Slave: 35
[  110.874196] esp32_sdio:process_rx_packet: new rx
[  110.886994] host rx: 9926700e: 03 00 00 00 17 00 0c 00 00 00 00 00 01 00 10 00  ................
[  110.895809] host rx: 6a2c0a0f: 0f 00 00 00 03 01 00 00 01 01 01 07 01 00 03 10  ................
[  110.904828] host rx: 9f502a77: 00 00 00                                         ...
[  110.912593] esp32_sdio:process_event_esp_bootup: Bootup Event tag: 3
[  110.917605] esp32_sdio:esp_validate_chipset: Chipset=ESP32 ID=00 detected over SDIO
[  110.925438] esp32_sdio:process_event_esp_bootup: Bootup Event tag: 0
[  110.931693] esp32_sdio:process_event_esp_bootup: Bootup Event tag: 1
[  110.937915] esp32_sdio:process_fw_data: ESP chipset's last reset cause:
[  110.944559] esp32_sdio:print_reset_reason: RTCWDT_RTC_RESET
[  110.950223] esp32_sdio:check_esp_version: ESP Firmware version: 1.0.3
[  110.956629] esp_commands_setup:1550 exiting
[  110.961316] esp32_sdio:esp_reg_notifier: Driver init is ongoing
[  110.966957] cmd_init_interface:1239
[  110.979552] cmd_init_interface:1258
[  110.981737] wait_and_decode_cmd_resp:187
[  110.990239] esp_cmd_work:326
[  111.029894] esp32_sdio:tx_process: new tx
[  111.032474] host tx: 9d6138a6: 00 00 01 00 08 00 0c 00 00 00 00 00 01 00 00 00  ................
[  111.054358] host tx: 16bab7ef: 00 00 00 00                                      ....
[  111.190104] esp32_sdio:read_packet: Len from Slave: 20
[  111.194287] esp32_sdio:process_rx_packet: new rx
[  111.206739] host rx: 571c84d5: 00 00 02 00 08 00 0c 00 00 00 00 00 01 02 00 00  ................
[  111.215716] host rx: 85d1f48a: 00 00 00 00                                      ....
[  111.223776] cmd_init_interface:1260 exiting
[  111.226584] wait_and_decode_cmd_resp:187
[  111.233198] esp_cmd_work:326
[  111.279402] esp32_sdio:tx_process: new tx
[  111.289334] host tx: a65f8323: 00 00 01 00 08 00 0c 00 00 00 00 00 03 00 00 00  ................
[  111.296778] host tx: 56701106: 00 00 00 00                                      ....
[... 574 lines cut ...]
I (42064) s->h: 0x3ffd538c   00 0f ac 02 00 0f ac 04  00 00                    |..........|
E (42109) FW_SDIO_SLAVE:
To Host
I (42109) s->h: 0x3ffd5400   01 01 f8 00 7c 5a 1c 29  21 04 08 0b b4 ff ff ff  |....|Z.)!.......|
I (42112) s->h: 0x3ffd5410   ef 5a 62 02 00 00 00 00  e0 00 00 00 83 81 37 df  |.Zb...........7.|
I (42121) s->h: 0x3ffd5420   04 00 00 00 64 00 31 04  00 0b 78 78 78 78 2d 4d  |....d.1...xxxx-M|
I (42131) s->h: 0x3ffd5430   6f 62 69 6c 65 01 08 82  84 8b 96 0c 12 18 24 03  |obile.........$.|
I (42140) s->h: 0x3ffd5440   01 0b 05 04 00 01 00 00  07 06 44 45 20 01 0d 14  |..........DE ...|
I (42150) s->h: 0x3ffd5450   2a 01 00 32 04 30 48 60  6c 2d 1a ad 01 03 ff ff  |*..2.0H`l-......|
I (42159) s->h: 0x3ffd5460   00 00 00 00 00 00 00 00  00 00 01 00 00 00 00 00  |................|
I (42169) s->h: 0x3ffd5470   00 00 00 00 00 3d 16 0b  00 00 00 00 00 00 00 00  |.....=..........|
I (42178) s->h: 0x3ffd5480   00 00 00 00 00 00 00 00  00 00 00 00 00 7f 08 04  |................|
I (42188) s->h: 0x3ffd5490   10 00 02 00 00 00 40 bf  0c b2 59 81 33 fa ff 00  |......@...Y.3...|
I (42197) s->h: 0x3ffd54a0   00 fa ff 00 00 c0 05 00  00 00 fc ff dd 18 00 50  |...............P|
I (42207) s->h: 0x3ffd54b0   f2 02 01 01 80 00 03 a4  00 00 27 a4 00 00 42 43  |..........'...BC|
I (42217) s->h: 0x3ffd54c0   5e 00 62 32 2f 00 dd 09  00 03 7f 01 01 00 00 ff  |^.b2/...........|
I (42226) s->h: 0x3ffd54d0   7f dd 0a 1c 5a 7c 04 01  00 22 91 04 f7 36 03 00  |....Z|..."...6..|
I (42236) s->h: 0x3ffd54e0   01 00 30 18 01 00 00 0f  ac 04 01 00 00 0f ac 04  |..0.............|
I (42245) s->h: 0x3ffd54f0   02 00 00 0f ac 02 00 0f  ac 04 00 00              |............|
E (42289) FW_SDIO_SLAVE:
To Host
I (42289) s->h: 0x3ffafd70   01 00 00 00                                       |....|

```
</details>

mantriyogesh, 2024-02-28:
Hello @pasco1995 ,

As you are using jumper wires and SDIO faces signal integrity issues while working on wires, can you please try below:

1. Use SDIO in 1-bit mode.
2. Lower the SDIO freq to 5MHz 
You can use module parameter `clockspeed` in MHz, also to do this:
https://github.com/espressif/esp-hosted/blob/3d4f2f7b5913e951a3af2083c117a88d3638e429/esp_hosted_ng/host/main.c#L36
https://github.com/espressif/esp-hosted/blob/3d4f2f7b5913e951a3af2083c117a88d3638e429/esp_hosted_ng/host/sdio/esp_sdio.c#L804

3. You can try a workaround in another branch, `feature/esp_as_host_mcu` , to mandate, send or receive always receive 1 block at a time.
Although it is not Linux branch, the SDIO driver should logically be similar.
https://github.com/espressif/esp-hosted/blob/504f4f203c27d5a9e5e3bb1b34f7101e43cd8f59/host/port/esp_hosted_config.h#L141-L142

Try in this order and stop once working:
(1) only,
(2) only,
(1) & (2) together,
(3) only
(1) (2) & (3) together

pasco1995, 2024-03-01:
Hi @mantriyogesh,

alle the logs above are caputered with ```clockspeed=1```.

<details close>
<summary>MMC1 Config</summary>

```
 cat /sys/kernel/debug/mmc1/ios
clock:          1000000 Hz
actual clock:   1000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    0 (legacy)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)
```

</details>

How do I configure the 1 bit mode ? Do I habe to change teh ESP32 software or just on the Linux side (device tree pin configuration, kernel module).

mantriyogesh, 2024-03-01:
Only device tree at Linux.

bus-width = <1>;

https://wiki.st.com/stm32mpu/wiki/SDMMC_device_tree_configuration#

pasco1995, 2024-03-04:
Hi @mantriyogesh 

the 1 bit Mode ist working!

<details>
<summary>Performancetest</summary>

```
ESP32* (1bit Mode SDIO@1MHz) & STM32MP135f-dk :
Befehl : iperf3.exe -c  192.168.137.253

Connecting to host 192.168.137.253, port 5201
[  4] local 192.168.137.1 port 62098 connected to 192.168.137.253 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   256 KBytes  2.09 Mbits/sec
[  4]   1.00-2.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec
[  4]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec
[  4]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec
[  4]   6.00-7.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec
[  4]   8.00-9.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec   768 KBytes   629 Kbits/sec                  sender
[  4]   0.00-10.00  sec   670 KBytes   549 Kbits/sec                  receiver 

====================================================================================================

ESP32* (1bit Mode SDIO@5MHz) & STM32MP135f-dk :
Befehl : iperf3.exe -c  192.168.137.253

Connecting to host 192.168.137.253, port 5201
[  4] local 192.168.137.1 port 62155 connected to 192.168.137.253 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   384 KBytes  3.14 Mbits/sec
[  4]   1.00-2.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   3.00-4.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   5.00-6.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   6.00-7.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   7.00-8.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   8.00-9.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  1.50 MBytes  1.26 Mbits/sec                  sender
[  4]   0.00-10.00  sec  1.40 MBytes  1.17 Mbits/sec                  receiver

====================================================================================================

ESP32* (1bit Mode SDIO@10MHz) & STM32MP135f-dk :
Befehl : iperf3.exe -c  192.168.137.253

Connecting to host 192.168.137.253, port 5201
[  4] local 192.168.137.1 port 62186 connected to 192.168.137.253 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   384 KBytes  3.14 Mbits/sec
[  4]   1.00-2.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   3.00-4.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   5.00-6.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   6.00-7.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   7.00-8.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   8.00-9.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  1.50 MBytes  1.26 Mbits/sec                  sender
[  4]   0.00-10.00  sec  1.41 MBytes  1.18 Mbits/sec                  receiver

====================================================================================================

ESP32* (1bit Mode SDIO@20MHz) & STM32MP135f-dk :
Befehl : iperf3.exe -c  192.168.137.253

Connecting to host 192.168.137.253, port 5201
[  4] local 192.168.137.1 port 62237 connected to 192.168.137.253 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   384 KBytes  3.14 Mbits/sec
[  4]   1.00-2.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   3.00-4.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   5.00-6.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   6.00-7.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   7.00-8.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   8.00-9.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  1.50 MBytes  1.26 Mbits/sec                  sender
[  4]   0.00-10.00  sec  1.39 MBytes  1.17 Mbits/sec                  receiver

* with deactivated logs

```

</details>

mantriyogesh, 2024-03-04:
The iperf speed is not changing with the clock, which is little surprising.

Can you please try the [Raw_TP_Testing.md](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md) test for tx and rx?

pasco1995, 2024-03-04:
Here are the logs:

<details close>
<summary>ESP Logs: Raw TP Mode 1 (Host to ESP)</summary>

```
I (3009) stats: 0-1 sec       0.00 kbits/sec
I (4009) stats: 1-2 sec       0.00 kbits/sec
I (5009) stats: 2-3 sec       6923.59 kbits/sec
I (5917) FW_MAIN: Scan request

I (6009) stats: 3-4 sec       6992.03 kbits/sec
I (7009) stats: 4-5 sec       7185.94 kbits/sec
I (8009) stats: 5-6 sec       7254.38 kbits/sec
I (9009) stats: 6-7 sec       7197.34 kbits/sec
I (10009) stats: 7-8 sec       7277.19 kbits/sec
I (11009) stats: 8-9 sec       7300.00 kbits/sec
I (11756) FW_MAIN: Scan request

I (12009) stats: 9-10 sec       7185.94 kbits/sec
I (13009) stats: 10-11 sec       7231.56 kbits/sec
I (14009) stats: 11-12 sec       7277.19 kbits/sec
I (15009) stats: 12-13 sec       7208.75 kbits/sec
I (16009) stats: 13-14 sec       7288.59 kbits/sec
I (17009) stats: 14-15 sec       7288.59 kbits/sec
I (18009) stats: 15-16 sec       7288.59 kbits/sec
I (18588) FW_MAIN: Scan request

I (19009) stats: 16-17 sec       7208.75 kbits/sec
I (20009) stats: 17-18 sec       7220.16 kbits/sec
I (21009) stats: 18-19 sec       7288.59 kbits/sec
I (22009) stats: 19-20 sec       7163.12 kbits/sec
I (23009) stats: 20-21 sec       7265.78 kbits/sec
I (24009) stats: 21-22 sec       7265.78 kbits/sec
I (25009) stats: 22-23 sec       7288.59 kbits/sec
I (26009) stats: 23-24 sec       7288.59 kbits/sec
I (27009) stats: 24-25 sec       7288.59 kbits/sec
I (28009) stats: 25-26 sec       7277.19 kbits/sec
```

</details>

<details close>
<summary>ESP Logs: Raw TP Mode 2 (ESP to Host)</summary>

```
I (3035) stats: 0-1 sec       2208.00 kbits/sec
I (4035) stats: 1-2 sec       1000.50 kbits/sec
I (5035) stats: 2-3 sec       1150.00 kbits/sec
I (6035) stats: 3-4 sec       1138.50 kbits/sec
I (6592) FW_MAIN: Scan request

I (7035) stats: 4-5 sec       1081.00 kbits/sec
I (8035) stats: 5-6 sec       977.50 kbits/sec
I (9035) stats: 6-7 sec       1138.50 kbits/sec
I (10035) stats: 7-8 sec       1150.00 kbits/sec
I (11035) stats: 8-9 sec       1150.00 kbits/sec
I (11603) FW_MAIN: Scan request

I (12035) stats: 9-10 sec       1081.00 kbits/sec
I (13035) stats: 10-11 sec       1023.50 kbits/sec
I (14035) stats: 11-12 sec       1138.50 kbits/sec
I (15035) stats: 12-13 sec       1150.00 kbits/sec
I (16035) stats: 13-14 sec       1150.00 kbits/sec
I (17035) stats: 14-15 sec       1150.00 kbits/sec
I (18035) stats: 15-16 sec       1150.00 kbits/sec
I (18593) FW_MAIN: Scan request

I (19035) stats: 16-17 sec       1081.00 kbits/sec
I (20035) stats: 17-18 sec       977.50 kbits/sec
I (21035) stats: 18-19 sec       1138.50 kbits/sec
I (22035) stats: 19-20 sec       1150.00 kbits/sec
I (23035) stats: 20-21 sec       1150.00 kbits/sec
I (24035) stats: 21-22 sec       1150.00 kbits/sec
I (25035) stats: 22-23 sec       1150.00 kbits/sec
I (26035) stats: 23-24 sec       1150.00 kbits/sec
I (27035) stats: 24-25 sec       1150.00 kbits/sec
I (28035) stats: 25-26 sec       1150.00 kbits/sec
```

</details>

The throughput from Host to ESP is higher. (Comparable with [ESP32 SPI throughput](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/README.md#5-throughput-performance))

mantriyogesh, 2024-03-04:
Host tx would be always little higher than host rx. But the difference you experience is extremely huge.
There is some issue in sdio host rx.

Possibly,  the isr performing slow for your Linux base sdio driver?

Do you have analyser?

mantriyogesh, 2024-05-08:
You might have to check your base Linux kernel driver for your SoC, how `cap-sdio-irq` is handled.

Another property we recommend using,`broken-cd`.
Have you enabled any debug logging? High debug prints (at esp and/or host) would slow down throughput a lot.

pasco1995, 2024-05-10:
Hi,

I found the bug in the PL18x Driver, which I am using for MMC/SDIO Communication with the ESP32.

When setting the `cap-sdio-irq` setting in the device tree, the driver sets the config `MMC_CAP2_SDIO_IRQ_NOTHREAD` to true. Therefore, there is no thread created for the handling of the interrupts. When disabling `MMC_CAP2_SDIO_IRQ_NOTHREAD` the thread is created and performance is as expected (between 10 and 12 Mbit/s).

<details>
<summary>Fix in drivers/mmc/host/mmci.c:2217</summary>

```
if (variant->use_sdio_irq && host->mmc->caps & MMC_CAP_SDIO_IRQ)
//    mmc->caps2 |= MMC_CAP2_SDIO_IRQ_NOTHREAD;                   <--- comment this to create thread in sdio irq mode
```

</details>

<details>
<summary>Performancetest</summary>

```
Connecting to host xxx.xxx.xxx.xxx, port 5201
[  4] local xxx.xxx.xxx.xxx port 52507 connected to xxx.xxx.xxx.xxx port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  1.12 MBytes  9.42 Mbits/sec
[  4]   1.00-2.00   sec  1.12 MBytes  9.43 Mbits/sec
[  4]   2.00-3.00   sec  1.50 MBytes  12.6 Mbits/sec
[  4]   3.00-4.00   sec  1.25 MBytes  10.5 Mbits/sec
[  4]   4.00-5.00   sec  1.38 MBytes  11.6 Mbits/sec
[  4]   5.00-6.00   sec  1.38 MBytes  11.5 Mbits/sec
[  4]   6.00-7.00   sec  1.50 MBytes  12.6 Mbits/sec
[  4]   7.00-8.00   sec  1.38 MBytes  11.5 Mbits/sec
[  4]   8.00-9.00   sec  1.38 MBytes  11.5 Mbits/sec
[  4]   9.00-10.00  sec  1.25 MBytes  10.5 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  13.2 MBytes  11.1 Mbits/sec                  sender
[  4]   0.00-10.00  sec  13.1 MBytes  11.0 Mbits/sec                  receiver
```

</details>

Thank you @mantriyogesh for your help and support!

closed 2024-05-10 by pasco1995

mantriyogesh, 2024-05-10:
Happy to help. :-)

Although, the Over-The-Air throughput is depending upon a lot of factors, a generally good OTA performance should exceed than the throughput numbers you have reached.

Throughput improvements:

1. What is SDIO clock you are using?
Current stable sdio clock can be found out using:
```sh
$ sudo cat /sys/kernel/debug/mmc1/ios
```

2. The debug prints disabled (specially per packet prints)

pasco1995, 2024-05-23:
Hi @mantriyogesh,

the clockspeed is at 50 MHz and Logs are all disabled - in kernel module and esp firmware.