espressif/esp-hosted #343: Will this be the solution for ESP32-P4 ?

diplfranzhoepfinger, 2024-02-25:
Will this be the solution for ESP32-P4 ?

Will it be integrated as smoothly as "you think P4 hast native WiFi?"

Tnx

mantriyogesh, 2024-02-25:
Hello @diplfranzhoepfinger ,

P4 doesn't have Wi-Fi and bluetooth capability.
We support P4 as host and any other ESP chipset as slave (which has Wi-Fi and bluetooth ).

We have seperate branch, 
https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host

Which is specially targetted for this.

We already use P4 as host and C6 as slave with SDIO as transport. Let me know, if this is what you needed

diplfranzhoepfinger, 2024-02-25:
> We already use P4 as host and C6 as slave with SDIO as transport. Let me know, if this is what you needed
exactly. 

will it be integrated in lwip ?

mantriyogesh, 2024-02-25:
With P4, lwip is working fine ( very stable) already.

diplfranzhoepfinger, 2024-02-25:
so cool. Great !

mantriyogesh, 2024-02-25:
Please wait till tomorrow, I will provide you small changes to use on top of IDF.
And a schematic we use to connect P4 to C6.

Basically, external pull-ups are needed in sd lines as mentioned in the ESP-Hosted documentation.

diplfranzhoepfinger, 2024-02-25:
no need, i do not yet have a Devkit for P4

mantriyogesh, 2024-02-25:
oh okay.

For evaluation, you can use any one of supported ESP chipset as slave and any ESP chipset as host.

ESP chipsets as slave needs to have Wi-Fi and/or Bluetooth.  supported ESP as slave:
ESP32/ ESP32-S2/S3/C2/C3/C6.

mantriyogesh, 2024-02-25:
Easy transport for evaluation is SPI bus, as it is easiest to setup with jumper wires (still needs to be <=10cm).

If you wish to test later, we can close this issue for now.
You can re-open whenever you test any of above.

closed 2024-02-25 by diplfranzhoepfinger