espressif/esp-hosted #344: esp32c3  how to Setting Coexistence Compile-time Options

xmlhp, 2024-02-26:
refer to this link:https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/coexist.html
@unbut host
idf.py , I had select  CONFIG_ESP_COEX_SW_COEXIST_ENABLE and BT_CONTROLLER_ENABLED
but the, I can find BT_NIMBLE_COEX_PHY_CODED_TX_RX_TLIM at 
Component config->Bluetooth->Controller Option
@kumekay

xmlhp, 2024-02-26:
[sdkconfig.zip](https://github.com/espressif/esp-hosted/files/14403137/sdkconfig.zip)
This is my sdkconfig

mantriyogesh, 2024-02-26:
Can you please state if you need BLE only or A2DP (Classic-Bt) also?

xmlhp, 2024-02-26:
BLE @mantriyogesh

mantriyogesh, 2024-02-27:
> idf.py , I had select CONFIG_ESP_COEX_SW_COEXIST_ENABLE and BT_CONTROLLER_ENABLED

Right ESP-Hosted needs the Bluetooth controller. The `CONFIG_ESP_COEX_SW_COEXIST_ENABLE` & `BT_CONTROLLER_ENABLED` should be by default enabled.

> but the, I can find BT_NIMBLE_COEX_PHY_CODED_TX_RX_TLIM at
Component config->Bluetooth->Controller Option

You assume, by this, you **cannot** find `BT_NIMBLE_COEX_PHY_CODED_TX_RX_TLIM`
Right, but you will see option, `BT_CTRL_COEX_PHY_CODED_TX_RX_TLIM`
[image]

https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/coexist.html#setting-coexistence-compile-time-options also lists, `BT_CTRL_COEX_PHY_CODED_TX_RX_TLIM`

But I think it is better to test the RF distribution as is first. If you have any specific issues, then can go in tuning these configs.

Also, please also be conscious while using any kind of power saving (Intention is that the modems of Wi-Fi and Bluetooth should be awake when in both connected state). The latest ESP-IDF master automatically takes care of many such things for you, automatically.

xmlhp, 2024-02-27:
[menuconfig.zip](https://github.com/espressif/esp-hosted/files/14413858/menuconfig.zip)
Please unzip menuconfig.zip. Bluetooth-contorl_options.png, the option **Conexistenc** is not found here?
@mantriyogesh

mantriyogesh, 2024-02-27:
1. Which is the flavor used: FG or NG?
2. **If FG**, Can you please try on latest IDF master? The document you had referred is ESP-IDF master in this https://github.com/espressif/esp-hosted/issues/344#issue-2153792193 post.
Can you also send the idf commit you use on master?

xmlhp, 2024-02-27:
1.NG
2 in esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter$ git log
.commit f32dc7495eb79891d667c8ce044f36cbdff22a70 (HEAD -> master)
Merge: 5042106 6f3c6e7
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Wed Jan 10 17:43:42 2024 +0800

    Merge branch 'bugfix/compilation_error' into 'master'

    esp_hosted_ng: Fix compilation error

    See merge request app-frameworks/esp_hosted!379

xmlhp, 2024-02-27:
hongli@ubuntu-server-22-04:~/esp32/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/esp-idf$ git log
commit f32dc7495eb79891d667c8ce044f36cbdff22a70 (HEAD -> master)
Merge: 5042106 6f3c6e7
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Wed Jan 10 17:43:42 2024 +0800

    Merge branch 'bugfix/compilation_error' into 'master'

    esp_hosted_ng: Fix compilation error

    See merge request app-frameworks/esp_hosted!379

xmlhp, 2024-02-27:
hongli@ubuntu-server-22-04:~/esp32/esp-hosted/.git/modules/esp_hosted_ng/esp/esp_driver/esp-idf$ git log
commit d3c99ed3b88eb4c7fb5cf29d6cca1f390ff107e1 (grafted, HEAD)
Author: Aditya Patwardhan <aditya.patwardhan@espressif.com>
Date:   Tue Aug 22 14:40:50 2023 +0800

    Merge branch 'update/version_5_1_1' into 'release/v5.1'

    Update version to 5.1.1

    See merge request espressif/esp-idf!25443

mantriyogesh, 2024-02-27:
My current idf commit is, ef87256e733e6740705cb618ec545eb342dcd50d
but it should work on any latest master (only if FG).

[image]

Let me try this same with NG.
NG with master would not work and have to stick with the IDF commit in directory, `esp_hosted_ng/esp/esp_driver/esp-idf`

mantriyogesh, 2024-02-27:
yes you are correct,

I had to patch `esp-idf` in NG to make this option visible.

### Fix Procedure
1. cd esp_hosted_ng/esp/esp_driver
2. cmake .
3. Patch ESP-IDF:
cd `esp_hosted_ng/esp/esp_driver/esp-idf`
Change:

```diff
diff --git a/components/bt/controller/esp32c3/Kconfig.in b/components/bt/controller/esp32c3/Kconfig.in
index 98274a2236..3d71069f5c 100644
--- a/components/bt/controller/esp32c3/Kconfig.in
+++ b/components/bt/controller/esp32c3/Kconfig.in
@@ -333,7 +333,7 @@ config BT_CTRL_MESH_DUPL_SCAN_CACHE_SIZE
 choice BT_CTRL_COEX_PHY_CODED_TX_RX_TLIM
     prompt "Coexistence: limit on MAX Tx/Rx time for coded-PHY connection"
     default BT_CTRL_COEX_PHY_CODED_TX_RX_TLIM_DIS
-    depends on ESP_WIFI_SW_COEXIST_ENABLE
+    depends on ESP_COEX_SW_COEXIST_ENABLE
     help
         When using PHY-Coded in BLE connection, limitation on max tx/rx time can be applied to
         better avoid dramatic performance deterioration of Wi-Fi.
@@ -351,7 +351,7 @@ endchoice

 config BT_CTRL_COEX_PHY_CODED_TX_RX_TLIM_EFF
     int
-    default 0 if (!ESP_WIFI_SW_COEXIST_ENABLE)
+    default 0 if (!ESP_COEX_SW_COEXIST_ENABLE)
     default 1 if BT_CTRL_COEX_PHY_CODED_TX_RX_TLIM_EN
     default 0 if BT_CTRL_COEX_PHY_CODED_TX_RX_TLIM_DIS

```

4. Patch root directory of ESP-Hosted
```diff
diff --git a/esp_hosted_ng/esp/esp_driver/CMakeLists.txt b/esp_hosted_ng/esp/esp_driver/CMakeLists.txt
index 86f9108d..fb55c052 100644
--- a/esp_hosted_ng/esp/esp_driver/CMakeLists.txt
+++ b/esp_hosted_ng/esp/esp_driver/CMakeLists.txt
@@ -3,21 +3,21 @@ cmake_minimum_required(VERSION 3.0)
 project(ESP-hosted)

 # Reset Git repository
-execute_process(
-    COMMAND git reset --hard
-)
+#execute_process(
+#    COMMAND git reset --hard
+#)

 # Initialize submodules for esp-idf
-execute_process(
-    COMMAND git submodule update --init --depth=1
-    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
-)
+#execute_process(
+#    COMMAND git submodule update --init --depth=1
+#    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
+#)

 # Initialize submodule for esp-idf in esp-idf directory
-execute_process(
-    COMMAND git submodule update --init --depth=1
-    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}/esp-idf
-)
+#execute_process(
+#    COMMAND git submodule update --init --depth=1
+#    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}/esp-idf
+#)

 # Determine which shell is compatible and execute the corresponding install script
 if(UNIX AND NOT APPLE)
```

5. Verify IDF to be used:
```sh
$ cd esp_hosted_ng/esp/esp_driver
$ which idf.py
<esp_hosted_root_dir>/esp_hosted_ng/esp/esp_driver/esp-idf/tools/idf.py
```

6. Build
```sh
$ cd esp_hosted_ng/esp/esp_driver/network_adapter
$ 
$ rm -rf sdkconfig build/
$ idf.py set-target esp32c3
$ idf.py menuconfig ##### Co-Ex option should be visible now #### change as you wish
$ idf.py build flash monitor
```

### Reasoning
Actually, refer below grep result:

```sh
 I  ~/c/h/e/e/e/esp-idf > d3c99ed3b8 * > grep -iIrs 'ESP_WIFI_SW_COEXIST_ENABLE'
./components/esp_coex/sdkconfig.rename:CONFIG_ESP_WIFI_SW_COEXIST_ENABLE       CONFIG_ESP_COEX_SW_COEXIST_ENABLE
```
these type of changes happen when IDF team cleans up the symbols, to make it easier for cross component dependencies (and this also leads to broken component dependency sometimes like this!)

I will check this further with Bluetooth internal team and let you know if this sis correct workaround and anymore changes are needed further.

xmlhp, 2024-02-27:
So, what cam I do?
Wait for your confirmation, update the host from git, and then compile the firmware of the esp side? Or do you follow your Workaround Procedure?

xmlhp, 2024-02-27:
So, what cam I do?
Wait for your confirmation, update the host from git, and then compile the firmware of the esp side? Or do you follow your **Workaround Procedure**?

mantriyogesh, 2024-02-27:
For now, You can continue with the above workaround.
If something not recommended doing, I will be able to respond back by day or two

xmlhp, 2024-02-27:
I think I'd better wait for your confirmation before proceeding.
Waiting for your feedback!

xmlhp, 2024-02-27:
Now, the phenomenon on our side looks like this:
1.WIFI is running in STA mode. SSID and password have been configured with wpa_supplicant, wpa_supplicant.conf. esp32c3 connects to the SSID, and gets the IP, and can access the Internet normally.
2. Bluetooth sends broadcasts. The broadcast interval is 1.28s
3. Bluetooth is connected and receives the SSID and password to configure the new WIFI.
4. Broadcast without Bluetooth wpa_cli disable_networ,remove_network. Stop the wpa_supplicant service. ifcconfig espsta0 down.
5.ifconfig espsta0 up, rewrite wpa_supplicant.cnf. Start the wpa_supplicant service, udhcp-i espsta0&

In this case, always connect to the newly configured SSID.

Please help to confirm, will this phenomenon still exist after wifi ble coexistence is configured?
@mantriyogesh

xmlhp, 2024-02-27:
Now, the phenomenon on our side looks like this:
1.WIFI is running in STA mode. SSID and password have been configured with wpa_supplicant, wpa_supplicant.conf. esp32c3 connects to the SSID, and gets the IP, and can access the Internet normally.
2. Bluetooth sends broadcasts. The broadcast interval is 1.28s
3. Bluetooth is connected and receives the SSID and password to configure the new WIFI.
4. Broadcast without Bluetooth wpa_cli disable_networ,remove_network. Stop the wpa_supplicant service. ifcconfig espsta0 down.
5.ifconfig espsta0 up, rewrite wpa_supplicant.cnf. Start the wpa_supplicant service, udhcp-i espsta0&

In this case, the newly configured SSID cannot be connected.

Please help to confirm, will this phenomenon still exist after wifi ble coexistence is configured?

mantriyogesh, 2024-02-27:
Can you please send:
Commands and output,

C3 side and host dmesg logs?
Possibly wpa supplicant debug logs also.

mantriyogesh, 2024-02-27:
https://github.com/espressif/esp-hosted/issues/344#issuecomment-1965865033
This workaround is confirmed to be fine.

xmlhp, 2024-02-27:
You can reduce the memory consumption by configuring the following options on menuconfig.

[CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-bt-ble-dynamic-env-memory): enable the configuration of dynamic memory for Bluetooth protocol stack.

[CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-esp-wifi-static-rx-buffer-num): reduce the number of Wi-Fi static RX buffers.

[CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-esp-wifi-dynamic-rx-buffer-num): reduce the number of Wi-Fi dynamic RX buffers.

[CONFIG_ESP_WIFI_TX_BUFFER](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-esp-wifi-tx-buffer): enable the configuration of dynamic allocation TX buffers.

[CONFIG_ESP_WIFI_DYNAMIC_TX_BUFFER_NUM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-esp-wifi-dynamic-tx-buffer-num): reduce the number of Wi-Fi dynamic TX buffers.

[CONFIG_ESP_WIFI_TX_BA_WIN](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-esp-wifi-tx-ba-win): reduce the number of Wi-Fi Block Ack TX windows.

[CONFIG_ESP_WIFI_RX_BA_WIN](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-esp-wifi-rx-ba-win): reduce the number of Wi-Fi Block Ack RX windows.

[CONFIG_ESP_WIFI_MGMT_SBUF_NUM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-esp-wifi-mgmt-sbuf-num): reduce the number of Wi-Fi Management Short Buffer.

[CONFIG_ESP_WIFI_RX_IRAM_OPT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-esp-wifi-rx-iram-opt): turning off this configuration option will reduce the IRAM memory by approximately 17 KB.

[CONFIG_LWIP_TCP_SND_BUF_DEFAULT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-lwip-tcp-snd-buf-default): reduce the default TX buffer size for TCP sockets.

[CONFIG_LWIP_TCP_WND_DEFAULT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-lwip-tcp-wnd-default): reduce the default size of the RX window for TCP sockets.

[CONFIG_LWIP_TCP_RECVMBOX_SIZE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-lwip-tcp-recvmbox-size): reduce the size of the TCP receive mailbox. Receive mailbox buffers data within active connections and handles data flow during connections。

[CONFIG_LWIP_TCP_ACCEPTMBOX_SIZE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-lwip-tcp-acceptmbox-size): reduce the size of the TCP accept mailbox. Accept mailbox queues incoming connection requests and manages the initiation of new connections.

[CONFIG_LWIP_UDP_RECVMBOX_SIZE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-lwip-udp-recvmbox-size): reduce the size of the UDP receive mailbox.

[CONFIG_LWIP_TCPIP_RECVMBOX_SIZE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-lwip-tcpip-recvmbox-size): reduce the size of TCPIP task receive mailbox

How can you easily find these options in menuconfig?

xmlhp, 2024-02-27:
[https://github.com/espressif/esp-hosted/issues/344#issuecomment-1966345881](url)
I can find wifi option.but I can't find CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY.

xmlhp, 2024-02-28:
[esp32c3_log(1).zip](https://github.com/espressif/esp-hosted/files/14432892/esp32c3_log.1.zip) c3 side log
[linux_host_log(1).zip](https://github.com/espressif/esp-hosted/files/14432893/linux_host_log.1.zip) linux side log

[https://github.com/espressif/esp-hosted/issues/344#issuecomment-1965880650](url)

1. Bluetooth radio is always on
2. First connection
**wpa_supplicant -Dnl80211 -i espsta0 /var/run/wpa_supplicant.conf &**
**udhcpc -i espsta0 &**
You can connect to the SSID.
The IP address can be obtained.
It takes only 3-4 seconds from connecting to getting an IP
3. The Bluetooth service is connected and receives the newly configured SSID and password, which is actually the first connection's SSID and password.
4.
Disable network: **wpa_cli -i espsta0 disable_network 0**
Remove network: **wpa_cli -i espsta0 remove_network 0**
Kill udhcpc: **killall-9 pidof udhcpc**
Disable wpa_supplicant:kill -9 pidof wpa_supplicant
ifconfig espsta0 down
5. Restart wifi:
**ifconfig espsta0 up
wpa_supplicant -Dnl80211 -i espsta0 -c /var/run/wpa_supplicant.conf &
udhpcpc -i espsta0 &**

From the diary: the SSID has been connected (in the diary: FW), and the IP timeout time has been obtained. The test takes 20 seconds.

Ask, how to modify, can change to reduce this time?
@mantriyogesh

mantriyogesh, 2024-03-05:
Sorry for the delay,

From logs I can see `[feeder][wifi_get_iface_ip]:ioctl failed for iface:espsta0,error:Cannot assign requested address`.

Let we investigate why the log is full of this trace.
@kapilkedawat ++

xmlhp, 2024-03-05:
[feeder][wifi_get_iface_ip]:ioctl failed for iface:espsta0,error:Cannot assign requested address.
After opening udhcpc, the program will try to query whether espsta0 has obtained IP? The linux system will print this if an IP address is not obtained

xmlhp, 2024-03-05:
The IP query is through the socket, **ioctl(sd, SIOCGIFADDR, &ifr)**

kapilkedawat, 2024-04-04:
Hi @xmlhp 
are you running dhcp client for getting the ip once station got connected?

xmlhp, 2024-04-07:
@kapilkedawat ，are you running dhcp client for getting the ip once station got connected?

xmlhp, 2024-04-07:
Hi @xmlhp
are you running dhcp client for getting the ip once station got connected?
@kapilkedawat ，

xmlhp, 2024-04-07:
@kapilkedawat ，Yes,I running dhcp client

kapilkedawat, 2024-04-08:
@xmlhp, sorry I missed that you are running udhcpd immediately after running wpa_supplicant. It's better if you start it after wpa_supplicant gets connected, the error logs won't show anymore for ip address.

>> 5. Restart wifi:
ifconfig espsta0 up
wpa_supplicant -Dnl80211 -i espsta0 -c /var/run/wpa_supplicant.conf &
udhpcpc -i espsta0 &
From the diary: the SSID has been connected (in the diary: FW), and the IP timeout time has been obtained. The test takes 20 seconds.

If this is the issue, could you please share logs of wpa_supplicant+wifi driver dmesg + esp32 firmware logs when the issue is coming? The logs provided [here](https://github.com/espressif/esp-hosted/files/14432892/esp32c3_log.1.zip) shows that the device got connected immediately.

xmlhp, 2024-04-09:
#If this is the issue, could you please share logs of wpa_supplicant+wifi driver dmesg + esp32 firmware logs when the issue is coming? The logs provided [here](https://github.com/espressif/esp-hosted/files/14432892/esp32c3_log.1.zip) shows that the device got connected immediately.
@kapilkedawat ,Yes,got connected d immediately.But obtained IP cost long time.
Connecting to the AP is fast, but getting the IP takes a long time.
After the device is powered on and successfully connected to the AP for the first time, it only takes 3-4 seconds to obtain the IP.
I want to reduce the time it takes to obtain an IP

kapilkedawat, 2024-04-09:
@xmlhp Is it possible for you to share the sniffer capture(along with host/firmware logs) when device is taking too long to get an IP? I can try to check what's happening during that time.

xmlhp, 2024-04-29:
If Bluetooth is turned on, it is in slave mode. It will take a long time to reconfigure the network and obtain the IP. If you turn off Bluetooth before reconfiguring the network, you will get the IP faster.
You can try this use case.

Is it possible for you to share the sniffer capture(along with host/firmware logs) when device is taking too long to get an IP?
How to sniffer capture?