espressif/esp-hosted #346: High ICMP Ping Latency Under Low Network Traffic Conditions

Philipansari, 2024-02-27:
I observed a significant increase in ICMP ping latency **after** a big file (~100MB) was transferred.

## My Setup

- ESP32c3 chip
- OpenWRT v21 linux-host
- First-Generation (latest master)
- WiFi over SPI
- calling `esp-interface-cmd softap_start` & `ifconfig ethap0 192.168.4.5` to host a network
- conncting a linux client to the softap

## Steps to reproduce

1. Start SoftAP mode and let a new client (192.168.4.6) connect to it.
2. Check ping latency with ping 192.168.4.5 and observe normal ping (~10ms).
3. Transfer a large file via SCP: scp root@192.168.4.6:/tmp/big_file /dev/null.
4. Check ping latency again and observe high latency (~5000ms).
5. Check again after a while and note no change.

## Observation

- Until the large file is transferred, the ping stays low.
- As long as a large file is being transferred, the ping stays low.
- As soon as the large file transfer stops, the ping increases (~1000ms to ~5000ms) but there is no packet loss.
- As soon as a large file is being transferred again, the ping returns to normal.
- Flooding the host with ping packets without transferring a file also brings the ping latency back to normal.
- Any high traffic will bring the ping latency back to normal. As soon as traffic stops, the ping latency increases again.
- if more clients are connected to the station, then the ping from one client to another client is not affected from the high latency

This indicates that during high transfer rates, the packets go through without issue. However, transferring only a few small packets (such as ICMP packets) results in them being held back.

### Verify SPI-bus

The same hardware was also used with the next-generation driver, expect that the linux-host connects itself to another station. However, transferring a huge file does not cause any problems afterwards, so the SPI-bus connection should be fine.

## Suggestions

- The GPIO handshake/data_ready signals may not be behaving as expected: Data is only transferred to the Linux host when the buffer is full, instead of sending the data as soon as it becomes available, which could hold back the data and cause high latency on low traffic.
- During the transfer of a large file, the internal buffers might adjust to handle high traffic, but after the file transfer, the buffers do not adjust back to normal operation, potentially leading to a buffer bloat problem ?

## To mention

This problem does not occur in a Raspberry Pi 4 setup. I tested the same source codes, and the latency was always low. I understand that you might not support other platforms, but perhaps someone has an idea where there might be a misconfiguration in my setup?

I have made no changes to the driver itself except for:

- changing GPIOs to match my hardware
- reduced to 10Mhz SPI-clock also for normal operation
- changed from SPI-mode 2 to SPI-mode 1

Philipansari, 2024-02-27:
While investigating this issue i observed that the problem is gone when the interrupt-service-routine of the handshake pin contains a print. Of course this will drastically reduce the speed of the driver and is no option for a work-around.
 
With the following modification the high latency problem is gone:

[esp_spi.c](https://github.com/espressif/esp-hosted/blob/3d4f2f7b5913e951a3af2083c117a88d3638e429/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L102-L107)
```{c}
static irqreturn_t spi_interrupt_handler(int irq, void * dev)
{
	up(&spi_sem);
	esp_info("x"); // added this line here
	esp_verbose("\n");
	return IRQ_HANDLED;
}
```

It is strange that the problem only arises after the first large data transmission.

mantriyogesh, 2024-02-28:
Thank you for all of the details.
Understood the issue fully. Thanks a lot for all the details shared, we will try to reproduce this problem (first on Raspberry pi and otherwise some other platform as possible)

Sorry for the long questions ahead, trying to understand the issue better, as you said Raspberry pi doesn't face this issue
Can you please help with below info:
1. Git commit used at C3 and host
2. Logs:
- If you could capture of the commands and their output
- C3 log
- host kern.log or dmesg from start
3. Host mpu details (for checking base spi driver on Linux)
4. Linux kernel version
5. Do you also have sdio as transport setup (I think not, but thought to ask if issue is also seen there)
6. Do you also use bluetooth at the same time, when issue occurs (I think not, but thought to clarify as rf time is shared)
7. Do you also observe the same behaviour on station mode? (Preferably bluetooth disabled and power save disabled on C3)
8. Does your Linux allowed to be in sleep or power save?
9. Finally, do you have a logic analyser?

Philipansari, 2024-02-28:
Thanks a lot for the support on this issue.

Sorry for the long comment.

1. commit: 3d4f2f7b5913e951a3af2083c117a88d3638e429
2. see below
3. `Freescale i.MX6 Ultralite` Model name `ARMv7 Processor rev 5 (v7l)`
4. v5.4.154
5. unfortunately not, sorry
6. bluetooth was never used/tested
7. see below, yes, indeed, it has the same problem
8. the linux itself is not in any sleep or power save mode
9. yes, i have

### Ping from linux-host to client, file transfer, ping again

```
$ ping 192.168.4.6
PING 192.168.4.6 (192.168.4.6): 56 data bytes
64 bytes from 192.168.4.6: seq=0 ttl=64 time=58.120 ms
64 bytes from 192.168.4.6: seq=1 ttl=64 time=81.676 ms
64 bytes from 192.168.4.6: seq=2 ttl=64 time=104.794 ms
64 bytes from 192.168.4.6: seq=3 ttl=64 time=179.349 ms
64 bytes from 192.168.4.6: seq=4 ttl=64 time=263.912 ms
64 bytes from 192.168.4.6: seq=5 ttl=64 time=207.331 ms
64 bytes from 192.168.4.6: seq=6 ttl=64 time=199.862 ms
64 bytes from 192.168.4.6: seq=7 ttl=64 time=18.787 ms
64 bytes from 192.168.4.6: seq=8 ttl=64 time=42.698 ms
64 bytes from 192.168.4.6: seq=9 ttl=64 time=66.202 ms
64 bytes from 192.168.4.6: seq=10 ttl=64 time=89.950 ms
64 bytes from 192.168.4.6: seq=11 ttl=64 time=113.528 ms
64 bytes from 192.168.4.6: seq=12 ttl=64 time=137.109 ms
64 bytes from 192.168.4.6: seq=13 ttl=64 time=162.851 ms
^C
--- 192.168.4.6 ping statistics ---
14 packets transmitted, 14 packets received, 0% packet loss
round-trip min/avg/max = 18.787/123.297/263.912 ms
```
```
$ scp philip@192.168.4.6:/tmp/tmp.iso /dev/null
philip@192.168.4.6's password: 
tmp.iso          0% 2224KB 772.9KB/s   33:20 ETA^C
```
```
$ ping 192.168.4.6
PING 192.168.4.6 (192.168.4.6): 56 data bytes
64 bytes from 192.168.4.6: seq=0 ttl=64 time=2185.167 ms
64 bytes from 192.168.4.6: seq=1 ttl=64 time=2209.065 ms
64 bytes from 192.168.4.6: seq=2 ttl=64 time=2029.460 ms
64 bytes from 192.168.4.6: seq=3 ttl=64 time=2053.233 ms
64 bytes from 192.168.4.6: seq=4 ttl=64 time=2077.249 ms
64 bytes from 192.168.4.6: seq=5 ttl=64 time=2100.408 ms
64 bytes from 192.168.4.6: seq=6 ttl=64 time=2124.813 ms
64 bytes from 192.168.4.6: seq=7 ttl=64 time=2148.502 ms
64 bytes from 192.168.4.6: seq=8 ttl=64 time=2176.771 ms
64 bytes from 192.168.4.6: seq=9 ttl=64 time=2194.951 ms
64 bytes from 192.168.4.6: seq=10 ttl=64 time=2013.807 ms
64 bytes from 192.168.4.6: seq=11 ttl=64 time=2149.801 ms
64 bytes from 192.168.4.6: seq=12 ttl=64 time=2157.804 ms
64 bytes from 192.168.4.6: seq=13 ttl=64 time=2196.044 ms
^C
--- 192.168.4.6 ping statistics ---
16 packets transmitted, 14 packets received, 12% packet loss
round-trip min/avg/max = 2013.807/2129.791/2209.065 ms
```

If more data than ~2MB is transferred, then the situation gets even worse:

```
$ scp philip@192.168.4.6:/tmp/tmp.iso /dev/null
[... 549 lines cut ...]
64 bytes from 10.8.4.4: seq=1 ttl=64 time=1018.539 ms
64 bytes from 10.8.4.4: seq=2 ttl=64 time=838.238 ms
64 bytes from 10.8.4.4: seq=3 ttl=64 time=1271.463 ms
64 bytes from 10.8.4.4: seq=4 ttl=64 time=281.996 ms
64 bytes from 10.8.4.4: seq=5 ttl=64 time=1319.891 ms
64 bytes from 10.8.4.4: seq=6 ttl=64 time=429.727 ms
64 bytes from 10.8.4.4: seq=7 ttl=64 time=862.875 ms
64 bytes from 10.8.4.4: seq=8 ttl=64 time=1007.921 ms
64 bytes from 10.8.4.4: seq=9 ttl=64 time=1108.313 ms
64 bytes from 10.8.4.4: seq=10 ttl=64 time=1008.716 ms
64 bytes from 10.8.4.4: seq=11 ttl=64 time=438.860 ms
64 bytes from 10.8.4.4: seq=12 ttl=64 time=974.666 ms
64 bytes from 10.8.4.4: seq=13 ttl=64 time=1201.375 ms
64 bytes from 10.8.4.4: seq=14 ttl=64 time=304.147 ms
64 bytes from 10.8.4.4: seq=15 ttl=64 time=1007.910 ms
64 bytes from 10.8.4.4: seq=16 ttl=64 time=659.758 ms
64 bytes from 10.8.4.4: seq=17 ttl=64 time=379.681 ms
64 bytes from 10.8.4.4: seq=18 ttl=64 time=1422.231 ms
64 bytes from 10.8.4.4: seq=19 ttl=64 time=1007.975 ms
64 bytes from 10.8.4.4: seq=20 ttl=64 time=548.048 ms
^C
--- 10.8.4.4 ping statistics ---
21 packets transmitted, 21 packets received, 0% packet loss
round-trip min/avg/max = 281.996/842.172/1422.231 ms
```

mantriyogesh, 2024-03-05:
We might need to reproduce this case, but it might take some time to address, as some other Linux like imx to be setup for this.

Is there any anomaly you observe on logic analyser? Possible to share the dumps?

Philipansari, 2024-03-08:
I tried to spot the root of this error by adding some prints and looking at the logs.

Adding this log function to the `skb` buffer which is [passed to the kernel via `netif_rx()`](https://github.com/espressif/esp-hosted/blob/415226f6a4fe9be5fb0f80231873a79428cc875f/esp_hosted_fg/host/linux/host_driver/esp32/main.c#L450)

```
void print_skb_data(struct sk_buff *skb) {
    // Ensure skb is not NULL
    if (!skb) return;
    printk(KERN_INFO "skb length: %u\n", skb->len);
    print_hex_dump(KERN_INFO, "", DUMP_PREFIX_NONE, 16, 1, skb->data, min(skb->len, 100U), true);
}
```
and logging ICMP reply/requests at kernel logs with this command

```
sudo iptables -I INPUT -p icmp --icmp-type echo-request -j LOG --log-prefix "ICMP Echo Request: "
```

gave me some insight where the problem occurred:

Even if the high latency problem occurs after a large file transfer, i am able to immediately send out anything without any delay. I verified it by sending single ICMP packages to a connected client and looking at the ICMP Echo request on its kernel logs. 

```
ping 192.168.4.6 -c 1
PING 192.168.4.6 (192.168.4.6): 56 data bytes

--- 192.168.4.6 ping statistics ---
1 packets transmitted, 0 packets received, 100% packet loss
```

On the client side i observed an immediate print of 

```
[18534.237240] ICMP Echo Request: IN=wlp0s20f3 OUT= MAC=88:d8:2e:4e:11:64:d4:f9:8d:34:35:19:08:00 SRC=192.168.4.5 DST=192.168.4.6 LEN=84 TOS=0x00 PREC=0x00 TTL=64 ID=35024 DF PROTO=ICMP TYPE=8 CODE=0 ID=5077 SEQ=0
```

So the problem doesn't occur at sending data, it **occurs during receive**.

Putting the skb log function in place gave me an interesting insight: the SPI-transaction (read_data) is happening right on point without any delay. As soon as a clients sends an ICMP request, the skb-buffer print on the linux-host side will be dumped out. But looking at the length of the buffer we can see that the package is obviously not a ICMP request, it is more like a delayed package from the data transfer:

```
... running file transmission (scp command)
[13043.078140] skb length: 1500
[13043.081274] 45 08 05 dc 57 54 40 00 40 06 54 64 c0 a8 04 06  E...WT@.@.Td....
[13043.088425] c0 a8 04 05 00 16 eb 6a bc 2d 51 20 a2 05 09 38  .......j.-Q ...8
[13043.095629] 80 10 01 f5 be 2b 00 00 01 01 08 0a f1 23 57 33  .....+.......#W3
[13043.102800] 4a ff d7 99 13 a1 9f eb eb 5c 87 1b 5c 5d 7c 6b  J........\..\]|k
[13043.109965] b1 5f 5e b4 d6 9f bd db 1c 71 82 d3 9a c4 15 49  ._^......q.....I
[13043.117108] 39 ec 70 0a 7c 22 e6 27 7a 6f bc 42 4b 98 44 44  9.p.|".'zo.BK.DD
[13043.124267] 1a 9c c8 87                                      ....
[13043.144277] skb length: 1500
[13043.147426] 45 08 05 dc 57 55 40 00 40 06 54 63 c0 a8 04 06  E...WU@.@.Tc....
[13043.154657] c0 a8 04 05 00 16 eb 6a bc 2d 56 c8 a2 05 09 38  .......j.-V....8
[13043.161829] 80 18 01 f5 4c 25 00 00 01 01 08 0a f1 23 57 33  ....L%.......#W3
[13043.168975] 4a ff d7 99 dd 1a 84 c1 46 e2 34 53 1a 09 77 0a  J.......F.4S..w.
[13043.176136] df cc 24 89 96 3c 4f 85 3b 84 95 fd 60 a7 a5 7a  ..$..<O.;...`..z
[13043.183296] 93 e9 2d cd 72 33 b9 46 99 12 45 67 f6 70 aa df  ..-.r3.F..Eg.p..
[13043.190454] 8c 1b 44 24                                      ..D$
[13043.202096] skb length: 1500
[13043.205008] 45 08 05 dc 57 56 40 00 40 06 54 62 c0 a8 04 06  E...WV@.@.Tb....
[13043.212222] c0 a8 04 05 00 16 eb 6a bc 2d 5c 70 a2 05 09 38  .......j.-\p...8
[13043.219393] 80 10 01 f5 eb c4 00 00 01 01 08 0a f1 23 57 33  .............#W3
[13043.226536] 4a ff d7 d0 bd 8e ca 75 68 ce c6 6a ba 16 8f 1e  J......uh..j....
[13043.233701] 23 9c ba 2e 9c 41 f4 d9 a9 76 5d b6 e0 dd bc 93  #....A...v].....
[13043.240863] a9 bf 41 9c 0f a3 93 f2 c9 a1 b9 cb cb 12 3f 00  ..A...........?.    # interrupted file-transmission with Ctrl+C (note the kernel-tick)
[13043.248002] f3 3d 5c ea                                      .=\.
[13045.633492] skb length: 1500
[13045.636542] 45 08 05 dc 57 57 40 00 40 06 54 61 c0 a8 04 06  E...WW@.@.Ta....    # executing ICMP requests manually in various intervals (look at the ticks), but data is still from file-transmission
[13045.643770] c0 a8 04 05 00 16 eb 6a bc 2d 62 18 a2 05 09 38  .......j.-b....8
[13045.650941] 80 18 01 f5 ae ac 00 00 01 01 08 0a f1 23 57 33  .............#W3
[... 213 lines cut ...]
I (146871) LEN: 56
I (147891) LEN: 56
I (150641) LEN: 100
I (150651) LEN: 120
I (150761) LEN: 56
I (151791) LEN: 56
I (152811) LEN: 56
I (173081) LEN: 1528
I (173141) LEN: 56
I (174311) LEN: 56
I (175341) LEN: 56
I (178201) LEN: 56
I (178261) LEN: 56
I (179221) LEN: 56
I (179301) LEN: 56
I (184561) LEN: 56
```

Note: the tick offset of the ESP32 and the linux-kernel logs is **12937.065 seconds**

Note: the ESP is sending out short packages than the linux-host is interpreting. For example looking at the package sent from ESP side at `151.791` and arrived at `13088.857` on the linux side is 56 bytes long (including 28 byte header) and the skb-buffer (removed header) is still 1500 bytes long. How can it be? Did i mixed up receiving and sending buffers?

Note: adding prints everywhere can also create strange behaviors since timings can not be adhered

I will check the transmissions again with a logic analyzer the next days, but i want to share the current results with you. Maybe you have an idea whats going wrong?

mantriyogesh, 2024-03-09:
Okay. Thank you again for all the logs.
Understood the scenario, great debug.
Wifi sniffer will solve a lot of questions and get clear understanding, what is this 1500 byte packet.

Some responses-
1. The length 1500 is confusing, what is the real packet? Can only check through the wifi sniffer. Also, what is the packet type (header->if_type) of this big packet?

2. There is always one extra packets from slave to host, for new session.
Let me try to explain some design considerations for spi.
A. SPI is driven by the host always.
B. The SPI should be full duplex
C. As being full duplex, even when either terminal (host or slave) doesn't hold a valid data (nothing to tx), it would send dummy data, with non valid `if_type`, the receiver terminal would discard this packet.
D. Slave needs to be always ready, as host can send data any time. Due to this, if slave doesn't have next valid tx, it will create a dummy packet and keep the spi driver queue ready. But the dummy packet length is always zero. So I don't think this 1500 length is dummy packet.
E. We need to idle the spi bus, when there is no valid data at either terminal
F. When slave has some new data to tx to host, data ready interrupt will be triggered. This is interrupt at host, which will schedule new spi transfer.
Data ready will be high until all packets are drained from slave to host (and host to slave also)
G. As the duration of ping would be 1 sec, which is very high, generally there would be one packet as dummy (last time we had to put in spi driver queue for slave spi readiness, when spi bus was going idle) followed by real data packet response
H. Dummy packet is just 1 packet after spi idle, should not much hog the spi as such and harmless.

4. As you are aware, I think you can cut down logs a bit, either side and print only necessary, as it will cause additional delays (I understand for debug, logs are easiest to do)

5. If the packets (header-if_type) is of type ESP_AP_IF and length unexpected, I think we need to understand what these messages are from the wifi sniffer capture.

7. Bluetooth and station mode is disabled ? Also, add back this dropping logic https://github.com/espressif/esp-hosted/commit/e4ee1a33514fd219b228b844205f499d5b7ee49f#diff-93cd143561776b7c16d5588f4e64301ed06856fc08b7df7a72630acf286d6ec4L224   for any broadcast /multicast data received at `AP` interface?

Philipansari, 2024-03-18:
Hi, i was able to track down the problem a bit further.

I adjust the test case: Instead of calling `scp` (which encrypts data and uses TCP) with a large binary file, i am using a self-written UDP stream of `0xFF` bytes without encryption. This way i do not need a wifi-sniffer to interpret the data. Additionally i observed whats happening on the SPI-bus by connecting a logic analyzer:

It clearly shows that the SPI-bus transmission itself is completely fine and there is no delay or buffered data written to the bus. But the kernel-prints on the linux-host are showing outdated (buffered) data on the kernel-logs right on the receiving function. This means i probably have an issue with the SPI-driver of my linux?

For clarification i made a small graphic to illustrate the setup:

[image]

Here you can take a look at the kernel logs (i wrote some comments what actually should happen): 
[kernel.log](https://github.com/espressif/esp-hosted/files/14635586/kernel.log)

And here is the capture of my logic analyzer (recorded with [Saleae Logic 2](https://www.saleae.com/pages/downloads))
The capture itself is taken on another transmission than shown in the kernel logs, because i disabled the kernel-prints for it)

The situation i observed also relates to the symptoms i found: removing and reloading the kernel-module (esp-hosted driver) **does not solve the buffer problem**, which is interesting because it will:
   1. re-initialize the allocated SPI-peripheral from the kernel
   2. restart the whole ESP32

without any effect.

Last to mention, the next-generation does not have this problem, even though the SPI-driver on my linux seems to have a problem. Does the implementation of the SPI-transmission layer differentiate significantly in the FG/NG? 
Can i merge the SPI-transmission layer from the NG into the FG as a workaround for my case?

mantriyogesh, 2024-03-18:
Well, this looks very confusing & much worrisome.

There are two differences, 
1. https://github.com/espressif/esp-hosted/blob/ee0b7414fadd6d255291059618929832bb23bf64/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L545C23-L545C50
2. spi_sem is used as semaphore instead of the spi as work queue.
Anyway, you can change this one by one or can also move to older commit.

Do let us know the result, just curious.

Philipansari, 2024-03-25:
Finally found the problem: **DMA** was disabled on the linux-system, which caused the SPI-transmission to be interrupted all the time. The unexpected interrupts caused the buffer-bloat situation.

For anyone who is curious if they have the same problem: You have to watch for idle-times of the CLK line during SPI transmissions, which can look like this: (first package is an ICMP ping request and the second one is the pong response)

[image]

After enabling DMA the CLK line does not idle during the transmission:

[image]

It seems the next-generation SPI-transmission layer is more robust against such cases.

closed 2024-03-25 by Philipansari