espressif/esp-hosted #347: stm32F4 SDIO+ESP32 FG

legendss13, 2024-02-28:
Hello:
I have successfully run through the ESP32-SDIO-AT firmware using the SDIO interface (4 wires) in stm32, but the data transfer speed of the AT command is too slow, and sometimes I need to transfer 300k data. Therefore, I want to use the hostedfg solution, but there is little relevant information and I don't know how to start, so there are currently two questions:
1. Do I need to modify my hardware circuit?
2. Where do I need to port the STM32 software?

In addition, the firmware of ESP32 has been compiled according to the instructions in the documentation, and below is the log on the ESP32 side：
bootloader[0m
[0;32mI (30) boot: compile time Feb 27 2024 16:34:42[0m
[0;32mI (30) boot: Multicore bootloader[0m
[0;32mI (35) boot: chip revision: v1.1[0m
[0;32mI (39) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (43) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (48) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (52) boot: Enabling RNG early entropy source...[0m
[0;32mI (5
[13:31:00.341]收←◆8) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (91) boot:  3 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (98) boot:  4 ota_0            OTA app          00 10 00110000 00100000[0m
[0;32mI (106) boot:  5 ota_1            OTA app          00 11 00210000 00100000[0m
[0;32mI (113) boot: End of partition table[0m
[0;32mI (118) boot: Defaulting to factory image[0m
[0;32mI (122) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=26844h (157764) map[0m

[13:31:00.393]收←◆[0;32mI (188) esp_image: segment 1: paddr=0003686c vaddr=3ffbdb60 size=05834h ( 22580) load[0m
[0;32mI (197) esp_image: segment 2: paddr=0003c0a8 vaddr=40080000 size=03f70h ( 16240) load[0m
[0;32mI (204) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=947e4h (608228) map[0m

[13:31:00.629]收←◆[0;32mI (424) esp_image: segment 4: paddr=000d480c vaddr=40083f70 size=1b560h (111968) load[0m

[13:31:00.693]收←◆[0;32mI (486) boot: Loaded app 
[13:31:00.714]收←◆from partition at offset 0x10000[0m
[0;32mI (487) boot: Disabling RNG early entropy source...[0m
[0;32mI (498) cpu_start: Multicore app[0m
[0;32mI (499) cpu_start: Pro cpu up.[0m
[0;32mI (499) cpu_start: Starting app cpu, entry point is 0x40081490[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (519) cpu_start: Pro cpu start user code[0m
[0;32mI (519) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (519) cpu_start: Application information:[0m
[0;32mI (523) cpu_start: Project name:     network_adapter[0m
[0;32mI (529) cpu_start: App version:      1[0m
[0;32mI (533) cpu_start: Compile time:     Feb 27 2024 16:34:17[0m
[0;32mI (540) cpu_start: ELF file SHA256:  8920c917a8fe0cf2...[0m
[0;32mI (545) cpu_start: ESP-IDF:          v5.1.2-dirty[0m
[0;32mI (551) cpu_start: Min 
[13:31:00.791]收←◆chip rev:     v0.0[0m
[0;32mI (556) cpu_start: Max chip rev:     v3.99 [0m
[0;32mI (560) cpu_start: Chip rev:         v1.1[0m
[0;32mI (565) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (573) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (578) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (584) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (591) heap_init: At 3FFCC358 len 00013CA8 (79 KiB): DRAM[0m
[0;32mI (597) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (603) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (610) heap_init: At 4009F4D0 len 00000B30 (2 KiB): IRAM[0m
[0;32mI (617) spi_flash: detected chip: generic[0m
[0;32mI (620) spi_flash: flash io: dio[0m
[0;32mI (625) coexist: coex firmware version: b6d5e8c[0m
[0;32mI (630) app_start: Starting scheduler on CPU0[0m
[0;32[0;32mI (635) app_start: Starting scheduler on CPU1[0m
mI (635) main_task: Started on CPU0[0m
[0;32mI (643) main_task: Calling app_main()[0m
[0;32mI (645) NETWORK_ADAPTER: *********************************************************************[0m
[0;32mI (653) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5                        [0m
[0;32mI (661) NETWORK_ADAPTER:                 Transport used :: SDIO only                     [0m
[0;32mI (667) NETWORK_ADAPTER: *********************************************************************[0m
[0;32mI (675) NETWORK_ADAPTER: Supported features are:[0m
[0;32mI (679) NETWORK_ADAPTER: - WLAN over SDIO[0m
[0;32mI (683) ESP_BT: - BT/BLE[0m
[0;32mI (685) ESP_BT:    - HCI Over SDIO[0m
[0;32mI (689) ESP_BT:    - BT/BLE dual mode[0m
[0;32mI (693) NETWORK_ADAPTER: capabilities: 0x1d[0m
[0;32mI (705) BTDM_INIT: BT controller compile version [ec4ac65][0m
[0;32mI (705) BTDM_INIT: Bluetooth MAC: 64:b7:08:3f:6f:2e[0m
[0;32mI (709) phy_init: phy_version 4780,16b31a7,Sep 22 2023,20:42:16[0m

[13:31:01.340]收←◆[0;32mI (1033) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 64:b7: 8:3f:6f:2e[0m
[0;32mI (1033) SDIO_SLAVE: Using SDIO interface[0m
[0;32mI (1035) SDIO_SLAVE: sdio_init: ESP32 SDIO timing: 0
[0m
I (1045) wifi:wifi driver task: 3ffda0f8, prio:23, stack:6656, core=1
I (1045) wifi:wifi firmware version: 91b9630
I (1047) wifi:wifi certification version: v7.0
I (1051) wifi:config NVS flash: disabled
I (1053) wifi:config nano formating: disabled
I (1057) wifi:Init data frame dynamic rx buffer num: 32
I (1061) wifi:Init static rx mgmt buffer num: 5
I (1063) wifi:Init management short buffer num: 32
I (1067) wifi:Init dynamic tx buffer num: 64
I (1071) wifi:Init static rx buffer size: 1600
I (1075) wifi:Init static rx buffer num: 16
I (1077) wifi:Init dynamic rx buffer num: 32
[0;32mI (1081) wifi_init: rx ba win: 6[0m
[0;32mI (1083) wifi_init: tcpip mbox: 32[0m
[0;32mI (1087) wifi_init: udp mbox: 6[0m
[0;32mI (1089) wifi_init: tcp mbox: 6[0m
[0;32mI (1093) wifi_init: tcp tx win: 5744[0m
[0;32mI (1097) wifi_init: tcp rx win: 5744[0m
[0;32mI (1099) wifi_init: tcp mss: 1440[0m
[0;32mI (1103) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (1107) wifi_init: WiFi RX IRAM OP enabled[0m
I (1113) wifi:mode : null

mantriyogesh, 2024-02-28:
Hello @legendss13 ,

If you have already used SDIO with ESP-AT, I think you should be able to run ESP-Hosted as well.

Answers to your questions:
1. Pull-ups are needed (See exact requirement in [porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md)
2. Porting layer: following the answer:

Porting layer is available for ESP-Hosted.

Ready port layer:
| Branch | port / doc | status | User level thoughts |
|:------:|:---------:|:-------------:|:----------------------------------------:|
| master | [doc](esp_hosted_fg/docs/MCU_based_host/MCU_based_readme.md) | should work, but no lwip | STM we used did not have SDIO, but had SDMMC. So we needed small upper HAL layer rewrite to get STM SDIO working | HAL layer may or may not be needed if STM now supports SDIO out of the box |
| [feature/esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host) | [Port dir](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host/host/port) | SDIO with ESP works superb | ESP as SDIO master port available, new port layer for STM need to be done |

The branch [feature/esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host) has clear port layer is different from master, 
This branch works very reliably for SDIO with lwip udp tx/rx touching 50Mbps Over The Air.

**SDIO master used:** ESP32 / ESP32-S3 / ESP32-P4 (because as if now, these only ESP chipsets expose SDIO master peripheral)
**SDIO slave used:** ESP32 or ESP32-C6 (because as if now, these are only chipsets that expose SDIO slave peripheral)
 

 
@claydonkey had done some SDIO porting for [feature/esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host) on STM32 in #287 . But Cannot fetch current status of it.

Could you please let us know which STM32 you use, and it has SDMMC support or SDIO also is supported out of the box?

legendss13, 2024-02-28:
I'm currently using STM32 F429IGT6.
As you said, my hardware connection should be fine. I already have the full LWIP stack available in my program and I've verified it in ETH without any problems.
My current thinking is: is it okay to directly connect the transceiver interface of SDIO with the transceiver task of LWIP?

/**
 * @brief This function should do the actual transmission of the packet. The packet is
 * contained in the pbuf that is passed to the function. This pbuf
 * might be chained.
 *
 * @param netif the lwip network interface structure for this ethernetif
 * @param p the MAC packet to send (e.g. IP packet including MAC addresses and type)
 * @return ERR_OK if the packet could be sent
 *         an err_t value if the packet couldn't be sent
 *
 * @note Returning ERR_MEM here if a DMA queue of your MAC is full can lead to
 *       strange results. You might consider waiting for space in the DMA queue
 *       to become available since the stack doesn't retry to send a packet
 *       dropped because of memory failure (except for the TCP timers).
 */

static err_t low_level_output(struct netif *netif, struct pbuf *p)
{
  uint32_t i = 0U;
  struct pbuf *q = NULL;
  err_t errval = ERR_OK;
  ETH_BufferTypeDef Txbuffer[ETH_TX_DESC_CNT] = {0};

  memset(Txbuffer, 0 , ETH_TX_DESC_CNT*sizeof(ETH_BufferTypeDef));

  for(q = p; q != NULL; q = q->next)
  {
    if(i >= ETH_TX_DESC_CNT)
      return ERR_IF;

    Txbuffer[i].buffer = q->payload;
    Txbuffer[i].len = q->len;

    if(i>0)
    {
      Txbuffer[i-1].next = &Txbuffer[i];
    }

    if(q->next == NULL)
    {
      Txbuffer[i].next = NULL;
    }

    i++;
  }

  TxConfig.Length = p->tot_len;
  TxConfig.TxBuffer = Txbuffer;
  TxConfig.pData = p;

  pbuf_ref(p);

  **HAL_ETH_Transmit_IT(&heth, &TxConfig);
  while(osSemaphoreAcquire(TxPktSemaphore, TIME_WAITING_FOR_INPUT)!=osOK)

  {
  }

  HAL_ETH_ReleaseTxPacket(&heth);**

  return errval;
}

/**
 * @brief Should allocate a pbuf and transfer the bytes of the incoming
 * packet from the interface into the pbuf.
 *
 * @param netif the lwip network interface structure for this ethernetif
 * @return a pbuf filled with the received packet (including MAC header)
 *         NULL on memory error
   */
static struct pbuf * low_level_input(struct netif *netif)
{
  struct pbuf *p = NULL;

  if(RxAllocStatus == RX_ALLOC_OK)
  {
    **HAL_ETH_ReadData(&heth, (void **)&p);**
  }

  return p;
}

legendss13, 2024-02-28:
Sorry, the code may not be clear：
[image]

legendss13, 2024-02-28:
> Sorry, the code may not be clear： ![752CF786-866C-49f1-843A-D91AA0E8D7B9](https://private-user-images.githubusercontent.com/42710532/308461612-20c44057-36e2-495e-88d6-6ad4aac2aa2c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxMDYyMjgsIm5iZiI6MTcwOTEwNTkyOCwicGF0aCI6Ii80MjcxMDUzMi8zMDg0NjE2MTItMjBjNDQwNTctMzZlMi00OTVlLTg4ZDYtNmFkNGFhYzJhYTJjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI4VDA3Mzg0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM4NDlmN2MxM2I3OTgxYTM3NGM0ZDRjNzIyZjBiMjcwMGRlZWYxMTJlODk2YzQ0N2U2ZWUyZDgzYjVhMTUzZDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.yvFNq8WoJpNWeauYk9w-zPaeefYb1x14oZFIfu4-28U)

I refer to the practice here https://github.com/drorgl/stm32-lwip-esp-hosted/tree/master/stm32-lwip

mantriyogesh, 2024-02-28:
1. https://github.com/espressif/esp-hosted/issues/347#issuecomment-1968403774 :
It is little old code base
2. It is only SPI. For SDIO how do you want to proceed?
3. As I check in https://www.mouser.sg/datasheet/2/389/stm32f427vg-1760963.pdf, your device support out of the box SDIO, so I suppose the HAL also would support.

> Sorry, the code may not be clear： [...]

right, this would be a hook function (may be from lwip or any networking interface).
You can hook it to send_to slave(...) or esp_hosted_tx(...)

Basically correct path, for from scratch porting. but the new branch, already takes care of this.

For example, in case of new branch, along with 
[esp_wifi_remote_21feb23.tar.gz](https://github.com/espressif/esp-hosted/files/14432174/esp_wifi_remote_21feb23.tar.gz)

if you browse,
you would see that the Wi-Fi netif ends(hooks) tx in _transport_drv.c_ : `transport_drv_sta_tx()` -> _sdio_drv.c_ : `esp_hosted_tx()` which sends the buffer in queue, _to_slave_queue_ which on dequeue, sent through the port function, `g_h.funcs->_h_sdio_write_block()` . This ends to the (ESP's) ported function _sdio_wrapper.c_ :  port function, **hosted_sdio_write_block()**

legendss13, 2024-02-29:
我们可能没在一个时区，抱歉不能及时回复。
目前我使用的F429 HAL 并没有看到SDIO 的驱动程序只有SD和MMC，因此我参考了esp-sdio中关于STM32的历程进行了sdio协议代码移植。
是的我看到了新的驱动程序，他看起来更加清晰独立，我会尝试移植使用它。
我希望未来能有一个完整的LWIP+hosted 的方案在stm32中展示。它可以是任意的一款芯片。这样将会很棒，来帮助像我一样的菜鸟节省时间。

legendss13, 2024-02-29:
We may not be in a time zone, sorry we can't reply in time.
The F429 HAL I'm using at the moment doesn't see SDIO drivers only SD and MMC, so I did a sdio protocol code port based on the STM32 journey in esp-sdio.
Yes I saw the new driver, he looks much clearer and independent, I will try to port it with it.
I hope to have a full LWIP+hosted solution to be shown in stm32 in the future. It can be any chip. It's going to be great to help rookies like me save time

legendss13, 2024-03-05:
Now I'm making new progress, but I don't know what that means. Because the STM32 program still can't be run. Can you give me a look at where I'm going to troubleshoot next?
Here are some **logs from ESP32**：

I (1063) wifi:Init dynamic rx buffer num: 32
[0; 32mI (1067) wifi_init: rx ba win: 6[0m
[0; 32mI (1069) wifi_init: tcpip mbox: 32[0m
[0; 32mI (1073) wifi_init: udp mbox: 6[0m
[0; 32mI (1075) wifi_init: tcp mbox: 6[0m
[0; 32mI (1079) wifi_init: tcp tx win: 5744[0m
[0; 32mI (1081) wifi_init: tcp rx win: 5744[0m
[0; 32mI (1085) wifi_init: tcp mss: 1440[0m
[0; 32mI (1089) wifi_init: WiFi IRAM OP enabled[0m
[0; 32mI (1093) wifi_init: WiFi RX IRAM OP enabled[0m
I (1099) wifi:mode : null

[17:13:05.019]← Close◆[0; 32mI (4985) NETWORK_ADAPTER: Start Data Path[0m

[17:13:05.168]← closed◆[0; 32mI (5105) NETWORK_ADAPTER: Initial set up done[0m
[0; 32mI (5105) slave_ctrl: event ESPInit[0m
[0; 32mI (5105) main_task: Returned from app_main()[0m

**logs from stm32：**
V sdio_transport: get_intr
D sdio_transport: clear_intr: 00080000
V STM32_sdio: Send size:4

V STM32_sdio: CMD53 response

V sdio_transport: get_intr
D sdio_transport: clear_intr: 00CC4000
V STM32_sdio: Send size:4

V STM32_sdio: CMD53 response

D sdio_transport: get_packet: slave len=23, max read size=1536
V sdio_transport: get_intr
D sdio_transport: clear_intr: 004C7000
V STM32_sdio: Send size:4

V STM32_sdio: CMD53 response

V sdio_transport: get_intr
D sdio_transport: clear_intr: 008C0000
V STM32_sdio: Send size:4

V STM32_sdio: CMD53 response

D sdio_transport: get_packet: slave len=34, max read size=1536
V sdio_transport: get_intr
D sdio_transport: clear_intr: 004C7000
V STM32_sdio: Send size:4

V STM32_sdio: CMD53 response

V sdio_transport: get_intr

[17:15:14.890]收←◆V sdio_transport: get_intr

[17:15:14.922]收←◆V sdio_transport: get_intr

[17:15:14.951]收←◆V sdio_transport: get_intr

[17:15:14.984]收←◆V sdio_transport: get_intr

[17:15:15.014]收←◆V sdio_transport: get_intr

STM32 keeps printing repeatedly： V sdio_transport: get_intr

legendss13, 2024-03-06:
Thank you for your patience, I've changed to interrupt to read the data.
But I found a new problem, and after I initialized it, I received the following data:

0x5 00 0xb 00 0xc 00 00 00 00 00 00 

0x3 00 0x16 00 0xc 00 00 00 00 00 00 00 0x1 0x8 00 0x63 0x74 0x72 0x6c 0x45 0x76 0x6e 

I don't know what this means, the code says it should be dirty data?Here's the original print:
[image]
[image]

I checked the code and the stm32 side didn't send data to esp-sdio before that。
Here's my initialization process：
[image]
The program is stuck at the mark。（Some related functions：）
[image]
[image]

legendss13, 2024-03-06:
The error here looks like something is not initialized mainly because it is not called here, where is it called when it is initialized?

[image]
[image]
[image]
[image]

legendss13, 2024-03-06:
Eventually leads to an error here：
[image]

mantriyogesh, 2024-03-06:
Actually, many of such semphore issues are fixed in 'feature/esp_as_mcu_host' branch.

Possibly the issue you face is because the semaphores are binary but should be counting.
But clearly, porting 'feature/esp_as_mcu_host' branch would have helped you much.

I am actually porting that branch for SPI, and once I am finished, I will be doing it for SDIO. but it might take 2 weeks or so..

legendss13, 2024-03-06:
Thanks for your reply,
I also looked at "feature/esp_as_mcu_host" before this, and it seems that the overall process is not much worse, and I am still in the verification testing stage, so I haven't ported that branch code.
I'll try to fix this semaphore, it looks like it's being called when the semaphore isn't initialized.

legendss13, 2024-03-06:
I have made a new progress, is this already connected to WiFi?Why does the AP prompt that there is no device connected?
[image]
And using DHCP does not obtain an IP address。

mantriyogesh, 2024-03-06:
It is connected to Wi-Fi (AP).

The master branch only showcases 'ARP'. It doesn't run full lwip.
So you may not be directly expect dhcp running.

You can try to run 'arping' as mentioned in the [ARP documentation](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/MCU_based_host/Getting_started.md#23-arp-testing-), to verify if it is communicating with Wi-Fi.

'arping' is not layer 3. it just showcase that the packets run through the Wi-Fi.
For lwip, you would still need to integrate lwip (feature/esp_as_mcu_host has it all integrated for esp32)

legendss13, 2024-03-06:
You're right, it's true that there is a problem with my lwip configuration, I have successfully obtained the IP after modifying it, and I will do the next TCP connection adaptation. Thank you for your patience and look forward to Hoseted+LWIP

legendss13, 2024-03-06:
Now I can send data to the MQTT server as normal. But three problems were encountered:
1. I'll come back when I'm connected to WiFi
[15:02:40.877] Received← ◆ 01 08 00 63 74 72 6c 52 65 73 70 02 08 00 08 02 10 cc 01 e2 0c 00 
NULL MAC returned
Failed to get wifi mac”。 The frequency of this problem is still quite high, and I don't know the specific reason (is it because the frequency of reconnecting the AP is fast?).
2. When I send data, the data below 1k can be sent normally, and if it is greater than this value, it will not be returned, I can make sure that my LWIP and mqtt work normally. Limited to ESP32?
3. Occasionally, CMD53 time out will fail to read data.

mantriyogesh, 2024-08-16:
Hello @legendss13 ,

did you get it through ahead?

legendss13, 2024-08-16:
yes... use sdio+lwip.

KL
***@***.***

mantriyogesh, 2024-08-16:
Are there any changes you need to make?
Can you please submit PR, if so, alternatively, can point to your fork or branch?

Someone else is also having exact issue as that of you.