espressif/esp-hosted #348: Is ESP32 EN RESET PIN a must have?

mike-2020, 2024-03-01:
I would like to use SDIO to connect ESP32 with host. Besides the PINs required by SDIO, the doc mentions an PIN to ESP32 EN, to reset ESP32. Is this a must-have PIN? Can I just pull up ESP32's EN pin, and so ESP32 will always be started once power is on?

mantriyogesh, 2024-03-01:
Is you host microcontroller or Linux based host?

> ESP32 will always be started once power is on

Yes, you can do that. Basically, rarely ESP goes in bad state if the communication on transport is broken in middle, so in case of Linux when the kernel module is loaded, we ask to reset the esp.

Issue of esp's invalid state, was observed much in case of spi transport.
If you are going to load the kernel driver on boot, then you can reset ESP (or power on ) that time.
Let us know if you get issues with sdio transport. (We had tested last time that reset pin was not much needed in case of sdio. But also we did not much test negative scenarios like incomplete transport transfer etc).

mike-2020, 2024-03-01:
The host is a SoC running Linux. The power on sequence will look like this:
1. ESP32 is powered on (ESP32 is powered by a dedicated LDO).
2. ESP32 enables PMIC (using a PIN), and then PMIC powers on the SoC host (PMIC provides power to SoC host).

You can see why I need ESP32 be always ON. If ESP32 is reset, during the reset, ESP32's PIN will go low, and then the PMIC will be reset, and then the SoC host will be reset.

Do you see any potential issues with this power sequence?

mantriyogesh, 2024-03-01:
Well now I understand the problem.
Anyway, as I understood, the Linux will be on after ESP is on, Linux power depends on ESP power state.

The kernel module esp32_sdio.ko will be inserted after linux boot sequence.
If you do not plan to remove (and reinsert as such later) the esp32_sdio.ko kernel module, I don't think there should be much problem.

Btw, using FG / NG?

mantriyogesh, 2024-05-29:
Can we close if answered your question?

mike-2020, 2024-05-30:
Yes, please.

closed 2024-05-30 by mantriyogesh