espressif/esp-hosted #349: Run this hosted solution along with some other app on ESP32

mike-2020, 2024-03-01:
I would like to run this solution on an ESP32 to provide WIFI network to my host SoC (Linux).

But at the same time, I also need to run some other App/Code on the same ESP32. And the App/Code also needs a WIFI network to use. Is it possible?

mantriyogesh, 2024-03-01:
It is technically possible, but very complex use case to support our of box.

Options to explore by yourself (can help in guiding, but it will not be supported in esp-hosted as such)

Opt (1) You can setup PPPoS to communicate in between the slave and host as separate netif i.e. separate network for slave-host ( but need lwip code also at slave, obviously with its own memory and heap requirement). Check [eppp-link](https://github.com/espressif/esp-protocols/blob/master/components/eppp_link/README.md)

Send packet from slave to host and reroute packet through normal lwip on sta netif through sdio through slave to wifi

Opt (2) if simple packets, you can inspect the packet received at sta and send simple response right from sta , without sending to host at all

Opt (3) if using FG, can reuse serial or priv if and send packet from slave to host and then host would take appropriate action

mike-2020, 2024-03-03:
I am not sure I follow your anwser, or you understand my question :-)

Requirement: (1) use ESP32 as a WIFI for a Linux host; (2) run some code (e.g. http client) on the same ESP32, and the code needs access WIFI network/internet.

My understanding is that this hosted solution would foward all recieved WIFI packets to host SoC via SPI/SDIO without checking MAC address in the packet. That is being said, if the solution can check packet's MAC address, send it to host SoC if it belongs to host, send it to ESP32 itself TCP/IP stack if it belongs to ESP32 itself, then the requirements can be fulfilled?

mantriyogesh, 2024-03-05:
Actually, we set the MAC address of slave ESP and host to be same, such that the Wi-Fi Rx at slave is done and packet relayed to host.

I am still thinking about your proposal if it could be do-able or not.

Another way I had thought for your use case, to use
**station (hosted) + softap (native)**

Station mode would work in hosted mode, whereas softap mode configured at slave (non hosted, not visible at Linux). This way, the new network can be used at  ESP and clients connected to softap can use this network. But be careful to remove/stub softap APIs from Linux as they will overwrite the slave softap.

But this will work fine if you use different network address.
For same network address, will be complicated use case and in that case something as you had suggested need to think of.

mantriyogesh, 2024-03-05:
Hello @mike-2020 ,
https://github.com/espressif/esp-hosted/issues/349#issuecomment-1974971362
is do-able.

It might not be just as straight forward with ESP-Hosted. But might need to customize the ESP-Hosted.

Theoretically, if you create bridge of two netif on ESP (ESP-IDF supports bridge), one with native station network interface, say 'sta0' and second say 'sta1' over Linux host.

ESP slave natively would connect the AP, and get 'sta0' as DHCP ip address (firstt netif in bridge).
Linux counterpart, sta1 will also trigger the dhclient, which will also get different ip address from AP using its own mac address.

Now theoretically, different ip addresses would be present on both esp and host.