espressif/esp-hosted #353: Performance comparation

mike-2020, 2024-03-11:
Look at the performance benchmark in the document. 

- ESP32 SDIO outperforms all other chips.
- In SPI mode, ESP32-C3 outperforms all other chips.
- No data for ESP32-C6 SDIO mode.

How to explain this result?
What is the bottleneck?
Can you also add comparation to a traditional/dedicated WIFI module (SDIO 2.0)? How much would the ESP32 hosted solution be slower?

mantriyogesh, 2024-03-11:
The SPI bottleneck is the max SPI slave clock supported.

### SPI
Current reading for SPI are taken with:
ESP32: 10MHz
Rest: **30MHz** SPI clock.

One can gradually increase to **max** SPI freq possible as:
ESP32: 10Mhz
ESP32-C2/C3/S2/S3 : 40Mhz
ESP32-C6: 26Mhz

If you are thinking why max slave frequency vary,
These actually limited by the APB clocks in use.

### SDIO
Some diff wrt SDIO:
ESP32-SDIO could be used if you wish to use classic-BT + BLE 4.2 together. Also dual code CPU.
ESP32-C6 could be preferred if only BLE 5.3 is needed

Current clocks used for reading are 40MHz, 4 bit SDIO.
You can gradually increase SDIO clock till max slave supported clock as 50MHz for both ESP32 and ESP32-C6.

Command `cat /sys/kernel/debug/mmc1/ios` could be used for reading current max SDIO clock.

SDIO supports HIGH_SPEED_MODE for both ESP32 and ESP32-C6. For high performance, one should concentrate on the configurations used for Wi-Fi, LWIP, etc.
For example, 
```
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=40
CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP_WIFI_TX_BA_WIN=32
CONFIG_ESP_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP_WIFI_RX_BA_WIN=32
CONFIG_ESP_WIFI_NVS_ENABLED=n
CONFIG_ESP_WIFI_ENABLE_WIFI_RX_STATS=n
CONFIG_ESP_WIFI_ENABLE_WIFI_TX_STATS=n
CONFIG_FREERTOS_HZ=1000
```
Some of these parameters related to Wi-Fi are dependent upon the ESP chipset used (and available memory there).
sdkconfig.defaults.<esp_chipset> files in https://github.com/espressif/esp-idf/tree/master/examples/wifi/iperf could be referred for base stable Wi-Fi configs supported.

[image]

Apart from ESP32-C6, other chipsets reading are little old. What I mean, they might perform better than the readings because of latest optimizations.

I will update ESP32-C6 performance figures in documenttation.

mantriyogesh, 2024-03-11:
Change added at a4337ac5e90bc048efc13ecee63c1e9070d32220

mantriyogesh, 2024-03-11:
With ESP32-C6 figures, this would be easier now.
Unless you have anything else in mind, can we close this issue?

mike-2020, 2024-03-12:
Thanks. This is very helpful.

closed 2024-03-12 by mike-2020