espressif/esp-hosted #354: esp32 wakeup from WOW Magic Packet

pritpalm-kuk, 2024-03-12:
1) We have tested esp32 wake up using the Netgear Nighthawk R7000 router
1.2) The esp32 is connected to the Netgear Nighthawk R7000 router
1.3) Laptop is connected to the Netgear Nighthawk R7000 router
1.3) We put the appliance to sleep and can wake it up within 40 mins of it being in sleep (100% reproducible)
1.4) We put the appliance to sleep and can not wake it up after 50 mins of the appliance being in sleep  (100% reproducible)  
2) We have tested esp32 wake up using the Sky Q Hub router
2.1) The esp32 is connected to the Sky Q Hub router
2.2) Laptop is connected to the Sky Q Hub router
2.3) We put the appliance to sleep and can only wake it up within 2 mins of being asleep (100% reproducible) 
2.4) We changed the Sky Q Hub router to allow inbound ICMPv6 Echo Requests.
2.5) With respect to point 2.4 above; we put the appliance to sleep and can wake it up within 20 mins of it being in sleep (100% reproducible).
2.6) With respect to point 2.5 above; after 20 mins the appliance can not be woken up.
3) We have three wake up sources; wakeup from dial, capacitance touch and esp32 wakeup-over-lan via magic packet implementation.
3.1) There are no issue when using dial and display capacitance touch to wake up the appliance
3.2) Using esp32 wake up via a ping (wakeup-over-lan via magic packet) we get Destination Host Unreachable.

From the above; the esp32 wake up implementation is not correct.

mantriyogesh, 2024-03-13:
@kapilkedawat ++

kapilkedawat, 2024-03-13:
Hi @pritpalm-kuk , Please do a wireless sniffer capture(from the bootup till issue occurs) and share that with us along with host and firmware logs for the whole session.

pritpalm-kuk, 2024-03-13:
Do you require the wireless sniffer log to have any filters?
I am using wireshark; is this ok; if not please recommend a wireless sniffer application

kapilkedawat, 2024-03-13:
Hi @pritpalm-kuk,

I mentioned in https://github.com/espressif/esp-hosted/issues/354#issuecomment-1993495631 please share the logs starting from bootup till issue occurs(please do not cut the logs in between or in starting).

Yes, wireless sniffer capture will be required, you will need a spare linux system which supports wireless sniffing capabilities.
This is the normal sequence of commands to put the system in sniffing mode.

$ sudo service NetworkManager stop #stop network manager to interact with wireless interface
$ sudo killall -15 wpa_supplicant #kill wpa_supplicant if its managing wireless interface
$ sudo ifconfig <wifi_iface> down
$ sudo iwconfig <wifi_iface> mode monitor
$ sudo iwconfig <wifi_iface> channel <ap_channel> #Please make sure you confirm the AP channel
$ sudo ifconfig <wifi_iface> up

These commands are OS specific and you may need to modify some based on your distro.
Once you put the interface in monitor mode, you can start wireless and select this interface to sniffer.

Also please explain your topology and from where to where you are trying to ping.

pritpalm-kuk, 2024-03-13:
> I mentioned in #354 (comment)

Hi Kapil

Thank you for the update:

When we issue the command sudo iwconfig <wifi_iface> mode monitor with
<wifi_iface>
set to espsta0; we get the following error:

*Error for wireless request "Set Mode" (8B06) :    SET failed on device
espsta0 ; Operation not supported.*

From the above we are unable to set up wireless sniffer capture
Please provide us an alternative megthod

Regards

Pritpal

-- 
This e-mail is intended only for the person to whom it is addressed and may 
contain confidential information. If an addressing or transmission error 
has misdirected this e-mail, please notify the author by replying to this 
e-mail. If you are not the intended recipient you must not use, disclose, 
print or rely on the contents of this e-mail and should immediately and 
permanently delete it.

Kenwood Limited is a company registered in England 
with registered number 872044. The registered office of Kenwood Limited is 
1 Kenwood Business Park, New Lane, Havant, Hampshire, PO9 2NH.

Kenwood 
Limited may monitor outgoing and incoming e-mails and other 
telecommunications on its e-mail and telecommunications systems for the 
purpose of ensuring compliance with the law and with Kenwood Limited's 
policies.

kapilkedawat, 2024-03-13:
Hi @pritpalm-kuk, as I mentioned in the previous comment, these commands needs to be run on a spare linux host which has wireless sniffing capabilies. 

>> **Yes, wireless sniffer capture will be required, you will need a spare linux system which supports wireless sniffing capabilities.**

pritpalm-kuk, 2024-03-13:
Hi Kapil

We have tried to use the wireless sniffer a rasberrypi.
I am afraid we get the same results.

Regards

Pritpal

-- 
This e-mail is intended only for the person to whom it is addressed and may 
contain confidential information. If an addressing or transmission error 
has misdirected this e-mail, please notify the author by replying to this 
e-mail. If you are not the intended recipient you must not use, disclose, 
print or rely on the contents of this e-mail and should immediately and 
permanently delete it.

Kenwood Limited is a company registered in England 
with registered number 872044. The registered office of Kenwood Limited is 
1 Kenwood Business Park, New Lane, Havant, Hampshire, PO9 2NH.

Kenwood 
Limited may monitor outgoing and incoming e-mails and other 
telecommunications on its e-mail and telecommunications systems for the 
purpose of ensuring compliance with the law and with Kenwood Limited's 
policies.

pritpalm-kuk, 2024-03-13:
Hi Kapil

What is wrong with using wireshark?

kapilkedawat, 2024-03-13:
Hello @pritpalm-kuk, yes, unfortunately raspberry pi doesn't support WiFi monitor mode natively, you have to use an external WiFi dongle which supports WiFi monitor mode. 

Let's say your network topology is as follows:

PC1 <--LAN/WiFi--> router <--WiFi -> ESP

we want to capture the packets over the air between router and ESP which will tell us whether router has forwarded the packet to ESP or not.

pritpalm-kuk, 2024-03-13:
Hi Kapil

Why can we not use wireshark?

pritpalm-kuk, 2024-03-13:
Also I have already shared the logs
see below 
[ep32-fw-log.txt](https://github.com/espressif/esp-hosted/files/14587525/ep32-fw-log.txt)
[kernel-log.txt](https://github.com/espressif/esp-hosted/files/14587526/kernel-log.txt)
[wireshark-log.txt](https://github.com/espressif/esp-hosted/files/14587527/wireshark-log.txt)

kapilkedawat, 2024-03-13:
@pritpalm-kuk Wireshark is an application to capture the packet on any of the network interface. For sniffing WiFi packets you need to start wireshark on WiFi interface. However you need to put the interface in monitor mode so that it can get the packets which are not destined to it. You can read more about it on internet.

>> [ep32-fw-log.txt](https://github.com/espressif/esp-hosted/files/14587525/ep32-fw-log.txt)
Firmware logs are not from bootup, please share from bootup till the issue occur.

Also please explain your topology in detail.

pritpalm-kuk, 2024-03-13:
Topology we are using is detailed below

PC (192.168.199.9) <--LAN/WiFi--> router <--WiFi -> ESP (192.168.199.8)

pritpalm-kuk, 2024-03-13:
please take a read of https://osqa-ask.wireshark.org/questions/31856/how-to-capture-packets-between-2-ips/

pritpalm-kuk, 2024-03-13:
Hi Kapil

Can you please attend the ESP Hosted Wake Up on Lan meeting @ 11:15am U.K
time.

Regards

Pritpal

-- 
This e-mail is intended only for the person to whom it is addressed and may 
contain confidential information. If an addressing or transmission error 
has misdirected this e-mail, please notify the author by replying to this 
e-mail. If you are not the intended recipient you must not use, disclose, 
print or rely on the contents of this e-mail and should immediately and 
permanently delete it.

Kenwood Limited is a company registered in England 
with registered number 872044. The registered office of Kenwood Limited is 
1 Kenwood Business Park, New Lane, Havant, Hampshire, PO9 2NH.

Kenwood 
Limited may monitor outgoing and incoming e-mails and other 
telecommunications on its e-mail and telecommunications systems for the 
purpose of ensuring compliance with the law and with Kenwood Limited's 
policies.

pritpalm-kuk, 2024-03-13:
Also this issue needs to be resolved as soon possible; hence level or urgency is required due first production release taking place next week

kapilkedawat, 2024-03-13:
HI @pritpalm-kuk, we need logs and capture to analyze the issue. Please share the logs and capture as soon as possible, once that is shared we will analyze them and based on the findings we can have a meeting if needed.

You can go through various links that I have shared below to capture wireless packets.

https://linuxhint.com/capture_wi-fi_traffic_using_wireshark/
https://developer.apple.com/documentation/network/recording_a_packet_trace/recording_a_wi-fi_packet_trace 
https://microchip.my.site.com/s/article/WiFi-Sniffer-data-capture-in-Linux-Ubuntu

Again, as explained in previous comments and reiterating again, please note that its critically important that you share full logs(from bootup till issue occurs) and sniffer capture.

pritpalm-kuk, 2024-03-13:
Hi Kapil

The esp32 ng hosted solution is from espressif.
Why are you expecting us to provide you the information you need whilst you
already know how to the setup the environment to get the logs you require.
We have clearly given you the issues we have reported.
You have our hardware!! and I have great expectations that you can use it
to replicate the issues we have noticed

Also make sure you attend the meeting at 11:16 am on the 14/03.2024 U.K
time.

Regards

Pritpal

-- 
This e-mail is intended only for the person to whom it is addressed and may 
contain confidential information. If an addressing or transmission error 
has misdirected this e-mail, please notify the author by replying to this 
e-mail. If you are not the intended recipient you must not use, disclose, 
print or rely on the contents of this e-mail and should immediately and 
permanently delete it.

Kenwood Limited is a company registered in England 
with registered number 872044. The registered office of Kenwood Limited is 
1 Kenwood Business Park, New Lane, Havant, Hampshire, PO9 2NH.

Kenwood 
Limited may monitor outgoing and incoming e-mails and other 
telecommunications on its e-mail and telecommunications systems for the 
purpose of ensuring compliance with the law and with Kenwood Limited's 
policies.

kapilkedawat, 2024-03-14:
Hi @pritpalm-kuk, 

We will attempt to reproduce this issue on the IMX dev board and will keep you updated on our findings. As we are still in the process of reproducing the issue, there is nothing specific to discuss in today's meeting. However, we will inform you if further discussion is needed in the future.

pritpalm-kuk, 2024-03-14:
We will still have the meeting; please can you attend the meeting

pritpalm-kuk, 2024-03-14:
Hi Kapil

I still expect to have the meeting at 11:30am uk time.
Please can you attend the meeting

Regards

Pritpal

-- 
This e-mail is intended only for the person to whom it is addressed and may 
contain confidential information. If an addressing or transmission error 
has misdirected this e-mail, please notify the author by replying to this 
e-mail. If you are not the intended recipient you must not use, disclose, 
print or rely on the contents of this e-mail and should immediately and 
permanently delete it.

Kenwood Limited is a company registered in England 
with registered number 872044. The registered office of Kenwood Limited is 
1 Kenwood Business Park, New Lane, Havant, Hampshire, PO9 2NH.

Kenwood 
Limited may monitor outgoing and incoming e-mails and other 
telecommunications on its e-mail and telecommunications systems for the 
purpose of ensuring compliance with the law and with Kenwood Limited's 
policies.

mantriyogesh, 2024-05-08:
Hello @pritpalm-kuk ,

Can we close this issue, if there are no further questions?