espressif/esp-hosted #355: esp32 ng disconnections is trying to connect to the network

pritpalm-kuk, 2024-03-13:
I can see that there are a bunch of disconnections while the esp32 is trying to connect to the network until it is finally able to connect to it.

Please see esp32 fw logs attached
[esp32.txt](https://github.com/espressif/esp-hosted/files/14588665/esp32.txt)

pritpalm-kuk, 2024-03-13:
From my last comment
How many connections on average are required to establish a successfull connection?

mantriyogesh, 2024-03-25:
Hello @pritpalm-kuk ,

I think you have attached the wrong log.

In the log attached we could not find the traces you mentioned.
In fact, there is no connection trigger, in the log.
[image]

Please attach the following logs where the issue is observed.
1. Complete ESP log
2. Dmesg log from bootup
3. Wpa supplicant log in debug
4. Possibly a sniffer log, so that we have clear idea of AP communication with ESP.

We have recorded a video where we showcased steps in detail and step wise manner.
You can follow the video, and repeat the steps to get the connection working/collect the log at issue and attach.

pritpalm-kuk, 2024-03-25:
> Screenshot_2024-03-25-22-09-33-713_com.brave.browser_beta-edit.jpg (view
> on web)

Hi Yogesh,

What is the function esp_wifi_internal_tx used for?

Regards

Pritpal

-- 
This e-mail is intended only for the person to whom it is addressed and may 
contain confidential information. If an addressing or transmission error 
has misdirected this e-mail, please notify the author by replying to this 
e-mail. If you are not the intended recipient you must not use, disclose, 
print or rely on the contents of this e-mail and should immediately and 
permanently delete it.

Kenwood Limited is a company registered in England 
with registered number 872044. The registered office of Kenwood Limited is 
1 Kenwood Business Park, New Lane, Havant, Hampshire, PO9 2NH.

Kenwood 
Limited may monitor outgoing and incoming e-mails and other 
telecommunications on its e-mail and telecommunications systems for the 
purpose of ensuring compliance with the law and with Kenwood Limited's 
policies.

pritpalm-kuk, 2024-03-25:
please refer to [esp32.txt](https://github.com/espressif/esp-hosted/files/14588665/esp32.txt)

mantriyogesh, 2024-03-25:
Please refer the logs needed at : https://github.com/espressif/esp-hosted/issues/355#issuecomment-2018111982

Unless full picture at that point, it will be direction-less to add any fix.
I hope you understand why the logs are needed.

We understand you face issue, we have tried this issue, which is not observed at our place.
Our reference logs and procedure to collect is already with you.
Please get all logs to proceed further.