espressif/esp-hosted #356: esp-hosted/esp_hosted_ng/host does not compile in kernel source tree

mike-2020, 2024-03-14:
Need add the line in bold. PWD does not work when compile the whole kernel:

EXTRA_CFLAGS += -I$(PWD)/include -I$(PWD) **-Idrivers/net/wireless/esp32/include -Idrivers/net/wireless/esp32**

mantriyogesh, 2024-03-14:
Can you point what part of code refers to 'drivers/net/wireless/esp32/' in the ESP-Hosted repo?

mike-2020, 2024-03-14:
This file: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/Makefile

mantriyogesh, 2024-03-14:
I understood that it is make file.
I was asking folder location, 'drivers/net/wireless/esp32' is not available in ESP-Hosted..

mantriyogesh, 2024-03-20:
Still trying to understand.

All what I wanted to say is that, the location 'drivers/net/wireless/esp32' is not recognized by the ESP-Hosted.
Maybe you had placed your files there. So in this case we do not need to change anything.

Is that correct @mike-2020 ?

mike-2020, 2024-03-20:
"drivers/net/wireless/esp32" is the linux kernel source tree. it looks like the "make" does not enter this directory before run gcc compiler (this is decided by high level Makefile). In such case, the Makefile statement above will not work.

kapilkedawat, 2024-03-26:
Hi @mike-2020 , are you trying to compile this as part of kernel? Current Makefile is written to compile this as a standalone module.