espressif/esp-hosted #357: ESP32-C2: SPI Clock Limited to 1MHz with ESP-Hosted-NG & Bluetooth Causes Panic

PokersKun, 2024-03-18:
Hello! I encountered an issue with ESP chip reboot when using imx6ull + esp32c2 as an ESP-Hosted-NG solution. Here are the steps I followed:

1. Configured the device tree of imx6ull, using ecspi1 as SPI master.
```
&ecspi1 {
	fsl,spi-num-chipselects = <1>;
	cs-gpios = <&gpio4 26 GPIO_ACTIVE_LOW>;
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_ecspi1>;
	status = "okay";
};

&iomuxc {
	pinctrl-names = "default";

	pinctrl_ecspi1: ecspi1grp {
		fsl,pins = <
			MX6UL_PAD_CSI_DATA07__ECSPI1_MISO	0x100b1		/* MISO */
			MX6UL_PAD_CSI_DATA06__ECSPI1_MOSI	0x100b1		/* MOSI */
			MX6UL_PAD_CSI_DATA04__ECSPI1_SCLK	0x100b1		/* SCLK */
			MX6UL_PAD_CSI_DATA05__GPIO4_IO26	0x80000000	/* CS1 */
			MX6UL_PAD_GPIO1_IO02__GPIO1_IO02	0x1b0b1		/* Reset */
			MX6UL_PAD_NAND_READY_B__GPIO4_IO12	0x1b0b1		/* Handshake */
			MX6UL_PAD_NAND_CLE__GPIO4_IO15		0x1b0b1		/* Data Ready */
		>;
	};
};
```
2. Compiled the slave driver and flashed it into esp32c2 (No code changes).
3. Modified the handshake and data ready pin definitions in the host driver and compiled.
[image]
4. Uploaded the driver to the imx6ull host and loaded the driver, displaying only the following two messages.
```
root@imx6ullevk:~# insmod esp32_spi.ko resetpin=2
[ 4623.958639] esp32_spi: esp_reset: Triggering ESP reset.
[ 4624.191389] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
```
5. After repeatedly adjusting SPI_MODE and SPI_CLK_MHZ according to the [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md), subsequent print messages only appeared when both the host and slave were set to 1MHz.
[image]
```
root@imx6ullevk:~# insmod esp32_spi.ko resetpin=2 clockspeed=1
[ 4762.841409] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [1]
[ 4764.536955] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[ 4764.544104] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[ 4764.554906] esp32_spi: esp_validate_chipset: Chipset=ESP32-C2 ID=0c detected over SPI
[ 4764.563188] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[ 4764.569829] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[ 4764.576258] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[ 4764.582829] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[ 4764.589619] esp32_spi: print_reset_reason: POWERON_RESET
[ 4764.594990] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[ 4764.609453] esp32_spi: esp_reg_notifier: Driver init is ongoing
[ 4764.800764] esp32_spi: esp_cfg80211_get_tx_power:
[ 4764.809019] esp32_spi: esp_cfg80211_get_tx_power:
[ 4765.017698] esp32_spi: init_bt: ESP Bluetooth init
[ 4765.022960] esp32_spi: print_capabilities: Capabilities: 0xe8. Features supported are:
[ 4765.037842] esp32_spi: print_capabilities:    * WLAN on SPI
[ 4765.043375] esp32_spi: print_capabilities:    * BT/BLE
[ 4765.048986] esp32_spi: print_capabilities:      - HCI over SPI
[ 4765.054762] esp32_spi: print_capabilities:      - BLE only
[ 4766.407943] Bluetooth: MGMT ver 1.22
```
6. When using the `scan on` command via `bluetoothctl`,
```
root@imx6ullevk:~# hciconfig hci0 up
root@imx6ullevk:~# bluetoothctl
Agent registered
[CHG] Controller 08:3A:8D:4A:0D:36 Pairable: yes
[bluetooth]# scan on
Discovery started
[... 165 lines cut ...]
[ 4901.274533] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[ 4901.281289] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[ 4901.287945] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[ 4901.294322] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[ 4901.301043] esp32_spi: print_reset_reason: SW_CPU_RESET
[ 4901.306283] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[ 4901.324161] esp32_spi: esp_reg_notifier: Driver init is ongoing
[ 4901.538557] esp32_spi: esp_cfg80211_get_tx_power:
[ 4901.540089] esp32_spi: esp_cfg80211_get_tx_power:
[ 4901.757700] esp32_spi: init_bt: ESP Bluetooth init
[ 4901.762972] esp32_spi: print_capabilities: Capabilities: 0xe8. Features supported are:
[ 4901.773254] esp32_spi: print_capabilities:    * WLAN on SPI
[ 4901.778897] esp32_spi: print_capabilities:    * BT/BLE
[ 4901.783965] esp32_spi: print_capabilities:      - HCI over SPI
[ 4901.789794] esp32_spi: print_capabilities:      - BLE only
[ 4903.059047] Bluetooth: MGMT ver 1.22
```

What could be the issue here? Is it due to the SPI_CLK_MHZ being too low? If so, how can I increase SPI_CLK_MHZ?

#
SoC: NXP i.MX 6ULL (1x Cortex-A7 800 MHz)
Linux Kernel: [6.1.55](https://github.com/nxp-imx/linux-imx/tree/lf-6.1.y)
ESP32 Chip: ESP32-C2 (ESP8684H4)
Interface: Wi-Fi and Bluetooth over SPI

PokersKun, 2024-03-18:
This is the result obtained in [Raw Throughput testing](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md):
1. Host --> ESP
```
*** Raw Throughput testing: Host --> ESP started ***

I (3485) stats: 0-1 sec       0.00 kbits/sec
I (4485) stats: 1-2 sec       11.41 kbits/sec
I (5485) stats: 2-3 sec       581.72 kbits/sec
I (6485) stats: 3-4 sec       570.31 kbits/sec
I (7485) stats: 4-5 sec       593.12 kbits/sec
I (8485) stats: 5-6 sec       581.72 kbits/sec
I (9485) stats: 6-7 sec       593.12 kbits/sec
I (10485) stats: 7-8 sec       581.72 kbits/sec
I (11485) stats: 8-9 sec       581.72 kbits/sec
I (12485) stats: 9-10 sec       593.12 kbits/sec
I (13485) stats: 10-11 sec       581.72 kbits/sec
I (14485) stats: 11-12 sec       581.72 kbits/sec
```
2. ESP --> Host
```
*** Raw Throughput testing: ESP --> Host started ***

I (3519) stats: 0-1 sec       897.00 kbits/sec
I (4519) stats: 1-2 sec       483.00 kbits/sec
I (5519) stats: 2-3 sec       586.50 kbits/sec
I (6519) stats: 3-4 sec       575.00 kbits/sec
I (7519) stats: 4-5 sec       586.50 kbits/sec

assert failed: queue_next_transaction spi_slave_api.c:411 (spi_trans->rx_buffer)
Stack dump detected
Core  0 register dump:
MEPC    : 0x40380608  RA      : 0x40389420  SP      : 0x3fccecd0  GP      : 0x3fcb0fb0  
0x40380608: panic_abort at /opt/work/esp/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/panic.c:452
0x40389420: __ubsan_include at /opt/work/esp/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x3fcb0268  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000077  S1      : 0x00000001  A0      : 0x3fcced0c  A1      : 0x3fcb281d  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fcb7000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fccee30  
S4      : 0x3fcb281c  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

panic_abort (details=details@entry=0x3fcced0c "assert failed: queue_next_transaction spi_slave_api.c:411 (spi_trans->rx_buffer)") at /opt/work/esp/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/panic.c:452
452         *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
#0  panic_abort (details=details@entry=0x3fcced0c "assert failed: queue_next_transaction spi_slave_api.c:411 (spi_trans->rx_buffer)") at /opt/work/esp/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/panic.c:452
#1  0x40389420 in esp_system_abort (details=details@entry=0x3fcced0c "assert failed: queue_next_transaction spi_slave_api.c:411 (spi_trans->rx_buffer)") at /opt/work/esp/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/port/esp_system_chip.c:84
#2  0x4038e5d6 in __assert_func (file=file@entry=0x3c075170 "", line=line@entry=411, func=<optimized out>, func@entry=0x3c075408 <__func__.2> "", expr=expr@entry=0x3c0751f4 "") at /opt/work/esp/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/newlib/assert.c:81
#3  0x4200af16 in queue_next_transaction () at /opt/work/esp/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c:411
#4  0x4200afca in spi_transaction_post_process_task (pvParameters=<error reading variable: value has been optimized out>) at /opt/work/esp/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c:450
#5  0x4038bbf6 in vPortTaskWrapper (pxCode=<optimized out>, pvParameters=<optimized out>) at /opt/work/esp/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:202
ELF file SHA256: 944f96fc8cc2e8f8
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 8093f1e1d80b1f7757bb91b1eb913f86f2b59e66ccdeabd9fcf16dd7a3bbdc5d

Rebooting...
```

During the ESP --> Host test, ESP panicked and caused a reset chip

mantriyogesh, 2024-03-20:
Hello @PokersKun ,

Can you please attach the sdkconfig ?

mantriyogesh, 2024-03-20:
If you remove https://github.com/espressif/esp-hosted/blob/ee0b7414fadd6d255291059618929832bb23bf64/esp_hosted_ng/esp/esp_driver/network_adapter/sdkconfig.defaults.esp32c2#L33

and 
```
rm -rf sdkconfig build
idf.py set-target esp32c2
idf.py flash monitor
```

Any differences?

PokersKun, 2024-03-20:
> If you remove  如果您删除
> 
> https://github.com/espressif/esp-hosted/blob/ee0b7414fadd6d255291059618929832bb23bf64/esp_hosted_ng/esp/esp_driver/network_adapter/sdkconfig.defaults.esp32c2#L33
> 
> and 和
> 
> ```
> rm -rf sdkconfig build
> idf.py set-target esp32c2
> idf.py flash monitor
> ```
> 
> Any differences? 有什么区别吗？

Hello Yogesh, thanks for your reply!
This is the [sdkconfig.txt](https://github.com/espressif/esp-hosted/files/14661793/sdkconfig.txt) file currently in use, after executing idf.py set-target esp32c2 only the Main XTAL frequency was changed to 26 MHz to match my chip.
Also, I don't see the configuration CONFIG_ESP_CACHE_MALLOC=y used in sdkconfig, and even after deleting it in sdkconfig.defaults.esp32c2 and rebuilding and reflashing it as you provided, I get the same results as described above without any difference.

PokersKun, 2024-03-20:
Hello @mantriyogesh ,
Tests have shown that this problem occurs when performing `scan on` via `bluetoothctl` after setting `SPI_CLK_MHZ` to `1MHz` in both the host and slave drivers, which is probably an ESP issue not related to the host.

Before porting to imx6ull, I tested this using a Raspberry Pi 4B (bookworm) as the host without modifying any code in ESP-Hosted-NG, and the result was that `scan on` works fine with `SPI_CLK_MHZ` set to `30MHz` by default in ESP-Hosted-NG.

However, after I modified the `SPI_CLK_MHZ` to `1MHz` on both the host and slave side, a similar panic message appeared on the Raspberry Pi 4B as shown below:

```
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x400812b8  RA      : 0x4200641a  SP      : 0x3fcd05f0  GP      : 0x3fcb0fb0  
0x400812b8: memcpy in ROM

TP      : 0x3fcb1a68  T0      : 0x400586f0  T1      : 0x40001a44  T2      : 0x00000000  
0x400586f0: multi_heap_internal_unlock in ROM
0x40001a44: __call_r_os_memblock_get in ROM

S0/FP   : 0x3fcd49e6  S1      : 0x0000000a  A0      : 0x00000000  A1      : 0x3fcd49e8  
A2      : 0x00000009  A3      : 0x00000042  A4      : 0x00000009  A5      : 0x00000001  
A6      : 0x00000018  A7      : 0x00000000  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fcd05f0: 0x00000000 0x00000000 0x3fcd0628 0x42006490 0x00000000 0x00000000 0x3fcd0628 0x42006e78
3fcd0610: 0x00000000 0x00000000 0x00000000 0x42006ebe 0x00000000 0x00000000 0x3fcd49d8 0x00000002
3fcd0630: 0x3fcd49d8 0x00180000 0x00000000 0x4038206e 0x00000000 0x00000000 0x00000000 0x4038bbf6
3fcd0650: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcd0670: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154
3fcd0690: 0x3fcd0570 0x00007f10 0x3fcb3708 0x3fcb3708 0x3fcd0690 0x3fcb3700 0x00000003 0x06631020
3fcd06b0: 0x84632c04 0x3fcd0690 0x00000000 0x00000016 0x3fccf68c 0x76636572 0x7361745f 0x4a01006b
3fcd06d0: 0x00440993 0x00000000 0x3fcd0680 0x00000016 0x00000000 0x3fcb9068 0x42000260 0x00000000
3fcd06f0: 0x3fcb7884 0x3fcb78ec 0x3fcb7954 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
3fcd0710: 0x00000000 0x4206aff6 0x00000000 0x00000000 0x00000000 0x3fcb94bc 0x00000000 0x00000000
3fcd0730: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcd0750: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcd0770: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcd0790: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcd07b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcd07d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x3f000000 0x00001000 0xa5a5a5a5 0xa5a5a5a5
3fcd07f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd0810: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd0830: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd0850: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd0870: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd0890: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd08b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd08d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd08f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd0910: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd0930: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd0950: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd0970: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd0990: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd09b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd09d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

Failed to run gdb_panic_server.py script: Command '['riscv32-esp-elf-gdb', '--batch', '-n', '/root/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/network_adapter.elf', '-ex', 'target remote | "/root/.espressif/python_env/idf5.1_py3.11_env/bin/python" -m esp_idf_panic_decoder --target esp32c2 "/tmp/tmpfk81byn8"', '-ex', 'bt']' returned non-zero exit status 1.
b'"/root/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/network_adapter.elf": not in executable format: file format not recognized\nTraceback (most recent call last):\n  File "<frozen runpy>", line 198, in _run_module_as_main\n  File "<fr [... cut ...]
I (2107) FW_MAIN: INIT Interface command

I (2108) wifi:mode : sta (08:3a:8d:40:54:70)
I (2109) wifi:enable tsf
I (2109) FW_CMD: Wifi Sta mode set

I (2195) FW_MAIN: Get MAC command
```

Does this indicate that the slave-side driver is not able to carry out its work stably at `1MHz` (or is it a specific issue only for this chip, ESP32-C2)?

But now I can successfully receive the ESP bootup event at imx6ull only if `SPI_CLK_MHZ` is `1MHz`, and it's not clear what's happening in between, is there any better solution?

mantriyogesh, 2024-03-21:
Hello @PokersKun ,

We tried to reproduce this locally, but we did not encountered this issue.
But we have XTAL as 40MHz.

Do you happen to have any  chip with crystal?
Otherwise, we will have to check other ways to debug or reproduce this at our place.

mantriyogesh, 2024-03-22:
Hello @PokersKun ,

Can you please send us the btmon log, ESP log and Linux log when issue happens?
If you have any sets of steps, could also help.

Also, can you try to run on 1.3 MHz or 2.6Mhz instead (in multiple of 26MHz)?

PokersKun, 2024-03-22:
Hello @mantriyogesh ,

I've asked our hardware engineers to try to replace the 26MHz crystal with a 40MHz one, and I'll continue to test and report back once the replacement is complete.

Also, the method of changing the XTAL frequency is the same as what you said, not directly changing the `sdkconfig` or `sdkconfig.h` file, the steps are as follows:

```shell
idf.py menuconfig
Component config  --->Hardware Settings  --->Main XTAL Config  --->Main XTAL frequency (40 MHz)  --->(X) 26 MHz
```

I tried changing the `SPI_CLK_MHZ` to `1.3MHz` and `2.6MHz` for both the host and slave drivers, but I got no different results than before.

1. after changing to `1.3MHz`, the result is exactly the same as step 5 of the issue to `1MHz`, and still get the panic message at `scan on`.

2. After changing to `2.6MHz`, consistent with step 4 of the issue, could not receive the ESP bootup event.

After that I re-changed to `1MHz` for testing, the Log for ESP and Linux is as shown in step 6 of the issue, below is the complete log file:
- Complete log file [esp-hosted-logs.pdf](https://github.com/espressif/esp-hosted/files/14718208/esp-hosted-logs.pdf) for ESP and Linux.
- Linux output log file [btmon_linux_log.txt](https://github.com/espressif/esp-hosted/files/14718245/btmon_linux_log.txt) for btmon and log file [btmon_debug.log](https://github.com/espressif/esp-hosted/files/14718248/btmon_debug.log) in btsnoop format.

Hopefully we can get a better solution, thanks for your great work!

mantriyogesh, 2024-03-25:
Hello @PokersKun 

Did you get a chance to test with these bins?

PokersKun, 2024-03-25:
> Hello @PokersKun ,
> 
> I tried to reproduce your scenario, but the things were working, the way it should be. I can attach all the logs, if you wish to compare with. (On Monday, as I forgot the host at office place)
> 
> [A] Setup: Platform used: Raspberry Pi 4B with ESP32-C2 Crystal: 26MHz (can be verified using `esptool.py` command, in case you don't know) It is office environment, meaning, it is noisy environment, with a lot of BLE devices. But C2 with continuous scan did not crash.
> 
> [B] How did I test:
> 
> 1. change XTAL freq of c2, as it was 26MHz XTAL.
> 2. Run `bluetoothctl` and `scan on` command.
> 3. `tail -F /var/log/messages` & `sudo btmon`  on host
> 4. Comment `adjust_spi_clock()`, keep SPI freq  `SPI_INITIAL_CLK_MHZ`to 10MHz  (unchanged) & test (1)
> 5. On top on (4), Change `SPI_INITIAL_CLK_MHZ` to `1` (means 1 MHz)
> 6. Verified the clock on SPI with analyser to be 1MHz and duty cycle was ~50% in (4) & (5)
> 
> [C] Questions: i) Can you confirm the crystal is 26MHz?
> 
> ```
> $ esptool.py chip_id --port <port>
> ```
> 
> ii) can you please test with below binaries? iii) Can you also check if your clock is stable with around 50% duty?
> 
> [D] C2 bins used Is it possible to use the binary I had used for flashing ESP32-C2? Base git commit: **Hosted**: [ee0b741](https://github.com/espressif/esp-hosted/commit/ee0b7414fadd6d255291059618929832bb23bf64) **IDF**: d3c99ed3b88eb4c7fb5cf29d6cca1f390ff107e1
> 
> sdkconfig for reference: [sdkconfig.txt](https://github.com/espressif/esp-hosted/files/14722972/sdkconfig.txt)
> 
> Binaries: [2024.03.22_gh_357_c2_26mhz_ble_scan_bins.tgz](https://github.com/espressif/esp-hosted/files/14723033/2024.03.22_gh_357_c2_26mhz_ble_scan_bins.tgz)

Hello @mantriyogesh ,

Thank you for your help!
Here are the answers over here to the questions you have asked:

i) The hardware is 100% sure that it is a 26MHz crystal.
The output of `esptool.py chip_id` is as follows:
```shell
esptool.py v4.7.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C2
Chip is ESP32-C2 (revision v1.0)
Features: WiFi, BLE
Crystal is 26MHz
MAC: 08:3a:8d:48:39:9c
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-C2 has no Chip ID. Reading MAC instead.
MAC: 08:3a:8d:48:39:9c
Hard resetting via RTS pin...
```

ii) Below are the test results from the binary you provided, the ESP32-C2 reports an panic the moment it boot up and goes into a loop reset:
```shell
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x16d0
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x2940
entry 0x403acb70
I (21) boot: ESP-IDF v5.1.1-1-gd3c99ed3b8-dirty 2nd stage bootloader
I (21) boot: compile time Mar 22 2024 21:13:26
I (22) boot: chip revision: v1.0
I (26) boot.esp32c2: MMU Page Size  : 32K
I (30) boot.esp32c2: SPI Speed      : 60MHz
[... 35 lines cut ...]
0x40380438: call_start_cpu0 at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/port/cpu_start.c:294

S8      : 0x00030020  S9      : 0x00010020  S10     : 0x00000005  S11     : 0x3fcde824  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

panic_abort (details=details@entry=0x3fcde60c "assert failed: esp_clk_init clk.c:70 (rtc_clk_xtal_freq_get() == CONFIG_XTAL_FREQ)") at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/panic.c:452
452         *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
#0  panic_abort (details=details@entry=0x3fcde60c "assert failed: esp_clk_init clk.c:70 (rtc_clk_xtal_freq_get() == CONFIG_XTAL_FREQ)") at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/panic.c:452
#1  0x40389420 in esp_system_abort (details=details@entry=0x3fcde60c "assert failed: esp_clk_init clk.c:70 (rtc_clk_xtal_freq_get() == CONFIG_XTAL_FREQ)") at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/port/esp_system_chip.c:84
#2  0x4038e5d6 in __assert_func (file=file@entry=0x3c070421 "", line=line@entry=70, func=<optimized out>, func@entry=0x3c07042c <__func__.0> "", expr=expr@entry=0x3c0703c8 "") at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/newlib/assert.c:81
#3  0x42000704 in esp_clk_init () at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/port/soc/esp32c2/clk.c:70
#4  0x40380506 in call_start_cpu0 () at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/port/cpu_start.c:573
ELF file SHA256: 3c10b7eb3d86b33d
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 18265a048ec850e239845026ed3db265b93695fa029ff6dd25e42139463108c0

Rebooting...
```

iii) My logic analyzer currently has some issues causing me to not be able to analyze the current duty cycle, I'm trying to fix this and get the clock stabilization information.

mantriyogesh, 2024-03-25:
[c2_26mhz_bins_25mar24.tgz](https://github.com/espressif/esp-hosted/files/14739750/c2_26mhz_bins_25mar24.tgz)

```
 I  ~/c/h2   *$…  esp_hosted_ng/esp/esp_driver/network_adapter  idf.py monitor                                                                                             139ms  Mon Mar 25 14:36:18 2024
Executing action: monitor
Serial port /dev/cu.usbserial-130
Connecting....
Detecting chip type... ESP32-C2
Running idf_monitor in directory /Users/yogesh/code/h2/esp_hosted_ng/esp/esp_driver/network_adapter
Executing "/Users/yogesh/.espressif/python_env/idf5.1_py3.11_env/bin/python /Users/yogesh/code/h2/esp_hosted_ng/esp/esp_driver/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-130 -b 74880 --toolchain-prefix riscv32-esp-elf- --target esp32c2 --revision 100 --decode-panic backtrace /Users/yogesh/code/h2/esp_hosted_ng/esp/esp_driver/network_adapter/build/network_adapter.elf -m '/Users/yogesh/.espressif/python_env/idf5.1_py3.11_env/bin/python' '/Users/yogesh/code/h2/esp_hosted_ng/esp/esp_driver/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.4.0 on /dev/cu.usbserial-130 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x16d0
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x293c
entry 0x403acb70
I (32) boot: ESP-IDF v5.1.1-1-gd3c99ed3b8-dirty 2nd stage bootloader
I (32) boot: compile time Mar 25 2024 14:31:19
I (33) boot: chip revision: v1.0
I (39) boot.esp32c2: MMU Page Size  : 32K
I (47) boot.esp32c2: SPI Speed      : 60MHz
I (54) boot.esp32c2: SPI Mode       : DIO
I (61) boot.esp32c2: SPI Flash Size : 2MB
I (69) boot: Enabling RNG early entropy source...
I (77) boot: Partition Table:
I (82) boot: ## Label            Usage          Type ST Offset   Length
I (94) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (105) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (117) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (128) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (140) boot:  4 ota_1            OTA app          00 11 00100000 000f0000
I (152) boot: End of partition table
I (158) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=1f548h (128328) map
I (200) esp_image: segment 1: paddr=0002f570 vaddr=3fcb07b0 size=00aa8h (  2728) load
I (201) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=6d480h (447616) map
I (309) esp_image: segment 3: paddr=0009d4a8 vaddr=3fcb1258 size=0210ch (  8460) load
I (311) esp_image: segment 4: paddr=0009f5bc vaddr=40380000 size=107b0h ( 67504) load
I (340) boot: Loaded app from partition at offset 0x10000
I (340) boot: Disabling RNG early entropy source...
I (358) cpu_start: Unicore app
I (358) cpu_start: Pro cpu up.
I (358) clk: *** CONFIG_XTAL_FREQ: 26, rtc_clk_xtal_freq_get(): 26 ***
I (380) cpu_start: Pro cpu start user code
I (381) cpu_start: cpu freq: 120000000 Hz
I (381) cpu_start: Application information:
I (385) cpu_start: Project name:     network_adapter
I (394) cpu_start: App version:      ee0b7414-dirty
I (402) cpu_start: Compile time:     Mar 25 2024 14:31:15
I (412) cpu_start: ELF file SHA256:  1dbb8c4948573d26...
I (421) cpu_start: ESP-IDF:          v5.1.1-1-gd3c99ed3b8-dirty
I (431) cpu_start: Min chip rev:     v1.0
I (438) cpu_start: Max chip rev:     v1.99
I (446) cpu_start: Chip rev:         v1.0
I (453) heap_init: Initializing. RAM available for dynamic allocation:
I (464) heap_init: At 3FCB7000 len 00025B70 (150 KiB): D/IRAM
I (474) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM
I (485) spi_flash: detected chip: gd
I (491) spi_flash: flash io: dio
W (497) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (517) sleep: Configure to isolate all GPIO pins in sleep state
I (528) sleep: Enable automatic switching of GPIO sleep configuration
I (538) coexist: coex firmware version: 80b0d89
I (547) coexist: coexist rom version de8c800
I (554) app_start: Starting scheduler on CPU0
I (562) main_task: Started on CPU0
[... 30 lines cut ...]
I (786) wifi_init: tcp tx win: 5744
I (793) wifi_init: tcp rx win: 5744
I (799) wifi_init: tcp mss: 1440
I (805) wifi_init: WiFi IRAM OP enabled
I (812) wifi_init: WiFi RX IRAM OP enabled
I (821) phy_init: phy_version 240,764b29e,Jun  9 2023,19:21:12
I (857) phy: libbtbb version: c8b653e, Jun  9 2023, 19:21:28
controller lib commit: [77e7dbb]
controller rom commit: [3314f9d]
I (860) FW_MAIN: ESP Bluetooth MAC addr: d8-bc-38- 3-46-76

I (868) FW_SPI: Using SPI interface
I (875) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (889) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1904) FW_MAIN: Initial set up done
I (1904) main_task: Returned from app_main()

```

Note a log: 
```
I (358) clk: *** CONFIG_XTAL_FREQ: 26, rtc_clk_xtal_freq_get(): 26 ***
```

Trying to understand, why these both values are not 26 in your case.

mantriyogesh, 2024-03-25:
I also provided network_adapter.elf and network_adapter.map file, which when used at your path, will show the BT with my sources of IDF.

PokersKun, 2024-03-25:
Hello @mantriyogesh ,

After using the `c2_26mhz_bins_25mar24.tgz` file you provided, there is no loop reset, but still can't receive the ESP bootup event at 10MHz, and get panicked at 1MHz using Bluetooth:

```shell
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x400812b8  RA      : 0x42006456  SP      : 0x3fcd05f0  GP      : 0x3fcb0fb0  
0x400812b8: memcpy in ROM
0x42006456: esp_vhci_host_send_packet at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/slave_bt.c:420

TP      : 0x3fcb1a18  T0      : 0x400586f0  T1      : 0x40001a44  T2      : 0x00000000  
0x400586f0: multi_heap_internal_unlock in ROM
0x40001a44: __call_r_os_memblock_get in ROM

S0/FP   : 0x3fcb9316  S1      : 0x0000000a  A0      : 0x00000000  A1      : 0x3fcb9318  
A2      : 0x00000009  A3      : 0x00000042  A4      : 0x00000009  A5      : 0x00000001  
A6      : 0x00000018  A7      : 0x00000000  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

0x400812b8 in ?? ()
#0  0x400812b8 in ?? ()
#1  0x42006456 in esp_vhci_host_send_packet (data=0x3fcb9316 "", len=<optimized out>) at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/slave_bt.c:420
#2  0x00000000 in ?? ()
Backtrace stopped: frame did not save the PC
ELF file SHA256: 1dbb8c4948573d26
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 18265a048ec850e239845026ed3db265b93695fa029ff6dd25e42139463108c0

Rebooting...
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0xc (RTC_SW_CPU_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x403807c0
0x403807c0: esp_restart_noos at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/port/soc/esp32c2/system_internal.c:96 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x16d0
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x293c
entry 0x403acb70
I (36) boot: ESP-IDF v5.1.1-1-gd3c99ed3b8-dirty 2nd stage bootloader
I (36) boot: compile time Mar 25 2024 14:31:19
I (37) boot: chip revision: v1.0
I (43) boot.esp32c2: MMU Page Size  : 32K
I (50) boot.esp32c2: SPI Speed      : 60MHz
I (58) boot.esp32c2: SPI Mode       : DIO
I (65) boot.esp32c2: SPI Flash Size : 2MB
I (72) boot: Enabling RNG early entropy source...
I (81) boot: Partition Table:
I (86) boot: ## Label            Usage          Type ST Offset   Length
I (97) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (109) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (120) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (132) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (144) boot:  4 ota_1            OTA app          00 11 00100000 000f0000
I (155) boot: End of partition table
I (162) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=1f548h (128328) map
I (204) esp_image: segment 1: paddr=0002f570 vaddr=3fcb07b0 size=00aa8h (  2728) load
I (205) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=6d480h (447616) map
[... 60 lines cut ...]
I (785) wifi_init: tcp mbox: 6
I (790) wifi_init: tcp tx win: 5744
I (797) wifi_init: tcp rx win: 5744
I (803) wifi_init: tcp mss: 1440
I (809) wifi_init: WiFi IRAM OP enabled
I (816) wifi_init: WiFi RX IRAM OP enabled
I (825) phy_init: phy_version 240,764b29e,Jun  9 2023,19:21:12
I (863) phy: libbtbb version: c8b653e, Jun  9 2023, 19:21:28
controller lib commit: [77e7dbb]
controller rom commit: [3314f9d]
I (866) FW_MAIN: ESP Bluetooth MAC addr:  8-3a-8d-48-39-9e

I (874) FW_SPI: Using SPI interface
I (881) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (895) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1910) FW_MAIN: Initial set up done
I (1910) main_task: Returned from app_main()
I (2195) FW_MAIN: INIT Interface command

I (2196) wifi:mode : sta (08:3a:8d:48:39:9c)
I (2197) wifi:enable tsf
I (2197) FW_CMD: Wifi Sta mode set

I (2295) FW_MAIN: Get MAC command
```

PokersKun, 2024-03-25:
Oddly enough, I'm getting the same problem on a Raspberry Pi 4B + ESP32-C2 DevKit using the file you provided:

```shell
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x400812b8  RA      : 0x42006456  SP      : 0x3fcd05f0  GP      : 0x3fcb0fb0  
0x400812b8: memcpy in ROM
0x42006456: esp_vhci_host_send_packet at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/slave_bt.c:420

TP      : 0x3fcb1a18  T0      : 0x400586f0  T1      : 0x40001a44  T2      : 0x00000000  
0x400586f0: multi_heap_internal_unlock in ROM
0x40001a44: __call_r_os_memblock_get in ROM

S0/FP   : 0x3fcd3e46  S1      : 0x0000000a  A0      : 0x00000000  A1      : 0x3fcd3e48  
A2      : 0x00000009  A3      : 0x00000042  A4      : 0x00000009  A5      : 0x00000001  
A6      : 0x00000018  A7      : 0x00000000  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

0x400812b8 in ?? ()
#0  0x400812b8 in ?? ()
#1  0x42006456 in esp_vhci_host_send_packet (data=0x3fcd3e46 "", len=<optimized out>) at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/slave_bt.c:420
#2  0x00000000 in ?? ()
Backtrace stopped: frame did not save the PC
ELF file SHA256: 1dbb8c4948573d26
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 18265a048ec850e239845026ed3db265b93695fa029ff6dd25e42139463108c0

Rebooting...
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403807c0
0x403807c0: esp_restart_noos at /opt/work/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/port/soc/esp32c2/system_internal.c:96 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x16d0
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x293c
entry 0x403acb70
I (36) boot: ESP-IDF v5.1.1-1-gd3c99ed3b8-dirty 2nd stage bootloader
I (36) boot: compile time Mar 25 2024 14:31:19
I (37) boot: chip revision: v1.0
I (43) boot.esp32c2: MMU Page Size  : 32K
I (50) boot.esp32c2: SPI Speed      : 60MHz
I (58) boot.esp32c2: SPI Mode       : DIO
I (65) boot.esp32c2: SPI Flash Size : 2MB
I (72) boot: Enabling RNG early entropy source...
I (81) boot: Partition Table:
I (86) boot: ## Label            Usage          Type ST Offset   Length
I (97) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (109) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (120) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (132) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (144) boot:  4 ota_1            OTA app          00 11 00100000 000f0000
I (155) boot: End of partition table
I (162) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=1f548h (128328) map
I (204) esp_image: segment 1: paddr=0002f570 vaddr=3fcb07b0 size=00aa8h (  2728) load
I (205) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=6d480h (447616) map
I (313) esp_image: segment 3: paddr=0009d4a8 vaddr=3fcb1258 size=0210ch (  8460) load
I (315) esp_image: segment 4: paddr=0009f5bc vaddr=40380000 size=107b0h ( 67504) load
[... 78 lines cut ...]
I (2143) wifi:enable tsf
I (2143) FW_CMD: Wifi Sta mode set

I (2251) FW_MAIN: Get MAC command

I (2564) FW_MAIN: Scan request

I (5961) FW_MAIN: Scan request

I (10971) FW_MAIN: Scan request

I (17960) FW_MAIN: Scan request

I (27960) FW_MAIN: Scan request

I (41970) FW_MAIN: Scan request

I (60960) FW_MAIN: Scan request

I (89959) FW_MAIN: Scan request

I (131959) FW_MAIN: Scan request
```

I see that the path to the IDF still shows my workspace, is this perhaps a problem with my IDF environment or a failure to erase the Flash of my ESP32-C2?

mantriyogesh, 2024-03-25:
1. What is the change on top of master that you have in your ESP-IDF setup?
```
$ cd esp_hosted_ng/esp/esp_driver/esp-idf
$ git status
$ git diff
$ git log
```

2. also, at ESP-Hosted (host),
```
$ cd <esp_hosted>
$ git status
$ git diff
$ git log
```

3. How do you make sure the freq is 10 or 1MHz? Any changes in code you have done?
Ideally, what I would do, is to comment 
https://github.com/espressif/esp-hosted/blob/f0ab2c261b104f89115ca9f8f851993b7f609b35/esp_hosted_ng/host/spi/esp_spi.c#L597
and change 
https://github.com/espressif/esp-hosted/blob/f0ab2c261b104f89115ca9f8f851993b7f609b35/esp_hosted_ng/host/spi/esp_spi.c#L20

5. Symbols getting locally parsed
> I see that the path to the IDF still shows my workspace, is this perhaps a problem with my IDF environment or a failure to erase the Flash of my ESP32-C2?

it's okay, for now. (It's not okay, but it doesn't matter right now. for post crash analysis, it is imp to have matching symbols).

The binary exposes the function pointer. But your (local) elf has symbols for those functions, so it matches and shows it.
If elf file and map file used as the one I sent, it would show up my paths.

6. Can you please check if the raw throughput in both directions is relatively correct as per your frequencies. (I know you have done it, but does it correctly change as you change the frequency?)

7. It is also very important if the clock is table and has around 50% duty cycle. But this might need logic analyser (any analyser is okay)

mantriyogesh, 2024-03-25:
Related to https://github.com/espressif/esp-hosted/issues/200 ?

PokersKun, 2024-03-25:
> Related to #200 ?

Yes, similar question to this one, but I guess he probably didn't test the bluetooth functionality because my `espsta0` also works fine at `1MHz`.

PokersKun, 2024-03-25:
> 1. What is the change on top of master that you have in your ESP-IDF setup?
> 
> ```
> $ cd esp_hosted_ng/esp/esp_driver/esp-idf
> $ git status
> $ git diff
> $ git log
> ```
> 
> 2. also, at ESP-Hosted (host),
> 
> ```
> $ cd <esp_hosted>
> $ git status
> $ git diff
> $ git log
> ```
> 
> 3. How do you make sure the freq is 10 or 1MHz? Any changes in code you have done?
>    Ideally, what I would do, is to comment
>    https://github.com/espressif/esp-hosted/blob/f0ab2c261b104f89115ca9f8f851993b7f609b35/esp_hosted_ng/host/spi/esp_spi.c#L597
>    
>    and change
>    https://github.com/espressif/esp-hosted/blob/f0ab2c261b104f89115ca9f8f851993b7f609b35/esp_hosted_ng/host/spi/esp_spi.c#L20
> 4. Symbols getting locally parsed
> 
> > I see that the path to the IDF still shows my workspace, is this perhaps a problem with my IDF environment or a failure to erase the Flash of my ESP32-C2?
> 
> it's okay, for now. (It's not okay, but it doesn't matter right now. for post crash analysis, it is imp to have matching symbols).
> 
> The binary exposes the function pointer. But your (local) elf has symbols for those functions, so it matches and shows it. If elf file and map file used as the one I sent, it would show up my paths.
> 
> 6. Can you please check if the raw throughput in both directions is relatively correct as per your frequencies. (I know you have done it, but does it correctly change as you change the frequency?)
> 7. It is also very important if the clock is table and has around 50% duty cycle. But this might need logic analyser (any analyser is okay)
> 8. Can you please share your setup picture also?
>    And check if the connections are correct and firm for all the gpios in between esp and host?
> 9. What are the SPI modes used at both places?

Hello @mantriyogesh ! 

In order to prevent problems with misinformation in communication, I reproduced all the steps I took with esp-hosted on a Raspberry Pi 4B + ESP32-C2 DevKit and documented them in that [document](https://github.com/PokersKun/my-file-station/blob/main/esp32-c2-debug-log.md), please check that document, thank you!

PokersKun, 2024-03-26:
Today I tried replacing the [ESP8684-DevKitC-02](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp8684/esp8684-devkitc-02/user_guide.html) (ESP32-C2) used on the Raspberry Pi 4B with a [ESP32-C6-DevKitC-1](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/user_guide.html), using the same Host driver and SPI clock frequency (1MHz), and got no panic when using Bluetooth, everything worked fine.

mantriyogesh, 2024-03-26:
Thanks for the update..
We are investigating the issue you have faced on c2.
By the way, thank you for the all logs you have accumulated in the [document](https://github.com/PokersKun/my-file-station/blob/main/esp32-c2-debug-log.md)

Can you please add 
https://github.com/espressif/esp-hosted/issues/357#issuecomment-2017698439 -> (1) in your log?

PokersKun, 2024-03-26:
> Thanks for the update.. We are investigating the issue you have faced on c2. By the way, thank you for the all logs you have accumulated in the [document](https://github.com/PokersKun/my-file-station/blob/main/esp32-c2-debug-log.md)
> 
> Can you please add [#357 (comment)](https://github.com/espressif/esp-hosted/issues/357#issuecomment-2017698439) -> (1) in your log?

Sure, it's been updated to [documentation](https://github.com/PokersKun/my-file-station/blob/main/esp32-c2-debug-log.md#2-git-in-esp-idf).

Also, when I executed the `esp-hosted/esp_hosted_ng/esp/esp_driver/setup.sh` file, the script automatically replaced the lib file inside the esp-idf for me, so I've attached the git info for `esp_wifi/lib`.

Thanks again for your great work!

mantriyogesh, 2024-03-26:
> Thanks again for your great work!

Thank you for your sweet words.

Anyway, we need to fix this ESP32-C2 issue you are facing.

The binary I sent earlier was tested on ES328684-DevKitM-1 v1.1 (26 MHz XTAL)
Whereas the hardware you were using is ESP8684-DevKitC-02 v1.0,
which just behave the same wrt software.

Clearly, we need to reproduce and get this fixed for your case.
Considering C6 is working fine, is this blocking for your current design/work?

We clearly need to fix this anomaly, This is just to treat the priority for this ESP32-C2 issue internally.

mantriyogesh, 2024-03-26:
Just to avoid any confusion,

> Clearly, we need to reproduce and get this fixed for your case

We have tested your scenario and tried to reproduce it multiple times, but it worked fine.
But something is difference in between our setups, which might be the reason for the issue, which we need to analyse and fix.

Actually, I noticed that you use 20cm wires, does it make any difference if <=10cm wires used?

PokersKun, 2024-03-29:
> Hello @PokersKun ,
> 
> We are able to reproduce our side. Somehow, it is only reproducing with ESP8684-DevKitC-02 V1.0
> 
> @Shreyas0-7 is already looking into this issue. We will fix this on master and update you once done.
> 
> Your detailed logs are much helpful and extremely appreciated.

Thank you for your contribution!

I apologize that I've been busy with other work lately and haven't been able to provide timely feedback on the progress.

We're trying to solve this problem under the current custom board solution (imx6ull+esp32c2) and the cable length of the pcb is definitely less than 6cm, so the cable length might not be the problem.

Currently we made new custom boards that can work up to 4MHz without the bluetooth panic problem (up to 5MHz, but extremely unstable and causes DMA errors, I'm not sure what causes this).

However, probably due to the difference between our custom board Linux system and Raspbian system, we are experiencing the problem of not being able to disable the auto-scan when using dbus to control wpa_supplicant, which could be caused by other issues, and I would need to create a new topic to describe the problem I'm experiencing.

Thanks again for your reply! @mantriyogesh

mantriyogesh, 2024-03-29:
Hello @PokersKun ,

We already have a fix for the issue (fixes the issue but still being tested for multiple scenarios).

This change would be merged in master in some time. For now, find the change just to get you unblocked.
[418.patch.txt](https://github.com/espressif/esp-hosted/files/14808065/418.patch.txt)

Your patience in this regard is highly appreciated.
Thank you.

mantriyogesh, 2024-03-29:
> we are experiencing the problem of not being able to disable the auto-scan when using dbus to control wpa_supplicant, which could be caused by other issues, and I would need to create a new issue to describe the problem I'm experiencing.

Surely, as now we can test on exact same platform you use, it should be easier if you share the steps or details in a new issue.

mantriyogesh, 2024-04-01:
Hello @PokersKun,

current issue is fixed as part of 589ef50d

mantriyogesh, 2024-04-01:
@PokersKun , Did you get a chance to verify this fix?

PokersKun, 2024-04-02:
Hello @mantriyogesh !

I have used the patch and after testing the Bluetooth Panic issue has gone away and works fine at 1MHz, both on the Raspberry Pi and our custom boards.

Thanks again to you and the team for your help!

closed 2024-04-02 by mantriyogesh