espressif/esp-hosted #360: what is the reson make the soft ap disappear for seconds?

LuciaferKelvin, 2024-04-07:
Now I am faced with a question. When I set the STA mode and let the ESP32 connect to an AP, the SoftAP sometimes disappears for a few seconds. The steps are as follows:
1.Open the SoftAP.
2.Let the ESP32 connect to an AP in STA mode.
I suspect that when the ESP32 connects to the AP, the SoftAP changes its channel to follow the STA channel. Is this correct? If this behavior is due to the channel change, how can we avoid this issue? I am using firmware release v0.4

mantriyogesh, 2024-04-08:
Hello @LuciaferKelvin ,

Can you please send:
1. the commands used , their output, full ESP log and dmesg log from start?
2. Git commit used at ESP and host
3. IDF git commit used

Also, is it possible to check if issue happens even on [ IDF latest stable](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/versions.html#releases) ?

We strongly recommend upgrading the ESP-Hosted to lastest master (which is based on 0.0.5), as we fix a lot of issues from 0.4. but this is not straight forward change and has design change how control path. With async req-resp and async event handler, you can better monitor the control path and connection with ESP.

Please always make sure you use exact same git commit at ESP and Host.

LuciaferKelvin, 2024-04-09:
Hello @mantriyogesh,
My workmate had sent the log to your email few hour last.
I had another question,
Is there has any command to get more information when station loses connection to the router ?

LuciaferKelvin, 2024-04-09:
We can't easily change the firmware version. That would take a lot of time because it requires changing the control path and implementing the upgrade functions.Also has some others reason can't to tell.

LuciaferKelvin, 2024-04-09:
We have already adapted to the relevant functions, so this will make the update more difficult.

mantriyogesh, 2024-04-09:
> We can't easily change the firmware version. That would take a lot of time because it requires changing the control path and implementing the upgrade functions.

Actually OTA was supported as part of 0.4.
So still do-able for firmwares out there.

mantriyogesh, 2024-04-09:
> Hello @mantriyogesh, My workmate had sent the log to your email few hour last. I had another question, Is there has any command to get more information when station loses connection to the router ?

In the 0.0.5 , we have events for example, 'sta disconnected' (and few others), which will directly let you know asynchronously that sta is disconnected and can connect again.

But events were not supported in 0.4.

LuciaferKelvin, 2024-04-09:
Could you provide a 0.0.5 firmware to me to test?

LuciaferKelvin, 2024-04-09:
We built the 0.0.5 firmware ourselves, but it does not support the 0.0.4 control path. The problem may be caused by the IDF version.

LuciaferKelvin, 2024-04-09:
The ESP crashed whenever we sent the control command.

mantriyogesh, 2024-04-09:
> We built the 0.0.5 firmware ourselves, but it does not support the 0.0.4 control path. The problem may be caused by the IDF version.

Basically, you need just the firmware, where 0.0.5 host would support upgrade to 0.4 firmware. Is that correct understanding?

SohKamYung-Espressif, 2024-04-09:
@LuciaferKel

> We built the 0.0.5 firmware ourselves, but it does not support the 0.0.4 control path.

> The ESP crashed whenever we sent the control command.

There were changes to the control path between v0.0.4 and v0.0.5. Because of this, it is not possible to use an app built using v.0.0.4 to control firmware built using v0.0.5.

However, it should be possible to do an OTA upgrade on a ESP32 device running v0.0.4 firmware to one running v0.0.5 firmware, but this is a two step process:

1. Build your v0.0.5 firmware (`network.bin`). Using your current v0.0.4 app, do an OTA upgrade using the v0.0.5 firmware to upgrade the ESP32 running v0.0.4 firmware. At the end of the upgrade process, any control path commands send by your app will now cause the ESP32 to crash and reboot: this is expected.
2. Build your app using v0.0.5. This app should now work with the v0.0.5 firmware.

The kernel driver can be either v0.0.4 or v0.0.5. It only passes control data between your app and the ESP32, so it should just work. But after step 2, the kernel driver to be used should be v0.0.5.

After this upgrade, your firmware and app should now be on v0.0.5, and you can now get the features in v.0.0.5 (network events).

LuciaferKelvin, 2024-04-09:
Does that mean we need to upgrade the master code while upgrading the ESP firmware?

LuciaferKelvin, 2024-04-09:
> @LuciaferKel
>
>
>
> There were changes to the control path between v0.0.4 and v0.0.5. Because of this, it is not possible to use an app built using v.0.0.4 to control firmware built using v0.0.5. [...]

1. That was we have meet when try to only upgrade the esp firmware.

LuciaferKelvin, 2024-04-09:
If we need to upgrade the firmware, we cannot ensure that both the master controller and the ESP can be upgraded simultaneously. This situation necessitates maintaining two control paths to support both v0.4 and v0.0.5. Additionally, we must determine the current firmware version used by the ESP.

mantriyogesh, 2024-04-09:
> Does that mean we need to upgrade the master code while upgrading the ESP firmware?

Because OTA was just introduced at the end of 0.4, the 0.4 to 0.0.5 auto upgrade needs to be done **manually**.
However, we are adding the changes to make it smooth transition so that host when upgraded, would automatically check and ask/upgrade to the newer firmware.

## Manual upgrade procedure

### Current software
There are three software components with you right now.
A. Firmware {0.4}
B. Host driver (kernel module) {0.4}
C. Host app {0.4}

### Desired software
From there I want to move to 
X. Firmware {master}
Y. Host driver (kernel module) {master}
Z. Host app {master}

##### Build master firmware
First step, Build master firmware. Basically build and make ready [X]

##### Firmware - manual upgrade
Now, we use existing [C] which is your app with {0.4}. Do OTA using [C] itself, but you would point your OTA target binary as [X].

Once this is done, ignore any crash happening at firmware (it would anyway keep on rebooting). Just check the OTA is complete.

##### Host - manual upgrade
Now, update your host code with [Y] & [Z].
Firmware would stop crashing as [X] is compatible with [Y] and [Z]

> This situation necessitates maintaining two control paths to support both v0.4 and v0.0.5.

Not really, Once you upgrade your firmware, you need to align your code to master 'control APIs'. Now on, you would only need to use master based control path.

> Additionally, we must determine the current firmware version used by the ESP.

Yes, you are right. This is missing, and we are adding the versioning system, wherein your code would identify if firmware needs to be upgraded based on the versioning number.

We are working on automatic upgrades and would introduce that on master shortly.

mantriyogesh, 2024-04-09:
> we cannot ensure that both the master controller and the ESP can be upgraded simultaneously

We will rectify this issue. But I totally understand your pain and how big this problem might be.

LuciaferKelvin, 2024-04-11:
I don't think the method could make it. That way could be very dangerous. If we only use the ESP station mode to connect to the internet, any problem could prevent the production from ever connecting to the internet. It will lose control. We should find out a way to make sure the master and ESP could be upgraded successfully at the same time. But now we don't want to upgrade the firmware. If the problem behavior is severe, we will think about upgrading. Thank you.

mantriyogesh, 2024-04-11:
We had tested this at our end multiple times, it just works always. Have you tried it?

Wifi restarts are removed in Latest stable IDF which would not disconnect connected stations like this..

mantriyogesh, 2024-04-11:
If you can come to master, it makes maintainable software at your place.
All recent fixes on master make the control path very robust.

The event subscription is a great feature, which helps the host to be worry free for many scenarios.

> If we only use the ESP station mode to connect to the internet, any problem could prevent the production from ever connecting to the internet.

I think I forgot to mention this, I totally understand your concern.
Actually, you can treat this as switch from control path 0.4 to 0.0.5.
Once slave is upgraded, it would just expect control path 0.0.5 for all its operations.
If you do not have any custom code at firmware, the upgrade should be stable and repetitive without issues.

If you could test this to sample esp-host first and then take a call if this could be done in production.

Also, just to note, current code allows to downgrade to 0.4 from master, if you think so later.
We are planning to make the new release, which we target, backward compatible to 0.0.5.

If you wish to still be on 0.4, we respect your decision, but some of the urgent fixes cannot be directly applied to your system

LuciaferKelvin, 2024-04-11:
Of course, we want to use the new version with fewer bugs. However, the new version should be compatible with the old version. If it is not compatible with the old version, it will introduce more unknown issues that we need to consider when deciding to upgrade.

LuciaferKelvin, 2024-04-11:
The most important issue we face when upgrading to the new version is the potential for the production system to lose its internet connection, which is a serious problem for both us and our customers. Therefore, we cannot make that decision lightly.

mantriyogesh, 2024-04-12:
Is there anything we need to do?
If not, we can close this issue for now.
You can reopen or create a new anytime later..

LuciaferKelvin, 2024-04-12:
sure

closed 2024-04-12 by LuciaferKelvin