espressif/esp-hosted #361: NG solution Keep scanning when network card is running

Xiehanxin, 2024-04-08:
same issue in  linux 5.1/4.4 , esp32 module keep scanning request
[image]

[image]

mantriyogesh, 2024-04-08:
@kapilkedawat ++

mantriyogesh, 2024-04-08:
@Xiehanxin Can you help communicate following information to customer?

Is the AP they are trying to connect nearby and are they able to connect the same AP SSID and password using other device like mobile phone, when kept nearby ESP?

### Some basic suggestions:
**[A]** use exact same git commit at both ESP and Linux
**[B]** Make sure there is only one process of wpa_supplicant running
Check https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng#311-wi-fi
It does `sudo killall wpa_supplicant` and then starts new instance of `wpa_supplicant`
**[C]** you can add ` -d` in wpa_supplicant command to get the debug logs. These debug logs will help a lot to understand what is ongoing at wpa_supplicant.
**[D]** Use master branch for your testing.

### Logs
1. ESP logs:
Set ESP side default logging to 'Debug' using
```
$ idf.py menuconfig
```
Navigate to `(Top) → Component config → Log output → Default log verbosity` and change log level to `Debug`
and re-flash ESP using latest master

Flash ESP and get **full** **textual** logs **from starting**

2. **Full textual** Linux dmesg log from bootup

3. wpa_supplicant debug **textual** logs with option ` -d` added (see above [C])

4. Possibly a full sniffer capture, which will pinpoint the issue (start sniffer before triggering connect)

kapilkedawat, 2024-04-08:
Hi @Xiehanxin, wpa_supplicant issues scan periodically to update its scan cache and find relevant AP provided in network configuration(which is an expected and default behavior).

What is the exact issue here? Is station not able to connect if correct credentials are provided or just that we are seeing periodical scan request.

Xiehanxin, 2024-04-08:
hi @kapilkedawat @mantriyogesh here is the debug log
[esp32.log](https://github.com/espressif/esp-hosted/files/14904811/esp32.log)
[nuc980.log](https://github.com/espressif/esp-hosted/files/14904812/nuc980.log)

kapilkedawat, 2024-04-08:
Hi @Xiehanxin, espsta0 is not able to find the AP during the scan. Is the AP present in the vicinity of client?

linjif, 2024-04-09:
There are multiple WiFi access points (APs) around the espsta0 device, and the PC can connect normally. However, there are instances where using the iw dev espsta0 scan command also fails to detect any AP hotspots.

linjif, 2024-04-09:
@kapilkedawat There are multiple WiFi access points (APs) around the espsta0 device, and the PC can connect normally. However, there are instances where using the iw dev espsta0 scan command also fails to detect any AP hotspots.

mantriyogesh, 2024-04-09:
The ESP wifi log can be confirmed by Kapil (the team  would be on regional holiday)

I am wondering why the capabilities are not populated:

```
esp32_sdio: print_capabilities: Capabilities: 0x0. Features supported are:
```

1. Are these capabilities always coming 0?
Surprisingly, In your earlier log, you had espsta0 Mac address updated fine.

2. As I remember, you used to connect using the PCB. But the pull up registers were not correct earlier. Are the pull up registers corrected (cmd, d0-d3) need to be minimum 10k ohm.

3. Is it possible to run raw throughout first, before jumping to wifi, just to make sure your transport is correct and stable?
Raw throughout: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md
Please test rx and tx both sides individually.

kapilkedawat, 2024-04-09:
Hi @linjif, given that transport is correct and verified and external antenna is attached properly. Please try with attached patch and share logs from starting till issue occurs.
[0001-esp_hosted_ng-Add-scan-logging.patch](https://github.com/espressif/esp-hosted/files/14919111/0001-esp_hosted_ng-Add-scan-logging.patch)

MacChu0315-Espressif, 2024-04-10:
Hi @kapilkedawat, i was trying to build the ng bin for our customer but encountered some issue. I used the latest esp-hosted and applied the patch you provied, can you take a look?
[image]

KenKang2015, 2024-04-10:
> The ESP wifi log can be confirmed by Kapil (the team would be on regional holiday)
>
> I am wondering why the capabilities are not populated:
>
> ```
> esp32_sdio: print_capabilities: Capabilities: 0x0. [...]
Hi @kapilkedawat  iIn the early stage, I used device debugging, but now I'm using development board debugging. T development board has pull up registers(cmd,d0-d3) for 10K ohm. and in development board,using  ESP32-WROOM-32E .Regarding the third point, I'll give it a try,

kapilkedawat, 2024-04-10:
> Hi @kapilkedawat, i was trying to build the ng bin for our customer but encountered some issue. I used the latest esp-hosted and applied the patch you provied, can you take a look? ![企业微信截图_17127192061990](https://private-user-images.githubusercontent.com/130628192/321076124-73aa39e1-fc40-49a4-8ef2-6ad8546eaa20.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI3MzQ1NzEsIm5iZiI6MTcxMjczNDI3MSwicGF0aCI6Ii8xMzA2MjgxOTIvMzIxMDc2MTI0LTczYWEzOWUxLWZjNDAtNDlhNC04ZWYyLTZhZDg1NDZlYWEyMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxMFQwNzMxMTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YjQ2Mjc5MDQ1MWQ5OWIwMmMxYTVlY2VhOGQ2NmNlMDNlY2EwOTk5ZTE1NjNkMmY1YmM3ZWM3N2M1Mzg1YWZhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.dY4tfUMi5-0pQU-5xa5LuoBFcSewn_9AtFScUePM58Y)

Please follow the instructions [here](https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng/esp/esp_driver#update-build-firmware-for-esp-hosted-ng-solution) first to setup the environment or use [export.sh](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/setup.sh) to setup the env. Apply the patch after that(since export.sh resets the changes) and the try to compile.

MacChu0315-Espressif, 2024-04-10:
> Please follow the instructions [here](https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng/esp/esp_driver#update-build-firmware-for-esp-hosted-ng-solution) first to setup the [...]

Hi @kapilkedawat, the firmware has been provided to the customer, thank you!

KenKang2015, 2024-04-10:
Hi @kapilkedawat,，The log of the firmware test using the new firmware is as follows:
[nuc980.log](https://github.com/espressif/esp-hosted/files/14931139/nuc980.log)
[esp322.log](https://github.com/espressif/esp-hosted/files/14931141/esp322.log)

kapilkedawat, 2024-04-11:
Hi @KenKang2015, from the logs, esp32 is forwarding beacons and probe response to host. 
> FW_CMD: handle_wpa_sta_rx_mgmt:654 beacon frame RSSI=-33
FW_CMD: handle_wpa_sta_rx_mgmt:659 probe response RSSI=-66
wifi:Send scan done event: status=0, apnum=9
FW_CMD: Wifi scan done event

Could you please **not** daemonize wpa_supplicant and add extra logging mode(**-d**) to see what's happening.

wpa_supplicant -i espsta0 -c wpa_supplicant.conf -dd &

Also please share content of wpa_supplicant.conf file.

KenKang2015, 2024-04-11:
Hi @kapilkedawat  using "pa_supplicant -i espsta0 -c wpa_supplicant.conf -dd &" for log
[nuc980.log](https://github.com/espressif/esp-hosted/files/14945371/nuc980.log)
[esp322.log](https://github.com/espressif/esp-hosted/files/14945373/esp322.log)

kapilkedawat, 2024-04-11:
Hi @KenKang2015, Please check 3 point of https://github.com/espressif/esp-hosted/issues/361#issuecomment-2045085662 . We need to verify the SDIO transport and its working correctly. I see that esp32 is sending the scan results whereas supplicant is not seeing any.

Also does customer has any local modification in the driver, if yes, we would like to check if that is not causing any issue.

KenKang2015, 2024-04-11:
Hi @kapilkedawat 

How do I modify the source code to achieve the functionality of "./rpi_init.sh sdio rawtp_esp_to_host"?  because I'm using cross-compilation.,compile on Ubuntu 22, run on NUC980 (ARM9).

kapilkedawat, 2024-04-11:
rawtp code is always compiled in, please use 
`sudo insmod esp32_sdio.ko $RESETPIN raw_tp_mode=1 # for host to esp`
`sudo insmod esp32_sdio.ko $RESETPIN raw_tp_mode=2 # for esp to host`

KenKang2015, 2024-04-13:
Hi @kapilkedawat,
with using "insmod esp32_sdio.ko $RESETPIN raw_tp_mode=2 # for esp to host" command ,the log is:
[nuc980-foresp2host.log](https://github.com/espressif/esp-hosted/files/14967455/nuc980-foresp2host.log)
[esp32-foresp2host.log](https://github.com/espressif/esp-hosted/files/14967456/esp32-foresp2host.log)
with using "insmod esp32_sdio.ko $RESETPIN raw_tp_mode=1 # for host to esp" command ,the log is:
[nuc980-forhost2esp.log](https://github.com/espressif/esp-hosted/files/14967459/nuc980-forhost2esp.log)
[esp32-forhost2esp.log](https://github.com/espressif/esp-hosted/files/14967462/esp32-forhost2esp.log)

Shreyas0-7, 2024-04-15:
Hey @KenKang2015 As we can check from logs the raw throughput between ESP and Host is 0, which implies there is some problem with connection of ESP and Host. I suspect this is reason why you are facing issues you mentioned earlier. Can you please verify your connection once again?

MacChu0315, 2024-04-18:
Hi @kapilkedawat, Our customer performed [Troubleshoot Instructions - 1](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Troubleshoot.md#1-host-fails-to-detect-sdio-esp-peripheral) and offers messages from linux and debug log from esp, can you take a look?

[messages_20240418.txt](https://github.com/espressif/esp-hosted/files/15024883/messages_20240418.txt)

[image]

jiajia13201845317, 2024-04-19:
Hi @kapilkedawat ， the customer provide test information in the  attachment, could you please help checkout ? Thanks !
[image]
[20240419_message.txt](https://github.com/espressif/esp-hosted/files/15038359/20240419_message.txt)

kapilkedawat, 2024-04-19:
Hi @jiajia13201845317, Please help customer to capture dmesg correctly. We need to dmesg when customer loaded the kernel module(ie insmod esp32_sdionew.ko). Provided logs doesn't show any of those logs. 

Also please confirm that the pull-up b/w esp and broad matches our [SDIO pull-up requirements](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/peripherals/sd_pullup_requirements.html).