espressif/esp-hosted #365: ESP-Hosted-FG:about increasing the throughput rate.

wky19971002, 2024-04-11:
[image]

1.In ESP32-C2, no bluetooth,spi communication,our actually measured throughput rate is only about 5M,this is obviously significantly lower than the throughput performance,so the first point I want to know what links will affect the throughput rate or how I should check it.
2.The second point I want to know about "increase Wi-Fi Queues",Which parameters in the code can I tweak for this.

mantriyogesh, 2024-04-11:
https://github.com/espressif/esp-hosted/blob/3c3fc1029b00ac46d98ab98495b7a817ddb5bc2d/esp_hosted_fg/esp/esp_driver/network_adapter/sdkconfig.defaults.esp32c2#L30-L31

These queues need to be increased.

Check [porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md) to further identify how to further improve the performance..

For example finding out how much maximum transport (spi) link speed, increasing clock etc.

Also please note that, the throughout heavily depends upon the environment you use, here raw transport link max throughout can give some indication how much max transport link reach.

wky19971002, 2024-04-11:
Thanks,I will try it.