espressif/esp-hosted #367: Switching SDIO speeds in feature branch

claydonkey, 2024-04-16:
First off. Long time no see.
I would like to propose a pull request of my stm32h7 host code. It is pretty complete except for the SDIOIT part and the topic of this issue.
I don't think I have things set up properly to transfer to high speed sdio after the initial negotiation with the esp32 slave. the stm32 host codebase is now updated to the currnet esp_as_mcu_host  but the command look a bit stale on my front, in telling the slave to switch to a faster clock speed:

```c

/**
**
* @brief Initialize GPSIO in SDMMC AF
*        Please check SDMMC_reg.h for pin definitions
*
* @param  None
* @retval None
*/
void sdio_gpio_init(void)
{

    HAL_SD_MspInit(hsd1);
}
int sdio_card_init(void *ctx)
{
    log_sdio_clock_settings();
    INIT_SDIO_DEBUG();
    uint8_t bus_width;
    SDMMC_CmdInitTypeDef cmd;
    HAL_StatusTypeDef errorstate;
    g_h.funcs->_h_evt_grp_setbits(wait_trans_evt_grp, SDIO_DATA_FREE_BIT);
    uint32_t sdmmc_clk = HAL_RCCEx_GetPeriphCLKFreq(RCC_PERIPHCLK_SDMMC);
 
    __HAL_RCC_SDMMC1_CLK_ENABLE();
    SD_InitTypeDef sdmmc_init = {
        .ClockEdge = SDMMC_CLOCK_EDGE_RISING,
        .ClockPowerSave = SDMMC_CLOCK_POWER_SAVE_DISABLE,
        .BusWide = SDMMC_BUS_WIDE_1B,
        .HardwareFlowControl = SDMMC_HARDWARE_FLOW_CONTROL_DISABLE,
        .ClockDiv = sdmmc_clk / (2U * SD_INIT_FREQ)};

    sdio_trans_ready_mutex = g_h.funcs->_h_create_semaphore(10);
    SDMMC_Init(hsd1->Instance, sdmmc_init);
    hsd1->Instance->DCTRL |= SDMMC_DCTRL_SDIOEN;
    hsd1->Instance->POWER |= SDMMC_POWER_PWRCTRL;

    /* wait 74 Cycles: required power up waiting time before starting
     the SD initialization sequence */

    HAL_Delay(1U + (74U * 1000U / (sdmmc_clk / (2U * sdmmc_init.ClockDiv))));

    /* Enable SDMMC */

    /* EXTI interrupt init*/
    //  HAL_NVIC_SetPriority(EXTI9_5_IRQn, 5, 0);
    // HAL_NVIC_EnableIRQ(EXTI9_5_IRQn);
    //   __SDMMC_ENABLE_IT(hsd1->Instance, SDMMC_IT_SDIOIT);
    __SDMMC_OPERATION_ENABLE(hsd1->Instance);
    HAL_Delay(1U + (74U * 1000U / (sdmmc_clk / (2U * sdmmc_init.ClockDiv))));
    /* Command 0 - SD reset */

    cmd.Argument = 0U;
    cmd.CmdIndex = SDMMC_CMD_GO_IDLE_STATE;
    cmd.Response = SDMMC_RESPONSE_NO;
    cmd.WaitForInterrupt = SDMMC_WAIT_NO;

    cmd.CPSM = SDMMC_CPSM_ENABLE;
    SDMMC_SendCommandWait(&cmd);
    /* Command 5 - IO_SEND_OP_COND */
    cmd.Argument = 0U;
    cmd.CmdIndex = SDMMC_CMD_SDMMC_SEN_OP_COND;
    SDMMC_SendCommandWait(&cmd);

    /* Command 5 - Set VDD Voltage Window: 3.2~3.4V */
    cmd.Argument = 0x00FF8000;
    SDMMC_SendCommandWait(&cmd);

    /* Command 3 - Get WiFi address (CMD3: SEND_RELATIVE_ADDR,
     * Ask the card to publish a new relative address (RCA)) */
    cmd.Argument = 0;
    cmd.CmdIndex = SDMMC_CMD_SET_REL_ADDR;
    cmd.Response = SDMMC_RESPONSE_SHORT;
    SDMMC_SendCommandWait(&cmd);

    hsd1->SdCard.RelCardAdd = SDMMC_GetResponse(hsd1->Instance, SDMMC_RESP1);
    /* Command 7 - Select WiFi (SELECT/DESELECT_CARD) */

    cmd.Argument = 0;
    cmd.CmdIndex = SDMMC_CMD_SET_BLOCKLEN;
    SDMMC_SendCommandWait(&cmd);
    hsd1->SdCard.BlockSize = SDMMC_GetResponse(hsd1->Instance, SDMMC_RESP1);

    /* Above sequence is needed while communicating to IO only device.
     * In case of failure/timeouts despite of maximum retry, IO would be unusable
     */

    ESP_LOGI(TAG, "RCA: %lu", hsd1->SdCard.RelCardAdd);

    cmd.Argument = hsd1->SdCard.RelCardAdd;
    cmd.CmdIndex = SDMMC_CMD_SEL_DESEL_CARD;
    SDMMC_SendCommandWait(&cmd);

    uint8_t card_cap = 0;

    sdio_io_read_byte(hsd1, SDIO_FUNC_0, SD_IO_CCCR_CARD_CAP, &card_cap);
    if (!(card_cap & CCCR_CARD_CAP_LSC) || (card_cap & CCCR_CARD_CAP_4BLS))
    {
        bus_width = CCCR_BUS_WIDTH_4;
    }

    sdio_io_read_byte(hsd1, SDIO_FUNC_0, SD_IO_CCCR_BUS_WIDTH, &bus_width);
    ESP_LOGI(TAG, "Bus Width 0x%02x. sdmmc_clk: %d ClockDiv %d", bus_width, sdmmc_clk, sdmmc_init.ClockDiv);
    sdio_io_write_byte(hsd1, SDIO_FUNC_0, SD_IO_CCCR_BUS_WIDTH, CCCR_BUS_WIDTH_4, NULL);
    sdio_io_write_byte(hsd1, SDIO_FUNC_0, 0x02, 0x02, NULL);
    sdio_io_write_byte(hsd1, SDIO_FUNC_0, 0x10, 0x00, NULL);
    sdio_io_write_byte(hsd1, SDIO_FUNC_0, 0x11, 0x02, NULL);
    sdio_io_write_byte(hsd1, SDIO_FUNC_0, 0x110, 0x00, NULL);
    sdio_io_write_byte(hsd1, SDIO_FUNC_0, 0x111, 0x02, NULL);
 
    sdmmc_init.BusWide = SDMMC_BUS_WIDE_4B;
    sdmmc_init.HardwareFlowControl = SDMMC_HARDWARE_FLOW_CONTROL_DISABLE;
   sdmmc_init.ClockDiv = 0;
    errorstate = SDMMC_Init(hsd1->Instance, sdmmc_init);
    hsd1->Instance->DCTRL |= SDMMC_DCTRL_SDIOEN;
    if (errorstate)
        return ESP_FAIL;

    sdio_io_read_byte(hsd1, SDIO_FUNC_0, SD_IO_CCCR_BUS_WIDTH, &bus_width);
    ESP_LOGI(TAG, "Bus Width 0x%02x. sdmmc_clk: %d ClockDiv %d", bus_width, sdmmc_clk, sdmmc_init.ClockDiv);
    hsd1->State = HAL_SD_STATE_READY;

   //  output CIS info from the slave
     if (hosted_sdio_print_cis_information(hsd1) != ESP_OK)
      {
         ESP_LOGW(TAG, "failed to print card info");
       }

   
    // initialise the card functions
    if (hosted_sdio_card_fn_init(hsd1) != ESP_OK)
    {
        ESP_LOGE(TAG, "hosted_sdio_init failed");
        return ESP_FAIL;
    }

    return ESP_OK;
}

```
the part in question is either 
```c

    sdio_io_write_byte(hsd1, SDIO_FUNC_0, SD_IO_CCCR_BUS_WIDTH, CCCR_BUS_WIDTH_4, NULL);
    sdio_io_write_byte(hsd1, SDIO_FUNC_0, 0x02, 0x02, NULL);
    sdio_io_write_byte(hsd1, SDIO_FUNC_0, 0x10, 0x00, NULL);
    sdio_io_write_byte(hsd1, SDIO_FUNC_0, 0x11, 0x02, NULL);
    sdio_io_write_byte(hsd1, SDIO_FUNC_0, 0x110, 0x00, NULL);
    sdio_io_write_byte(hsd1, SDIO_FUNC_0, 0x111, 0x02, NULL);
 
    sdmmc_init.BusWide = SDMMC_BUS_WIDE_4B;
    sdmmc_init.HardwareFlowControl = SDMMC_HARDWARE_FLOW_CONTROL_DISABLE;
   sdmmc_init.ClockDiv = 0; << strange behaviour as I can't seem to get communication at any other clockspeed
  ```

or (and my memory is hazy on this)

```c
 if (hosted_sdio_card_fn_init(hsd1) != ESP_OK)
```

where do we tell the slave to change speed?

I hope everything is good with you and I hope to commence my project successfully as from when we last chatted.

mantriyogesh, 2024-04-17:
Hello @claydonkey ,

Good to see you back.. Hope you had great holidays.

Points to cover:
- Write bytes code snippet
- Slave to use 'High speed SDIO'

## Write bytes code snippet
The code in discussion came into question probably because, 
code in master branch refers to:

https://github.com/espressif/esp-hosted/blob/c7dd9e1dabc33fe5a3c7f9a93b6955de194b0bfe/esp_hosted_fg/host/stm32/driver/transport/sdio/sdio_ll.c#L464-L469

which seems stale. Using function0 is not correct in some argument. e.g.
```
sdio_io_write_byte(hsd1, SDIO_FUNC_0, 0x110, 0x00, NULL);
```
as it is func1 
Please refer to the https://github.com/espressif/esp-hosted/blob/7df1a1452364dc62d009e0e4fa911308bdb617ab/host/port/sdio_wrapper.c#L159-L165

Also, please note that func0 is not really used in ESP-Hosted.

mantriyogesh, 2024-04-17:
## Slave to use 'High speed SDIO'

Actually, the default speed at slave is by default to be 'High speed' if you use latest idf stable releases for slave building.
As part of the two steps frequency, SDIO probe/initialization/detection would be done at <=400KHz frequency.
Once CIS is populated, host can directly jump to the slave capability, I assume (@SohKamYung-Espressif will add comment further on this).

By default, slave would expose high speed capability:
https://github.com/espressif/esp-hosted/blob/7df1a1452364dc62d009e0e4fa911308bdb617ab/slave/main/Kconfig.projbuild#L227-L238

Although, you want to mandate slave you can add 
```
CONFIG_ESP_SDIO_HIGH_SPEED=y
```
in https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/slave/sdkconfig.defaults.esp32

then
```
$ rm -rf sdkconfig build
$ idf.py set-target esp32
$ idf.py flash menuconfig
```
The _sdkconfig_ file would show
```
CONFIG_ESP_SDIO_HIGH_SPEED=y
```
and `ESP_SDIO_DEFAULT_SPEED` would be commented

SohKamYung-Espressif, 2024-04-17:
@claydonkey This is from the SDIO Simplified Spec. You have to set the EHS (Enable High Speed) bit in the slave, and then switch the host to high speed.

[image]

claydonkey, 2024-04-17:
Thanks guys. So it was actually running at full speed. I just wasn't sure. That pesky problem with the SDIOT is still present so I am continuous polling interrupts but would I be correct in saying that shouldn't affect the transmission speed? I have a fully fledged app using lwip an sending protoc packets over UDP. The bandwidth is quite a bit slower than I would hope at 4Mbytes/sec.Al I can think of now is the physical connection is not good enough (the stm32 is running at a max of 32Mhz before failing...

mantriyogesh, 2024-04-18:
Hello @claydonkey ,

Software part
-----------------------

### Polling
I think polling is still fine, but the speed you get is low, not sure if you have logic analyser to measure the actual clock released from host.

Also, the throughout depends upon polling interval.  Polling could be optimised when there is already write ongoing from host.

### Throughout
Have you tried raw throughout test on this setup (both tx and rx individually)?

I remember you had got a better throughout in earlier issue raised, was that and this setup different?

Hardware part
-----------------------
Is the setup connected with jumper cables? 
I hope that you already have the appropriate pull-ups enabled for expected sdio lines.

To mitigate polling issue, you can also just new external interrupt using any unused GPIO. In this case, slave  would trigger the interrupt, when it has some data to send and clear. This interrupt now host is sure of incoming data and can start the sdio read sdio (better to read interrupt when write is not ongoing)

mantriyogesh, 2024-04-18:
May we know, which feature/esp_as_mcu, you are using as reference?

Also can you please send the host and ESP log (just to double confirm if any obvious incompatibility of codes)

claydonkey, 2024-04-18:
Hello @mantriyogesh 

> Have you tried raw throughout test on this setup (both tx and rx individually)?

 I hate to be a bore but I have forgotten how to do this...

> I remember you had got a better throughout in earlier issue raised, was that and this setup different?

I think it may be a case of refactoring, I havent changed anything fundamentally but I have ditched the iperf I was using before and will have to look at an earlier revision to test that again...

> I have a fully fledged app using lwip an sending protoc packets over UDP

The overhead I beleve is quite small. I am even using compression but when profiles, the result of 100ms for 1000 packets seems pretty trivial?!?

> Is the setup connected with jumper cables? I hope that you already have the appropriate pull-ups enabled for expected sdio 
the harware is a very sory ribbon cable between 2 sdmmc sockets.

> To mitigate polling issue, you can also just new external interrupt using any unused GPIO. 

 This seems like a good idea. It doesnt look like it's implemented in the esp slave code. Should I plonk it in myself?

```sh
W ********************************************************
W ********************************************************
W ********             ESP-Hosted                 ********
W ********************************************************
W ********************************************************
W Mcu CLK Freq: 280000000Hz
W Peripheral CLK Freq: 280000000Hz
W Transport - SDIO
D rpc_init
D rpc_slaveif_init
D Starting tx thread
D LwIP stack has been initialized
D esp-netif has been successfully initialized
D created task for loop 0x240967a8
D created event loop 0x240967a8
D esp_netif_add_to_list 0x2409e598
D esp_netif_add_to_list netif added successfully (total netifs: 1)
D check: remote, if=0x0 fn=0x9000c9fd

D call api in lwip: ret=0x0, give sem
I remote_esp_wifi_init_slave
I Add ESP-Hosted channel IF[1]: S[0] Tx[0x9002779d] Rx[0x9000ec65]
I Add ESP-Hosted channel IF[2]: S[0] Tx[0x900276ed] Rx[0x9000ec65]
I Attempt connection with slave: retry[0]
I Reset slave using GPIO[16]
I Starting SDIO process rx task
I RCA: 98304
I Bus Width 0x40. sdmmc_clk: 40000000 ClockDiv 98
I Bus Width 0x42. sdmmc_clk: 40000000 ClockDiv 0
W Mcu CLK Freq: 280000000Hz
W Peripheral CLK Freq: 40000000Hz
W Transport - SDIO
W SYSCLK Source: 3
W AHB Prescaler: 0
W APB1 Prescaler: 64
W APB2 Prescaler: 1024
W Flash Latency: 6
W Oscillator Type: 63
W HSE State: 65536
W LSE State: 0
W HSI State: 1
W LSI State: 0
W PLL State: 2
W PLL Source: 2
W PLL M: 12
W PLL N: 280
W PLL P: 2
W PLL Q: 2
[... 362 lines cut ...]
D Intr in Read Task: 004C7000
D Intr in Read Task: 00CC4000
D Intr in Read Task: 00CC7000
D Intr in Read Task: 004C7000
D Intr in Read Task: 00CC4000
D Intr in Read Task: 004C7000
D Intr in Read Task: 008C0000
D Intr in Read Task: 004C7000
D Intr in Read Task: 00CC4000
D Intr in Read Task: 004C7000
D Intr in Read Task: 00C00000
D Intr in Read Task: 004C7000
D Intr in Read Task: 00CC4000
D Intr in Read Task: 00CC7000
D Intr in Read Task: 004C7000
D Intr in Read Task: 00400000
D Intr in Read Task: 008C4000
D Intr in Read Task: 004C7000
D Intr in Read Task: 00CC4000
D Intr in Read Task: 004C7000
D Intr in Read Task: 00CC4000
D Intr in Read Task: 00CC7000
D Intr in Read Task: 00CC7000
D Intr in Read Task: 004C7000
```

mantriyogesh, 2024-04-19:
> I hate to be a bore but I have forgotten how to do this...

https://github.com/espressif/esp-hosted/blob/7df1a1452364dc62d009e0e4fa911308bdb617ab/host/port/esp_hosted_config.h#L198-L212

> All this comments below are not useful now as I have got iperf reinstated and I am getting nice rx at 20Mbits/sec in the new branch. So the problem could well be overhead in the main program...

Noted, thank you for the update.
Have you tried the TCP iperf? What I meant, TCP has to have acknowledgement, so rx and tx are interleaved in that sense.

Also, in general, there are  good amount of throughout optimization changes went in since your last commit, 
7dc00559a1d7991f06e07cc3ff1ede1a9a427ff3

Once you have some backup, you can move to latest to get these throughout numbers bumped.

To get overview of features, you can check 
[esp_hosted_config](https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/host/port/esp_hosted_config.h)

And [esp_hosted_config_bt](https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/host/port/esp_hosted_config_bt.h)

Regarding the outbound GPIO for triggering sdio as interrupt, it should be fairly easy change.
I am not right now sure how quickly I can provide the solution, but it should be fairly simple.

1. Interrupt addition in stm32 ioc/configuration
2. Enable sdio driver rw lock in host: https://github.com/espressif/esp-hosted/blob/7df1a1452364dc62d009e0e4fa911308bdb617ab/host/port/sdio_wrapper.c#L36-L42
3. ESP to trigger interrupt (I think easiest to tweak IDF sdio driver code, where existing d1 interrupt is being raised)

mantriyogesh, 2024-04-29:
Any update @claydonkey ?

claydonkey, 2024-04-29:
sorry @mantriyogesh. I am still here. I have suffered a bit of a setback I won't spare you the details but I am currently unable to type at any speed (my index finger suffered an accident). I have made all manner of changes and cleanup to my host code but not really addressed the issue at hand (no pun intended). Please bear with me. I really massively appreciate all your help.

mantriyogesh, 2024-04-29:
Oh dear. Take care.
Can you fork your repo? I can try with some stm and sort this ahead with external GPIO on top of your code, if that helps.