espressif/esp-hosted #368: default transmission power of WiFi and Bluetooth for the esp32-c3 chip

linchanghe123, 2024-04-17:
Hi, mantriyogesh

What are the default transmission power of WiFi and Bluetooth for the esp32-c3 chip in esp_hosted_ng and esp_hosted_fg modes, respectively?

mantriyogesh, 2024-04-17:
Hello @linchanghe123 ,

## Wi-Fi
If not configured, default power for Wi-Fi for any ESP chipset would be:
`min[max tx power of chip, max tx power allowed by regulatory]`

NG: https://github.com/espressif/esp-hosted/issues/329
FG: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/c_support/test.c#L95-L98

Let me confirm for Bluetooth

linchanghe123, 2024-04-18:
Thank you. Looking forward to your further reply.

> Hello @linchanghe123 ,
>
> ## Wi-Fi
> If not configured, default power for Wi-Fi for any ESP chipset would be: `min[max tx power of chip, max tx power allowed by regulatory]`
>
> NG: #329 FG: [...]

linchanghe123, 2024-04-29:
Bluetooth power is temporarily not being monitored, close issue.

closed 2024-04-29 by linchanghe123

mantriyogesh, 2024-04-29:
The default bluetooth transmit power for all ESP chipset is +9dBm.