espressif/esp-hosted #369: ESP-Hosted-FG: support SDIO block packets to wait WIFI tx  by ESP32C6

hanchq, 2024-04-18:
While ESP wifi buffer is full,  'esp_wifi_internal_tx' return a error, It will cause Host's data been discarded. We hope that ESP can privote a feedback to Host, so that Host will block packets to wait WIFI tx. Please help to support it. 
Looking forward to your response.

mantriyogesh, 2024-04-18:
Hello @hanchq ,

By the time this feedback is reached, the buffer was passed from host to slave and cleared from host.
So in short, there is no perfect way to get this done.

However, in the different repo, where we integrated MCU solution with ESP as slave and ESP as MCU host,
please refer: https://github.com/espressif/esp-hosted/blob/7df1a1452364dc62d009e0e4fa911308bdb617ab/slave/main/app_main.c#L469-L480

Is your application UDP or TCP based?

Two points:
1. (Right now only in feature/esp_as_mcu_host branch) When throttling is enabled from host, we can specify MIN and MAX threshold. Host would monitor slave buffering/queues. When slave queues towards wifi > MAX threshold, we start discarding incoming packet (proactive failure, to avoid in transit failure). We resume accepting packets when slave load falls below MIN threshold.
2. At the least, you can add:
```c
		int retry_wifi_tx = MAX_WIFI_STA_TX_RETRY;
		do {
			ret = esp_wifi_internal_tx(ESP_IF_WIFI_STA, payload, payload_len);
			retry_wifi_tx--;
		} while (ret && retry_wifi_tx);
```
Please just do not just add `vTaskDelay(XXX);` **without throttling**, as it may spike up memory at slave momentarily to accommodate new items.

mantriyogesh, 2024-04-24:
Any news @hanchq on this?