espressif/esp-hosted #370: esp32s3+sdio

snikeguo, 2024-04-19:
为什么没有esp32s3+sdio这种组合方式？同样的2.4g wifi4，sdio，我认为s3会比c6快吧？毕竟主频高，双核。为什么不做这个呢。

mantriyogesh, 2024-04-19:
Hello @snikeguo,

Your question is correct, but ESP32-S3 only supports sdio as master. It doesn't support sdio as slave. Due to this reason, we do not support S3 as sdio slave

snikeguo, 2024-04-19:
> 你好 ，
> 
> 你的问题是正确的，但 ESP32-S3 只支持 sdio 作为主站。它不支持 sdio 作为从属。因此，我们不支持将 S3 作为 sdio 从站

那可不可以s3做主机，stm32做从机？tcp rx tx速度能做到多少？

snikeguo, 2024-04-19:
我不确定stm32h7能不能做从机。。。看来只能用c6

mantriyogesh, 2024-04-19:
> 那可不可以s3做主机，stm32做从机？tcp rx tx速度能做到多少？

I am afraid out-of-the-box it is not supported. Although, we would think to adapt it that way.

For ESP32-C6 as SDIO **slave** with ESP32-P4 as **SDIO master**, With UDP iperf2, we reached around 55-58Mbps in Tx and 75Mbps in Rx.
This is tested in [feature/esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host) branch.

### Porting host code to ESP-Hosted for STM32
- Branch: [feature/esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host)
The porting layer is exposed at [host/port](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host/host/port), where you can add STM32H7 functionality.
- Branch: [master](https://github.com/espressif/esp-hosted/tree/master)

Although master being old branch, the transport layer port should be easily usable: [esp_hosted_fg/host/stm32](https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg/host/stm32)

使用谷歌翻译:

恐怕开箱即用的它不受支持。尽管如此，我们还是会考虑以这种方式进行调整。

对于 ESP32-C6 作为 SDIO **从站**，ESP32-P4 作为 **SD​​IO 主站**，使用 UDP iperf2，我们在 Tx 中达到约 55-58Mbps，在 Rx 中达到 75Mbps。
这是在 [feature/esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host) 分支中进行测试。

### 将主机代码移植到 STM32 的 ESP-Hosted
- 分支：[feature/esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host)
移植层公开在 [host/port](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host/host/port)，您可以在其中添加 STM32H7 功能。
- 分支：[master](https://github.com/espressif/esp-hosted/tree/master)

虽然 master 是旧分支，但传输层端口应该很容易使用：[esp_hosted_fg/host/stm32](https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg/host/stm32)

mantriyogesh, 2024-04-19:
> 我不确定stm32h7能不能做从机。。。看来只能用c6

I think STM32 can also support as sdio slave, but ESP-Hosted would need to change the role (and code for slave and host) accordingly. ESP32 as slave and ESP32 as host both codes are available. But the ESP32 always acted as SDIO slave till now in ESP-Hosted. Technically, it is feasible, but need some time to do.

使用谷歌翻译:
我认为 STM32 也可以支持 sdio 从机，但 ESP-Hosted 需要相应地更改角色（以及从机和主机的代码）。 ESP32 作为从机和 ESP32 作为主机两种代码均可用。但到目前为止，ESP32 在 ESP-Hosted 中始终充当 SDIO 从机。从技术上讲，这是可行的，但需要一些时间。

snikeguo, 2024-04-19:
What is the difference between master and feature/esp_as_mcu_host?

snikeguo, 2024-04-19:
I think I should use ESP32C6 as an SDIO slave and STM32H7 as an SDIO master. Should I use the “Master ” branch or the 
 “feature/esp_as_mcu_host” branch?

mantriyogesh, 2024-04-19:
### Branch master
**Pros**
- We added STM32 support as SPI and SDIO out of the box in master, for specific STM32 model at that time.
- The transport layer is ready available for consumption

**Cons**
- No LWIP or Bluetooth integrated, you would need to do it on yourself
- arping is tested for network stack integration 'PoC'

### Branch feature/esp_as_mcu_host
as STM32 nodes keep evolving, it became tedious to maintain all models. So we created new branch, where we showcase all esp chipsets as host. It becomes easier to anyone to refer this host side solution & test it to verify.
There was underlying issue with STM32 that it doesn't directly support SDIO, instead we had to tweak the HAL layer of `SD-MMC` to support. But this over the time complicated things to maintain. afterall, we could maintain `Espressif` code and not `STM32` HAL copy.

**Pros**
- Major fixes in transports available
- lwip, bluetooth out-of-the-box supported
- throughputs are highly optimized
- Distinct and easy port layer available for non-ESP chipset as host

**Cons**
- STM32 directly as host is not added, left to user. afterall, not all stm32
- Some host side code is just used right from ESP-IDF as it was readily available in ESP as host environment, which needs separation effort.
  - We have done some part of separation already, but it is still in development and has lower priority
  - At the same time, we can share what we have, if that helps for your adaption

mantriyogesh, 2024-04-19:
Some suggestions:
1. Evaluate ESP as host first! Use ESP32-C6 as slave and some other ESP as host (SDIO host supported over S3/P4/ESP32, otherwise ESP32/S2/S3/C2/C3/C6 support SPI) and evaluate the solution, where you get idea, what exactly you can achieve
2. Use base branch as feature/esp_as_mcu_host and add a porting layer from master, it might not really be straight forward, as some ESP-IDF code may need to be added in host side code

You can also refer to #367 & #270 where @claydonkey has put good & quality efforts and could help you in this regard.

snikeguo, 2024-04-19:
> Some suggestions:
>
> 1. Evaluate ESP as host first! Use ESP32-C6 as slave and some other ESP as host (SDIO host supported over S3/P4/ESP32, otherwise ESP32/S2/S3/C2/C3/C6 support SPI) and evaluate the [...]

I have to use STM32H723. I am working on a CANFD BUS to WIFI conversion device. Currently, only STM32H723 has 3 CANFD channels

snikeguo, 2024-04-19:
Do you have the PORT code for STM32 under the feature/esp_as_mcu_host branch?
[image]
Do I need to port this file?

mantriyogesh, 2024-04-19:
Check files in

https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg/host/stm32/driver/transport/sdio

Check port for STM32 sdio using sdmmc hal changes. See files sdio_ll.c, sdio_ll.h
You would need to integrate the hal functions from these files, such that functions in sdio_wrapper.c are ported.

Simplified sdio slave communication reference documentation is at https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/peripherals/esp_slave_protocol.html

mantriyogesh, 2024-04-19:
Although chances are less for STM32, but if at all your STM32 allows direct sdio communication, you can use it directly instead of SDMMC HAL tweaks.