espressif/esp-hosted #376: When will esp_hosted_ng  spi support AP mode

kangpei1, 2024-05-07:
i need esp32 as a ap mode device,so i want to know When will esp_hosted_ng  spi support AP mode,thank you.

mantriyogesh, 2024-05-08:
We're working on adding softAP support for NG, but it'll take some time to merge it into the master branch.
If you want to test it early, I've attached [2024.04.24_softap_patch.tgz](https://github.com/espressif/esp-hosted/files/15247233/2024.04.24_softap_patch.tgz)
 for you to try. Please check the readme.txt for instructions.

kangpei1, 2024-05-10:
> We're working on adding softAP support for NG, but it'll take some time to merge it into the master branch. If you want to test it early, I've attached [...]

Thank you very muck ,I will try it.

kangpei1, 2024-05-22:
@mantriyogesh Hi , I download this patch,and try in ,but I get this log ,why esp32 inint sta mode
log in my linux host NUC980
[image]

log in esp32
[image]

mantriyogesh, 2024-05-22:
@Shreyas0-7 , can you please check this?

Shreyas0-7, 2024-05-22:
Hey @kangpei1 default mode is sta mode you can start AP mode using hostapd
- write your configuration for your AP in file such as host.conf 
- start AP mode using `hostapd host.conf` command

PercyHide, 2024-08-05:
hi @Shreyas0-7 @mantriyogesh  I have solved the linux host network init faild problem, I try start ap mode,but its not success. 

**1,my host.conf**
[image]
2,my linux log
[image]

PercyHide, 2024-08-05:
sorry for my mistak, I useed mater so its not work, now I have creat a wifi ,but I cant connect it, this is my log in esp32
[image]

PercyHide, 2024-08-05:
my linux log
[image]

Shreyas0-7, 2024-08-05:
@PercyHide Is it possible for you to attach sniffer capture, also please post hostapd logs and dmesg logs

PercyHide, 2024-08-06:
1,my linux log
[linux_log.txt](https://github.com/user-attachments/files/16504334/linux_log.txt)
2,my host.conf
[image]
3,my esp32 log
[image]

PercyHide, 2024-08-06:
i think i have the same problem with https://github.com/espressif/esp-hosted/issues/415.
1,when i used 2024.04.24_softap_patch.tgz ,i can have a wifi ,but i cant connect it.
2,when i used master , i cant get wlan0, and i get the log
[image]

mantriyogesh, 2024-08-06:
Hello @PercyHide , I doubt your SPI porting has some issues. Porting is one time activity, which you would have already performed, but may still have done issues. 

If you import selective code from
https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L637

does it help?

Please include this line and dependent code from FG file to ng file and reflash ESP and try again

mantriyogesh, 2024-08-06:
What I meant, before evaluation of networking (be it sta or softap), I think it is worth to evaluate the underlying transport.

If above doesn't work, do you have logic analyser?

PercyHide, 2024-08-06:
> What I meant, before evaluation of networking (be it sta or softap), I think it is worth to evaluate the underlying transport.
> 
> If above doesn't work, do you have logic analyser?

i am sorry,i used esp_host_ng, and i dont have logic analyser.

PercyHide, 2024-08-06:
the sta mode i used is ok ,and i want to confirm did the master have support soft ap mode?

PercyHide, 2024-08-06:
I have a question

The v1.0.3 matser in esp_spi.c

funtion :static void esp_spi_work(struct work_struct *work)

hardware_type dont have be a value; its different in last version, so  when i used v1.0.3 ,i cant get wlan0, my spi didt work normal because trans.cs_change didt be 1, when i Assign trans.cs_change value of 1， my spi work normal, i can get wlan0.

[image]

PercyHide, 2024-08-06:
> Hello @PercyHide , I doubt your SPI porting has some issues. Porting is one time activity, which you would have already performed, but may still have done issues.
>
> If you import selective code from [...]

i have try it , i also cant connect the ap

mantriyogesh, 2024-08-06:
right, this message is not from transport, but from:

https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c#L858

which is registered in slave only, https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c#L926

I will leave @Shreyas0-7 to further check on this in the AP debug direction, as this frame is received from slave side Wi-Fi driver, as I can observe.

BTW, You can ignore https://github.com/espressif/esp-hosted/issues/376#issuecomment-2270264669

PercyHide, 2024-08-06:
> right, this message is not from transport, but from:
> 
> https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c#L858
> 
> which is registered in slave only,
> 
> https://github.com/espressif/esp-hosted/blob/6ddb670edcf15108be92d7e8e4fc6f32542ef5e4/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c#L926
> 
> I will leave @Shreyas0-7 to further check on this in the AP debug direction, as this frame is received from slave side Wi-Fi driver, as I can observe.
> 
> BTW, You can ignore [#376 (comment)](https://github.com/espressif/esp-hosted/issues/376#issuecomment-2270264669)

OK, Thank you , What can i do next?

mantriyogesh, 2024-08-06:
@Shreyas0-7 PTAL..

Shreyas0-7, 2024-08-06:
@PercyHide Is it possible for you to capture sniffer?

PercyHide, 2024-08-06:
> @PercyHide Is it possible for you to capture sniffer?

Sorry , i cant

Shreyas0-7, 2024-08-06:
I suspect connection is not going through because RSN IE is not present in assoc req
`
wlan0: STA 50:84:92:04:75:9f IEEE 802.11: authenticated
`
`
wlan0: STA 50:84:92:04:75:9f IEEE 802.11: No WPA/RSN IE in association request
`
To confirm this I might need sniffer capture

PercyHide, 2024-08-06:
> I suspect connection is not going through because RSN IE is not present in assoc req `wlan0: STA 50:84:92:04:75:9f IEEE 802.11: authenticated` `wlan0: STA 50:84:92:04:75:9f IEEE 802.11: No WPA/RSN IE in association request` To confirm this I might need sniffer capture

Maybe you can tell me ,what should i do.