espressif/esp-hosted #380: about  use  esp_hosted_ng

lipengvita, 2024-05-11:
1 I have compile the esp_hosted_ng  both esp32_spi.ko    and esp32c3  bin  file
2  when esp32-spi.ko  I change the data read and handshake pin
3  the linux side use T113_i  system . esp32c3 is esp32c3-zero
4 connect mosi miso clk  nss handshak  dataread   witch esp32c-zero , first  linux boot and insmod esp32_spi.ko  I get the message
esp32_spi:spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]
then I boot the esp32c3 , but I can find network inteface in linux system

mantriyogesh, 2024-05-11:
@lipengvita ,

Please share ESP and host side full textual logs, to analyse the issue.

lipengvita, 2024-05-11:
host side just one line logs
  sp32_spi:spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]
 esp  side 
 FW_MAIN: Initial set up done  
I (10636) main_task: Returned from app_main()
I check the code  host side need a message from esp side 
when host side recevie esp side boot event I will bring up interface . I guess it maybe have  not recevice the message

lipengvita, 2024-05-13:
I  set the linux  spi  bus  clock  30M  an  insmod esp32c3-spi.ko  with clockspeed-30    the atach is logic analyzer capacture , pealse give adiv
[2024-05-13_09-24-09.zip](https://github.com/espressif/esp-hosted/files/15288376/2024-05-13_09-24-09.zip)
e

mantriyogesh, 2024-05-13:
Can you please point me the software (with version ) to use for opening this file?

lipengvita, 2024-05-13:
[image]

lipengvita, 2024-05-13:
i want change the clock speed to 10M,
I change the linux host dts spi clock 10 M
and change the  esp/esp_driver/network_adapter/network_adapter/main/spi_slave_api.c     #define SPI_CLK_MHZ 10 
Do it ok

mantriyogesh, 2024-05-13:
Do you mean with 10M it is working fine?

lipengvita, 2024-05-13:
In 10M clock spi no data So I want to sure the spi clock change is correct.  30M spi master send data ,slave not send data

lipengvita, 2024-05-13:
additionnay . My esp32c3 device not connect reset with linux host and powered by pc   ， after I insmod  linux esp32-spi.ko ,I reset the esp32c32 device by touch the esp32c3 reset buttion

lipengvita, 2024-05-13:
Do the reset pin is mandatory?

mantriyogesh, 2024-05-13:
I think you should check [porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) first.

Check your jumper cables lengths, check if they are firm connected, etc.

From your last image, the clock seems not correct. the duty cycle sent from host should be around 50% to use SPI correctly.

mantriyogesh, 2024-05-13:
> Do the reset pin is mandatory?

For SPI, yes.
But even in SPI, for first transaction it is not mandatory.

mantriyogesh, 2024-05-13:
Can you please share:
1. dmesg log from bootup
2. esp log from start
3. git commit used at ESP and host

Anyway after stepwise following porting guide, you should be able to rule out generally faced issues.

lipengvita, 2024-05-13:
I change the linux host dts spi clock 10 M
and change the esp/esp_driver/network_adapter/network_adapter/main/spi_slave_api.c #define SPI_CLK_MHZ 10    Is  this  correct？

mantriyogesh, 2024-05-13:
> I change the linux host dts spi clock 10 M and change the esp/esp_driver/network_adapter/network_adapter/main/spi_slave_api.c #define SPI_CLK_MHZ 10 Is this correct？

This is correct, but if your underlying spi is not ported correctly, it may not work.
Check the porting guide for the details.

lipengvita, 2024-05-13:
If the length of the Dupont wire is greater than 6cm, will the device not return to normal operation?

mantriyogesh, 2024-05-13:
High freq would fail.
10MHz should work with 10cm cables fine.

Please check below points:
1. Answers for https://github.com/espressif/esp-hosted/issues/380#issuecomment-2106686608.
2. Check SPI clock stability from host.
3. GPIOs correctly working
4. output of `ls /dev/spidev*`

We will wait for full logs and answers for above questions.

lipengvita, 2024-05-15:
@mantriyogesh  doese this ' 4 output of ls /dev/spidev* '  mean when the esp32_spi.ko  is insmod the /dev/spidev* can be find?

lipengvita, 2024-05-15:
Do I need add this in code , I see a issue you answer  content add this 
[image]

lipengvita, 2024-05-15:
@mantriyogesh   espc32c3 IO2 is download gpio  ,user I02 as MISO do it ok?

mantriyogesh, 2024-05-15:
Please send above logs first.
Without textual logs, it would be incorrect to suggest something.

lipengvita, 2024-05-15:
@mantriyogesh   esp log 
I (259) esp_image: segment 4: paddr=000c77d4 vaddr=4038ccI (446) FW_MAIN: - WLAN                                                                                                                       over SPI
I (456) FW_BT: - BT/BLE
I (456) FW_BT:    - HCI Over SPI
I (456) FW_BT:    - BLE only
I (466) FW_MAIN: Capabilities: 0xe8
pp rom version: 9387209
net80211 rom version: 9387209
I (486) wifi:wifi driver task: 3fca15c0, prio:23, stack:6656, core=0
I (486) wifi:wifi firmware version: fddc5e5
I (486) wifi:wifi certification version: v7.0
I (486) wifi:config NVS flash: disabled
I (486) wifi:config nano formating: disabled
I (496) wifi:Init data frame dynamic rx buffer num: 32
I (496) wifi:Init management frame dynamic rx buffer num: 32
I (506) wifi:Init management short buffer num: 32
I (506) wifi:Init dynamic tx buffer num: 32
I (516) wifi:Init static tx FG buffer num: 2
I (516) wifi:Init static rx buffer size: 1600
I (516) wifi:Init static rx buffer num: 10
I (526) wifi:Init dynamic rx buffer num: 32
I (526) wifi_init: rx ba win: 6
I (536) wifi_init: tcpip mbox: 32
I (536) wifi_init: udp mbox: 6
I (536) wifi_init: tcp mbox: 6
I (546) wifi_init: tcp tx win: 5744
I (546) wifi_init: tcp rx win: 5744
I (556) wifi_init: tcp mss: 1440
I (556) wifi_init: WiFi IRAM OP enabled
I (556) wifi_init: WiFi RX IRAM OP enabled
I (566) BLE_INIT: BT controller compile version [963cad4]
I (566) phy_init: phy_version 970,1856f88,May 10 2023,17:44:12
I (616) BLE_INIT: Bluetooth MAC: f4:12:fa:39:c4:06

I (616) FW_MAIN: ESP Bluetooth MAC addr: f4-12-fa-39-c4- 6

I (616) FW_SPI: Using SPI interface
I (616) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (626) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (30636) FW_SPI: Initial  QueueSend  254
E (30636) FW_SPI:  queue new transaction
  linux host log  [ 1952.174756] spi data ready interrupt handler  this  I add in esp32_spi.ko

closed 2024-05-15 by lipengvita