espressif/esp-hosted #383: esp_hosted_ng, there are some problems with scanning wifi

mengleia, 2024-05-15:
system: linux kernel 6.1.36
esp driver: ESP-Hosted-NG_release_v1.0.2
esp-hosted:  5042106
 
On our imx8mm board, we can already scan and connect to wifi. Recently, it was found that some ssid could not be scanned out, and it was determined to be 2.4g.

mantriyogesh, 2024-05-15:
Hello @mengleia ,

Can you please retry using master branch at both sides?

mengleia, 2024-05-15:
hi @mantriyogesh 
Yes, Im using the master branch, but it is not the latest commit. Just now I had the same problem with the latest code test

mantriyogesh, 2024-05-15:
I meant latest commit on master branch at both the sides.
We had fixed multiple similar issues you are facing.

If you still face the same issue on **latest master**, please help with:
1. ESP log with 'DEBUG' level
Run `idf.py menuconfig` and change default log level:
Component config → Log output → Default log verbosity 
Change to `Debug`
2. dmesg log of host from bootup
3. wpa_supplicant debug logs (add -d in wpa_suplicant command)
4. Sniffer capture (start before triggering connect to capture EAPOL frames)

If we have all these logs, we can understand the current wifi communication from ESP to AP and would help you to prove the case.
If you miss one or some of them, we might guess and suggest, as case understood from the logs.

mengleia, 2024-05-15:
I did not set up the esp32 development environment, I used ESP-Hosted-NG_release_v1.0.2 firmware. 
Just now I found that the ssid that could not be scanned before could be scanned, and I did not make any modification. I spent the whole afternoon yesterday testing.
I found that some AP with good signal strength were often not on the scan list, and I used the wpa_cli -i espsta0 scan_r command
[image]

mantriyogesh, 2024-05-15:
@kapilkedawat ++

mengleia, 2024-05-16:
Hello, is there any solution? This problem is very serious, will lead to the user can not connect to the desired AP, multiple searches can not find the AP, now the hardware has been determined, so it must be dealt with on the software

kapilkedawat, 2024-05-16:
Hi @mengleia, could you please retry with latest firmware on master branch? You can compile the firmware using the steps mentioned in https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/README.md

mengleia, 2024-05-16:
@kapilkedawat  Thank you for your reply. Do you have the firmware of only sdio with the latest code compiled? Because the compilation environment here is somewhat complicated, I don't want to build the compilation environment of esp32 again. If not, I have to build the development environment of esp32

kapilkedawat, 2024-05-16:
[esp32_sdio_firmware.zip](https://github.com/espressif/esp-hosted/files/15332144/esp32_sdio_firmware.zip)
Please retry with attached firmware and latest master host code. Incase your esp32 needs external antennas, please make sure they are attached.

mengleia, 2024-05-16:
Thank you very much. I will give you a reply after testing for some time

mengleia, 2024-05-17:
@kapilkedawat I used the latest code test, in the latest same environment, I compared the scanning AP of rtl8723bu and esp32, and found that the same AP signal strength difference is quite large, the number of esp32 scan AP is relatively small, and the antenna has been installed normally. Is it related to reception sensitivity?
[image]
[image]

kapilkedawat, 2024-05-17:
Hi @mengleia, thank you for your response.

Could you please provide following info.
1. rtl8723bu is located at similar location as esp32?
2. have you retried to check the antenna pins properly?
3. How are you powering up esp32? 

Could you also please flash [scan](https://github.com/espressif/esp-idf/tree/master/examples/wifi/scan) example from esp-idf and run it(you can connect a serial interface to get the logs). This will give us some info whether is it related to driver or chip itself is getting these values from hardware.

mengleia, 2024-05-17:
@kapilkedawat 
1. Yes, I have an imx8mm development board that uses the rtl8723bu, and I am testing the rtl8723bu and esp32 in the same environment
2. We are using esp32-wroom-32ue, the antenna has been inserted normally, is there anything special in the hardware?
3. The esp32 restarts when the driver is loaded. The following is a print of me loading the driver and performing a scan operation
[image]
[esp32_run.log](https://github.com/espressif/esp-hosted/files/15344244/esp32_run.log)

kapilkedawat, 2024-05-17:
@mengleia this is indeed odd, you shouldn't see this much difference in RSSI if both are at similar location. Did you try esp-idf example by any chance that I mentioned in https://github.com/espressif/esp-hosted/issues/383#issuecomment-2116625973 ? Please try that on this chip and let us know the output. 

Do you use similar antenna on both the chips(realtek and esp)? Also do you have any ESP32-WROOM-32 with internal intenna, if yes could you please try to run the scan example on that as well.

mengleia, 2024-05-17:
@kapilkedawat  yes, the both the chips(realtek and esp) at similar location, I tried switching the two antennas, and nothing changed. i have not ESP32-WROOM-32 with internal intenna.
You know, I use the esp32 as a wifi module, I didn't build a development environment. If there is a need, I will go to see how to compile esp32 program.

kapilkedawat, 2024-05-17:
Hi @mengleia , yes please build the scan example and share the logs with us.

mengleia, 2024-05-18:
Hi @kapilkedawat, Here are my scan results using esp32 scan example and rtl. Both machines were on my desktop, and the test environment and time were conducted almost simultaneously.
[esplog.txt](https://github.com/espressif/esp-hosted/files/15360183/esplog.txt)
[image]

kapilkedawat, 2024-05-18:
Hi @mengleia, seems like esp32 reception is having some issue.
1. Others devices also show similar RSSI as rtl?
2. How far is `Veryark_yyq` from esp32? is it in line of site?

So far I can only suspect antenna connector or antenna causing this issue.

mengleia, 2024-05-20:
@kapilkedawat 
1. phone and Pc can show similar RSSI
2. Veryark yyq and esp32 are in the same room, about 3.2m

kapilkedawat, 2024-05-22:
Hi @mengleia, if you have another chip and antenna connector, could you please try with that?