espressif/esp-hosted #384: esp32c3  run  wpa_supplicant  , ping run in command ,  the ap close and reopen    get the panic info

lipengvita, 2024-05-15:
PING 192.168.43.1 (192.168.43.1): 56 data bytes
ping: sendto: Network is unreachable
#@chint espsta0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
ping 192.168.43.1
PING 192.168.43.1 (192.168.43.1): 56 data bytes
ping: sendto: Network is unreachable
#@chint [  236.328839] Internal error: Oops: 817 [#1] PREEMPT SMP ARM
[  236.334996] Modules linked in: esp32_spi(O) sunxi_ce [last unloaded: 8188eu]
[  236.342916] CPU: 1 PID: 34 Comm: kworker/1:1 Tainted: G           O      5.4.61 #12
[  236.351502] Hardware name: Generic DT based system
[  236.356915] Workqueue: ESP_CMD_WORK_QUEUE esp_cmd_work [esp32_spi]
[  236.363871] PC is at esp_cmd_work+0x78/0x170 [esp32_spi]
[  236.369833] LR is at arch_spin_lock+0xc/0x48
[  236.374619] pc : [<bf010b38>]    lr : [<c01502c4>]    psr: a00f0013
[  236.381647] sp : ce987f20  ip : 00000000  fp : 00000000
[  236.387505] r10: bf0188e4  r9 : bf0188a0  r8 : bf0188b0
[  236.393364] r7 : bf0188a8  r6 : 00000000  r5 : cd83ff40  r4 : bf018838
[  236.400684] r3 : 00000122  r2 : cd83ff54  r1 : bf0188a8  r0 : 00000000
[  236.408007] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  236.416013] Control: 10c5387d  Table: 4d4d006a  DAC: 00000051
[  236.422458] Process kworker/1:1 (pid: 34, stack limit = 0x46e93011)
[  236.429487] Stack: (0xce987f20 to 0xce988000)
[  236.434378] 7f20: ce93db00 cedbe9c0 bf0188e0 cedc6000 00000000 00000000 bf0188e4 c012fdcc
[  236.443559] 7f40: ce93db00 bf0188e0 ce93db00 ce93db14 cedbe9c0 ce986000 cedbe9d8 c0c02d00
[  236.452741] 7f60: c0c0b92c c0130394 00000000 ce816b80 ce986000 ce92af00 ce93db00 ce875ec4
[  236.461920] 7f80: c0130198 ce816b9c 00000000 c0134ba0 ce92af00 c0134a8c 00000000 00000000
[  236.471095] 7fa0: 00000000 00000000 00000000 c01010e8 00000000 00000000 00000000 00000000
[  236.480262] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  236.489438] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[  236.498653] [<bf010b38>] (esp_cmd_work [esp32_spi]) from [<c012fdcc>] (process_one_work+0x16c/0x20c)
[  236.508914] [<c012fdcc>] (process_one_work) from [<c0130394>] (worker_thread+0x1fc/0x2dc)
[  236.518094] [<c0130394>] (worker_thread) from [<c0134ba0>] (kthread+0x114/0x128)
[  236.526397] [<c0134ba0>] (kthread) from [<c01010e8>] (ret_from_fork+0x14/0x2c)
[  236.534496] Exception stack(0xce987fb0 to 0xce987ff8)
[  236.540163] 7fa0:                                     00000000 00000000 00000000 00000000
[  236.549339] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  236.558514] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  236.565940] Code: e3550000 0afffff8 e895000c e5823004 (e5832000)
[  236.572825] ---[ end trace 92657f3d950d96ff ]---
[  236.578009] Kernel panic - not syncing: Fatal exception in interrupt
[  236.585146] CPU0: stopping
[  236.588191] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D    O      5.4.61 #12
[  236.596488] Hardware name: Generic DT based system
[  236.601879] [<c010e2f8>] (unwind_backtrace) from [<c010a8bc>] (show_stack+0x10/0x14)
[  236.610577] [<c010a8bc>] (show_stack) from [<c07b1030>] (dump_stack+0x7c/0x98)
[  236.618687] [<c07b1030>] (dump_stack) from [<c010c6b0>] (handle_IPI+0xc0/0x168)
[  236.626896] [<c010c6b0>] (handle_IPI) from [<c03a7134>] (gic_handle_irq+0x70/0x78)
[  236.635394] [<c03a7134>] (gic_handle_irq) from [<c01021cc>] (__irq_svc+0x6c/0xa8)
[  236.643787] Exception stack(0xc0c01f40 to 0xc0c01f88)
[  236.649457] 1f40: 000c5b44 cedb0574 00000000 c0115280 00000001 c0c00000 c0c03e28 c0c03e64
[  236.658638] 1f60: 00000000 cefff380 c0b2f0c0 00000000 c0c76830 c0c01f90 c0107fd8 c0107fdc
[  236.667814] 1f80: 600f0013 ffffffff
[  236.671731] [<c01021cc>] (__irq_svc) from [<c0107fdc>] (arch_cpu_idle+0x2c/0x38)
[  236.680034] [<c0107fdc>] (arch_cpu_idle) from [<c013e920>] (do_idle+0xb8/0x120)
[  236.688241] [<c013e920>] (do_idle) from [<c013ec24>] (cpu_startup_entry+0x18/0x1c)
[  236.696737] [<c013ec24>] (cpu_startup_entry) from [<c0b00c68>] (start_kernel+0x340/0x3d0)
[  236.705913] invalid reason or reasonbase NULL
[  236.710808] rtc set bootreason panic reboot fail
[  236.716006] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---

mantriyogesh, 2024-05-15:
@kapilkedawat ++

kapilkedawat, 2024-05-18:
Hi @lipengvita , what do you mean by ap close and open here? Are you running supplicant or hostapd?

lipengvita, 2024-05-20:
Hi  @kapilkedawat  I just run supplicant  , use esp32c3 mode connect an ap ssid  , when I close the ap and reopen the ap ,esp32c3 mode host side crash

kapilkedawat, 2024-05-20:
Hi @lipengvita, are you using latest code?

lipengvita, 2024-05-21:
@kapilkedawat  thks  I user the lastest code  ， I can not get the crash again

closed 2024-05-21 by lipengvita