espressif/esp-hosted #386: esp_hosted_ng STA+AP Mode does not work simultaneously

danielhartwig, 2024-05-16:
I started the hostapd with the host.conf then i started a scan on the wlan0 device which is an esp32 connected via SPI to an Raspberry Pi 3B. The Scan via "sudo iw wlan0 scan" results in "command failed: Operation not permitted (-1)".

####
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on
#dtoverlay=sdio,poll_once=off
dtoverlay=disable-bt
# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1
[... 593 lines cut ...]
[08:49:55:182] <0x1b>[0m␍␊

[08:49:55:182] <0x1b>[0;32mI (169916) FW_CMD: Setting GTK [1]␍␊

[08:49:55:182] <0x1b>[0m␍␊

[08:49:55:214] <0x1b>[0;32mI (169956) FW_MAIN: Set default key request␍␊

[08:49:55:214] <0x1b>[0m␍␊

[08:51:41:055] <0x1b>[0;32mI (275794) FW_MAIN: Scan request␍␊

[08:51:41:055] <0x1b>[0m␍␊

[08:51:41:055] <0x1b>[0;32mI (275794) FW_CMD: Scan failed ret=[0xffffffff]␍␊

[08:51:41:055] <0x1b>[0m␍␊

[08:58:48:914] <0x1b>[0;32mI (703654) FW_MAIN: Scan request␍␊

[08:58:48:914] <0x1b>[0m␍␊

[08:58:48:914] <0x1b>[0;32mI (703654) FW_CMD: Scan failed ret=[0xffffffff]␍␊

[08:58:48:930] <0x1b>[0m␍␊

mantriyogesh, 2024-05-16:
@Shreyas0-7 , Can you please check this?

mantriyogesh, 2024-05-16:
@danielhartwig ,

Apologies from my side, The NG softAP binary shared earlier, only handles softAP functionality and doesn't work with Station interface.
This functionality is complex and yet development in progress for STA+AP, which may take time.

You can either use ESP-Hosted-FG, for STA+AP use case.

ESP-Hosted-FG supports the station+softAP and can be easily evaluated using simple [C](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/c_demo.md) or [python](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/python_demo.md) user space application.

Full FG doc: https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg

lipengvita, 2024-05-16:
@mantriyogesh  Dose ng supplort ap mode ,how to config ?

mantriyogesh, 2024-05-16:
Hello @danielhartwig ,

`Station + softap` combination is not supported in the *NG* binary I had shared earlier.
Individually Station or Softap should work.

If requirement is for `Station + softap`, please use FG.

danielhartwig, 2024-05-17:
Hello @mantriyogesh, 
do have a Release Date or Month for this feature-set  ?

> If requirement is for Station + softap, please use FG.

we are thinking about this possibility

mantriyogesh, 2024-05-17:
FG: supports station & softap simultaneously.

NG: it may get delayed as it is a complicated scenario. No specific timeline to commit.

wzmnop, 2025-04-30:
> FG: supports station & softap simultaneously.
> 
> NG: it may get delayed as it is a complicated scenario. No specific timeline to commit.

Hi @mantriyogesh,
Is there any updates on supporting station & softap simultaneously for NG？
I tried to create virtual interface for AP mode, but failed.
`iw dev wlan0  interface add ap0 type __ap
command failed: Operation not supported (-122)
`
The command failed due to lack of driver support for virtual interface creation.
`
static struct cfg80211_ops esp_cfg80211_ops = {
#if 0
    .add_virtual_intf = esp_cfg80211_add_iface,
    .del_virtual_intf = esp_cfg80211_del_iface,
#endif
    .....
};
`

mantriyogesh, 2025-04-30:
Right now, it is not supported on NG, @danielhartwig .

We do not have a timeline for this to commit. If your use case is sta+ap, I would suggest to use FG for now.

@kapilkedawat Can you please brief the complications involved at NG firmware side?

wzmnop, 2025-04-30:
> Right now, it is not supported on NG, [@danielhartwig](https://github.com/danielhartwig) .
>
> We do not have a timeline for this to commit. If your use case is sta+ap, I would suggest to use FG for [...]

@mantriyogesh 

I'm using linux host. If use FG，does that mean I can't use socket to program?

mantriyogesh, 2025-04-30:
FG and NG differ only in how Wi-Fi is controlled.

NG uses the standard method: Wi-Fi is managed by `wpa_supplicant`, and `libnl` is used to control connections.
FG, on the other hand, provides RPC APIs to control Wi-Fi directly—for example, calling `connect_ap(ssid, password)` would initiate the connection and return 0 on success or an error code otherwise.

See the key differences here: https://github.com/espressif/esp-hosted?tab=readme-ov-file#22-key-differences

The difference is only at OSI Layer 2 (Wi-Fi link layer). From Layer 3 onward (IP, TCP/UDP), everything is handled by Linux. That means you can use sockets normally in both FG and NG. The networking stack above Layer 2 is the same.

FG allows the APIs to be called from either C or Python application.
FG Demo application is available at https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg#32-demo-apps

The FG python demo GIF can be easily understand at: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/python_demo.md#modes-supported

danielhartwig, 2025-04-30:
@mantriyogesh Could you close this Ticket please.
@wzmnop open for your purpose a ticket please. other advice for your Problem. Instead of using an ESP32-based solution i would look after an cheap Realtek Wifi USB Module like https://www.alibaba.com/product-detail/Realtek-RTL8731BU-Professional-Supplier-Lower-Power_1600914925732.html

This Module supports the STA+AP Mode too.

I used an Azurewave 8997 Module instead. Please be aware that if they release the NG Version with this new Feature it could have some Childhood diseases. I would prefer the USB Wifi solution before using an non wpa-supplicant conform solution.

mantriyogesh, 2025-04-30:
@danielhartwig,
Obviously you are free to evaluate which works the best for you.

I understand you might have bitter experience, as ESP not having softap + sta support on NG.
We also wish to add it, but we cannot give false commitments. Further, Unless the things work fine we would not open any feature.

Please let us know if you have some specific issues, we can look at. We would be happy to help you in that case.

closed 2025-04-30 by mantriyogesh