espressif/esp-hosted #387: dose esp-hosted-ng  support ap mode

lipengvita, 2024-05-16:
dose esp-hosted-ng  support ap mode  ？

lipengvita, 2024-05-16:
If support， how to config   it

mantriyogesh, 2024-05-16:
For softAP functionality, please evaluate: 
[2024.05.16_softap_patch.tgz](https://github.com/espressif/esp-hosted/files/15331423/2024.05.16_softap_patch.tgz)

mantriyogesh, 2024-05-16:
It is same code path provided in other issue you referenced.
Please check the readme.txt in the tarball above.

lipengvita, 2024-05-16:
@mantriyogesh  Can I build the esp32c3 slave bin myself

mantriyogesh, 2024-05-16:
[softAP_patch_over_040b72285c6d09f18c5cf769e86d3cb206571283.patch.gz](https://mattermost.espressif.cn:6689/files/4gq3ywag4pd4xrgtth49wte4te/public?h=LkZ_jHht344G2kF3j6MJ3F6yMXT0VbHwE58EYMq2Spg)

This is patch to be used over ESP-Hosted 040b72285c6d09f18c5cf769e86d3cb206571283 commit.
You need to :

1. clone master
2. Clean repo and submodules so that no local or chanegd files
3. Apply patch
4. cd esp_hosted_ng/esp/esp_driver
5. *Comment* https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_ng/esp/esp_driver/CMakeLists.txt#L6-L20
6. `cmake .` 
7. cd esp-idf
8. `. ./export.sh`
9. `cd ../network_adapter/`
10. `idf.py set-target esp32c3`
11. `idf.py flash monitor`

lipengvita, 2024-05-16:
@mantriyogesh    host side build  error    /data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c: In function 'process_mgmt_tx_status':
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:661:9: error: variable 'status' has initializer but incomplete type
  struct cfg80211_tx_status status = {
         ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:662:3: error: unknown field 'buf' specified in initializer
   .buf = data,
   ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:662:10: warning: excess elements in struct initializer
   .buf = data,
          ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:662:10: note: (near initialization for 'status')
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:663:3: error: unknown field 'len' specified in initializer
   .len = len,
   ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:663:10: warning: excess elements in struct initializer
   .len = len,
          ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:663:10: note: (near initialization for 'status')
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:664:3: error: unknown field 'ack' specified in initializer
   .ack = ack,
   ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:664:10: warning: excess elements in struct initializer
   .ack = ack,
          ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:664:10: note: (near initialization for 'status')
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:661:28: error: storage size of 'status' isn't known
  struct cfg80211_tx_status status = {
                            ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:672:2: error: implicit declaration of function 'cfg80211_mgmt_tx_status_ext' [-Werror=implicit-function-declaration]
  cfg80211_mgmt_tx_status_ext(&priv->wdev, &status, GFP_ATOMIC);
  ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:661:28: warning: unused variable 'status' [-Wunused-variable]
  struct cfg80211_tx_status status = {
                            ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c: In function 'process_ap_mgmt_rx':
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:677:9: error: variable 'info' has initializer but incomplete type
  struct cfg80211_rx_info info = {0};
         ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:677:34: warning: excess elements in struct initializer
  struct cfg80211_rx_info info = {0};
                                  ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:677:34: note: (near initialization for 'info')
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:677:26: error: storage size of 'info' isn't known
  struct cfg80211_rx_info info = {0};
                          ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:689:2: error: implicit declaration of function 'cfg80211_rx_mgmt_ext' [-Werror=implicit-function-declaration]
  cfg80211_rx_mgmt_ext(&priv->wdev, &info);
  ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:677:26: warning: unused variable 'info' [-Wunused-variable]
  struct cfg80211_rx_info info = {0};
                          ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c: In function 'cmd_add_station':
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:1913:9: error: 'struct station_parameters' has no member named 'link_sta_params'
  if (sta->link_sta_params.supported_rates_len && sta->link_sta_params.supported_rates) {
         ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:1913:53: error: 'struct station_parameters' has no member named 'link_sta_params'
  if (sta->link_sta_params.supported_rates_len && sta->link_sta_params.supported_rates) {
                                                     ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:1916:56: error: 'struct station_parameters' has no member named 'link_sta_params'
   memcpy(&cmd_config->sta_param.supported_rates[2], sta->link_sta_params.supported_rates, 10);
                                                        ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:1919:9: error: 'struct station_parameters' has no member named 'link_sta_params'
  if (sta->link_sta_params.ht_capa) {
         ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:1922:48: error: 'struct station_parameters' has no member named 'link_sta_params'
   memcpy(&cmd_config->sta_param.ht_caps[2], sta->link_sta_params.ht_capa, 26);
                                                ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:1924:9: error: 'struct station_parameters' has no member named 'link_sta_params'
  if (sta->link_sta_params.vht_capa) {
         ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:1927:49: error: 'struct station_parameters' has no member named 'link_sta_params'
   memcpy(&cmd_config->sta_param.vht_caps[2], sta->link_sta_params.vht_capa, 12);
                                                 ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:1929:9: error: 'struct station_parameters' has no member named 'link_sta_params'
  if (sta->link_sta_params.he_capa) {
         ^
/data/test_bin/git_softap/esp-hosted/esp_hosted_ng/host/esp_cmd.c:1933:48: error: 'struct station_parameters' has no member named 'link_sta_params'
   memcpy(&cmd_config->sta_param.he_caps[3], sta->link_sta_params.he_capa, 24);
                                                ^
cc1: some warnings being treated as errors

mantriyogesh, 2024-05-16:
Output of `uname -a`

lipengvita, 2024-05-16:
Linux chint 5.4.61 #13 SMP PREEMPT Thu May 16 13:05:20 CST 2024 armv7l GNU/Linux

lipengvita, 2024-05-16:
@mantriyogesh

mantriyogesh, 2024-05-17:
Hello @lipengvita ,

@kapilkedawat is working on it and it may take some time to resolve. But it essentially is just kernel version difference in Linux host used. Clearly, we would need to support older kernels as well.

kapilkedawat, 2024-05-17:
@lipengvita sorry for the trouble, please try out attached patch.
[0001-esp_hosted_ng-Fixed-some-compilation-issues.patch](https://github.com/espressif/esp-hosted/files/15344381/0001-esp_hosted_ng-Fixed-some-compilation-issues.patch)

Since this is still work in progress, feel free to report any issues that you encounter.

lipengvita, 2024-05-17:
@kapilkedawat   compile esp side error 
[86/86] cd /data/esp_ap/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/bootloader/esp-idf/esptool_p...ootloader 0x0 /data/esp_ap/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/bootlootloader binary size 0x5030 bytes. 0x2fd0 bytes (37%) free.
[900/901] Generating binary image from built executableFAILED: .bin_timestamp /data/esp_ap/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/.bin_timestamp
cd /data/esp_ap/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build && /opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/components/esptool_py/esptool/esptool.py --chip esp32c3 elf2image --f_freq 80m --flash_size 4MB --elf-sha256-offset 0xb0 --min-rev 3 --min-rev-full 3 --max-rev-full 99 -o /data/esp_ap/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/network_adapter.bin /data/esp_ap/eng/esp/esp_driver/network_adapter/build/network_adapter.elf && /opt/esp/tools/cmake/3.24.0/bin/cmake -E echo "Generated /data/esp_ap/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/network_adapter.s/cmake/3.24.0/bin/cmake -E md5sum /data/esp_ap/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/network_adapter.bin > /data/esp_ap/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/.bin
esptool.py v4.7.dev1
Creating esp32c3 image...
Merged 8 ELF sections

A fatal error occurred: Segment loaded at 0x42087664 lands in same 64KB flash mapping as segment loaded at 0x4208739e. Can't generate binary. Suggest changing linker script or ELF to merge sections.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /data/esp_ap/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/log/idf_py_stderr_output_27623 and /data/esp_ap/esp-hosted/esp_hosted_ng/k_adapter/build/log/idf_py_stdout_output_27623

kapilkedawat, 2024-05-17:
Hi @lipengvita, I tried to compile this for esp32c3 and didn't face any issues. Please try with following steps.
1. Clone a frash hosted repo.
2. git checkout d224c4b13bfea15d05237673e9e56c6c814095f3
3. apply patches shared.
4. git submodule update -f
5. go to `esp_hosted/esp_hosted_ng/esp/esp_driver` and run `bash setup.sh`, please see the output whether it's getting succeed or not
6. go to esp-idf and run `. ./export.sh`
7. now go to `esp_hosted/esp_hosted_ng/esp/esp_driver/network_adapter` folder and run `idf.py set-target esp32c3`
8. idf.py build

Let us know if issue persist after that.

lipengvita, 2024-05-20:
Hi  @kapilkedawat 
1  I run git clone https://github.com/espressif/esp-hosted.git
 2  git checkout d224c4b13bfea15d05237673e9e56c6c814095f3  
  fatal: reference is not a tree: d224c4b13bfea15d05237673e9e56c6c814095f3
 maybe  d224c4b13bfea15d05237673e9e56c6c814095f3  not a valid branch

lipengvita, 2024-05-21:
HI @kapilkedawat  I think maybe the wifi lib have some problem  ，Please give me the lib you used

lipengvita, 2024-05-21:
HI @kapilkedawat  not the patch provide me the lib.tgz support ap mode

kapilkedawat, 2024-05-22:
Hi @lipengvita, sorry for the delay, we will provide another binaries shortly.

Shreyas0-7, 2024-05-22:
Hey @lipengvita 
Please follow these steps to try AP mode on chip esp32c3
- run git clone https://github.com/espressif/esp-hosted.git
- `cd /esp-hosted/esp_hosted_ng/esp/esp_driver/`
- If your system has cmake run `cmake .` else `chmod u+x setup.sh`, `./setup.sh` This step will download submodule esp-idf
- `cd network_adapter`
- Download patch [ap_support.zip](https://github.com/espressif/esp-hosted/files/15392567/ap_support.zip) unzip it.
- apply using `git am --binary ap_support.patch`
- run `cmake .` else `chmod u+x setup.sh`, `./setup.sh` for coping the libs
- go to esp-idf and run `. ./export.sh`
-  again cd into folder `esp_hosted/esp_hosted_ng/esp/esp_driver/network_adapter` and run `idf.py set-target esp32c3`
- `idf.py build`

lipengvita, 2024-05-22:
@Shreyas0-7  Hi  cmake .   get fallow error
Submodule path 'components/esp_phy/lib': checked out '2d319a382336cf0522ea4bb5a3fbd6701a8633c6'
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: unpack-objects failed
Fetched in submodule path 'components/esp_wifi/lib', but it did not contain 0a25c10a587df080550995e18fc10c996abc7ecf. Direct fetching of that commit failed.

lipengvita, 2024-05-22:
@Shreyas0-7  Dose new  esp-hosted must be compile under idf 5.1.3 ?   I've been using idf5.1.1 to compile it before

Shreyas0-7, 2024-05-22:
@lipengvita Yes, all packages required for that will be installed by cmake or setup.sh command step

lipengvita, 2024-05-23:
@Shreyas0-7  For testing , I modify your code  ap_main.c
   while( 1)
 {
    sleep(3);
    send_bootup_event_to_host(capa);
    ESP_LOGI(TAG,"Initial set up done");
}
 and I find  after sending it a few times
 the system maybe hang, system not print  "Initial set up done" again

lipengvita, 2024-05-23:
@Shreyas0-7 @kapilkedawat @mantriyogesh  Hi I use  the patch not work well
linux side log
   20.239778] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [1]
#@chint [   32.203691] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[   32.211477] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[   32.218654] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[   32.227472] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[   32.234676] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[   32.241838] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[   32.249002] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[   32.256461] esp32_spi: print_reset_reason: POWERON_RESET
[   32.262454] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[   32.270162] esp32_spi: esp_reg_notifier: Driver init is ongoing
[   37.448351] esp32_spi: wait_and_decode_cmd_resp: Command[0x3] timed out
[   37.455808] esp32_spi: cmd_get_mac: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   37.465557] esp32_spi: esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[   37.474641] esp32_spi: process_event_esp_bootup: network iterface init failed
[   44.145115] random: crng init done
[   44.148945] random: 7 urandom warning(s) missed due to ratelimiting
[   70.776632] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[   70.828598] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[   70.835758] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[   70.844729] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[   70.851898] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[   70.859065] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[   70.866202] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[   70.873669] esp32_spi: print_reset_reason: POWERON_RESET
[   70.879651] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[   70.887317] esp32_spi: esp_reg_notifier: Driver init is ongoing
[   76.008359] esp32_spi: wait_and_decode_cmd_resp: Command[0x3] timed out
[   76.015801] esp32_spi: cmd_get_mac: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   76.025514] esp32_spi: esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[   76.034332] esp32_spi: process_event_esp_bootup: network iterface init failed
esp side log

I (7696) FW_SPI: send_bootup_event_to_host 189

I (7697) FW_SPI: send_bootup_event_to_host 194

I (7701) FW_SPI: send_bootup_event_to_host 205

I (7706) FW_SPI: send_bootup_event_to_host 237

I (7712) FW_SPI: send_bootup_event_to_host 253

I (7717) FW_SPI: send_bootup_event_to_host 256

I (7722) FW_SPI: send_bootup_event_to_host 262

I (7728) FW_MAIN: Initial set up done
I (7732) main_task: Returned from app_main()
I (7911) FW_MAIN: INIT Interface command

I (7912) wifi:mode : sta (f4:12:fa:39:c4:04)
I (7913) wifi:enable tsf
I (7913) FW_CMD: Wifi Sta mode set

I (7991) FW_MAIN: Get MAC command

lipengvita, 2024-05-24:
@Shreyas0-7  At first, my host test couldn't receive the reset message, so I wanted to let the esp test loop send a message to see where the question was going on, and found that the esp test stopped sending a few packets of data, and then I tuned through the reset message. I found that the host test couldn't receive the get mac message, and I guessed if the ESP test couldn't send a few packets of messages。
At first, my host test couldn't receive the reset message  I modify the set code the host side can get reset message
the code  I  add  the code 

        /* Initialize SPI slave interface */
        ret=spi_slave_initialize(ESP_SPI_CONTROLLER, &buscfg, &slvcfg, DMA_CHAN);
        assert(ret==ESP_OK);

        +gpio_set_drive_capability(GPIO_SCLK, GPIO_DRIVE_CAP_3);
        +gpio_set_drive_capability(GPIO_MISO, GPIO_DRIVE_CAP_3);
        +gpio_set_pull_mode(GPIO_MISO, GPIO_PULLDOWN_ONLY);

mantriyogesh, 2024-05-24:
I assume you are using jumper cables, longer than 10 cm, with esp32-c3?

lipengvita, 2024-05-24:
@mantriyogesh  Yes, as you say ,I use iumper cables,  longer the 10CM  and esp32-c3  .
     Does it have any effect, my sta mode worked fine in the previous version but not worked in the ap patch version

mantriyogesh, 2024-05-24:
Appreciate your debugging efforts.

Longer wires delay the communication, so need to power more. Drive strength might not be needed if you use smaller wires or PCB. Longer wires also would little bit smaller throughout in long run, as GPIO transmission delays are accounted.

> my sta mode worked fine in the previous version but not worked in the ap patch version

@Shreyas0-7 is checking it. Can you please provide as much logs possible, from whole procedure? It would fasten the resolution process.

Would it be possible to smaller the wires as much as possible say 5-6 cm? Please keep all wires the same length. Prefer whole wires, but if not better to shorten and solder.

Also sending bootup event more than once would not be correct. I think before continuing ahead,

1. Try the back to back  ESP to reset with 6-7 sec delay in between (let the drive strength be high, for now) using rst button and assess the stability 
2. Check your [raw throughput test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md) in either direction. If it is very low, you might still need some efforts on the transport
3. Last but not the least, check the [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md) for some common considerations

lipengvita, 2024-05-24:
@mantriyogesh Thank you very much for your answer, smaller the wires  cannot be realized for the time being, my development board spi interface is just in the middle of the board, and we will directly connect  without the jumper wire later. Now what I'm doing is just preliminary work to verify the function . I want to sure the ap mode  and sta mode work fine

mantriyogesh, 2024-05-24:
Okay.

I think you can stick with smaller frequencies like 5-10MHz by the time bing.

lipengvita, 2024-05-24:
@mantriyogesh  @Shreyas0-7  Hi I change the frequencies to 5MHz and 10Mhz ，I get the same result but litte   ，host side wait   for   packets  timeout
host side
[  168.013768] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[  168.056705] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[  168.063864] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[  168.072800] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[  168.079998] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[  168.087184] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[  168.094321] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[  168.101845] esp32_spi: print_reset_reason: POWERON_RESET
[  168.107860] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[  168.115529] esp32_spi: esp_reg_notifier: Driver init is ongoing
[  173.206431] esp32_spi: wait_and_decode_cmd_resp: Command[0xF] timed out
[  173.213872] esp32_spi: cmd_get_tx_power: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[  173.436425] esp32_spi: init_bt: ESP Bluetooth init
[  173.442464] esp32_spi: print_capabilities: Capabilities: 0xe8. Features supported are:
[  173.451470] esp32_spi: print_capabilities:    * WLAN on SPI
[  173.457714] esp32_spi: print_capabilities:    * BT/BLE
[  173.463455] esp32_spi: print_capabilities:      - HCI over SPI
[  173.470009] esp32_spi: print_capabilities:      - BLE only
[  183.050897] esp32_spi: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  169.254.186.131
[  188.086424] esp32_spi: wait_and_decode_cmd_resp: Command[0xD] timed out
[  188.093863] esp32_spi: cmd_set_ip_address: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
esp side
I (7731) FW_MAIN: Initial set up done
I (7735) main_task: Returned from app_main()
I (7872) FW_MAIN: INIT Interface command
I (7873) wifi:mode : sta (f4:12:fa:39:c4:04)
I (7874) wifi:enable tsf
I (7874) FW_CMD: Wifi Sta mode set
I (7912) FW_MAIN: Get MAC command
I (7952) FW_MAIN: Tx power command
I (22813) FW_MAIN: Set IP Address

Shreyas0-7, 2024-05-24:
Hey @lipengvita if possible can you please attach host side logs

Shreyas0-7, 2024-05-24:
Hey @lipengvita can you please proceed and try using AP mode?

lipengvita, 2024-05-24:
Hi @Shreyas0-7   I use hostapd get follow message   
host side
api-server    api-server-2  api-server_1  esp32_spi.ko  test.sh
#@chint insmod /mnt/UDISK/esp32_spi.ko clockspeed=10
[   27.749044] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]
#@chint [   38.523008] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[   38.530799] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[   38.537936] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[   38.546797] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[   38.553978] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[   38.561161] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[   38.568316] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[   38.575741] esp32_spi: print_reset_reason: POWERON_RESET
[   38.581732] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[   38.589428] esp32_spi: esp_reg_notifier: Driver init is ongoing
[   43.688327] esp32_spi: wait_and_decode_cmd_resp: Command[0xF] timed out
[   43.695785] esp32_spi: cmd_get_tx_power: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   43.720738] random: crng init done
[   43.724577] random: 7 urandom warning(s) missed due to ratelimiting
[   43.918322] esp32_spi: init_bt: ESP Bluetooth init
[   43.924388] esp32_spi: print_capabilities: Capabilities: 0xe8. Features supported are:
[   43.933441] esp32_spi: print_capabilities:    * WLAN on SPI
[   43.939705] esp32_spi: print_capabilities:    * BT/BLE
[   43.945436] esp32_spi: print_capabilities:      - HCI over SPI
[   43.951944] esp32_spi: print_capabilities:      - BLE only

#@chint
#@chint ifconfig
br-lan    Link encap:Ethernet  HWaddr F6:31:AF:1B:57:E5
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr F6:31:AF:1B:57:E5
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:39

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:336 errors:0 dropped:0 overruns:0 frame:0
          TX packets:336 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:20160 (19.6 KiB)  TX bytes:20160 (19.6 KiB)

wlan0     Link encap:Ethernet  HWaddr F4:12:FA:39:C4:04
          inet6 addr: fe80::b0c7:8d26:be3d:9a71/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:10 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

#@chint hosa[   55.100762] esp32_spi: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  169.254.186.131

-sh: hosa: command not found
#@chint hostapd /data/cfg/hostapd.conf &
[1] 1477
#@chint [   72.008319] esp32_spi: wait_and_decode_cmd_resp: Command[0xF] timed out
[   72.015755] esp32_spi: cmd_get_tx_power: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   77.048318] esp32_spi: wait_and_decode_cmd_resp: Command[0xF] timed out
[   77.055755] esp32_spi: cmd_get_tx_power: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[... 79 lines cut ...]

I (7865) FW_MAIN: Tx power command

I (24306) FW_MAIN: Set IP Address

I (36166) FW_MAIN: Tx power command

I (46267) FW_MAIN: Tx power command

I (56367) FW_MAIN: Tx power command

I (66428) FW_MAIN: Set multicast mac address list

I (66468) FW_MAIN: Set MODE command

E (66468) FW_CMD: Setting mode=2

I (66469) wifi:flush txq
I (66469) wifi:stop sw txq
I (66470) wifi:lmac stop hw txq
I (66474) wifi:mode : softAP (f4:12:fa:39:c4:05)
I (66478) wifi:Total power save buffer number: 20
I (66481) wifi:Init max length of beacon: 752/752
I (66486) wifi:Init max length of beacon: 752/752
I (66491) FW_CMD: Unregistered event: 3

Shreyas0-7, 2024-05-27:
Hey @lipengvita [RawTp](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md)
Can you please your raw throughput performance using this guide. This will check your hardware connection is fine.

lipengvita, 2024-05-28:
Hi  @Shreyas0-7  Thank you very much , To avoid the hardware connection reason . I will  test it later . about two week later . I can test esp32c3 as wifi modules not use wire connect but pin to pin

lipengvita, 2024-06-24:
@Shreyas0-7  Now I can use pin to pin to connect linux and esp32 modules  ，But  I get the same error . 
 I  use this patch https://github.com/sergey-suloev/esp-hosted/commit/e51d5a0e41878c26a51c01b91dc73734d952c1b6 Now the esp32 modules can play as ap mode , But when the client connect the ap  the ap side get some error 
n0: STA 70:1a:b8:91:ac:1c IEEE 802.11: No WPA/RSN IE in association request
wlan0: STA 70:1a:b8:91:ac:1c IEEE 802.11: authenticated
wlan0: STA 70:1a:b8:91:ac:1c IEEE 802.11: No WPA/RSN IE in association request
wlan0: STA 70:1a:b8:91:ac:1c IEEE 802.11: authenticated
wlan0: STA 70:1a:b8:91:ac:1c IEEE 802.11: No WPA/RSN IE in association request
wlan0: STA 70:1a:b8:91:ac:1c IEEE 802.11: authenticated
wlan0: STA 70:1a:b8:91:ac:1c IEEE 802.11: No WPA/RSN IE in association request
 Could  you give me some advice

Shreyas0-7, 2024-06-24:
Hey @lipengvita it seems like station which is trying to connect isn't providing WPA/RSN ie which is required by hostapd for connection. To confirm this can you please capture the sniffer packet if possible.
Also please share hostapd configuration which you are using for AP

lipengvita, 2024-06-24:
@Shreyas0-7  This is my hostpad config，I  don't have a capture tools

interface=wlan0
ctrl_interface=/var/run/hostapd
ssid=110114119
channel=6
wpa=1
wpa_passphrase=1234567890
beacon_int=100
wpa_key_mgmt=WPA-PSK

lipengvita, 2024-06-24:
@  hostap config  
interface=wlan0
ctrl_interface=/var/run/hostapd
ssid=110114119
channel=6
wpa=2
wpa_passphrase=1234567890
beacon_int=100
wpa_key_mgmt=WPA-PSK

esp32 side log

I (990438) FW_CMD: process_mgmt_tx: broadcast address, sending response immediately

I (990488) FW_MAIN: Add key request

I (990488) FW_CMD: set_key_internal:1600

I (990488) FW_CMD: Setting GTK [1]

I (990528) FW_MAIN: Set default key request

E (1005916) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

E (1005963) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as 0

E (1006973) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

E (1007038) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as 0

E (1008048) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

E (1008113) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as 0

E (1009132) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

E (1009173) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as 0

E (1014556) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

E (1014773) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as 0

E (1015790) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

E (1015845) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as 0

E (1016420) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as c0

I (1016469) FW_MAIN: AP station command

I (1016469) FW_CMD: process_ap_station:got station add command

I (1016469) mac: 0x3fcbaf18   70 1a b8 91 ac 1c                                 |p.....|
aid=0
I (1016478) supported_rates: 0x3fcbaf38   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (1016489) ht_rates: 0x3fcbaf44   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (1016499) ht_rates: 0x3fcbaf54   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (1016509) vht_rates: 0x3fcbaf60   00 00 00 00 00 00 00 00  00 00 00 00 00 00        |..............|
I (1016519) he_rates: 0x3fcbaf70   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (1016529) he_rates: 0x3fcbaf80   00 00 00 00 00 00 00 00  00 00 00                 |...........|
I (1016538) FW_CMD: add_station_node_ap: not station add cmd, handle later

I (1016609) FW_MAIN: AP station command

I (1016609) FW_CMD: process_ap_station:got station add command

I (1016610) mac: 0x3fcbaf18   70 1a b8 91 ac 1c                                 |p.....|
aid=0
I (1016619) supported_rates: 0x3fcbaf38   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (1016629) ht_rates: 0x3fcbaf44   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (1016639) ht_rates: 0x3fcbaf54   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (1016649) vht_rates: 0x3fcbaf60   00 00 00 00 00 00 00 00  00 00 00 00 00 00        |..............|
I (1016659) he_rates: 0x3fcbaf70   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (1016669) he_rates: 0x3fcbaf80   00 00 00 00 00 00 00 00  00 00 00                 |...........|
I (1016678) FW_CMD: add_station_node_ap: not station add cmd, handle later

I (1016729) FW_MAIN: disconnect request

I (1016729) FW_CMD: Disconnect request: reason [2], interface=1

@Shreyas0-7

Shreyas0-7, 2024-06-24:
@lipengvita 
We might need capture to debug this failure furthermore, also are you using wpa_supllicant as a station?
Here are the steps to use Wireshark. First, put your laptop in Monitor mode and then sniff the WLAN interface.
Here are some resources that might help you to capture 80211 packets.
If you are using a Windows machine, try: https://www.youtube.com/watch?v=As9BUoOMk_A
If you are using an iOS machine, try: https://www.macupdate.com/app/mac/56809/airtool/old-versions
If you are using a Linux machine, try:

ifconfig wlp0s20f3 down
iwconfig wlp0s20f3 mode monitor
ifconfig wlp0s20f3 up
iwconfig wlp0s20f3 channel 6
wireshark

Here wlp0s20f3 can be any other NIC which is attached to your laptop 
Please note that a lot of WiFi cards (NIC) don't support putting them in monitor mode. So it is vital to check the compatibility of your laptop's NIC.
Hope that helps.

jborl, 2024-09-19:
Hi there,

any progress related to this topic? I'm interested in an AP/Station mode solution (configurable at run-time). I tried the patches from May listed above but without success, yet. I could help with more testing if required.
Cheers juergen

mantriyogesh, 2024-09-19:
@kapilkedawat

kapilkedawat, 2024-09-19:
@jborl AP mode is available in latest hosted-ng code. Please enable https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/Makefile#L16 to enable AP support. Feel free to report if you face any issues.