espressif/esp-hosted #389: It takes a long time to obtain an IP address through multiple settings

mrhan1992, 2024-05-20:
The software version we use on the ESP32-C6 module is esp_hosted_v5.1.1_commit_d07a9242f49. When tested in a Linux environment, it takes a long time from test_station_mode_connect to obtaining the router IP address. Generally, the time to obtain the IP is 2-3 seconds. But when setting multiple times, it will take 8-12 seconds! The test is called as follows:
```
    test_station_mode_connect(&m_config);
    system("dhclient ethsta0 -r -q");
    system("dhclient ethsta0 -v -nw -q");
```
How should we solve this problem?

mantriyogesh, 2024-05-20:
We do not recognise the revision `esp_hosted_v5.1.1_`
But can under stand the commit `d07a9242f49`  - Oct 7

Issue is fixed in 54fb6dcddda49198d179e7d2985e6a9ae99d40ff
Test with latest master.

Use exact same code at esp and host.

mrhan1992, 2024-05-24:
Hi yogesh
Thanks to you reply , But the new latsed version is still have this present when I was test, Do you have some suggest to me ?

mantriyogesh, 2024-05-24:
Please test both sides with 2eb1fff5b7b18af20087ace3a35ba596172acdd5 commit.
If you face issues, please get us the logs both side, where you face issues

jinnie11, 2024-05-27:
[driver-log.txt](https://github.com/espressif/esp-hosted/files/15454338/driver-log.txt)
[[COM3] 2024-05-27 16-23-35.668.log](https://github.com/espressif/esp-hosted/files/15454341/COM3.2024-05-27.16-23-35.668.log)
@mantriyogesh

mantriyogesh, 2024-05-28:
hello @mrhan1992 ,

The reason code in disconnect is 8.
basically, WIFI_REASON_ASSOC_LEAVE.

Could you please test this software with some other AP or hotspot?
Try to reduce the interference as much as possible from other AP/Bluetooth/Microwave oven.

If you still face this issue despite, Please enable the debug logs by running
```
idf.py menuconfig
```
[image]
```
and then idf.py flash monitor
```

Also please collect the sniffer logs starting before connect (to capture EAPOL M1-M4 messages)

mantriyogesh, 2024-05-28:
Also, please include both ESP and host logs from startup.

ESP logs
- Include debug logs as mentioned above
- From flashing
Host log
- dmesg from bootup
Sniffer logs
- Run sniffer on your channel
- Start sniffer before triggering connect

mantriyogesh, 2024-05-28:
Can you please check if your code triggers disconnect from some leg?
What happens if you comment it.

mantriyogesh, 2024-05-29:
Can you please send us the complete logs?
Please include the commands run and their output.

Also, any simple script to reproduce the issue on latest master would definitely help if we can reproduce at our end.