espressif/esp-hosted #390: about BT

lipengvita, 2024-05-21:
[60582.356568] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[60582.373269] esp32_spi: prepare_command_request: command queue init is not done yet
[60582.381851] esp32_spi: cmd_get_tx_power: Failed to get command node
[60582.537901] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[60582.545081] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[60582.554430] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[60582.561676] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[60582.569106] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[60582.576261] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[60582.583968] esp32_spi: print_reset_reason: POWERON_RESET
[60582.590061] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[60582.597987] esp32_spi: esp_reg_notifier: Driver init is ongoing
[60583.137189] esp32_spi: init_bt: ESP Bluetooth init
[60583.143340] esp32_spi: print_capabilities: Capabilities: 0xe8. Features supported are:
[60583.153338] esp32_spi: print_capabilities:    * WLAN on SPI
[60583.159824] esp32_spi: print_capabilities:    * BT/BLE
[60583.165525] esp32_spi: print_capabilities:      - HCI over SPI
[60583.172375] esp32_spi: print_capabilities:      - BLE only
Dose this mead bt  ok？
hciconfig hci0 up 
hcitool lescan 
Set scan parameters failed: Input/output error

mantriyogesh, 2024-05-21:
The hciconfig is deprecated since long and no longer maintained:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=b1eb2c4cd057624312e0412f6c4be000f7fc3617

Deprecated tools by BlueZ:
https://archlinux.org/packages/extra/x86_64/bluez-deprecated-tools/#:~:text=http://www.bluez.org,list%20for%20bluez-deprecated-tools

Can you please get this scenario tested using bluetoothctl?

lipengvita, 2024-05-21:
thks  I get it

closed 2024-05-21 by lipengvita