espressif/esp-hosted #391: build esp-hosted-ng error

XZ-Niupitang, 2024-05-23:
[image]
I have installed the esp-idf V5.2.1 compilation environment locally and followed the installation steps step by step. The lib copy of setup.sh has also been completed, but this error is still reported. What is the reason?

mantriyogesh, 2024-05-23:
Have you followed https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/README.md#update-build-firmware-for-esp-hosted-ng-solution  for firmware flashing on master branch latest commit?

If so, it should work. Can you please show the output log if it fails?

XZ-Niupitang, 2024-05-23:
The local esp-idf compilation environment is installed by itself and get_idf (export.sh) is declared in /etc/profile. There is no problem compiling the routines in the example. I did not use git clone to download the source code directly, but just After downloading the code-zip of esp-hosted to local decompression and then pressing setup to execute it, the wifi library reported at the beginning is undefined and does not appear after replacing the files in esp-idf-v5.2.1/components/esp_wifi/lib/. Instead, this new question
[image]
Every time I compile, I get stuck at the position in the picture.

XZ-Niupitang, 2024-05-23:
> 您是否遵循[了 https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/README.md#update-build-firmware-for-esp-hosted-ng-solution](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/README.md#update-build-firmware-for-esp-hosted-ng-solution) 在主分支最新提交时刷新固件？
> 
> 如果是这样，它应该有效。如果失败，您能否显示输出日志？
The local esp-idf compilation environment is installed by itself and get_idf (export.sh) is declared in /etc/profile. There is no problem compiling the routines in the example. I did not use git clone to download the source code directly, but just After downloading the code-zip of esp-hosted to local decompression and then pressing setup to execute it, the wifi library reported at the beginning is undefined and does not appear after replacing the files in esp-idf-v5.2.1/components/esp_wifi/lib/. Instead, this new question
[image]
Every time I compile, I get stuck at the position in the picture.

mantriyogesh, 2024-05-23:
Currently, for NG, you cannot use your own IDF. the IDF would be used from the commit pushed as part of esp_hosted_ng/esp/esp_driver/esp-idf

But you do not need to do it manually, it is already automated with instructions.
If you do `get_idf()` and it is different from the one in  `esp_hosted_ng/esp/esp_driver/esp-idf`, it is expected to fail.

So please run one by instruction carefully from, https://github.com/espressif/esp-hosted/issues/391#issuecomment-2126696076
Please attach **textual** log of **all** steps.

XZ-Niupitang, 2024-05-24:
> Currently, for NG, you cannot use your own IDF. the IDF would be used from the commit pushed as part of esp_hosted_ng/esp/esp_driver/esp-idf
> 
> But you do not need to do it manually, it is already automated with instructions. If you do `get_idf()` and it is different from the one in `esp_hosted_ng/esp/esp_driver/esp-idf`, it is expected to fail.
> 
> So please run one by instruction carefully from, [#391 (comment)](https://github.com/espressif/esp-hosted/issues/391#issuecomment-2126696076) Please attach **textual** log of **all** steps.

Successfully compiled and downloaded to the board, thank you for your support

closed 2024-05-24 by XZ-Niupitang