espressif/esp-hosted #392: WiFi transmission rate

linchanghe123, 2024-05-24:
Hi mantriyogesh,

Using ng mode, When connecting to WiFi with good WiFi signals, the tested UDP transmission rate is only about 100KB/S. The TCP transmission rate is approximately 300KB/S.

 Is there any way to increase the transmission rate? Attempted to increase the SPI bus frequency from 10M to 40M, but it seems to have little effect.

TCP：
[image]

UDP:
[image]

mantriyogesh, 2024-05-24:
This is excessively low.

Worth to separate Wi-Fi and transport speeds.

To isolate and test your transport, test the [raw transport throughput](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md) in tx and rx direction.

closed 2024-05-27 by linchanghe123

mantriyogesh, 2024-05-27:
I see the issue closed, but not sure if it was working for you.

Have you managed to check your raw transport throughput?
Do let us know if anything required/ unclear.