espressif/esp-hosted #394: Error using esp-hosted-ng on Raspberry Pi 4B

XZ-Niupitang, 2024-05-24:
[  181.559569] esp32_spi: spi_dev_init: Failed to obtain GPIO for Handshake pin, err:-517

There is no problem compiling and installing the Raspberry Pi host driver, but this problem is reported during loading. Do I need to modify the device tree of the Raspberry Pi? The problem is that this Handshake pin is not used by other drivers, which is strange.

mantriyogesh, 2024-05-24:
We provide the code for latest kernel version raspberry pi.
Are you using Fg driver?

For older kernels, please refer https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Troubleshoot.md#42-reset-pin-and-spi-gpio-numbering

XZ-Niupitang, 2024-05-24:
> We provide the code for latest kernel version raspberry pi. Are you using Fg driver?
> 
> For older kernels, please refer https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Troubleshoot.md#42-reset-pin-and-spi-gpio-numbering

My Raspberry Pi system information is

Linux version 6.6.20+rpt-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07)

XZ-Niupitang, 2024-05-24:
> We provide the code for latest kernel version raspberry pi. Are you using Fg driver?
> 
> For older kernels, please refer https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Troubleshoot.md#42-reset-pin-and-spi-gpio-numbering

#define HANDSHAKE_PIN           22
#define SPI_IRQ                 gpio_to_irq(HANDSHAKE_PIN)
#define SPI_DATA_READY_PIN      27
#define SPI_DATA_READY_IRQ      gpio_to_irq(SPI_DATA_READY_PIN)
#define SPI_BUF_SIZE            1600

This is the pin in esp_spi.h, do I need to modify the wiring?

mantriyogesh, 2024-05-24:
change all pin numbers as:

```
new_pin = old_pin +512
```

So 6 becomes 518., and so on.

This is change applicable for Raspberry Pi latest kernels (bookworm onwards)

mantriyogesh, 2024-05-24:
If you start the softap (instead of connecting in sta mode), and setup some routes, to auto-forward your NIC packets on Linux to softap & _vice-versa_, it is equivalent to router.

XZ-Niupitang, 2024-05-25:
> If you start the softap (instead of connecting in sta mode), and setup some routes, to auto-forward your NIC packets on Linux to softap & _vice-versa_, it is equivalent to router.

Thank you for your answer, but I still haven't succeeded, but I guess I didn't find the correct way to enable the use of hostapd and dns tools, but you mentioned enabling softap in your reply. What does this mean? Is there a detailed instruction manual?

XZ-Niupitang, 2024-05-25:
> If you start the softap (instead of connecting in sta mode), and setup some routes, to auto-forward your NIC packets on Linux to softap & _vice-versa_, it is equivalent to router.

[image]
This is the current situation and I'm confused as to what's going on.

mantriyogesh, 2024-05-25:
I am not sure what is inside create_ap.

Also, softap is supported with FG, using https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md#115-ctrl_cmd_t--wifi_start_softapctrl_cmd_t-req

But this is FG specific control path command, to be used with c demo in FG.

For NG, softap is not yet supported. If you still wish to evaluate, check issue 387

XZ-Niupitang, 2024-05-29:
> I am not sure what is inside create_ap.
>
> Also, softap is supported with FG, using [...]

OK, I have switched to FG, but the following log is reported when loading the driver. Is it caused by a pin error? How can I modify it?

`[   93.373370] esp32_spi: loading out-of-tree module taints kernel.
[   93.374281] esp32_spi: esp_reset: Triggering ESP reset.
[   93.374466] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[534] DataReady[539]
[   93.374469] esp32_spi: esp_spi_thread: esp spi thread created
[   93.374738] esp32_spi: spi_dev_init: ESP host driver claiming SPI bus [0],chip select [0] with init SPI Clock [10]
[   94.289139] esp32_spi: process_event: INIT event rcvd from ESP
[   94.289155] esp32_spi: process_init_event: EVENT: 2
[   94.289162] esp32_spi: process_init_event: EVENT: 1
[   94.289165] esp32_spi: adjust_spi_clock: ESP Reconfigure SPI CLK to 26 MHz
[   94.289169] esp32_spi: process_init_event: EVENT: 0
[   94.289173] esp32_spi: process_init_event: EVENT: 3
[   94.289794] esp32_spi: process_capabilities: ESP peripheral capabilities: 0xe8
[   94.496588] esp32_spi: process_init_event: esp boot-up event processed`

mantriyogesh, 2024-05-29:
This is the typical ESP boot-up event. Once you (re)load the driver, it would restart the ESP using [reset pin](https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L110) used.

> Is it caused by a pin error? How can I modify it?
I am still trying to understand, why is this error. This means the driver loaded and ESP (most likely ESP32-C6) first tx message is received at host.

XZ-Niupitang, 2024-05-29:
> This is the typical ESP boot-up event. Once you (re)load the driver, it would restart the ESP using [reset [...]

I thought it was the driver SPI pin setting error, but now there is a problem. After resetting espc6, I still can't load the fg driver normally. /dev/esps0 is missing in the system. After rechecking all the steps, I didn't find anything abnormal. Both host and esp started normally. 
ESP32c6 log is as follows：
[image]
Rpi 4B log is as follow:
[image]
It seems that the driver is stuck and does not run after the boot-up event is processed.

mantriyogesh, 2024-05-29:
The serial driver should be created as soon as you insert the kernel module. https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/host/linux/host_driver/esp32/esp_serial.c#L248-L296

It is function of kernel module and not transport / ESP reboot anymore.

I am not sure why this should happen:  last line, 'Waiting for driver to reconnect..'
Are you connecting ESP from Raspberry Pi USB port?

mantriyogesh, 2024-05-29:
What happens if you do https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Getting_started.md

mantriyogesh, 2024-05-29:
Can you try to change:
https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/host/linux/host_driver/esp32/esp_serial.c#L277

With:

```c
		devs[i].dev = device_create(cl, NULL, dev_num, NULL, "esps%d", i);
		if (IS_ERR(devs[i].dev) {
			printk(KERN_ERR "Err while creating serial drv: %d\n", PTR_ERR(devs[i].dev));
			return  PTR_ERR(devs[i].dev);
		}
```

XZ-Niupitang, 2024-05-29:
> Can you try to change:
>
> https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/host/linux/host_driver/esp32/esp_serial.c#L277
>
> With:
>
> ```c
> 		devs[i].dev [...]
[image]
[image]

After adding the above statement, this error is reported. What happened?

XZ-Niupitang, 2024-05-29:
> The serial driver should be created as soon as you insert the kernel module. [...]

No, esp32c6 USB is connected to another computer host，This is consistent with the previous link method using ng

XZ-Niupitang, 2024-05-29:
> Images in [#394 (comment)](https://github.com/espressif/esp-hosted/issues/394#issuecomment-2136598907) show, clearly there are some issues in your SPI communication.
> 
> 1. Please go through the [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md) to see frequently faced issues and their solution while porting or timing issues. Please see [SPI Timing issues - Tuning](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#spi-timing-issues----tuning) section for some easy timing tuning.
>    On similar notes: Use smaller (<10cm), all same length, good jumper cables. Use same SPI mode both ESP and host. Lower SPI frequency (<5MHz)
> 2. Enable checksum at both slave and host
> 3. Perform [transport throughput test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md) first, before jumping to demo app.
>    Unless the transport is working fine, anything above would simply always or inttermittently fail

[image]
[image]
It may be my misunderstanding. I did not find the following statement in the system log, but the driver should have been mounted successfully.

mantriyogesh, 2024-05-29:
Send **textual logs** :
 - ESP 
   - From start of flashing
 - Host
   - dmesg from bootup

The network interface is just added once slave boot-up msg received from slave. So expected to show that ethsta0 is listed.
You can see the mac address is still all zeros.
What I meant, Only one message, doesn't confirm that the complete driver is ported fine.

https://github.com/espressif/esp-hosted/issues/394#issuecomment-2136638407 still holds true.

XZ-Niupitang, 2024-05-29:
> Send **textual logs** :
> 
> * ESP
>   
>   * From start of flashing
> * Host
>   
>   * dmesg from bootup
> 
> The network interface is just added once slave boot-up msg received from slave. So expected to show that ethsta0 is listed. You can see the mac address is still all zeros. What I meant, Only one message, doesn't confirm that the complete driver is ported fine.
> 
> [#394 (comment)](https://github.com/espressif/esp-hosted/issues/394#issuecomment-2136638407) still holds true.

Well, you may be right. After successfully entering the CLI mode, I control the module through commands. I can see that the commands are responded to normally by esp32c6, but the CLI will directly exit with an error. There is still a problem with SPI communication...
[image]
[image]

XZ-Niupitang, 2024-05-29:
> Send **textual logs** :
>
> * ESP
>
>   * From start of flashing
> * Host
>
>   * dmesg from bootup
>
> The network interface is just added once slave boot-up msg received from slave. So expected to show that [...]

How can I check the cause of the problem? bus error is too brief

mantriyogesh, 2024-05-29:
Please ensure the transport layer is ported correctly, as improper porting may cause the application to fail.
If you jump to an application without correctly porting the transport, I am not sure how would you come to the solution.

Application just would fail & it cannot ( & need not) understand kernel module handling of spi.
If foundation (transport) is not correct, application will fail in expected ways.

I have stated possible ways to debug & correct the transport in https://github.com/espressif/esp-hosted/issues/394#issuecomment-2136638407

XZ-Niupitang, 2024-05-29:
> Please ensure the transport layer is ported correctly, as improper porting may cause the application to fail. If you jump to an application without correctly porting the transport, I am not sure how would you come to the solution.
> 
> Application just would fail & it cannot ( & need not) understand kernel module handling of spi. If foundation (transport) is not correct, application will fail in expected ways.
> 
> I have stated possible ways to debug & correct the transport in [#394 (comment)](https://github.com/espressif/esp-hosted/issues/394#issuecomment-2136638407)

To be precise, I didn't do any porting work. I still did the initialization and module control based on the Raspberry Pi 4B and the examples, so it's quite strange.

mantriyogesh, 2024-05-29:
Oh that's really strange, We test on Raspberry Pi 4B only, works perfect.

Have you followed the https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SPI_setup.md#12-raspberry-pi-software-setup?

Can you please reboot the Raspberry Pi and get the **textual and full** logs at esp and Raspberry Pi?
Can you please send the picture/photo of your setup, where spi lines are seen?

mantriyogesh, 2024-05-29:
> To be precise, I didn't do any porting work

You do not need any extra porting for Raspberry. Solution is ported already for Raspberry Pi.

XZ-Niupitang, 2024-05-29:
> Oh that's really strange, We test on Raspberry Pi 4B only, works perfect.
> 
> Have you followed the https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SPI_setup.md#12-raspberry-pi-software-setup?
> 
> Can you please reboot the Raspberry Pi and get the **textual and full** logs at esp and Raspberry Pi (dmesg from bootup, commands run and output) ? Can you please send the picture/photo of your setup, where spi lines are seen?
> 
> Also attach `/boot/config.txt` or `/boot/firmware/config.txt`?

Raspberry Pi dmesg out：
[image]

idf.py monitor :
[image]

boot/config.txt
[image]

mantriyogesh, 2024-05-29:
Please send us the picture of your setup, with jumper wires and SPI connections visible.

To limit freq, change 
https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L124

To say `5`

Instead, better, comment: 
https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L206

and change https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L34

to `5`

mantriyogesh, 2024-05-29:
Is the `/dev/esps0` created after running `bash rpi_init.sh spi`

```sh
$ ls /dev/esps0
```

XZ-Niupitang, 2024-05-29:
> Is the `/dev/esps0` created after running `bash rpi_init.sh spi`
> 
> ```shell
> $ ls /dev/esps0
> ```

I can find this interface,This is my hardware wiring diagram

[image]

XZ-Niupitang, 2024-05-29:
> Is the `/dev/esps0` created after running `bash rpi_init.sh spi`
> 
> ```shell
> $ ls /dev/esps0
> ```

There is another very strange question, as shown in the figure below: Why are there only two last characters left after setting ssid to esp32c6?

[image]
[image]

mantriyogesh, 2024-05-29:
> https://github.com/espressif/esp-hosted/issues/394#issuecomment-2136931783

Oh! the wires are very big for frequency of 26MHz.

From porting guide:

[image]

XZ-Niupitang, 2024-05-29:
> Oh. It looks like bug in python app, we will check this
> 
> Can you please try with [C-demo-app](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/c_demo.md)

Well, it is indeed a problem with the Python demo. C_Demo controls everything normally, but it is troublesome to manually configure dnsmasq after opening softAp. Is there any better way?

mantriyogesh, 2024-05-29:
`dnsmaq` is just an example, which we demo for DHCP server.
It really depends upon Linux distribution and varies greatly. Actually, it is out of scope for ESP-Hosted, as such.

You can use any alternative as deemed necessary.

XZ-Niupitang, 2024-05-29:
> `dnsmaq` is just an example, which we demo for DHCP server. It really depends upon Linux distribution and varies greatly. Actually, it is out of scope for ESP-Hosted, as such.
> 
> You can use any alternative as deemed necessary.

Thanks for your help buddy. The verification on Raspberry Pi has been completed. I guess there will be more problems in the subsequent transplantation. LOL

closed 2024-05-29 by XZ-Niupitang