espressif/esp-hosted #395: Can I change the MAC address of Bluetooth via esp-hosted-fg?

wmdscjhdpy, 2024-05-25:
I would like to be able to change the bluetooth mac address, I am using hosted-fg on linux. i searched for information and only found idf and how to set it on the AT firmware, does hosted-fg currently support configuring the bluetooth mac address?

mantriyogesh, 2024-05-25:
Right now, it is not added, but if you need it only at slave, you can do that by calling 
`esp_iface_mac_addr_set(ESP_MAC_BT, <mac_bytes>)`

We agree that this is something should be allowed from host.

If impatient, this could be done with small hacks. The underlying control api `wifi_set_mac()` handles the mac address change already, which allows mac address change with 'mode' and 'mac_addr'. You can re-use this api till we add generic api. the wifi_set_mac req from host terminates request at slave at:
https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_control.c#L1352-L1359

Here handle one mode mode, of bluetooth

user space might also need change at: 
https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/host/control_lib/src/ctrl_core.c#L1218-L1222

to include bluetooth case.

I am not sure if you would be able to follow this.
Anyway, If you want to wait for us, We would add this support, and let you know. It might take some time although.

wmdscjhdpy, 2024-05-25:
Thank you for your reply.
Let me follow your note: since esp-idf provides this interface on esp32c3 devices, but at the moment this interface is not exposed to the host operation API. So I can use the esp32c3 firmware code in the repository, add the interface to the call to the `esp_iface_mac_addr_set` function in the corresponding code area you hinted at and compile my own firmware. and add the new enumeration type to reuse the `wifi_set_mac` interface in the host program to accomplish my purpose. Is my understanding off?
I happen to have used the `wifi_set_mac` function in development and modified the MAC address of the wifi. 
It doesn't seem too hard if the firmware on the slave side just needs to add a function for bluetooth to modify the MAC address and then add a new enumeration.

mantriyogesh, 2024-05-25:
Yes, absolutely correct.
The change should be simple enough, if just reuse wifi_set_mac for now for bluetooth with new enum.

 If you want to do, can proceed else wait for us. I think I will change the ctrl API name as well from wifi_set_mac to just set_mac and wifi_get_mac to just get_mac, just to reuse the API for Bluetooth.

If you think you have done it, you can also add pr. We will add and correct if anything additional need to be done on top of your commit.

Rokachy, 2025-06-11:
Is this change included in newest SDK version?