espressif/esp-hosted #396: About porting esp_hosted_fg to other real-time operating systems

colazhb, 2024-05-25:
1. May I ask which branch is more convenient for porting or closer to a more complete version.
2. Attempted to port the main branch esp_hosted_fg directory to the rt thread system and found that many files were coupled with the freeftros library. For example, the use of QueueHandle_t in queues. Also, if you see a related file such as platform_wrapper. c, do you want to modify this file when porting to other systems? This file contains a lot of business related code and non porting related code.
3. Which files need attention or modification during migration?

mantriyogesh, 2024-05-25:
Actually host code in master is very much tied to FreeRTOS.

We took some efforts to make it generic with adding a port layer at host, but in different branch. Please check details, what it supports, what are current limitations etc in https://github.com/espressif/esp-hosted/issues/363

colazhb, 2024-05-27:
Thank you very much. I'll keep trying

mantriyogesh, 2024-07-01:
Did you able to use ESP-Hosted-MCU using https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host branch?