espressif/esp-hosted #397: WiFi Retransmissions

zereo08, 2024-05-28:
Hello!
I ported esp-hosted-fg to AzureRTOS, but i have some problems concerning transfer stability.
When transferring data via TCP or HTTP, retransmissions occur on various points during the transfer, regardless of file size (tested with 1MB, 10MB and 100MB test files). 
These retransmissions can take up to several seconds and reduce the overall performance -> are there any known issues concerning this topic?
I deactivated BT, only WiFi is running.
I also only reach 300kB/s over TCP and ~550kB/s over HTTP, is there any way to make those faster as well?
Tried to look through topics, but solutions found so far did not work for me unfortunately...

Here is an excerpt of wireshark concerning retransmissions:

[image]

Thanks in advance!

mantriyogesh, 2024-05-28:
Hello @zereo08 ,

The 2.4mbps looks very low speed.

Some questions:
1. Is there anything running at slave other than that of Hosted?
2. By AzureRTOS, I asuume it is MCU (and not MPU like Linux)
3. What is the base branch used? Is it from `master` branch or `feature/esp_as_mcu_host` branch?
4. What is the commit you have ported? 
5. Logs at both sides, ESP and MCU host?

zereo08, 2024-05-28:
Thanks for the quick response!

1) There is nothing else running on slave except Hosted
2) Yes, i use STM32H753ZI as host
3) I used master branch, because i did not see the other branch, thanks for pointing it out!
4) commit is i think 29208b6
5) can you tell me which logs you are interested in, so i can narrow it down?

mantriyogesh, 2024-05-28:
Master branch is little outdated but still have STM integration. As you have lwip integrated, I think you can check out

 https://github.com/espressif/esp-hosted/issues/363#issuecomment-2120146155

In this we are showcasing ESP as slave and another ESP as host (not stm32 yet). But it has a port layer.

Please note, there are a lot of changes in the code and esp_as_mcu_host has a lot of bug fixes and master is old.
But then at the same time, this new branch is not exactly backward compatible with master. 

Master branch:
We are slowly keeping master only for Linux new features. But only maintenance for MCU as such.

Why?
The complication with master is that it gets broken easily for MCU because of changes in Linux part. 

Going forward, feature/esp_as_mcu_host branch is dedicated for the any MCU as host adaption and features.

zereo08, 2024-05-28:
Ok thanks for the information, i will look into it, but i think porting the whole project again is gonna be tedious, so maybe i will try the comment you provided.

Thanks nontheless, have a nice day!

closed 2024-05-28 by zereo08

mantriyogesh, 2024-05-28:
I think at the least, you can incorporate two fixes:

Host side: 

https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/host/stm32/driver/transport/spi/spi_drv.c#L305-L338

 To change:

```c
static void check_and_execute_spi_transaction(void)
{
	uint8_t * txbuff = NULL;
	uint8_t is_valid_tx_buf = 0;
	GPIO_PinState gpio_handshake = GPIO_PIN_RESET;
	GPIO_PinState gpio_rx_data_ready = GPIO_PIN_RESET;

        xSemaphoreTake(mutex_spi_trans, portMAX_DELAY); /*changed line*/
	/* handshake line SET -> slave ready for next transaction */
	gpio_handshake = HAL_GPIO_ReadPin(GPIO_HANDSHAKE_GPIO_Port,
			GPIO_HANDSHAKE_Pin);

	/* data ready line SET -> slave wants to send something */
	gpio_rx_data_ready = HAL_GPIO_ReadPin(GPIO_DATA_READY_GPIO_Port,
			GPIO_DATA_READY_Pin);

	if (gpio_handshake == GPIO_PIN_SET) {

		/* Get next tx buffer to be sent */
		txbuff = get_tx_buffer(&is_valid_tx_buf);

		if ( (gpio_rx_data_ready == GPIO_PIN_SET) ||
		     (is_valid_tx_buf) ) {

			/* Execute transaction only if EITHER holds true-
			 * a. A valid tx buffer to be transmitted towards slave
			 * b. Slave wants to send something (Rx for host)
			 */
			//// xSemaphoreTake(mutex_spi_trans, portMAX_DELAY);
			spi_trans_func[hardware_type](txbuff);
			//// xSemaphoreGive(mutex_spi_trans);
		}
	}
        xSemaphoreGive(mutex_spi_trans); /*changed line*/
}
```

mantriyogesh, 2024-05-28:
2. Slave:
Add 

https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L636
 And 

https://github.com/espressif/esp-hosted/blob/2eb1fff5b7b18af20087ace3a35ba596172acdd5/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L544-L566

These fixes may or may not affect the throughout. But are important

zereo08, 2024-05-28:
The mutex change on the host side did the trick on my end and almost doubled the transfer speed, which is sufficient in my case!!

Have an awesome day and thanks again!