espressif/esp-hosted #398: esp32c6 mini GPIO question

XZ-Niupitang, 2024-05-30:
Can the SPI pins of the driver in esp-host-fg be changed freely?
[image]
[image]

mantriyogesh, 2024-05-30:
For ESP32-C6, yes, just try to avoid the bootstrapping pins to be used

mantriyogesh, 2024-05-30:
Hello @XZ-Niupitang ,

Unless they are already in use (say by sdio or any other function), any on bootstrapping gpios could be used.

Check https://docs.espressif.com/projects/esp-idf/en/v5.2.1/esp32c6/api-reference/peripherals/spi_slave.html#gpio-matrix-and-io-mux

When you use any other gpios than that of IO_MUX, they would route through GPIO matrix.
Obviously there would be minor delays in that, but overall ESP32-C6 should withstand. The max freq we tested for ESP32-C6 as spi slave is 26MHz.

However, the ESP32-C6 is used on PCB. for jumper cables, lower frequencies are suggested to be used.

XZ-Niupitang, 2024-05-30:
> Hello @XZ-Niupitang ,
>
> Unless they are already in use (say by sdio or any other function), any on bootstrapping gpios could be used.
>
> Check [...]

After trying to modify the interface, I found that I could no longer communicate with the Raspberry Pi. The driver of the Raspberry Pi was not modified. The c6 program was modified as follows. The startup log is as follows:
[image]
[image]
[image]
The actual hardware wiring is only changed in the C6 test board, the Raspberry Pi is unchanged.

mantriyogesh, 2024-05-30:
Setup photo please?

`git diff` at both places?

XZ-Niupitang, 2024-05-31:
> Setup photo please?
> 
> `git diff` at both places?

The only changes made are:
[image]

XZ-Niupitang, 2024-05-31:
It's ok, it looks like the modification was successful, sometimes life is fucked up

closed 2024-05-31 by XZ-Niupitang

FlSenicar, 2024-09-19:
Hello @XZ-Niupitang 
Can you tell me what you did to fix this issue? I am currently experiencing the same issue after remapping the SPI to other Gpios. 
The data seems to be shifted by one bit. Whenever I move the SPI back to the original IOs everything works fine.
Thank you very much!