espressif/esp-hosted #399: When does ng support idf version 5.2 or later

Ericiiiiii, 2024-05-31:
Now I use c6 to develop hostng and some other services, but my business needs to use version 5.2. Currently, host only supports version 5.1.3 idf, so my business cannot be developed together with hostng. I very much request hostng to support the version above 5.2 as soon as possible ，thanks。

mantriyogesh, 2024-05-31:
@kapilkedawat

Ericiiiiii, 2024-06-05:
When is the idf version expected to be upgraded?

kapilkedawat, 2024-06-05:
Hi @Ericiiiiii , could you please explain the need of IDF v5.2? Is there any specific feature that you want from IDFv5.2?

Ericiiiiii, 2024-06-05:
features：
TWAI: Added new TWAI driver APIs to support multiple controller instances
[https://github.com/espressif/esp-idf/releases/tag/v5.2](url)
I need not only hostng, but also this dual twai feature

Ericiiiiii, 2024-06-05:
> kapilkedawat
features：
TWAI: Added new TWAI driver APIs to support multiple controller instances
[https://github.com/espressif/esp-idf/releases/tag/v5.2](url)
I need not only hostng, but also this dual twai feature

Ericiiiiii, 2024-06-07:
hi@kapilkedawat， Is there a plan to upgrade the idf version when I need to synchronize development progress，thansk

mantriyogesh, 2024-07-01:
@kapilkedawat PTAL

Ericiiiiii, 2024-07-23:
is there any progress？ @mantriyogesh @kumekay

mantriyogesh, 2024-07-23:
@kapilkedawat ?