espressif/esp-hosted #401: Does esp32-c3 hosted-fg have an interface for querying the firmware version number?

ShengjieCh, 2024-06-06:
I hope to obtain the version number of the currently used firmware in the upper-layer business. Is there any way to achieve this?

mantriyogesh, 2024-06-12:
Hello @ShengjieCh ,

The current code doesn't have this support. however, it should be easy to add a new get-set using protobuf using https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/common/proto/esp_hosted_config.proto .

Protobuf is standard message serialisation interface.
Many customers have added their own commands and response as per their own requirement.

As this is generic request, we can also add for you but it may take some time.

ShengjieCh, 2024-07-05:
Will host-fg support it? We don't want to make modifications ourselves. Hope it is consistent with the realease version.

mantriyogesh, 2024-07-05:
Yes. We plan to add this. Might take couple weeks, unless urgent.

We are also going to improve upon release numbering.
New release version would be XY-A.B.C.D.E

where, X & Y are characters like, 'FG' OR 'NG' (two letters).
A, B, C, D, E are numbers.

For your ease, we will first implement control  request-response (app would demo request and response) using existing release numbering, i.e. A.B.C

Basically upon running a 'get_fw_release' command in user app would fetch it from esp.

**Later**, new versioning would be in place (hopefully this would be major release number change, so that many customers would not be affected)

ShengjieCh, 2024-07-11:
yes, this function is rather urgent.

mantriyogesh, 2024-07-30:
https://github.com/espressif/esp-hosted/issues/405#issuecomment-2257506378

Please verify and help to close the issue.