espressif/esp-hosted #404: SSID Length Error

XZ-Niupitang, 2024-06-17:
When using c_Demo to drive WiFi to turn on AP mode, the set WIFI name is VIN-LAAAPLDL2R1000154, but after starting the hotspot, the Wifi name is wrong. I checked the macro definition of SSID_LENGTH and it seems that there is no problem. How can I solve it?
[image]
[image]

mantriyogesh, 2024-06-17:
Thank you @XZ-Niupitang ,

Let we test this and come back.

XZ-Niupitang, 2024-06-19:
> Thank you @XZ-Niupitang ,
> 
> Let we test this and come back.

Hi, is there any result for this problem?

XZ-Niupitang, 2024-06-19:
> Thank you @XZ-Niupitang ,
> 
> Let we test this and come back.

Now I have encountered some other problems during use. After successfully mounting the driver and starting the AP mode, when connecting to WIFI and using the network to transmit data, the CPU load rate will directly increase to 100%. What is the reason for this?

mantriyogesh, 2024-06-19:
Can you please post the top output and the all *textual* logs of all possible windows (e.g. ESP, iperf server/client, dmesg,top)

XZ-Niupitang, 2024-06-19:
> Can you please post the top output and the all _textual_ logs of all possible windows (e.g. ESP, iperf server/client, dmesg,top)

Sorry, I opened all logs during compilation and checked dmesg to find the problem. After modifying it, the CPU usage is still high.

mantriyogesh, 2024-06-19:
Please attach the logs mentioned in https://github.com/espressif/esp-hosted/issues/404#issuecomment-2178495411 to this issue so that we also can review..

Also post your git commit at esp and host. if you have any diff, add up git diff both places.

XZ-Niupitang, 2024-06-20:
> Please attach the logs mentioned in [#404 (comment)](https://github.com/espressif/esp-hosted/issues/404#issuecomment-2178495411) to this issue so that we also can review..
> 
> Also post your git commit at esp and host. if you have any diff, add up git diff both places.
`static void esp_spi_transaction(void)
{
	struct spi_transfer trans;
	struct sk_buff *tx_skb = NULL, *rx_skb = NULL;
	u8 *rx_buf;
	int ret = 0;
	volatile int trans_ready, rx_pending;

	ktime_t start, end;
	ktime_t start1, end1;
	ktime_t start2, end2;
	ktime_t start3, end3;
	ktime_t start4, end4;
	ktime_t start5, end5;
	ktime_t startall, endall;	
	s64 elapsed_ns = 0;

	
	mutex_lock(&spi_lock);
	startall = ktime_get();

	start = ktime_get();

	trans_ready = gpio_get_value(HANDSHAKE_PIN);
	rx_pending = gpio_get_value(SPI_DATA_READY_PIN);
	end = ktime_get();
	elapsed_ns = ktime_to_ns(ktime_sub(end, start));
	printk(KERN_INFO "gpio_get_value Execution time: %lld ns\n", elapsed_ns);
	elapsed_ns = 0;

	if (trans_ready) 
	{
		start1 = ktime_get();
		if (data_path) 
		{
			tx_skb = skb_dequeue(&spi_context.tx_q[PRIO_Q_SERIAL]);
			
			if (!tx_skb)
			{
				tx_skb = skb_dequeue(&spi_context.tx_q[PRIO_Q_BT]);
			}
			if (!tx_skb)
			{
				tx_skb = skb_dequeue(&spi_context.tx_q[PRIO_Q_OTHERS]);
			}
			if (tx_skb) 
			{
				if (atomic_read(&tx_pending))
					atomic_dec(&tx_pending);

				if (atomic_read(&tx_pending) < TX_RESUME_THRESHOLD) {
					esp_tx_resume();
#if TEST_RAW_TP
					esp_raw_tp_queue_resume();
#endif
				}
			}
		}
		end1 = ktime_get();
		elapsed_ns = ktime_to_ns(ktime_sub(end1, start1));
		printk(KERN_INFO "data_path Execution time: %lld ns\n", elapsed_ns);
		elapsed_ns = 0;

		if (rx_pending || tx_skb) 
		{
			start2 = ktime_get();
			memset(&trans, 0, sizeof(trans));
			trans.speed_hz = spi_context.spi_clk_mhz * NUMBER_1M;

			/* Setup and execute SPI transaction
			 * 	Tx_buf: Check if tx_q has valid buffer for transmission,
			 * 		else keep it blank
			 *
			 * 	Rx_buf: Allocate memory for incoming data. This will be freed
			 *		immediately if received buffer is invalid.
			 *		If it is a valid buffer, upper layer will free it.
			 * */

			/* Configure TX buffer if available */

			if (tx_skb) 
			{
				// esp_info("2222 \n");
				trans.tx_buf = tx_skb->data;
				esp_hex_dump_dbg("spi_tx: ", trans.tx_buf, 32);
				
			} 
			else 
			{
				tx_skb = esp_alloc_skb(SPI_BUF_SIZE);
				trans.tx_buf = skb_put(tx_skb, SPI_BUF_SIZE);
				memset((void*)trans.tx_buf, 0, SPI_BUF_SIZE);
			}
			end2 = ktime_get();
			elapsed_ns = ktime_to_ns(ktime_sub(end2, start2));
			printk(KERN_INFO "tx_skb tx_buf Execution time: %lld ns\n", elapsed_ns);
			elapsed_ns = 0;

			/* Configure RX buffer */
			start5 = ktime_get();
			rx_skb = esp_alloc_skb(SPI_BUF_SIZE);
			rx_buf = skb_put(rx_skb, SPI_BUF_SIZE);

			memset(rx_buf, 0, SPI_BUF_SIZE);

			trans.rx_buf = rx_buf;
			trans.len = SPI_BUF_SIZE;

			end5 = ktime_get();
			elapsed_ns = ktime_to_ns(ktime_sub(end5, start5));
			printk(KERN_INFO "SPI_BUF_SIZE Execution time: %lld ns\n", elapsed_ns);
			elapsed_ns = 0;

#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 15, 0))
			if (hardware_type == ESP_PRIV_FIRMWARE_CHIP_ESP32) 
			{
				trans.cs_change = 1;
			}
#endif
			start3 = ktime_get();
			ret = spi_sync_transfer(spi_context.esp_spi_dev, &trans, 1);
			if (ret) 
			{
				esp_err("SPI Transaction failed: %d\n", ret);
				dev_kfree_skb(rx_skb);
				dev_kfree_skb(tx_skb);
			} 
			else 
			{

				/* Free rx_skb if received data is not valid */
				start4 = ktime_get();
				if (process_rx_buf(rx_skb)) 
				{
					dev_kfree_skb(rx_skb);
				}

				if (tx_skb)
					dev_kfree_skb(tx_skb);
				end4 = ktime_get();
				elapsed_ns = ktime_to_ns(ktime_sub(end4, start4));
				printk(KERN_INFO "process_rx_buf Execution time: %lld ns\n", elapsed_ns);
				elapsed_ns = 0;
			}
			end3 = ktime_get();
			elapsed_ns = ktime_to_ns(ktime_sub(end3, start3));
			printk(KERN_INFO "spi_sync_transfer Execution time: %lld ns\n", elapsed_ns);
			elapsed_ns = 0;
		}
	} 
	else 
	{
		up(&spi_sem);
		// usleep_range(10,100);
	}

	endall = ktime_get();
	elapsed_ns = ktime_to_ns(ktime_sub(endall, startall));
	printk(KERN_INFO "all Execution time: %lld ns\n", elapsed_ns);
	elapsed_ns = 0;

	mutex_unlock(&spi_lock);
}
`
I added time printing to the esp_spi_transaction function and found that it takes more than 30 milliseconds to execute this function once. What time-consuming operations are there in this function?

The system log is as follows
[ 2158.976045] gpio_get_value Execution time: 2605 ns
[ 2158.980850] data_path Execution time: 2255 ns
[ 2158.985489] tx_skb tx_buf Execution time: 320 ns
[ 2158.990119] SPI_BUF_SIZE Execution time: 2395 ns
[ 2158.997986] process_rx_buf Execution time: 5070 ns
[ 2159.002784] spi_sync_transfer Execution time: 8049060 ns
[ 2159.008090] all Execution time: 32050255 ns

XZ-Niupitang, 2024-06-20:
> Please attach the logs mentioned in [#404 (comment)](https://github.com/espressif/esp-hosted/issues/404#issuecomment-2178495411) to this issue so that we also can review..
> 
> Also post your git commit at esp and host. if you have any diff, add up git diff both places.

[ 5949.331948] all Execution time: 3271465 ns
[ 5949.336042] gpio_get_value Execution time: 2175 ns
[ 5949.340841] skb_dequeue Execution time: 700 ns
[ 5949.345286] esp_alloc_skb Execution time: 2355 ns
[ 5949.349991] spi_sync_transfer Execution time: 3264720 ns
[ 5949.355307] process_rx_buf Execution time: 2300 ns
[ 5949.360102] skb_put Execution time: 990 ns
[ 5949.364207] SPI Transaction Len : 1600

root@j7-evm:~# [ 1126.027280] esp32_spi: esp32_probe: reset esp32!
[ 1126.237173] esp32_spi: esp32_probe: ESP32C6 reset performed
[ 1126.243853] esp32_spi: spi_init: ESP: SPI host config: GPIOs: Handshake[406] DataReady[405]
[ 1126.248351] esp32_spi: esp_spi_thread: esp spi thread created
[ 1126.252757] esp32_spi: spi_dev_init: ESP host driver claiming SPI bus [5],chip select [3] with init SPI Clock [26]
[ 1126.473208] esp32_spi: spi_init: HOST state is READY
[ 1127.067889] all Execution time: 3030 ns
[ 1127.071725] gpio_get_value Execution time: 2670 ns
[ 1127.076508] skb_dequeue Execution time: 0 ns
[ 1127.080770] esp_alloc_skb Execution time: 0 ns
[ 1127.085216] spi_sync_transfer Execution time: 0 ns
[ 1127.090002] process_rx_buf Execution time: 0 ns
[ 1127.094523] skb_put Execution time: 0 ns
[ 1127.098436] SPI Transaction Len : 0
[ 1127.104254] esp32_spi: process_event: INIT event rcvd from ESP
[ 1127.105056] all Execution time: 3138070 ns
[ 1127.110080] esp32_spi: process_init_event: EVENT: 2
[ 1127.114269] gpio_get_value Execution time: 1565 ns
[ 1127.119024] esp32_spi: process_init_event: EVENT: 1
[ 1127.119025] esp32_spi: process_init_event: EVENT: 0
[ 1127.119027] esp32_spi: process_init_event: EVENT: 3
[ 1127.119562] esp32_spi: process_capabilities: ESP peripheral capabilities: 0xe8

It seems that all the time is consumed in SPI transmission. However, most of the transmitted information is 1600 bytes long. In addition to modifying the underlying SPI driver, I currently have no better optimization solution. Do you have any good method for reference?

mantriyogesh, 2024-06-24:
We would need to analyse this in detail @XZ-Niupitang . We would also reproduce this behaviour if the cpu utilization hitting such a high.

@SohKamYung-Espressif can you PTAL?

XZ-Niupitang, 2024-06-25:
> We would need to analyse this in detail @XZ-Niupitang . We would also reproduce this behaviour if the cpu utilization hitting such a high.
> 
> @SohKamYung-Espressif can you PTAL?

https://github.com/XZ-Niupitang/TDA4_ESPWIFI/tree/main

I uploaded the transplanted TDA4 driver source code and esp32c6 program source code. Please take a look and point out any errors. Thanks.

mantriyogesh, 2024-06-25:
Hello @XZ-Niupitang ,

@SohKamYung-Espressif is on business travel right now, He would be back this week and check the issue in detail.
Thank you for sharing the code base. Can you please let us know the base git commit you have referred for eitther side?

XZ-Niupitang, 2024-06-26:
> Hello @XZ-Niupitang ,
> 
> @SohKamYung-Espressif is on business travel right now, He would be back this week and check the issue in detail. Thank you for sharing the code base. Can you please let us know the **base** **ESP-Hosted** git commit you have referred for either side?

The base ESP-Hosted last commit was：
commit **af480c54895044fa33c82ecde285a8c05ac76fbb** (HEAD -> master)
Author: Kapil Gupta <kapil.gupta@espressif.com>
Date:   Fri May 17 11:38:14 2024 +0530

SohKamYung-Espressif, 2024-06-26:
> The base ESP-Hosted last commit was：
> commit **af480c54895044fa33c82ecde285a8c05ac76fbb** (HEAD -> master)
> Author: Kapil Gupta [kapil.gupta@espressif.com](mailto:kapil.gupta@espressif.com)
> Date: Fri May 17 11:38:14 2024 +0530

Hello, @XZ-Niupitang

Could you point me to your top-of-the-tree commit in https://github.com/espressif/esp-hosted/commits/master/ ? I did a `git log` and that commit you quoted is not present.

SohKamYung-Espressif, 2024-06-26:
> I used htop to check the current CPU

Can you run `htop` and enable the option to show kernel threads? On my version of `htop`, this is the `K` key (capital K). This is to verify which kernel threads are using CPU resources on your system.

The ESP SPI driver normally starts up the kernel thread `esp32_spi` to do spi transmissions.

XZ-Niupitang, 2024-06-26:
> Hello, @XZ-Niupitang
>
> Could you point me to your top-of-the-tree commit in https://github.com/espressif/esp-hosted/commits/master/ ? I did a `git log` and that commit you quoted is not present.

I have not committed any code on this branch. My current code is based on commit af480c54895044fa33c82ecde285a8c05ac76fbb ，As I replied before。

XZ-Niupitang, 2024-06-26:
> Can you run `htop` and enable the option to show kernel threads? On my version of `htop`, this is the `K` key (capital K). This is to verify which kernel threads are using CPU resources on your [...]

Yes, that's right, esp_spi occupies 100% of the core. In AP mode, as long as you use wifi to transfer files, this thread will always occupy a core.

SohKamYung-Espressif, 2024-06-26:
> I have not committed any code on this branch. My current code is based on commit af480c54895044fa33c82ecde285a8c05ac76fbb

I am unable to find this commit in the esp-hosted master branch.

Could you check https://github.com/espressif/esp-hosted/commits/master/ and let me know the name of the commit?

SohKamYung-Espressif, 2024-06-26:
Can you try this patch to see if it fixes the SoftAP SSID issue?

[softap_ssid_fix_1.patch](https://github.com/user-attachments/files/15983114/softap_ssid_fix_1.patch)

XZ-Niupitang, 2024-06-26:
> I am unable to find this commit in the esp-hosted master branch.
>
> Could you check https://github.com/espressif/esp-hosted/commits/master/ and let me know the name of the commit?

[image]
[image]
[image]

XZ-Niupitang, 2024-06-26:
> Can you try this patch to see if it fixes the SoftAP SSID issue?
> 
> [softap_ssid_fix_1.patch](https://github.com/user-attachments/files/15983114/softap_ssid_fix_1.patch)

Thanks, I will add this patch and test it again

XZ-Niupitang, 2024-06-26:
> Can you try this patch to see if it fixes the SoftAP SSID issue?
> 
> [softap_ssid_fix_1.patch](https://github.com/user-attachments/files/15983114/softap_ssid_fix_1.patch)

Thanks for your support, the SSID length issue has been resolved, but the high CPU usage issue still exists. Please see the video below.

https://github.com/espressif/esp-hosted/assets/103943684/e2919695-77d4-441d-bb03-7c1a2b9775b9

SohKamYung-Espressif, 2024-06-26:
> the high CPU usage issue still exists

I have tested using iperf2 using a Raspberry Pi 4 as the host (our reference linux system).

With the SPI bus at 26 MHz, I am seeing the `esp32_spi` kthread peak at 20% of cpu (usually it is about 10-15%). This is while iperf2 throughput is around 14 Mbits/s. 

I have compared your linux driver code against the master branch, and I do not see any major differences.

One possibility is hardware DMA was not enabled for SPI transfers on your SoC. The CPU would then busy doing the SPI transfers.

XZ-Niupitang, 2024-06-26:
> > the high CPU usage issue still exists
> 
> I have tested with iperf2 using a Raspberry Pi 4 as the host (our reference linux system).
> 
> With the SPI bus at 26 MHz, I am seeing the `esp32_spi` kthread peak at 20% of cpu (usually it is about 10-15%). This is while iperf2 throughput is around 14 Mbits/s.
> 
> I have compared your linux driver code against the master branch, and I do not see any major differences.
> 
> One possibility is hardware DMA was not enabled for SPI transfers on your SoC. The CPU would then be busy doing the SPI transfers.

Thanks for your reply. I will continue to check the SPI driver. I also found that SPI transmission takes up a lot of time.

XZ-Niupitang, 2024-06-26:
> I have tested with iperf2 using a Raspberry Pi 4 as the host (our reference linux system).
>
> With the SPI bus at 26 MHz, I am seeing the `esp32_spi` kthread peak at 20% of cpu (usually it is about [...]

[image]
Thank you very much for providing the solution. It is indeed caused by not enabling DMA in the device tree.

closed 2024-06-26 by XZ-Niupitang

SohKamYung-Espressif, 2024-06-28:
@XZ-Niupitang FYI, the patch I provided to you has been merged into master:  https://github.com/espressif/esp-hosted/commit/abe8f45c577f26a88d98528e27dfd027472fa8f4

XZ-Niupitang, 2024-06-28:
> @XZ-Niupitang FYI, the patch I provided to you has been merged into master: [abe8f45](https://github.com/espressif/esp-hosted/commit/abe8f45c577f26a88d98528e27dfd027472fa8f4)

Thanks for the support

SohKamYung-Espressif, 2024-06-28:
@XZ-Niupitang

> It is indeed caused by not enabling DMA in the device tree

Could you post the SPI section from your device tree here, as a reference, in case other people encounter the same problem.

XZ-Niupitang, 2024-06-28:
> @XZ-Niupitang
> 
> > It is indeed caused by not enabling DMA in the device tree
> 
> Could you post the SPI section from your device tree here, as a reference, in case other people encounter the same problem.

Of course, I'd love to：

	main_spi6: spi@2160000 {
		compatible = "ti,am654-mcspi","ti,omap4-mcspi";
		reg = <0x00 0x02160000 0x00 0x400>;
		interrupts = <GIC_SPI 190 IRQ_TYPE_LEVEL_HIGH>;
		#address-cells = <1>;
		#size-cells = <0>;
		clocks = <&k3_clks 272 1>;
		power-domains = <&k3_pds 272 TI_SCI_PD_EXCLUSIVE>;
		dmas = <&main_udmap 0xc624>, <&main_udmap 0x4624>,
			   <&main_udmap 0xc625>, <&main_udmap 0x4625>,
			   <&main_udmap 0xc626>, <&main_udmap 0x4626>,
			   <&main_udmap 0xc627>, <&main_udmap 0x4627>;
		dma-names = "tx0", "rx0" , "tx1", "rx1", "tx2", "rx2" , "tx3", "rx3";
	};