espressif/esp-hosted #405: Can esp32-c3 support independent control of turning on and off wifi and bluetooth?

ShengjieCh, 2024-06-17:
Because of power consumption requirements, I hope to be able to control the switches of wifi and bluetooth separately; when wifi is not needed, turn off wifi and turn it on when wifi is needed.

mantriyogesh, 2024-06-17:
Using FG or NG?

ShengjieCh, 2024-06-17:
host-fg

mantriyogesh, 2024-06-18:
Sure. We plan these additional APIs, say,
```
wifi_setup()
wifi_cleanup()
bluetooth_setup()
bluetooth_cleanup()
```
in the control path APIs, to use from application layer.
These separate APIs would help conserve power and memory.

We would add this support, but it may take some time, considering current work.

ShengjieCh, 2024-06-18:
May I ask when support will be available?

mantriyogesh, 2024-06-18:
It's small feature, but may take time till 1-Jul. Would that work for you?

ShengjieCh, 2024-06-18:
This is helpful, thanks!

ShengjieCh, 2024-07-01:
Can it be completed this week?pleted this week?

mantriyogesh, 2024-07-01:
Yes..

ShengjieCh, 2024-07-08:
May I ask when this change will be included in the release?

mantriyogesh, 2024-07-08:
It would be added once reviewed.
Did you try it at your end? I have tested it for simple and complex combinations already.

ShengjieCh, 2024-07-11:
not yet...    When is it expected to be added to the realease version?

mantriyogesh, 2024-07-11:
Merged now, will reflect in 10 mins.

ShengjieCh, 2024-07-19:
Do I need to update the firmware?
[image]

mantriyogesh, 2024-07-19:
Firmware not flashed?

ShengjieCh, 2024-07-19:
Where can I get the latest firmware?

mantriyogesh, 2024-07-19:
Flash using the code:
- if windows: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/setup_windows11.md
- else: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SPI_setup.md#222-source-compilation

Just to remind, please use latest FG master at both sides.

ShengjieCh, 2024-07-22:
Hello, please tell me what's the strategy for this firmware update, Will there be a release version? Because we don't want to use our own locally compiled firmware.
[image]

mantriyogesh, 2024-07-22:
Yes, we are going to update ota as per versioning system, the code is already in review, but it may take another week to review and test the change.

ShengjieCh, 2024-07-22:
Thanks, it's helpful!

ShengjieCh, 2024-07-30:
hello, mantriyogesh, may I ask what day of the week it will be updated?

mantriyogesh, 2024-07-30:
Versioning support change pulled to master a minute ago, please help checking it if it answers your questions.

mantriyogesh, 2024-07-30:
Please note, the binaries are yet not updated, they would update on next release.
Not yet planned for next release, the blocking bit right now is OTA with new release numbering. once it is in place, we will push new binaries.

mantriyogesh, 2024-07-30:
FG existing ota, should work anyway as is.

ShengjieCh, 2024-08-02:
hi, mantriyogesh: 
After actual measurement, the power consumption measured after disabling wifi is similar to that of enabling wifi. Is this in line with expectations? What is the specific implementation of disable/enable wifi?
disable wifi(0.277W)：[image]
enable wifi(0.283W)：[image]

The log of disable wifi is as follows
[image]

mantriyogesh, 2024-08-02:
Question would be what mode wifi was used when enabled. 

Power consumption
Wifi active tx  > wifi active rx > wifi idle > wifi disable

Similar would be the case with bluetooth. 

Obviously wifi peaks world be much higher than bluetooth. 

There is single rf time shared between wifi and bluetooth, giving many more combinations, but importantly, wifi and bluetooth both disabled would also be one imp case.

For power reduction, some more cases to consider:
- deep sleep
- light sleep
- wifi modem sleep
- wifi tx max power
- wifi state idle/scan/connected 
- lowering cpu freq, also DFS - spi peripherals 

Specific requirements might be helpful to understand.
But in general, you might need to consider application design to use specific sleep modes in specific wifi and/or bluetooth states.

Optionally, you can also get in touch with our dedicated support team at https://www.espressif.com/en/contact-us/sales-questions

ShengjieCh, 2024-08-02:
After enable wifi, the softAP mode is used, but no STA is connected to it, it is just broadcasting

mantriyogesh, 2024-08-02:
With softap mode, there is power save is not possible.
In this case, power consumption would be expected to be high, irrespective of whether you use `STA mode` and/or `Bluetooth`.

ShengjieCh, 2024-08-02:
But is it useless to disable wifi? First I enable wifi, then enable softap mode. After that, I call stop softap and disable wifi. Will the power consumption become lower? I can’t see the wifi broadcast at this time.

mantriyogesh, 2024-08-02:
If softAP is stopped followed by disable Wi-Fi, the power would be lowered wrt Wi-Fi.
Try disabling Bluetooth and Wi-Fi both and then measure power without starting softAP.
For power consumption, consider these factors : https://github.com/espressif/esp-hosted/issues/405#issuecomment-2264546193 keeping RF current state in the mind.

mantriyogesh, 2024-08-02:
For proper power saving, you might need to consider multiple options.

- Sleep modes:
  - https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-reference/system/sleep_modes.html

- PM:
  - https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-reference/system/power_management.html

- Power Measurement considerations:
  - https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-guides/current-consumption-measurement-modules.html?highlight=power%20save

- Wi-Fi power save:
  - https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-guides/wifi.html?highlight=power%20save#esp32-c3-wi-fi-power-saving-mode

- Bluetooth:
  - https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-reference/bluetooth/controller_vhci.html#_CPPv420esp_ble_tx_power_set20esp_ble_power_type_t17esp_power_level_t
  - https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-reference/bluetooth/controller_vhci.html#_CPPv419esp_bt_sleep_enablev

These need to be configured correctly.
This broader scope of power save we had already considered developing slave power save feature which customisable and suited for ESP-Hosted case, but this would take tremendous time, considering all the permutation combination to verify. This change is first targeted on https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host (MCU as host solution). Once streamlined, we will consider putting to master as well.

But for now, you might have to check above links which describes a lot of power saving options available.
While measuring the power, you also might need to account a lot of factors, for example, LED over dev kit(if using one), etc.

In simple words, let us know your exact requirements.
To answer your earlier question, what code is involved, please find the commit https://github.com/espressif/esp-hosted/commit/4c1be0f6bb2ae58f91b6b8c3587e5e52a72e2048 which is added for enable/disable functionality of Wi-Fi & Bluetooth. Slave specific code, where we disable the controllers upon receiving the disable request, is here:

https://github.com/espressif/esp-hosted/commit/4c1be0f6bb2ae58f91b6b8c3587e5e52a72e2048#diff-7ecd9e625393ef2d3c307c0950ef3e2186fa3f5a7ffce1afac1e7081ee350aec

ShengjieCh, 2024-08-06:
> If softAP is stopped followed by disable Wi-Fi, the power would be lowered wrt Wi-Fi. Try disabling Bluetooth and Wi-Fi both and then measure power without starting softAP. For power consumption, consider these factors : [#405 (comment)](https://github.com/espressif/esp-hosted/issues/405#issuecomment-2264546193) keeping RF current state in the mind.

hi, mantriyogesh:
How much is the power consumption expected to be reduced after disabling wifi? What is the order of magnitude? (<10mW, 10~50mW, 50~100mW, >100mW); After I tried to comment out initialise_wifi(), only the BT function is available. At this time Power consumption dropped from 280mW to 74mW, which is what I expected.
[image]
[image]

mantriyogesh, 2024-08-06:
Let me check and get back on this

ShengjieCh, 2024-08-07:
hi, mantriyogesh, is there any progress?

ShengjieCh, 2024-08-07:
I want to know what is the difference between disable wifi and not initializing wifi? Why is there such a big difference in power consumption? Is the function of disable wifi just a change of some parameters and cannot actually turn off wifi?
[image]
[image]

mantriyogesh, 2024-08-07:
We have not measured the power readings recently.

esp_wifi_stop() should stop the RF but keep the Wi-Fi driver with dormant state. FreeRTOS tasks , interrupts etc are retained wrt Wi-Fi driver.
Additionally, Not power wise, but memory wise, the complete Wi-Fi driver's context is still occupied & not freed.

So we wish to not use Wi-Fi, better to de-init(), as it would also free out the memory and slightly lower power consumption further.

As I said earlier, Power consumption is dependent upon a lot of factors, including what application is doing, Rf state etc.

In nutshell, esp_wifi_stop() should get you drastic power consumption difference.
The difference from esp_wifi_stop() to esp_wifi_deinit() should be very small in comparison.

There are a lot of ESP-IDF power saving options available as said earlier.

To further localise this problem, just justt wifi and disable bluetooth completely and try:

```bash
idf.py menuconfig
```
and components->bluetooth => unselect (disable) => flash and reboot.

Now, try to enable and disable wifi and re-do power measurements.

In any case, ESP-Hosted doesn't have code or control over ESP-IDF, in fact ESP-Hosted lies over ESP-IDF.

Even if you face issue, directly or indirectly, ESP-IDF Wi-Fi driver team would need to check this issue.

If you can showcase the issue, we can ask Wi-Fi team to check internally.
Issue would be hold, unless we see response from ESP-IDF wifi driver team.

Alternatively, you can raise https://github.com/espressif/esp-idf/issues issue, and would directly be checked by Wi-Fi driver team there.

ShengjieCh, 2024-08-07:
An issue has been raised. Please help to ask Wi-Fi team to check, thanks! 
https://github.com/espressif/esp-idf/issues/14322