espressif/esp-hosted #407: The communication between the ESP32C3 development board and the board failed

oldmylei, 2024-06-21:
Hello, I tried to connect our board to the ESP32 development board, and there was an error when loading the driver.

The following is the diagram of the hardware connection. We connected seven signals with the ESP32 in total, and one GND signal was not drawn:
[image]

When I finished this, I built and burned the ESP32 board image under esp_hosted_ng/esp/esp_driver/network_adapter, and compiled the driver of our board under esp_hosted_ng/host. Below is the log of the ESP32C3 development board and the dmesg content information of our board:
[esp32_dmesg.txt](https://github.com/user-attachments/files/15922495/esp32_dmesg.txt)
[esp32_log.txt](https://github.com/user-attachments/files/15922498/esp32_log.txt)

According to the log, it seems that my board is not properly wired to the ESP32 development board? Or the ESP32 needs to start some kind of mode.

Could you take a look, please? Thank you

mantriyogesh, 2024-06-21:
Thank you for sharing detailed logs and clear issue description @oldmylei .

Currently ESP driver code not able to claim spi bus.
Meaning possibly you are using incorrect spi bus and/or incorrect cs.
Other possibility that spidev(default Linux user space driver for spi) is still using the spi bus. If the bus is available only then hosted kernel driver can claim.. if spidev is still enabled, (check `$ ls /dev/spidevX.Y` output. X is bus, Y is CS. If bus-cs mapping file is present, spidev has claimed it. You might need to disable spidev in device tree or SoC config)

Please review carefully, https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md

oldmylei, 2024-06-21:
[image]
I read this tutorial and it required me to turn off the default SPI driver. Here is the SPI node configuration I used. I actually disabled it and went to the device node that did not generate /dev/spidevX.Y

[image]

mantriyogesh, 2024-06-21:
Although we are not experts for device tree of your Linux SoC, From experience, It looks like you are disabling spi bus altogether. you need to only disable spidev (disable the user space driver only. not the bus itself)

oldmylei, 2024-06-21:
[image]
[image]

I think I should understand what you mean. Do you mean to mask only the default driver of SPI as I did in the picture above?

mantriyogesh, 2024-06-21:
Change status disabled for (inside) **spidev**. not outside it.
Outside would disable 'spi6' altogether.

I am not sure of compatible. But I think, let it be as is for now.

mantriyogesh, 2024-06-21:
You would need to port the spi bus number to 6 and cs to 0(?).
Also connect the smallest wires possible with spi6 and CS0 of spi 6 with slave' connections.

There are three extra wires/GPIOs apart from 4 wire spi, that you need.
Please refer porting guide in earlier comment.

Once those handshake, dataready and resetpin are configured, test them.
reset pin on toggle should reset slave.
data_ready and handshake are interrupt at host, so manually low to high (edge) interrupt should either print in hosted code (check porting guide)

mantriyogesh, 2024-06-21:
these are typical steps while porting..

oldmylei, 2024-06-21:
I set handshake, dataready and resetpin as normal output GPIO
[image]

IO90 is handshake, then IO91 is dataready, I didn't connect the resetpin signal because the ESP32C3 has a reset button on the development board
After doing this, I still get the following error when loading the compiled generated driver

[image]

oldmylei, 2024-06-24:
[image]
Good morning, according to the documentation, it seems that only 4.X kernel adaptation is described. My system kernel is version 5.10, can it adapt to this module?

mantriyogesh, 2024-06-24:
HCI over SPI is supported 'over' kernel version 4.7.0.
Meaning, 4.7.0 + is supported for HIC over SPI (bluetooth over spi).

You kernel version is supported.

mantriyogesh, 2024-06-24:
> I didn't connect the resetpin signal because the ESP32C3 has a reset button on the development board

Reset not only a buttin, but GPIO on devkit header, named 'RST'

Can you please run command,
```
ls /dev/spidev*
```
-> 1. Without disabling spidev
-> 2. After disabling spidev

Also can you please post the current device tree for spi?

oldmylei, 2024-06-28:
good morning, I noticed that under esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host there isa spidev_disabler.dts file, the following is its contents
[image]
Maybe I should add the following information to my device tree?
[image]

oldmylei, 2024-06-28:
[image]
When my device tree is set this way, my machine generates spidev6.0 nodes
[image]

[image]
When my device tree is set this way, no spi nodes are generated

oldmylei, 2024-06-28:
In addition, I set the reset pin to 89, which proved to be effective, the ESP32 development board reset when I entered the following command
[image]
[image]

oldmylei, 2024-06-28:
Hello, I have solved the problem of loading the driver, but I did not generate the WIFI node after loading the esp32_spi.ko driver, the following is the detailed log

root@myd-jd9x:~# insmod esp32_spi.ko resetpin=89
root@myd-jd9x:~#
root@myd-jd9x:~#
root@myd-jd9x:~# lsmod
Module                  Size  Used by    Tainted: G
esp32_spi              49152  0
lt8912b_hdmi           16384  0
yt6801                368640  0
xrp_hw_semidrive       16384  0
lm75                   20480  0
pvrsrvkm             1273856  0
ngbe                  217088  0
root@myd-jd9x:~#
root@myd-jd9x:~#
root@myd-jd9x:~# ifconfig
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:168 errors:0 dropped:0 overruns:0 frame:0
          TX packets:168 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:12416 (12.1 KiB)  TX bytes:12416 (12.1 KiB)

root@myd-jd9x:~# ifconfig -a
can0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:52

can1      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:53

can2      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:54

can3      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:55

enp1s0    Link encap:Ethernet  HWaddr 00:55:7B:B5:7D:F7
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:50 Base address:0x8000

eth0      Link encap:Ethernet  HWaddr 56:B8:E5:23:5E:F3
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:46

eth1      Link encap:Ethernet  HWaddr 76:AF:C2:A6:A3:16
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:47

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:168 errors:0 dropped:0 overruns:0 frame:0
          TX packets:168 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:12416 (12.1 KiB)  TX bytes:12416 (12.1 KiB)

teql0     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          NOARP  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tunl0     Link encap:UNSPEC  HWaddr 00-00-00-00-30-30-00-30-00-00-00-00-00-00-00-00
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@myd-jd9x:~#

oldmylei, 2024-06-28:
This is the info information for loading the driver
[image]

mantriyogesh, 2024-06-28:
ifconfig listing is later stage. Also, have you changed the name of network intterface as such? (by default, it would be espsta0 for ng)
I think something is wrong before reaching to ifconfig stage. dmesg and esp side logs would let you know the issues.

mantriyogesh, 2024-06-28:
As resetpin is not hooked, please reset esp manually and check the logs in the dmesg again..

Reset pin is mandatory.

For now,
1. rmmod esp32_spi
2. insmod esp32_spi resetpin=89
3. check esp is being reset on step (2)
4. also check the GPIO handshake and data ready correctly generating interrupts. 
For this, Can add log at https://github.com/espressif/esp-hosted/blob/8db84e42e71823e6a04ed55eef06d3ccacd8e4f6/esp_hosted_ng/host/spi/esp_spi.c#L60
and https://github.com/espressif/esp-hosted/blob/8db84e42e71823e6a04ed55eef06d3ccacd8e4f6/esp_hosted_ng/host/spi/esp_spi.c#L69

and manually trigger posedge interrupt, to see these pins are correct.

mantriyogesh, 2024-06-28:
Also , please test **spidev** test (before disabling spidev), to verify the underlying spi driver is working fine or not for your bus and chip select combination.

mantriyogesh, 2024-06-28:
In summary, do the porting (using porting guide).
Verify all GPIOs work as they are intended to be working.
Verify SPI lines are working fine (both hardware and software/config issues would need tto rule out first)

oldmylei, 2024-06-28:
All I know for sure is that the resetpin I set will reset the ESP32C3 properly

mantriyogesh, 2024-06-28:
sure. other pins working as they expected to be?

Unless all GPIOs work as they need to, porting is incomplete and solution would not work as such

oldmylei, 2024-06-28:
[image]
I added the above printing information, the result did not print these sentences

[image]
Is it proof that my handshake and dateready pins are not connected correctly, I am sure that I have set both pins to GPIO output mode and are GPIO90 and GPIO91 pins

oldmylei, 2024-06-28:
I noticed the spi/esp_spi.h file, do I need to change the HANDSHAKE_PIN and SPI_DATA_READY_PIN values?
[image]

mantriyogesh, 2024-06-28:
> https://github.com/espressif/esp-hosted/issues/407#issuecomment-2196405152

This is mentioned in porting guide..
[image]

oldmylei, 2024-06-28:
[image]
After I changed the values of HANDSHAKE_PIN and SPI_DATA_READY_PIN, an interrupt did occur and the debugging information above, but still no wifi node was generated

mantriyogesh, 2024-06-28:
Worth to check the porting guide step-wise.
Porting is prerequisite for ESP-Hosted project. We have added common issues and ways to fix them in this guide.

oldmylei, 2024-07-01:
Good morning. I have a question. Can the CS signal of SPI be set to GPIO to control it?

mantriyogesh, 2024-07-01:
Sorry, are you referring to the CS at slave (on ESP32-C6) ?
If you mean can you change the CS to different GPIO, yes, you can do it.

Or do you mean you wish to omit it from SPI slave controller and manually handle it?

oldmylei, 2024-07-01:
I added the following print information to the esp_inetaddr_event function in the esp_cfg80211.c file
[image]
But the log information is not printed, is there some reason why this function is not called?

oldmylei, 2024-07-01:
In addition, I traced this function to the end, which was called by the esp_add_card function, and then the esp_add_card function was called by the process_event_esp_bootup function in the spi/esp_spi.c file, specifically why this function was not called. How do we track it?

mantriyogesh, 2024-07-01:
@kapilkedawat ++

kapilkedawat, 2024-07-01:
@oldmylei as Yogesh mentioned, please check the porting guide and make sure connections are set correctly. Only after that we need check further for cfg functions. Unless connections are not correct you won't get the boot event and cfg functions won't be called.

mantriyogesh, 2024-07-01:
@oldmylei I think instead of jumping directly to end solution, as we know it is not working, can set back a little and progress step wise. 

Assuming, pins resetpin, handshake and data ready are doing good, we can focus on verifying the transport instead of complicating with network and wifi layers. 

Have you tested spidev testing before on the spi line, possibly with full duplex? What I meant, did we verify spi line in either direction?

In any way, Can you perform raw throughout test in either direction (tx and rx) using [transport_test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md).

If this test fails, above stated spidev test would be mandatory. If this test passes, please attach the logs for ESP & host for both directions (rx and tx separately)

oldmylei, 2024-07-03:
[image]
Good morning, I have written an spi test program, and the above is the verification result (I short-connected the two feet MISO and MOSI for verification), and the result shows that the received data is normal

[image]
In addition, I would like to ask whether I need to change the parameter information of esp_board.chip_select if my CS chip selects the signal using GPIO control

mantriyogesh, 2024-07-03:
We have never tried in Linux. As part MCU we used to control when the default supported CS was not correct.
But as you are using Linux, It is better to let the SPI controller driver handle this own their own.

However, if you think you must do this, It would be your own debugging and handling the timings correct. Logic analyser would help you in getting there in big way.

Is there any specific rationale behind that you could share with us why you wish to handle it manually?

If you wish to control the pin manually, you need to find a way to inform your linux driver, **not to use** chip_select. Also, modify lower linux driver or if supported through config, disable handling of chip select.

mantriyogesh, 2024-07-03:
Also worth tto check if your platform SPI driver supports `SPI_CS_HIGH` mode

oldmylei, 2024-07-03:
[image]
In fact, there is a notification driver on my underlying driver, which is the pin of my CS chip selection signal

mantriyogesh, 2024-07-03:
I did not understand your latest comment.

In any case, this is all experiments you wish to do it.
As scope of ESP-Hosted, it can only function if the base spi transport is stable.

So I would request to test your manual changes with some other simple user space driver first in either direction rx and tx.

Once you verify the spi driver is stable , you can add ESP-Hosted layer on top to satisfy your use case.

oldmylei, 2024-07-09:
hi,mantriyogesh
I have now switched to the SDIO+UART interface on the ESP32,However, after I connected the hardware (a total of eight signals, namely SD2.CLK CMD DATA0~DATA3 resetpin and GND) and loaded the driver, there was no data interactive display. The printed result is as follows

[image]

mantriyogesh, 2024-07-09:
First only try with sdio, instead of sdio+uart.

As uart pins are not connected, you cannot expect uart to be working. Please check again setup document in detail.

Also share the setup picture for sdio.

mantriyogesh, 2024-07-09:
Troubleshoot your issues at first level using [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md)

Once you keep every requirement fulfilled in porting guide, sdio would work.

oldmylei, 2024-07-10:
I am following the diagram below to connect the signal

[image]

Also, I made progress when loading the driver last night, but reported the following error
[image]

I want to change mmc1 to mmc2, here is my configuration in the device tree, but I can't find the source code where to configure this

[image]

mantriyogesh, 2024-07-10:
1. -110 is timeout.
Please share dmesg from bootup.

Additionally,
2. Can you please try 'broken-cd;'

3. Have you added necessary pull-ups to necessary lines?
4. Can you please share the photo of your setup?

oldmylei, 2024-07-10:
[log.txt](https://github.com/user-attachments/files/16154319/log.txt)
Above is dmesg from bootup.

The second point I don't understand is the need to add 'broken-cd; 'Is this parameter on my device tree? Are there any examples?

For the third point, I connected the other six SD2 signals except CLK to pull-up resistors

Fourth, I don't quite understand what you mean. Do you need me to share anything

mantriyogesh, 2024-07-10:
1. Possible incorrect/unsupported Device Tree blob present in your Device tree (Just mild warning)
```
[    7.124899] mmcblk mmc1:0001: no of_node; not parsing pinctrl DT
```
seems to have some incorrect config. But I would rather prefer to **ignore** if the mmc device instance is tried to probe by kernel fine (looks like so in your case)

2. ESP sdio device is sensed by your kernel, so basic connections should be fine (still need to judge connections for signal integrity issues discussed in the end of this comment)
[image]

3. Please make sure you connect **resetpin** to 'EN' of ESP32 as mentioned in porting guide:
[image]

4. Entry, with 'mc1:0001:**2**' failing can be ignored, since we do not use `function 2` of sdio. Although, 'mc1:0001:1' is failing with `-110`. Try polling using 'broken-cd'
```
[  277.643854] esp_sdio: probe of mmc1:0001:2 failed with error -110
```

> The second point I don't understand is the need to add 'broken-cd; 'Is this parameter on my device tree? Are there any examples?

5. yes, This is to poll the probing, if it fails to probe slave sdio.
you can find example from https://wiki.stmicroelectronics.cn/stm32mpu/index.php?title=SDMMC_device_tree_configuration&oldid=70703.

> For the third point, I connected the other six SD2 signals except CLK to pull-up resistors

6. What is pull-up value for all sdio pins?  Recommendation in [porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md#13-additional-pull-ups)
[image]

7. Please cross-check if you need to burn eFuse for voltage expectations.
For ESP32, most likely you would need to do this irreversible action. It highly depends on exact module/chipset used. Please check : [Bootstrap conflict on SDIO dat2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#conflicts-between-bootstrap-and-sdio-on-dat2)

> Fourth, I don't quite understand what you mean. Do you need me to share anything

8. **Jumper cables & SDIO**
Is your hardware PCB or you are using wires? SDIO is known for signal integrity issues if used with wires. If you wish to evaluate with wires only, there are some suggestions:
- shorten wires as much as possible
- if you can do 2-3 cm, great
- Good quality wires only
- Use exact same length for all the pins.

9. 1-bit mode SDIO
If still wish to evaluate using jumper cables, can use 1-bit SDIO mode. In 1-bit mode, dat0 and dat1 are used. Still other unused pins dat2 and dat3 need to be pulled-up.
In this case also point (8) is applicable. No changes need to be done at ESP. For Linux, device tree need to be changed with 'bus-width=<1>' and re-flash the device tree  & reboot.

### Some general suggestions while using Device Tree (DT)
- Beware and very conservative while adding any known or unknown config in your device tree.
- Do not introduce too many changes at a time and test, you wouldn't know what line makes the issues
- Although it is generic config across Linux systems, Not all Linux system driver implement all device tree options
- At times, the low level Linux drivers are also buggy or not thought of interoperability for any specific config

oldmylei, 2024-07-10:
I think the point is, I want to use mmc2 to process this signal from SDIO, but it is clearly recognized by mmc1, how can I modify this

mantriyogesh, 2024-07-10:
This is not in the scope of ESP-Hosted.
Getting the platform up with device tree or config as per your Linux, is **prerequisite** for ESP-Hosted project as such.

But we understand that you are stuck till getting platform up, to evaluate ESP-Hosted as such.
So please consider this as help and not support as such.

1. Backup existing device tree (source and built dtb)
2. Identify the procedure to recover your device if 'boot-loop' happens while executing below steps
3. Find dtsi for your board. DSI you can consider as reference-book for your Linux all possible hardwares
4. Create small dts with pointing the 'mmc2' device blob as reference and update the config similar to mmc1
5. build with new dts (make sure it is visible to your build environment) and flash the new dtb to specific place needed as per your Linux SoC.

Please bear in mind, Espressif is not responsible should you encounter any hardware failures or bootloops or any data loss.

oldmylei, 2024-07-12:
[image]
Hello, now the driver is loaded at this stage, but there seems to be a problem with the communication, can you help to check

mantriyogesh, 2024-07-12:
Please help by sharing a photo of your setup.

oldmylei, 2024-07-12:
[image]

mantriyogesh, 2024-07-12:
Picture of where ESP is connected to Linux.

oldmylei, 2024-07-12:
[image]

oldmylei, 2024-07-12:
[image]
I have connected pull-up resistors for several signals of SD2 in the part I circled in the figure, and do you think the DuPont wire we connected is too long

mantriyogesh, 2024-07-12:
Yes, Wires should at all not be used.
If you use wires, the communication issues might happen.

If you can reduce to **3-4 cm**, a little better!!
Still, occasional signal integrity issues might be there.

- I highly suggest: point (9) SDIO 1-bit mode for **evaluation**  
  - Less data pins = less noise = less signal integrity issues
- Fallback to SPI for wires                                                             OR
- You have base communication already tested, create **PCB** with SDIO connections and then use 4-bit SDIO mode.

### Things to keep in mind, while creating PCB for SDIO
- You keep the points for tapping all SDIO signals
- Keep extra GPIOs (maybe floating/NC) in between ESP and host, for some future use
- **Ensure lengths of SDIO are not very high from host to slave**
- Pull-ups added with 51KOhm for expected lines
- Keep exact same length of connections in PCB for all the SDIO lines.

oldmylei, 2024-07-12:
Good evening, I have cut the length of Dupont wire, the following is the error appeared in ESP32, can you see
[image]

oldmylei, 2024-07-12:
[image]
When executing the above command, the above ESP32 error will appear

oldmylei, 2024-07-12:
[image]

mantriyogesh, 2024-07-12:
Check your power to ESP board or gpios one by one
https://esp32.com/viewtopic.php?t=6469

oldmylei, 2024-07-12:
[image]
If I do not execute this command, the ESP32 module will not crash for a long time, I think it should have nothing to do with power supply

[55 more replies]

closed 2024-07-22 by oldmylei