espressif/esp-hosted #408: How to judge the wifi connection process, the password is wrong

xmlhp, 2024-06-21:
OS：linux,  hosted-ng
We use wpa_supplicant to connect to the AP, and write the ssid and password to /wpa_supplicant.conf.
Is there any way to know that the password in wpa_supplicant.conf is wrong during the wifi association verification?
Thanks!

mantriyogesh, 2024-06-21:
@kapilkedawat ++

mantriyogesh, 2024-06-24:
It is done as part of wpa supplicant, not esp-hosted.
Please refer the output from gpt:

# Determining Incorrect Password in wpa_supplicant.conf

There are ways to determine if the password in `wpa_supplicant.conf` is incorrect during the WiFi association process. The primary method is to check the logs generated by `wpa_supplicant`. These logs provide detailed information about the WiFi association process, including any errors that occur due to incorrect credentials.

## Steps to Check for Incorrect Password

### 1. Enable Verbose Logging in wpa_supplicant
Ensure that `wpa_supplicant` is configured to produce detailed logs. You can do this by running `wpa_supplicant` with the `-d` (debug) or `-dd` (more detailed debug) flag, depending on how much information you need.

```sh
sudo wpa_supplicant -c /etc/wpa_supplicant/wpa_supplicant.conf -i wlan0 -dd
```

### 2. Check the Log Output
Observe the output in the terminal. If the password is incorrect, you will typically see messages related to authentication failures. Look for lines that indicate issues such as:

```
wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="YourSSID" auth_failures=1 duration=10 reason=WRONG_KEY
```

This message indicates that the password (key) is incorrect.

### 3. Use Syslog or Log Files
On many systems, `wpa_supplicant` logs are also written to system log files, such as `/var/log/syslog` or `/var/log/messages`. You can search these logs for relevant error messages:

```sh
grep wpa_supplicant /var/log/syslog
```

Look for lines indicating authentication failures due to wrong passwords.

### 4. Monitor wpa_supplicant Using journalctl
If your system uses `systemd`, you can use `journalctl` to monitor `wpa_supplicant` logs in real-time:

```sh
sudo journalctl -u wpa_supplicant -f
```

Similar to the previous methods, look for messages indicating authentication problems.

By examining the detailed logs from `wpa_supplicant`, you can determine whether the password in `wpa_supplicant.conf` is incorrect and take appropriate action to correct it.

closed 2024-07-01 by mantriyogesh