espressif/esp-hosted #409: spi+NG  Firmware version: 1.0.3 failed

hustzju, 2024-06-21:
I have tried to port spi+NG solution to my board. I use 1.0.3 version,but it does not work.
host side log:
root@myd-y6ull14x14:~# insmod  esp32_spi.ko resetpin=44 raw_tp_mode=1
[   62.261172] esp32_spi: loading out-of-tree module taints kernel.
[   62.515531] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
root@myd-y6ull14x14:~# [   64.502214] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[   64.509381] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[   64.519797] esp32_spi: esp_validate_chipset: Chipset=ESP32 ID=00 detected over SPI
[   64.527798] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[   64.534208] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[   64.540839] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[   64.547437] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[   64.554104] esp32_spi: print_reset_reason: POWERON_RESET
[   64.559578] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[   64.590606] esp32_spi: esp_reg_notifier: Driver init is ongoing
[   69.604735] esp32_spi: wait_and_decode_cmd_resp: Command[0x1] timed out
[   69.611444] esp32_spi: cmd_init_interface: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   69.625454] esp32_spi: esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[   69.633557] esp32_spi: process_event_esp_bootup: network iterface init failed

esp32 side log:
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7096
load:0x40078000,len:15584
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
[0;32mI (31) boot: ESP-IDF v5.1.1-1-gd3c99ed-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time Jun 21 2024 18:32:38[0m
[0;32mI (33) boot: Multicore bootloader[0m
[0;32mI (37) boot: chip revision: v3.0[0m
[0;32mI (41) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (45) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (50) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (54) boot: Enabling RNG early entropy source...[0m
[0;32mI (60) boot: Partition Table:[0m
[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (93) boot:  3 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (100) boot:  4 ota_0            OTA app          00 10 00110000 00100000[0m
[0;32mI (108) boot:  5 ota_1            OTA app          00 11 00210000 00100000[0m
[0;32mI (116) boot: End of partition table[0m
[0;32mI (120) boot: Defaulting to factory image[0m
[0;32mI (125) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=20b50h (133968) map[0m
[0;32mI (181) esp_image: segment 1: paddr=00030b78 vaddr=3ffbdb60 size=05430h ( 21552) load[0m
[0;32mI (190) esp_image: segment 2: paddr=00035fb0 vaddr=40080000 size=0a068h ( 41064) load[0m
[0;32mI (208) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=8ccc4h (576708) map[0m
[0;32mI (416) esp_image: segment 4: paddr=000cccec vaddr=4008a068 size=14fb0h ( 85936) load[0m
[0;32mI (468) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (468) boot: Disabling RNG early entropy source...[0m
[0;32mI (479) cpu_start: Multicore app[0m
[0;32mI (479) cpu_start: Pro cpu up.[0m
[0;32mI (480) cpu_start: Starting app cpu, entry point is 0x40081448[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (497) cpu_start: Pro cpu start user code[0m
[0;32mI (497) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (497) cpu_start: Application information:[0m
[0;32mI (502) cpu_start: Project name:     network_adapter[0m
[0;32mI (508) cpu_start: App version:      release/ng-v1.0.2-134-gce3c50a-[0m
[0;32mI (515) cpu_start: Compile time:     Jun 21 2024 18:32:27[0m
[0;32mI (521) cpu_start: ELF file SHA256:  85e42e0717a1bb23...[0m
[0;32mI (527) cpu_start: ESP-IDF:          v5.1.1-1-gd3c99ed-dirty[0m
[0;32mI (533) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (538) cpu_start: Max chip rev:     v3.99 [0m
[0;32mI (543) cpu_start: Chip rev:         v3.0[0m
[0;32mI (548) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (555) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (561) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (567) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (573) heap_init: At 3FFC78A8 len 00018758 (97 KiB): DRAM[0m
[0;32mI (579) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (586) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (592) heap_init: At 4009F018 len 00000FE8 (3 KiB): IRAM[0m
[0;32mI (599) spi_flash: detected chip: generic[0m
[0;32mI (603) spi_flash: flash io: dio[0m
[0;32mI (607) coexist: coex firmware version: 80b0d89[0m
[0;32mI (612) app_start: Starting scheduler on CPU0[0m
[0;32[0;32mI (617) app_start: Starting scheduler on CPU1[0m
mI (617) main_task: Started on CPU0[0m
[0;32mI (626) main_task: Calling app_main()[0m
[0;32mI (630) stats: *********************************************************************[0m
[0;32mI (638) stats:                 ESP-Hosted Firmware version :: 1.0.3                        [0m
[0;32mI (647) stats:                 Transport used :: SPI only                      [0m
[0;32mI (655) stats: *********************************************************************[0m
[0;32mI (664) FW_MAIN: Supported features are:[0m
[0;32mI (668) FW_MAIN: - WLAN over SPI[0m
[0;32mI (672) FW_BT: - BT/BLE[0m
[0;32mI (675) FW_BT:    - HCI Over SPI[0m
[0;32mI (679) FW_BT:    - BT/BLE dual mode[0m
[0;32mI (684) FW_MAIN: Capabilities: 0xf8[0m
I (699) wifi:wifi driver task: 3ffbd5f0, prio:23, stack:6656, core=1
I (701) wifi:wifi firmware version: fddc5e5
I (702) wifi:wifi certification version: v7.0
I (704) wifi:config NVS flash: disabled
I (708) wifi:config nano formating: disabled
I (712) wifi:Init data frame dynamic rx buffer num: 40
I (717) wifi:Init management frame dynamic rx buffer num: 40
I (722) wifi:Init management short buffer num: 32
I (726) wifi:Init dynamic tx buffer num: 64
I (730) wifi:Init static rx buffer size: 1600
I (735) wifi:Init static rx buffer num: 16
I (738) wifi:Init dynamic rx buffer num: 40
[0;32mI (743) wifi_init: rx ba win: 32[0m
[0;32mI (746) wifi_init: tcpip mbox: 32[0m
[0;32mI (750) wifi_init: udp mbox: 6[0m
[0;32mI (754) wifi_init: tcp mbox: 6[0m
[0;32mI (757) wifi_init: tcp tx win: 5744[0m
[0;32mI (762) wifi_init: tcp rx win: 5744[0m
[0;32mI (766) wifi_init: tcp mss: 1440[0m
[0;32mI (770) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (774) wifi_init: WiFi RX IRAM OP enabled[0m
[0;32mI (780) BTDM_INIT: BT controller compile version [946b762][0m
[0;32mI (786) BTDM_INIT: Bluetooth MAC: c8:f0:9e:08:bf:76[0m
[0;32mI (793) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07[0m
[0;32mI (1196) FW_MAIN: ESP Bluetooth MAC addr: c8-f0-9e- 8-bf-76
[0m
[0;32mI (1196) FW_SPI: Using SPI interface[0m
[0;32mI (1197) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (1205) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (2217) FW_MAIN: Initial set up done[0m
[0;32mI (2217) main_task: Returned from app_main()[0m
[0;32mI (2335) FW_MAIN: INIT Interface command
[0m
I (2339) wifi:mode : sta (c8:f0:9e:08:bf:74)
I (2340) wifi:enable tsf
[0;32mI (2340) FW_CMD: Wifi Sta mode set
[0m

before v1.0.3, I have tried to download the released asset of v1.0.2, it could work. but it seemed to have a slow network speed. Because the released asset doesn't have raw throughput testing, so I upgrade v1.0.3, and compiled it . Then it doesn't work like above.

mantriyogesh, 2024-06-21:
You need to always use same git commit for host and slave.

If you use 1.0.3 at slave, do you use same commit at host ng?

mantriyogesh, 2024-06-21:
@kapilkedawat ++

mantriyogesh, 2024-06-24:
What is the git commit you using at host and slave?

[ 69.604735] esp32_spi: wait_and_decode_cmd_resp: Command[0x1] timed out
[ 69.611444] esp32_spi: cmd_init_interface: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[ 69.625454] esp32_spi: esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[ 69.633557] esp32_spi: process_event_esp_bootup: network iterface init failed

This possibly means you have issues while porting.
Can you please try running [spi only throughput test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md) in Tx and Rx, both directions?

Also, please ensure that 'checksum' is enabled while flashing slave using 'idf.py menuconfig'.

hustzju, 2024-06-24:
1. I have download the latest code of your git , just modify handshake and ready pin number which is 1.0.3 version. And it occurs error like above. I'm sure 'checksum' is enabled while flashing slave using 'idf.py menuconfig'.
2. when host and slave both on v1.0.1 ( also like v1.0.2 ), and the same hardware of esp32 and host, it could work, espsta0 appears. but throughput is very small. It seems that the speed was very fast at first, but later the response failed
[host_log.txt](https://github.com/user-attachments/files/15951382/host_log.txt)

kapilkedawat, 2024-06-24:
Could you please try to run [spi only throughput test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md) b/w esp32 and host both side.

mantriyogesh, 2024-06-24:
Only use laster master code at both ESP and host side.

You are trying to test iperf 3, which involves many variables, like wifi strength, interface, transport stability etc.

We are trying to isolate the transport performance first. Once transport is stable we can involve wifi etc. so spi only throughout test in either side is important to be tested with both side latest master.

hustzju, 2024-06-24:
When use  laster master code at both ESP and host side, something goes wrong. I have already talked about it above, this is log:
[spi_v1.0.3log.txt](https://github.com/user-attachments/files/15952706/spi_v1.0.3log.txt)

mantriyogesh, 2024-06-24:
yes,

https://github.com/espressif/esp-hosted/blob/b1422afc4fe0a2ea1ffd6b5459edddc4d968ee71/esp_hosted_ng/host/main.c#L324-L327

should be bypassed or optional if raw tp test ongoing..
@kapilkedawat

hustzju, 2024-06-24:
when use v1.0.3, I load spi driver without  raw_tp_mode parameter ,just "insmod esp32_spi_103.ko resetpin=44" . And Why don't the same errors occur on V1.0.2 and V1.0.1?

hustzju, 2024-06-27:
@mantriyogesh  the issue of v1.0.3 , Will you still follow？
Or could you give me source code with raw_tp_mode  supported  of  v1.0.2？ I can test the raw throughput on v1.0.2

kapilkedawat, 2024-06-27:
@Shreyas0-7 could you please share raw tp on 1.0.2 once to verify?

Shreyas0-7, 2024-06-27:
Hey @hustzju can you please checkout to this commit 80193079 
For this you have to follow these steps to setup raw throughput 
- On Host side:
    1. While setting up the host, pass `rawtp` argument to `rpi_init.sh`.
        - e.g if you are setting Wi-Fi over SDIO and you want to test raw TP over SDIO interface, then compile and load host driver as below:
        ```sh
        $ cd esp_hosted_ng/host/
        $ ./rpi_init.sh sdio rawtp
        ```

- On ESP side:
	1. Go to `esp_hosted_ng/esp/esp_driver/network_adapter/main/stats.h`
	2. Enable raw throughput tetsing feature by making `TEST_RAW_TP` value to `1`.
	3. There are two directions to test raw throughput and at a time, throughput can be tested only in one direction (simplex).
	    - ESP to Host : For this, make `TEST_RAW_TP__ESP_TO_HOST` value to 1
	    - Host to ESP : For this, make `TEST_RAW_TP__ESP_TO_HOST` value to 0
	4. Build and flash ESP firmware again.

hustzju, 2024-06-27:
OK，tks，I will try

hustzju, 2024-06-27:
I have two application related issues
1. The connection between esp32 and the router may be disconnected sometimes, host side get logs:
"Disconnect event for ssid Test wifi [reason:2]" or " [reason:3]". what is define of the error reason code?
2. How to make the module automatically reconnect to WiFi if disconnected?

kapilkedawat, 2024-08-06:
Hi @hustzju, wpa_supplicant will retry the connection automatically. The disconnection may be coming from AP, so nothing can be done in that case, however do share detailed logs if you are facing this frequently.