espressif/esp-hosted #411: `iw espsta0 scan` doesn't return anything

naguirre, 2024-06-22:
Hello. I'm trying to set up ESP Hosted NG.
My hardware is a custom board I developed based on an ESP32-C3-Mini and an Allwinner f1c200s.
I built the latest version of esp-hosted-ng (`b1422af`) firmware, and the Linux kernel version on the F1c200s is `6.9.2`. 
I followed the porting guide and started by validating the SPI communication between ESP32-C3 and F1c200s with `spidev`.
So far so good, I was able to validate the communication at 30Mhz between the two devices.

I noticed, though, that when the SPI slave uses `MODE2`, the data received on F1c200s are invalid. Whether I'm using `MODE0` or `MODE1`, the data are correct. So, for the next tests, I'm using `MODE1` for esp32 and `MODE2` for the host.

With `spidev` I didn't notice any issue with the communication after those changes. I was able to read and write data from the ESP32-C3 to the F1c200s and vice versa.
I used the ESP32 [SPI slave example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_slave/receiver) and developed a small app for the F1c200s.

I then flashed the ESP Hosted NG firmware, and built `esp32_spi`. 

The reset GPIO line is fine, and I can see the ESP32 resetting when the line is asserted even if I had to raise the pin to 1 and increase the reset time a bit instead of just getting it back to input.

```diff
diff --git a/esp_hosted_ng/host/main.c b/esp_hosted_ng/host/main.c
index 258d2bd..30cf84b 100644
--- a/esp_hosted_ng/host/main.c
+++ b/esp_hosted_ng/host/main.c
@@ -961,7 +961,9 @@ static void esp_reset(void)
 
                        /* HOST's resetpin set to LOW */
                        gpio_set_value(resetpin, 0);
-                       udelay(200);
+                       mdelay(200);
+                       gpio_set_value(resetpin, 1);
+
 
                        /* HOST's resetpin set to INPUT */
                        gpio_direction_input(resetpin);
```

The Handhsake and data-ready pins look good as well, and I can see the interrupts being triggered on the F1c200s side.

The main problem is that the command `iw espsta0 scan` does not work. It hangs, and I never get a result.

The logs from ESP32 : 
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x170c
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f9c
entry 0x403cc710
I (30) boot: ESP-IDF v5.1.3-dirty 2nd stage bootloader
I (30) boot: compile time Jun 22 2024 15:20:15
I (30) boot: chip revision: v0.4
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 factory          factory app      00 00 00010000 00100000
I (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (102) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (109) boot: End of partition table
I (113) boot: Defaulting to factory image
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=1faf8h (129784) map
I (148) esp_image: segment 1: paddr=0002fb20 vaddr=3fc94600 size=004f8h (  1272) load
I (148) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=8be94h (573076) map
I (246) esp_image: segment 3: paddr=000bbebc vaddr=3fc94af8 size=02de4h ( 11748) load
I (249) esp_image: segment 4: paddr=000beca8 vaddr=40380000 size=145d8h ( 83416) load
[... 227 lines cut ...]
[ 3500.580000] spi_interrupt_handler
[ 3500.610000] spi_data_ready_interrupt_handler
[ 3502.640000] Bluetooth: hci0: Opcode 0x1002 failed: -110
[ 3578.640000] spi_interrupt_handler
[ 3578.740000] spi_data_ready_interrupt_handler
[ 3578.750000] spi_interrupt_handler
[ 3578.770000] spi_interrupt_handler
[ 3578.790000] spi_interrupt_handler
[ 3578.800000] spi_interrupt_handler
[ 3579.950000] spi_data_ready_interrupt_handler
[ 3579.970000] spi_interrupt_handler
```

The command I ran from the host to get these logs : 

```
modprobe esp32_spi resetpin=97 clockspeed=1
# ip link show espsta0
# ip link set dev espsta0 up
12: espsta0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN qlen 1000
    link/ether f0:f5:bd:f6:80:c8 brd ff:ff:ff:ff:ff:ff
# iw espsta0 scan
```

I'm wondering if you have any idea because I'm starting to run out of ideas now.

mantriyogesh, 2024-06-24:
Thank you @naguirre , for detailed debug and following the documentation. Much appreciated..

I still see that you had to run the different SPI modes at slave and host.
I highly suspect the timing issues. How to confirm?

Enable checksum at slave and try to run raw throughput test in TX and RX using [transport test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md).

Additionally, you can change the logs to info and verify the tx from esp matches to rx at host and vice versa.
esp logs you already know.
host side logs:

host tx: https://github.com/espressif/esp-hosted/blob/b1422afc4fe0a2ea1ffd6b5459edddc4d968ee71/esp_hosted_ng/host/spi/esp_spi.c#L322
You can optionally add log in 'else' case (As spi is full duplex, when host doesn't have any valid data to tx while doing rx from slave, it would send dummy data, which would be ignored at receiver end)

host rx:
After `spi_sync_transfer()` att rx print at: https://github.com/espressif/esp-hosted/blob/b1422afc4fe0a2ea1ffd6b5459edddc4d968ee71/esp_hosted_ng/host/spi/esp_spi.c#L351
add 
```c
esp_hex_dump_verbose("Rx: ", trans.rx_buf, 32);
```

This would help you analyse any mismatch between the tx and rx message either side.
As you are **enabling checksum** (using `idf.py menuconfig` at slave), if any message silently getting ignored would be printing error, tto understand any underlying transport timing issue.

mantriyogesh, 2024-06-24:
I can see the scan request was sent from host to slave. but cannot confirm if the slave tx (i.e. host rx is correctly populated). Above logging would clear the things up.

naguirre, 2024-06-26:
Hello, Thanks for your answer!

I checked that the checksum is activated on the esp side. I guess it was already the case, as I can see `CONFIG_ESP_SPI_CHECKSUM=y` in the sdkconfig file of the network adapter firmware.
I added logs host side :
```
[  956.970000] esp32_spi: esp_reset: Triggering ESP reset.
[  957.190000] >>>>>>>>>> ESP32 SPI Driver
[  957.190000] esp32_spi: spi_dev_init: I am here
[  957.200000] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]
[  958.670000] spi_data_ready_interrupt_handler
[  958.670000] spi_interrupt_handler
[  958.680000] esp_spi_work
[  958.680000] tx: 05314711: 00 4e c9 3f 18 fe c7 3f 00 00 00 00 00 00 00 00  .N.?...?........
[  958.690000] tx: 019f74f5: 00 00 00 00 00 00 00 00 01 00 00 00 01 00 00 00  ................
[  958.700000] rx: 26e9af94: 03 00 00 00 1a 00 0c 00 52 01 00 00 01 00 13 00  ........R.......
[  958.710000] spi_interrupt_handler
[  958.710000] rx: 692066d0: 12 00 00 00 03 01 05 02 01 01 00 01 e8 01 07 01  ................
[  958.720000] esp_spi_work
[  958.720000] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[  958.730000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[  958.740000] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[  958.740000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[  958.750000] esp32_spi: adjust_spi_clock: ESP Reconfigure SPI CLK to 1 MHz
[  958.760000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[  958.760000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[  958.770000] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[  958.780000] esp32_spi: print_reset_reason: POWERON_RESET
[  958.780000] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[  958.790000] esp32_spi: esp_reg_notifier: Driver init is ongoing
[  958.800000] esp32_spi: esp_cfg80211_add_iface: Updated priv[0] to c18ec480
[  958.800000] esp32_spi: esp_cmd_work: Processing Command [0x1]
[  958.800000] esp_spi_work
[  958.800000] tx: aaaeb17c: 00 00 01 00 08 00 0c 00 16 00 00 00 01 00 00 00  ................
[  958.810000] tx: ff07565a: 00 00 00 00 02 7c 42 12 71 b1 94 73 39 d7 24 7c  .....|B.q..s9.$|
[  958.840000] rx: b697a9b5: 7f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  958.850000] rx: 5c82b24d: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  958.880000] spi_data_ready_interrupt_handler
[  958.880000] esp_spi_work
[  958.890000] tx: f500cf2c: e4 e2 27 92 f1 d0 e2 df c5 5b 1c fe 86 7d 3e e0  ..'......[...}>.
[  958.890000] tx: 97f1166c: 13 f4 a1 59 f0 70 90 d9 f1 dc 00 de 10 db 6f 0c  ...Y.p........o.
[  958.920000] rx: b22f6956: 7f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  958.920000] spi_interrupt_handler
[  958.930000] rx: 957f2dd7: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  958.940000] esp_spi_work
[  958.940000] tx: a0efded6: fd 64 e6 29 44 fe 0c 9d 63 0d 12 de 31 ba ac 3b  .d.)D...c...1..;
[  958.950000] tx: f5c9f57e: 68 3d 5a 2a a0 af 0d 27 56 d3 ba b8 e3 1f 18 47  h=Z*...'V......G
[  958.980000] rx: 01716a3d: 00 00 02 00 08 00 0c 00 19 00 00 00 01 02 00 00  ................
[  958.980000] spi_interrupt_handler
[  958.990000] rx: f516340a: 00 00 00 00 64 e9 c9 3f 54 00 00 00 88 e9 c9 3f  ....d..?T......?
[  959.000000] esp32_spi: process_rx_packet: Checksum enabled
[  959.000000] esp_spi_work
[  959.010000] esp32_spi: wait_and_decode_cmd_resp: Resp for command [0x1]
[  959.010000] esp32_spi: esp_cmd_work: Processing Command [0x3]
[  959.010000] esp_spi_work
[  959.010000] tx: 71e93f08: 00 00 01 00 08 00 0c 00 18 00 00 00 03 00 00 00  ................
[  959.020000] tx: 8a4ce7d9: 00 00 00 00 f3 0d 19 1e f6 7b b8 c1 86 a1 2b 98  .........{....+.
[  959.040000] rx: 56c04c4d: 7f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  959.040000] spi_interrupt_handler
[  959.050000] rx: 0a609169: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  959.060000] esp_spi_work
[  959.080000] spi_data_ready_interrupt_handler
[  959.090000] esp_spi_work
[  959.090000] tx: b78c14fa: 36 7c 8b a0 9f 19 c1 72 40 fe 5f e1 1c 3f 78 49  6|.....r@._..?xI
[  959.100000] tx: 4cff4a9a: e7 f8 e9 9c ee cd 11 e5 08 bf 75 5b 47 3b d6 ad  ..........u[G;..
[  959.120000] rx: 9825e848: 7f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  959.120000] spi_interrupt_handler
[  959.130000] rx: 2932b7ed: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  959.140000] esp_spi_work
[  959.150000] tx: 296c4a53: 47 11 ed f1 06 af c7 9b 96 ae 84 4c a5 28 96 18  G..........L.(..
[  959.160000] tx: aae17cc2: 50 14 3a bf c2 c9 b3 e9 fa 55 ce bb fc a6 e5 ca  P.:......U......
[... 416 lines cut ...]
I (40424) Rx Cmd: 0x3fcb886c   04 00 00 00 00 00 00 00  ff ff ff ff ff ff 00 00  |................|
I (40434) Rx Cmd: 0x3fcb887c   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (40444) Rx Cmd: 0x3fcb888c   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (40453) Rx Cmd: 0x3fcb889c   00 00 00 00                                       |....|
I (40462) FW_MAIN: Scan request

I (40508) FW_SPI: 0x3fcb8ea4   ff 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (40508) FW_SPI: 0x3fcb8eb4   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
[... cut ...]
I (41723) FW_SPI: 0x3fcb8ea4   ff 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (41723) FW_SPI: 0x3fcb8eb4   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (41779) FW_SPI: 0x3fc9e2ac   00 00 03 00 50 01 0c 00  f1 42 00 00 01 01 4c 01  |....P....B....L.|
I (41780) FW_SPI: 0x3fc9e2bc   24 5a 4c 99 d2 4a 05 0b  c4 ff ff ff 7b 0a 7c 02  |$ZL..J......{.|.|
```

There is indeed one checksum mismatch, but I can't see a difference on the SPI TX Side for this frame.

kapilkedawat, 2024-06-27:
Hi @naguirre, could you please enable following logs.. 
https://github.com/espressif/esp-hosted/blob/abe8f45c577f26a88d98528e27dfd027472fa8f4/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c#L653 
https://github.com/espressif/esp-hosted/blob/abe8f45c577f26a88d98528e27dfd027472fa8f4/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c#L658

Also please recompile the firmware with debug logging enabled. https://github.com/espressif/esp-idf/blob/master/components/log/Kconfig.level#L26

mantriyogesh, 2024-06-27:
In addition to @kapilkedawat  response,

> Checksum mismatch: 44cc 444c

Is very important issue to handle. This print really means that spi communication is not correct and timing issues are there in between host and slave.

1. Can you please post setup picture closeup where ESP and host connections are clearly visible?
2. In addition to logs, can you please take logic analyser reading for pins: SPI 4 pins + Handshake + Data Ready?

naguirre, 2024-06-29:
Hello, 
I can't reproduce the CRC error I had last time. I'm wondering if it was not something really weird that happened only once. Since then, this message never shows up again. Also, I retried the SPI connection with spidev, and it looks really stable, without missing bytes or corrupted ones.

Everything is soldered on a PCBA, ESP32-C3-Mini, and F1c200s. So, I can send a picture, but all the SPi signals are routed on internal layers of the PCB. So that's hard to make a picture of that :D

For the same reason, and because I didn't route Test points, I can't plug a logic analyzer on those signals :/

I activated all the debugging you asked for. 
Sometimes, I'm able to get the `iw espsta0 scan` to return the list of Wi-Fis. But I'm not able to get wpa_supplicant working. Most of the time, I get the same result as the one I pushed before, where scanning never returns.

I see this message often: `esp32_spi: esp_cmd_work: Busy in another cmd execution` Is this something I have to worry about, or is it only a warning not related ?

Here the logs : 
Host : 
```
[   69.160000] >>>>>>>>>> ESP32 SPI Driver
[   69.160000] esp32_spi: spi_dev_init: I am here
[   69.170000] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]
[   71.030000] spi_data_ready_interrupt_handler
[   71.030000] spi_interrupt_handler
[   71.040000] esp_spi_work
[   71.040000] tx: c715f386: 7b 8f 3a a7 80 f6 4c 3e 64 43 dc 7b e9 53 21 b6  {.:...L>dC.{.S!.
[   71.050000] tx: 611f8ad4: c3 f4 34 53 7b 89 db 19 03 21 2d db 1b 4b ed d5  ..4S{....!-..K..
[   71.060000] rx: 1fb67d38: 03 00 00 00 1a 00 0c 00 52 01 00 00 01 00 13 00  ........R.......
[   71.070000] spi_interrupt_handler
[   71.070000] rx: d42367f0: 12 00 00 00 03 01 05 02 01 01 00 01 e8 01 07 01  ................
[   71.080000] esp_spi_work
[   71.080000] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[   71.090000] esp32_spi: esp_commands_teardown: Cleaning up command queues
[   71.090000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[   71.100000] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[   71.100000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[   71.110000] esp32_spi: adjust_spi_clock: ESP Reconfigure SPI CLK to 1 MHz
[   71.120000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[   71.120000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[   71.130000] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[   71.140000] esp32_spi: print_reset_reason: POWERON_RESET
[   71.140000] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[   71.150000] esp32_spi: esp_reg_notifier: Driver init is ongoing
[   71.160000] esp32_spi: esp_cfg80211_add_iface: Updated priv[0] to c18ec480
[   71.160000] esp32_spi: cmd_init_interface: Queueing CMD_INIT_INTERFACE
[   71.170000] esp32_spi: esp_cmd_work: Processing Command [0x1]
[   71.170000] esp_spi_work
[   71.170000] tx: dc25a6d4: 00 00 01 00 08 00 0c 00 16 00 00 00 01 00 00 00  ................
[   71.180000] tx: 32544093: 00 00 00 00 2c e2 7d e6 da cf ef 34 11 83 ac 31  ....,.}....4...1
[   71.200000] rx: 1fd173c3: 7f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   71.200000] spi_interrupt_handler
[   71.210000] rx: b017a301: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   71.220000] esp_spi_work
[   71.260000] spi_data_ready_interrupt_handler
[   71.260000] esp_spi_work
[   71.270000] tx: 9e1cdc16: ef 15 03 cf f9 e6 8b a5 e3 89 e3 6a 35 af 54 2b  ...........j5.T+
[   71.270000] tx: b2cec522: da 15 93 57 5a 16 de fc 94 cd 2b 95 01 86 ca 00  ...WZ.....+.....
[   71.300000] rx: 4b2b1a28: 7f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   71.300000] spi_interrupt_handler
[   71.310000] rx: e3d497fd: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   71.320000] esp_spi_work
[   71.320000] tx: 5fc2a9a4: f0 4c da f3 81 57 4b 07 cf d9 ee 0f 68 73 d1 e8  .L...WK.....hs..
[   71.330000] tx: 4f754aeb: f0 dc 59 97 5b 73 67 5f 6e 65 51 ce c6 f8 bc 3f  ..Y.[sg_neQ....?
[   71.350000] rx: 9c69e883: 00 00 02 00 08 00 0c 00 19 00 00 00 01 02 00 00  ................
[   71.350000] spi_interrupt_handler
[   71.370000] rx: 1946fb4d: 00 00 00 00 c4 eb c9 3f 54 00 00 00 e8 eb c9 3f  .......?T......?
[   71.380000] esp32_spi: process_rx_packet: Checksum enabled
[   71.380000] esp32_spi: process_cmd_resp: Queuing work for command : 0x1
[   71.380000] esp_spi_work
[   71.380000] esp32_spi: esp_cmd_work: Busy in another cmd execution
[   71.380000] esp32_spi: wait_and_decode_cmd_resp: Resp for command [0x1]
[   71.380000] esp32_spi: cmd_get_mac: Queueing CMD_GET_MAC
[   71.380000] esp32_spi: esp_cmd_work: Processing Command [0x3]
[   71.380000] esp_spi_work
[... 746 lines cut ...]

I (24893) FW_SPI: 0x3fcb8ac0   ff 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (24893) FW_SPI: 0x3fcb8ad0   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (24949) FW_SPI: 0x3fcb9104   00 00 03 00 57 01 0c 00  11 47 00 00 01 01 53 01  |....W....G....S.|
I (24949) FW_SPI: 0x3fcb9114   26 5a 4c 99 d2 4a 05 0b  c1 ff ff ff df 4b 7b 01  |&ZL..J.......K{.|
D (24967) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (24968) wifi:perform scan: ss_state=0x9, chan<12,0>, dur<360,360>
I (25012) FW_SPI: 0x3fc9e660   00 00 03 00 50 01 0c 00  91 43 00 00 01 01 4c 01  |....P....C....L.|
[... cut ...]
D (25341) event: running post WIFI_EVENT:1 with handler 0x4200a27a and context 0x3fca21d4 on loop 0x3fca0620
I (25392) FW_SPI: 0x3fcba24c   ff 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (25392) FW_SPI: 0x3fcba25c   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (25449) FW_SPI: 0x3fc9ebcc   00 00 03 00 04 00 0c 00  14 00 00 00 01 00 00 00  |................|
I (25449) FW_SPI: 0x3fc9ebdc   00 00 00 00 c4 eb c9 3f  54 00 00 00 e8 eb c9 3f  |.......?T......?|
```

naguirre, 2024-06-29:
My bad the checksum error is there : 
```
[   94.090000] esp32_spi: process_rx_packet: Checksum enabled
[   94.100000] esp32_spi: process_rx_packet: Checksum mismatch: 1bf0 1b70
```

mantriyogesh, 2024-06-29:
Can you please add below code (and needed code) from FG to set HS down on CS assertion:
https://github.com/espressif/esp-hosted/blob/8db84e42e71823e6a04ed55eef06d3ccacd8e4f6/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L636

naguirre, 2024-06-29:
Hum, it looks much more stable with this! 
Scanning wifi passes every time!

naguirre, 2024-06-29:
I removed all the logs, set the CLK back to 30Mhz, and configured SPI to MODE_2 but added `register_hs_disable_pin` as suggested
I also rebased on top of master.

Raw TP mode. give me these results : 
```
*** Raw Throughput testing: ESP --> Host started ***

I (3452) stats: 0-1 sec       10844.50 kbits/sec
I (4452) stats: 1-2 sec       12155.50 kbits/sec
I (5452) stats: 2-3 sec       11994.50 kbits/sec
I (6452) stats: 3-4 sec       11914.00 kbits/sec
I (7452) stats: 4-5 sec       12006.00 kbits/sec
I (8452) stats: 5-6 sec       12017.50 kbits/sec
I (9452) stats: 6-7 sec       11431.00 kbits/sec
I (10452) stats: 7-8 sec       12029.00 kbits/sec
I (11452) stats: 8-9 sec       12006.00 kbits/sec
```
That looks good :-D 

I'm still unable to get wpa_supplicant running, though, but I'm guessing it's a kernel configuration issue.
I'll let you know.

mantriyogesh, 2024-06-30:
> I removed all the logs, set the CLK back to 30Mhz, and configured SPI to MODE_2 but added `register_hs_disable_pin` as suggested I also rebased on top of master.
>
> Raw TP mode. give me these results [...]

what is tx and rx raw throughput? I wish to compare tx and rx raw throughput..

12Mbps for 30MHz clock is little less (not very less, so okay). But this might relate to host clock wasting. only taping with logic analyzer can further optimize..

naguirre, 2024-07-01:
I tried to measure it but I got only 0kbps and the Host is unresponsive when I load the driver with `raw_tp_mode=1`
I did more tests with RAW spi and spidev and found some byte corruptions. So, there is definitively something wrong at the HW level. I'll continue to investigate!

mantriyogesh, 2024-07-01:
Sure. I think worth to keep https://github.com/espressif/esp-hosted/issues/411#issuecomment-2198265906 and test always in your ESP-Hosted-NG code.

However, lower SPI driver (at the least, spidev full duplex test if supported) should work anyway (ESP-Hosted would also fail if this fails, as you correctly pointed).

What is the 'custom' base Linux hardware? It is just to map similar issues if someone else faces in future.

We would wait for your observations.

naguirre, 2024-07-07:
This is the schematics of the "custom" hardware I'm developing: 
https://github.com/naguirre/mds_hardware_v1/blob/main/network_player/network_player.pdf
In the same repo, you will find the whole design developed with Kicad.

naguirre, 2024-07-07:
I did a lot of tests with spidev and minimal esp32-c3 SPI slave code. With these tests, the HW looks pretty stable. I sent a file of random bytes with chunks of 4096 bytes, protected with a CRC, and I sent back this buffer to the Linux host. 
I correctly sent and received 1Gbytes of data without any CRC errors. I found that using SPI_MODE_3 for both the Linux host and ESP slave is correct. 
I tried speed up to 80Mhz and it's working pretty well.

I tried to observe with a logic analyzer, but I can only solder wires on vias, and I broke a board trying to do so :/ At least MISO signal looks to be unusable now.

I switched on another board. I got the same result with this new one.

With this SPI_MODE_3 and the changes you suggest I'm able to scan wifis, but wpa_supplicant failed : 
Maybe it will ring a bell?
```
#  modprobe esp32_spi resetpin=97 clockspeed=30
#  wpa_supplicant -D nl80211 -i wlan0 -c ~/open.conf -ddd
wpa_supplicant v2.10
random: Trying to read entropy from /dev/random
Successfully initialized wpa_supplicant
Initializing interface 'wlan0' conf '/root/open.conf' driver 'nl80211' ctrl_interface 'N/A' bridge 'N/A'
Configuration file '/root/open.conf' -> '/root/open.conf'
Reading configuration file '/root/open.conf'
Line: 1 - start of a new network block
ssid - hexdump_ascii(len=8):
     74 65 73 74 2d 6d 64 73                           test-mds
key_mgmt: 0x4
Priority group 0
   id=0 ssid='test-mds'
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-14-72:1
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:9
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported cipher 00-0f-ac:11
nl80211: Supported cipher 00-0f-ac:12
nl80211: Supports Probe Response offload in AP mode
nl80211: key_mgmt=0x1ff0f enc=0x76f auth=0x7 flags=0x32350a0 rrm_flags=0x0 probe_resp_offloads=0x0 max_stations=0 max_remain_on_chan=5000 max_scan_ssids=10
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 4 iftype 2 (STATION)
nl80211: Subscribe to mgmt frames with non-AP handle 0xb6a5d390
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0xb6a5d390 match=0104 multicast=0
nl80211: kernel reports: Registration to specific type not supported
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=2): 01 04
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0xb6a5d390 match=040a multicast=0
nl80211: kernel reports: Registration to specific type not supported
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=2): 04 0a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0xb6a5d390 match=040b multicast=0
nl80211: kernel reports: Registration to specific type not supported
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=2): 04 0b
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0xb6a5d390 match=040c multicast=0
nl80211: kernel reports: Registration to specific type not supported
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=2): 04 0c
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0xb6a5d390 match=040d multicast=0
nl80211: kernel reports: Registration to specific type not supported
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=2): 04 0d
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0xb6a5d390 match=090a multicast=0
nl80211: kernel reports: Registration to specific type not supported
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=2): 09 0a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0xb6a5d390 match=090b multicast=0
nl80211: kernel reports: Registration to specific type not supported
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=2): 09 0b
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0xb6a5d390 match=090c multicast=0
nl80211: kernel reports: Registration to specific type not supported
[... 57 lines cut ...]
nl80211: Register frame match - hexdump(len=2): 13 01
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0xb6a5d390 match=1305 multicast=0
nl80211: kernel reports: Registration to specific type not supported
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=2): 13 05
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0xb6a5d390 match=7e506f9a1a multicast=0
nl80211: kernel reports: Registration to specific type not supported
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=5): 7e 50 6f 9a 1a
nl80211: Failed to register Action frame processing - ignore for now
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
netlink: Operstate: ifindex=4 linkmode=1 (userspace-control), operstate=5 (IF_OPER_DORMANT)
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=4 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 4 iftype 2 (STATION)
nl80211: Unsubscribe mgmt frames handle 0x3e2d5b19 (mode change)
wlan0: Failed to initialize driver interface
Failed to add interface wlan0
wlan0: Cancelling scan request
wlan0: Cancelling authentication timeout
Off-channel: Clear pending Action frame TX (pending_action_tx=0
QM: Clear all active DSCP policies
wlan0: CTRL-EVENT-DSCP-POLICY clear_all
```

kapilkedawat, 2024-07-08:
Hi @naguirre, could you share dmesg as well? Also did you make sure that no other instance of wpa_supplicant/hostapd is running? Please share `open.conf` as well.

naguirre, 2024-07-08:
There is no other instance of wpa_supplicant/hostapd running.
dmesg doesn't show any interesting errors

```
# cat open.conf
network={
    ssid="test-mds"
    key_mgmt=NONE
}
```

dmesg :
```
[   66.430000] esp32_spi: loading out-of-tree module taints kernel.
[   66.870000] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [30]
[   66.880000] spi_interrupt_handler
[   68.770000] spi_data_ready_interrupt_handler
[   68.770000] spi_interrupt_handler
[   68.780000] spi_interrupt_handler
[   68.780000] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[   68.790000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[   68.790000] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[   68.800000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[   68.810000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[   68.810000] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[   68.820000] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[   68.830000] esp32_spi: print_reset_reason: POWERON_RESET
[   68.830000] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[   68.850000] esp32_spi: esp_reg_notifier: Driver init is ongoing
[   68.860000] spi_interrupt_handler
[   68.880000] spi_data_ready_interrupt_handler
[   68.890000] spi_interrupt_handler
[   68.890000] spi_interrupt_handler
[   68.900000] spi_interrupt_handler
[   68.910000] spi_data_ready_interrupt_handler
[   68.910000] spi_interrupt_handler
[   68.920000] spi_interrupt_handler
[   68.930000] spi_interrupt_handler
[   68.930000] spi_data_ready_interrupt_handler
[   68.940000] spi_interrupt_handler
[   68.940000] spi_interrupt_handler
[   69.170000] esp32_spi: init_bt: ESP Bluetooth init
[   69.170000] esp32_spi: print_capabilities: Capabilities: 0xe8. Features supported are:
[   69.180000] esp32_spi: print_capabilities: 	 * WLAN on SPI
[   69.190000] esp32_spi: print_capabilities: 	 * BT/BLE
[   69.190000] esp32_spi: print_capabilities: 	   - HCI over SPI
[   69.200000] esp32_spi: print_capabilities: 	   - BLE only
[   69.220000] spi_interrupt_handler
[   69.220000] spi_data_ready_interrupt_handler
[   69.230000] spi_interrupt_handler
[   69.230000] spi_interrupt_handler
[   69.240000] spi_interrupt_handler
[   69.240000] spi_data_ready_interrupt_handler
[   69.250000] spi_interrupt_handler
[   69.250000] spi_interrupt_handler
[   69.260000] spi_interrupt_handler
[   69.260000] spi_data_ready_interrupt_handler
[   69.270000] spi_interrupt_handler
[   69.270000] spi_interrupt_handler
[   69.280000] spi_interrupt_handler
[   69.280000] spi_data_ready_interrupt_handler
[   69.290000] spi_interrupt_handler
[   69.290000] spi_interrupt_handler
[   69.300000] spi_interrupt_handler
[   69.300000] spi_data_ready_interrupt_handler
[   69.310000] spi_interrupt_handler
[   69.310000] spi_interrupt_handler
[   69.320000] spi_interrupt_handler
[   69.320000] spi_data_ready_interrupt_handler
[   69.330000] spi_interrupt_handler
[... 120 lines cut ...]
   25 root     [kworker/R-ipv6_]
   26 root     [spi0]
   28 root     [kworker/0:2-eve]
   29 root     [spi1]
   30 root     [ubi_bgt0d]
   34 root     [ubifs_bgt0_0]
   52 root     /sbin/syslogd -n
   56 root     /sbin/klogd -n
   71 root     /sbin/udevd -d
   96 dbus     dbus-daemon --system
  101 root     /usr/libexec/bluetooth/bluetoothd -n
  128 root     /usr/sbin/dropbear -R
  129 root     /sbin/getty -L console 0 vt100
  131 root     /usr/sbin/dropbear -R -2 9
  134 root     -sh
  142 root     /usr/sbin/dropbear -R -2 9
  143 root     -sh
  146 root     [kworker/R-ESP_S]
  148 root     [kworker/0:1H]
  149 root     [kworker/R-ESP_C]
  150 root     [kworker/u5:0-hc]
  151 root     [kworker/u5:1-hc]
  152 root     [kworker/u5:2-hc]
  162 root     ps -ef
```

naguirre, 2024-07-14:
Hello, @kapilkedawat do you need more info ?

kapilkedawat, 2024-07-15:
Hi @naguirre, the logs don't have any failure info but seems to me that wpa_supplicant is unable to create eapol socket interface (https://w1.fi/cgit/hostap/tree/src/drivers/driver_nl80211.c#n2356). Could you please put a print here and confirm if that's actually the case?

naguirre, 2024-07-17:
indeed, good catch :
`nl80211: socket(PF_PACKET, SOCK_DGRAM) failed: Address family not supported by protocol`

I added `CONFIG_PACKET=y` in my kernel config and it's much better now 👍

kapilkedawat, 2024-07-19:
Hi @naguirre, please close the issue if no further items. Thanks for using esp-hosted, feel free to contact us here if you face any more issues.

closed 2024-07-20 by naguirre