espressif/esp-hosted #412: esp32c6 not de facto supported by ESP-Hosted-NG

mmusil25, 2024-06-23:
**Abstract**

I followed the instructions to flash the ESP-Hosted-NG binary on my ESP32-c6 and I was unable to flash due to a chip support error. 

**Details**

I have an ESP32-C6 dev kit ([link to the exact product I bought](https://a.co/d/05Zau978)). I'm following the ESP-Hosted-NG software setup guide ([link here](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md#11-host-software-setup)). I am on step 1.2 specifically on the sub-step "Run the flashing command from flashing_cmd.txt file." I unpacked the [.tgz file from release 1.0.2](https://github.com/espressif/esp-hosted/releases/tag/release%2Fng-v1.0.2) and navigated to ESP-Hosted-NG-release_1.0.2/ and noticed that only "esp32" and "esp32c3"
 subdirectories existed. However, on the ESP-Hosted-NG landing page in section 1.2 it [says that ESP32-C6 is supported.](https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng#12-supported-esp-boards). So, I assumed that I could use the "esp32" directory. 

I then navigated to the "sdio_only" sub directory. I used the following flashing command per the instructions. 

`
esptool.py -p /dev/ttyACM1 -b 115200 --before default_reset --after hard_reset --chip esp32               write_flash --flash_mode dio --flash_size detect --flash_freq 40m           0x1000 bootloader.bin           0x8000 partition-table.bin              0xd000 ota_data_initial.bin                 0x10000 network_adapter.bin
`

And I get this error

`A fatal error occurred: This chip is ESP32-C6 not ESP32. Wrong --chip argument?`

So I changed the `--chip` argument to `--chip esp32-c6" and I also tried `--chip esp32c6" and both result in this error. 

`A fatal error occurred: bootloader.bin is not an ESP32-C6 image. Use --force to flash anyway.`

Should I force to flash or am I doing something wrong here? Why isn't there a directory for esp32c6?

Thanks!

mantriyogesh, 2024-06-24:
no no, Actually, you cannot flash ESP32 binaries to ESP32-C6.
For ESP32-C6 , we suggest:
1. Use **master** code at both host and slave. Ignore 1.0.2 and use latest master, as multiple fixes are gone in master.
Current master: [b1422af](https://github.com/espressif/esp-hosted/commit/b1422afc4fe0a2ea1ffd6b5459edddc4d968ee71)

2. Build and flash  esp32-c6 from sources using 
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/README.md
Please note, any local changes would be reverted in this procedure.

kapilkedawat, 2024-08-06:
Hi @mmusil25, please let us know if you were able to proceed further or still facing the issue.