espressif/esp-hosted #413: esp-hosted-ng  network interface not add

lipengvita, 2024-06-24:
Here is the linux side log  
  34.813006] esp32_spi: loading out-of-tree module taints kernel.
[   35.038599] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [30]
#@chint [   36.491873] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[   36.499666] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[   36.506798] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[   36.515626] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[   36.522803] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[   36.529968] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[   36.537101] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[   36.544566] esp32_spi: print_reset_reason: POWERON_RESET
[   36.550563] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[   36.558395] esp32_spi: esp_reg_notifier: Driver init is ongoing
[   41.687998] esp32_spi: wait_and_decode_cmd_resp: Command[0xF] timed out
[   41.695448] esp32_spi: cmd_get_tx_power: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   41.917990] esp32_spi: init_bt: ESP Bluetooth init
[   41.924014] esp32_spi: print_capabilities: Capabilities: 0xe8. Features supported are:
[   41.932999] esp32_spi: print_capabilities:    * WLAN on SPI
[   41.939289] esp32_spi: process_cmd_resp: Command response not expected
[   41.939306] esp32_spi: print_capabilities:    * BT/BLE
[   41.952507] esp32_spi: print_capabilities:      - HCI over SPI
[   41.959054] esp32_spi: print_capabilities:      - BLE only
[   44.007996] Bluetooth: hci0: command 0x1003 tx timeout
[   44.014574] Bluetooth: hci0: unexpected event for opcode 0x1003
esp32 log 
I (30) boot: ESP-IDF e7771c75-dirty 2nd stage bootloader
I (30) boot: compile time Jun 24 2024 11:16:29
I (30) boot: chip revision: v0.4
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 factory          factory app      00 00 00010000 00100000
I (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (102) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (109) boot: End of partition table
I (114) boot: Defaulting to factory image
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=1f9f8h (129528) map
I (148) esp_image: segment 1: paddr=0002fa20 vaddr=3fc94600 size=005f8h (  1528) load
I (148) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=8bcc0h (572608) map
I (247) esp_image: segment 3: paddr=000bbce8 vaddr=3fc94bf8 size=02ce4h ( 11492) load
I (249) esp_image: segment 4: paddr=000be9d4 vaddr=40380000 size=145d8h ( 83416) load
I (275) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...
I (286) cpu_start: Unicore app
I (286) cpu_start: Pro cpu up.
I (295) cpu_start: Pro cpu start user code
I (295) cpu_start: cpu freq: 160000000 Hz
I (295) cpu_start: Application information:
I (298) cpu_start: Project name:     network_adapter
I (304) cpu_start: App version:      release/ng-v1.0.2-262-gb1422af-
I (311) cpu_start: Compile time:     Jun 24 2024 11:16:56
I (317) cpu_start: ELF file SHA256:  54a70b160e7d93c0...
I (323) cpu_start: ESP-IDF:          e7771c75-dirty
I (328) cpu_start: Min chip rev:     v0.3
I (333) cpu_start: Max chip rev:     v1.99
I (338) cpu_start: Chip rev:         v0.4
I (343) heap_init: Initializing. RAM available for dynamic allocation:
I (350) heap_init: At 3FC9B950 len 000246B0 (145 KiB): DRAM
I (356) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (363) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
I (371) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (378) spi_flash: detected chip: gd
I (381) spi_flash: flash io: dio
W (385) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (399) sleep: Configure to isolate all GPIO pins in sleep state
I (405) sleep: Enable automatic switching of GPIO sleep configuration
I (412) coexist: coex firmware version: 77cd7f8
I (417) coexist: coexist rom version 9387209
I (423) app_start: Starting scheduler on CPU0
I (427) main_task: Started on CPU0
I (431) main_task: Calling app_main()
I (435) stats: *********************************************************************
I (443) stats:                 ESP-Hosted Firmware version :: 1.0.3
I (452) stats:                 Transport used :: SPI only
I (460) stats: *********************************************************************
I (469) FW_MAIN: Supported features are:
I (473) FW_MAIN: - WLAN over SPI
I (477) FW_BT: - BT/BLE
I (480) FW_BT:    - HCI Over SPI
I (484) FW_BT:    - BLE only
I (488) FW_MAIN: Capabilities: 0xe8
pp rom version: 9387209
net80211 rom version: 9387209
I (498) wifi:wifi driver task: 3fca1df0, prio:23, stack:6656, core=0
I (504) wifi:wifi firmware version: e3cf69a
I (507) wifi:wifi certification version: v7.0
I (511) wifi:config NVS flash: disabled
I (515) wifi:config nano formating: disabled
I (519) wifi:Init data frame dynamic rx buffer num: 40
I (524) wifi:Init static rx mgmt buffer num: 5
I (528) wifi:Init management short buffer num: 32
I (532) wifi:Init dynamic tx buffer num: 40
I (536) wifi:Init static tx FG buffer num: 2
I (540) wifi:Init static rx buffer size: 1600
I (544) wifi:Init static rx buffer num: 20
I (548) wifi:Init dynamic rx buffer num: 40
I (552) wifi_init: rx ba win: 32
I (556) wifi_init: tcpip mbox: 32
I (560) wifi_init: udp mbox: 6
I (563) wifi_init: tcp mbox: 6
I (567) wifi_init: tcp tx win: 5760
I (571) wifi_init: tcp rx win: 5760
I (575) wifi_init: tcp mss: 1440
I (579) wifi_init: WiFi IRAM OP enabled
I (584) wifi_init: WiFi RX IRAM OP enabled
I (589) BLE_INIT: BT controller compile version [9359a4d]
I (595) BLE_INIT: Bluetooth MAC: 24:58:7c:b2:c4:5e

I (601) phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21
I (639) FW_MAIN: ESP Bluetooth MAC addr: 24-58-7c-b2-c4-5e

I (639) FW_SPI: Using SPI interface
I (640) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (648) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1658) FW_MAIN: Initial set up done
I (1658) main_task: Returned from app_main()
I (1776) FW_MAIN: INIT Interface command

I (1777) wifi:mode : sta (24:58:7c:b2:c4:5c)
I (1778) wifi:enable tsf
I (1778) FW_CMD: Wifi Sta mode set

I (1816) FW_MAIN: Get MAC command

I (1856) FW_MAIN: Tx power command

mantriyogesh, 2024-06-24:
@kapilkedawat PTAL..

kapilkedawat, 2024-06-24:
Hi @lipengvita, could you please recheck the connections(command timeouts are happening)? Pls also run [spi only throughput test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md)  b/w esp32 and host both side.

closed 2024-07-04 by lipengvita