espressif/esp-hosted #415: esp32c3  esp-host-ng  ap mode crash

lipengvita, 2024-06-25:
@Shreyas0-7 
I use the ap patch ,  when set ap as open mode get follow mesage
when use encryption mode , client can connect the ap
esp side log
I (389866) wifi:flush txq
I (389868) wifi:stop sw txq
I (389871) wifi:lmac stop hw txq
tx cb unregister ret=0
tx cb register ret=0
I (389879) wifi:mode : softAP (24:58:7c:b2:c4:5d)
I (389883) wifi:Total power save buffer number: 20
I (389887) wifi:Init max length of beacon: 752/752
I (389891) wifi:Init max length of beacon: 752/752
I (389942) FW_CMD: process_mgmt_tx: broadcast address, sending response immediately

I (391113) FW_MAIN: Set multicast mac address list

I (395663) FW_MAIN: Set IE command

I (395663) FW_CMD: Setting 0 IEs len=9

I (395663) BEACON: 0x3fcba61c   7f 07 00 00 00 00 01 00  01                       |.........|
I (395723) FW_MAIN: Set IE command

I (395723) FW_CMD: Setting 1 IEs len=9

I (395723) PROBE: 0x3fcbac60   7f 07 00 00 00 00 01 00  01                       |.........|
I (395773) FW_MAIN: Set IE command

I (395773) FW_CMD: Setting 2 IEs len=9

I (401833) FW_MAIN: Set IE command

I (401833) FW_CMD: Setting 0 IEs len=9

I (401833) BEACON: 0x3fcba61c   7f 07 00 00 00 00 01 00  01                       |.........|
I (401893) FW_MAIN: Set IE command

I (401893) FW_CMD: Setting 1 IEs len=9

I (401893) PROBE: 0x3fcbadb4   7f 07 00 00 00 00 01 00  01                       |.........|
I (401933) FW_MAIN: Set IE command

I (401933) FW_CMD: Setting 2 IEs len=9

E (404618) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

E (404743) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as 0

I (404803) FW_MAIN: disconnect request

I (404803) FW_CMD: Disconnect request: reason [2], interface=1

I (404843) FW_MAIN: AP station command

I (404843) FW_CMD: process_ap_station:got station add command

I (404843) mac: 0x3fcbac5c   70 1a b8 91 ac 1c                                 |p.....|
aid=1
I (404852) supported_rates: 0x3fcbac7c   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (404862) ht_rates: 0x3fcbac88   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (404872) ht_rates: 0x3fcbac98   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (404882) vht_rates: 0x3fcbaca4   00 00 00 00 00 00 00 00  00 00 00 00 00 00        |..............|
I (404892) he_rates: 0x3fcbacb4   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (404902) he_rates: 0x3fcbacc4   00 00 00 00 00 00 00 00  00 00 00                 |...........|
I (404912) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4203dd3e  RA      : 0x4203de20  SP      : 0x3fca1ed0  GP      : 0x3fc95000
TP      : 0x3fc828ec  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x00010c00
S0/FP   : 0x3fcb9fc0  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x00000000
A2      : 0x0000008b  A3      : 0x4005893e  A4      : 0x0000000c  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000012  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x3fc9e5f0
S8      : 0x3fc9e5f0  S9      : 0x0000008b  S10     : 0x3fcdf934  S11     : 0x00000000
T3      : 0x0000000b  T4      : 0x3fca1dbd  T5      : 0x0000000f  T6      : 0x00000005
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000001
MHARTID : 0x00000000

Stack memory:
3fca1ed0: 0x00000000 0x3fce0000 0x3ff1b000 0x3fce0000 0x3fce0000 0x00000001 0x00000000 0x00000000
3fca1ef0: 0x3fc9e5f0 0x3fc98f48 0x3fcb9fc0 0x420460b4 0x3ff1b000 0x3fcb9bd4 0x3fcbac54 0x3fcbac5c
3fca1f10: 0x3fce0000 0x00000000 0x00000000 0x3fcbac5c 0x00000000 0x00000000 0x3fcbac54 0x4200a8fe
3fca1f30: 0x00000000 0x3fce0000 0x3fce0000 0x3ff1b5a8 0x3fc98f48 0x3fcb9bd4 0x3fcb77d4 0x42041b58
3fca1f50: 0x3fc98f48 0x00000000 0x3fc9ac90 0x3ff1b5a8 0x3fc98f48 0xffffffff 0x3fcb9bd4 0x4203f0e2
3fca1f70: 0x00000000 0x3fcdf918 0x3fce0000 0x4004094c 0x00000000 0x00000000 0x00000006 0x3fcb9bd4
3fca1f90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca1fb0: 0x00000000 0x00000000 0x00000000 0x4038bdfa 0x00000000 0x00000000 0x00000000 0x00000000
3fca1fd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca1ff0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fca1840 0x3fc9bb74 0x3fc97e50 0x3fc97e50
3fca2010: 0x3fca2000 0x3fc97e48 0x00000002 0x3fc9ff38 0x3fc9ff38 0x3fca2000 0x00000000 0x00000017
3fca2030: 0x3fca05fc 0x69666977 0x4f046d00 0xc13e2a47 0x00d297d7 0x00000000 0x3fca1ff0 0x00000017
3fca2050: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc9c244 0x3fc9c2ac 0x3fc9c314 0x00000000
3fca2070: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4207d490 0x00000000 0x00000000
3fca2090: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca20b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca20d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca20f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2110: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2130: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2150: 0x3f000000 0x00000018 0x3c09da2c 0x3fcb97cc 0x3fcb9a7c 0x00000000 0x3fcb9a24 0x3fca05e8
3fca2170: 0x00000010 0x3c09e984 0x3fca2188 0x00000000 0x00000000 0x0000000c 0x3fca2198 0x00000000
3fca2190: 0x3fca2180 0x00000014 0x4200aa98 0x00000000 0x00000000 0xb33fffff 0x3fca2190 0x00000bfc
3fca21b0: 0x6f6d706f 0x00006564 0x2e617473 0x64697373 0x00000000 0x2e617473 0x68747561 0x65646f6d
3fca21d0: 0x00010000 0x00000000 0x00000004 0x00000002 0x3fc993fc 0x2e617473 0x64697373 0x00000000
3fca21f0: 0x2e617473 0x68747561 0x65646f6d 0x00000000 0x2e617473 0x00240701 0x00000000 0x00000000
3fca2210: 0x00000000 0x3fc99400 0x2e617473 0x68747561 0x65646f6d 0x00000000 0x2e617473 0x64777370
3fca2230: 0x00000000 0x2e617473 0x00010002 0x00000000 0x0000000a 0x00000000 0x3fc9942a 0x2e617473
3fca2250: 0x64777370 0x00000000 0x2e617473 0x006b6d70 0x2e617473 0x6e616863 0x00000000 0x00410603
3fca2270: 0x00000000 0x00000000 0x00000000 0x3fc9942b 0x2e617473 0x006b6d70 0x2e617473 0x6e616863
3fca2290: 0x00000000 0x6f747561 0x6e6f632e 0x0000006e 0x00200704 0x00000000 0x00000000 0x00000000
3fca22b0: 0x3fc9946c 0x2e617473 0x6e616863 0x00000000 0x6f747561 0x6e6f632e 0x0000006e 0x69737362

linux side log
@chint [  391.200126] esp32_spi: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=00
[  391.249737] esp32_spi: esp_cfg80211_change_iface: current iface type=1 new iface type=0
[  391.385351] esp32_spi: esp_cfg80211_change_iface: wdev iftype=2, ret=0
[  392.516116] esp32_spi: esp_set_mac_address: 374 00:11:22:33:44:55
[  392.645479] esp32_spi: esp_cfg80211_change_iface: current iface type=0 new iface type=1
[  392.735231] esp32_spi: esp_cfg80211_change_iface: wdev iftype=3, ret=0
[  392.857615] br-lan: port 2(wlan0) entered blocking state
[  392.863616] br-lan: port 2(wlan0) entered disabled state
[  392.870193] device wlan0 entered promiscuous mode
[  392.875637] br-lan: port 2(wlan0) entered blocking state
[  392.881610] br-lan: port 2(wlan0) entered forwarding state
[  393.287450] esp32_spi: process_cmd_resp: Command response not expected=25

kapilkedawat, 2024-06-25:
Hi @lipengvita, is crash happening in open mode?

lipengvita, 2024-06-26:
@kapilkedawat  Yes  , It crash in  open mode ,and In  Encryption mode ，the sta can connect the ap

Shreyas0-7, 2024-06-26:
Hey @lipengvita can you please provide your hostapd configuration and which esp chip you are using

lipengvita, 2024-06-27:
@Shreyas0-7    follow is the hostapd conf
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
hw_mode=g
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 120 240
beacon_int=100
dtim_period=2
channel=1
chanlist=1
radio_config_id=2de78496a172cdfcf8133f30932a36b0
interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
auth_algs=1
wpa=0
ssid=hello_1
bridge=br-lan
wds_bridge=
snoop_iface=br-lan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
config_id=3dd1c6a4aa08dd4a278453b16d5c0c3e
bssid=00:02:03:04:05:06
In addition, if it is configured in encrypted mode ， sta can connect the ap
[image]

lipengvita, 2024-06-27:
@Shreyas0-7  I config the ap encrypted wpa2 ，but I use the phone get the ap encrypted mode is wep

Shreyas0-7, 2024-06-27:
Hey @lipengvita thanks for pointing out these issues, they were fixed in internal MR, will provide you a patch for the same.

Shreyas0-7, 2024-06-27:
[Uploading ap_support.zip…]() Can you please try with this patch?

lipengvita, 2024-06-28:
@Shreyas0-7   your patch can not apply  I found ap support merge to master . I download code and compile  do some test
I found when I  set esp32c3 as  ap mode  ,Encryption mode  is psk2 ,  when a client connect the ap . esp32 crash two  follow is the esp32 side log
I (5045) FW_MAIN: Set multicast mac address list

I (5084) FW_MAIN: Add key request

I (5084) FW_CMD: set_key_internal:1606

I (5084) FW_CMD: Setting GTK [1]

I (5125) FW_MAIN: Set multicast mac address list

I (5164) FW_MAIN: Set default key request

I (5204) FW_MAIN: Tx power command

E (5737) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

E (5789) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as 0

E (6797) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

E (7337) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

E (7857) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

I (10831) FW_MAIN: disconnect request

I (10831) FW_CMD: Disconnect request: reason [2], interface=1

I (10871) FW_MAIN: AP station command

I (10871) FW_CMD: process_ap_station:got station add command

I (10871) mac: 0x3fcba63c   70 1a b8 91 ac 1c                                 |p.....|
I (10879) FW_CMD: aid=1

I (10883) supported_rates: 0x3fcba65c   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (10893) ht_rates: 0x3fcba668   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (10903) ht_rates: 0x3fcba678   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (10912) vht_rates: 0x3fcba684   00 00 00 00 00 00 00 00  00 00 00 00 00 00        |..............|
I (10922) he_rates: 0x3fcba694   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (10932) he_rates: 0x3fcba6a4   00 00 00 00 00 00 00 00  00 00 00                 |...........|
I (10942) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4203dc30  RA      : 0x4203dd12  SP      : 0x3fca1ed0  GP      : 0x3fc95000
TP      : 0x3fc828d4  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x0000e200
S0/FP   : 0x3fcb8bb8  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x00000000
A2      : 0x0000008b  A3      : 0x4005891e  A4      : 0x0000000c  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000012  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x3fc9db58
S8      : 0x3fc9db58  S9      : 0x0000008b  S10     : 0x3fcdf934  S11     : 0x00000000
T3      : 0x0000000a  T4      : 0x3fca1dbd  T5      : 0x0000000c  T6      : 0x00000005
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000001
MHARTID : 0x00000000

Stack memory:
3fca1ed0: 0x00000000 0x3fce0000 0x3ff1b000 0x3fce0000 0x3fce0000 0x00000001 0x00000000 0x00000000
3fca1ef0: 0x3fc9db58 0x3fc98f48 0x3fcb8bb8 0x42045fd4 0x3ff1b000 0x3fce0000 0x3fcdf934 0x3fcba63c
3fca1f10: 0x3fce0000 0x00000000 0x00000000 0x3fcba63c 0x00000000 0x00000000 0x3fcba634 0x4200a90e
3fca1f30: 0x3fce0000 0x3fce0000 0x3fce0000 0x3ff1b5a8 0x3fc98f48 0x3fcb8b44 0x3fcb77d4 0x42041a4a
3fca1f50: 0x3fc98f48 0x00000000 0x3fc9ac90 0x3ff1b5a8 0x3fc98f48 0xffffffff 0x3fcb8b44 0x4203efd4
3fca1f70: 0x00000000 0x3fcdf918 0x3fce0000 0x4004094c 0x00000000 0x00000000 0x00000006 0x3fcb8b44
3fca1f90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca1fb0: 0x00000000 0x00000000 0x00000000 0x4038bdfa 0x00000000 0x00000000 0x00000000 0x00000000
3fca1fd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca1ff0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fca1e10 0x3fc9bb74 0x3fc97e50 0x3fc97e50
3fca2010: 0x3fca2000 0x3fc97e48 0x00000002 0x3fc9ff38 0x3fc9ff38 0x3fca2000 0x00000000 0x00000017
3fca2030: 0x3fca05fc 0x69666977 0x86e1dd00 0xfeb737bf 0x0019cb41 0x00000000 0x3fca1ff0 0x00000017
3fca2050: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc9c244 0x3fc9c2ac 0x3fc9c314 0x00000000
3fca2070: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4207d404 0x00000000 0x00000000
3fca2090: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca20b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca20d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca20f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2110: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2130: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2150: 0x3f000000 0x00000014 0x00000003 0x3fcb8b60 0x00000000 0x00000000 0x3fca2150 0x00000010
3fca2170: 0x3c09e994 0x3fca2184 0x00000000 0x00000000 0x0000000c 0x3fca2194 0x00000000 0x3fca217c
3fca2190: 0x00000018 0x4200aaa8 0x00000000 0xe100ffff 0x00000000 0xb33fffff 0x3fca218c 0x00000bfc
3fca21b0: 0x6f6d706f 0x00006564 0x2e617473 0x64697373 0x00000000 0x2e617473 0x68747561 0x65646f6d
3fca21d0: 0x00010000 0x00000000 0x00000004 0x00000002 0x3fc993fc 0x2e617473 0x64697373 0x00000000
3fca21f0: 0x2e617473 0x68747561 0x65646f6d 0x00000000 0x2e617473 0x00240701 0x00000000 0x00000000
3fca2210: 0x00000000 0x3fc99400 0x2e617473 0x68747561 0x65646f6d 0x00000000 0x2e617473 0x64777370
3fca2230: 0x00000000 0x2e617473 0x00010002 0x00000000 0x0000000a 0x00000000 0x3fc9942a 0x2e617473
3fca2250: 0x64777370 0x00000000 0x2e617473 0x006b6d70 0x2e617473 0x6e616863 0x00000000 0x00410603
3fca2270: 0x00000000 0x00000000 0x00000000 0x3fc9942b 0x2e617473 0x006b6d70 0x2e617473 0x6e616863
3fca2290: 0x00000000 0x6f747561 0x6e6f632e 0x0000006e 0x00200704 0x00000000 0x00000000 0x00000000
3fca22b0: 0x3fc9946c 0x2e617473 0x6e616863 0x00000000 0x6f747561 0x6e6f632e 0x0000006e 0x69737362
ELF file SHA256: 57be2c5f97dc127c

Shreyas0-7, 2024-06-28:
@lipengvita Hey, can you please configuration for this as well? Also when you compiled the new code which support AP did you run `cmake .` command in `esp_driver` folder? These command will make sure you have new libraries. Also please help with the hostapd version you are using on linux side.

lipengvita, 2024-07-01:
@Shreyas0-7   follow is my steps
 1 run git clone https://github.com/espressif/esp-hosted.git
 2 cd /esp-hosted/esp_hosted_ng/esp/esp_driver/
 3  cmake . else chmod u+x setup.sh, ./setup.sh 
 4  go to esp-idf and run . ./export.sh
 5 cd into folder esp_hosted/esp_hosted_ng/esp/esp_driver/network_adapter and run idf.py set-target esp32c3
idf.py build
my hostapd version
 hostapd -v
hostapd v2.10-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors

Shreyas0-7, 2024-07-01:
Hey @lipengvita and after following these steps are still facing the issue?

lipengvita, 2024-07-01:
@Shreyas0-7  Yes  . I still face the issue , esp32 crash

Shreyas0-7, 2024-07-10:
Hey @lipengvita we have asked internal team to test the issue

lipengvita, 2024-07-15:
@Shreyas0-7  Hey  Has the internal team already started testing?What is their current status?

Shreyas0-7, 2024-07-15:
Hey @lipengvita we are not able to reproduce the same issue but we have raised a MR which fixes some issues on softAP can you please check with latest master.

lipengvita, 2024-07-15:
@Shreyas0-7 Ok ，I will try it later

lipengvita, 2024-07-24:
@Shreyas0-7  I have try the master . It do not work too ,   the network interface can not be added
The host size log 
root@/# insmod /mnt/UDISK/esp32_spi.ko resetpin=113 clockspeed=5
[   38.542310] esp32_spi: loading out-of-tree module taints kernel.
[   38.763474] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [5]
root@/# [   40.229548] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[   40.237346] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[   40.244520] esp32_spi: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[   40.253345] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[   40.260472] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[   40.267636] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[   40.274800] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[   40.282223] esp32_spi: print_reset_reason: POWERON_RESET
[   40.288219] esp32_spi: check_esp_version: ESP-Hosted Version: NG-1.0.3.0.0
[   40.296397] esp32_spi: esp_reg_notifier: Driver init is ongoing
[   45.432870] esp32_spi: wait_and_decode_cmd_resp: Command[0x3] timed out
[   45.440329] esp32_spi: cmd_get_mac: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   45.450053] esp32_spi: esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
[   45.458877] esp32_spi: process_event_esp_bootup: network iterface init failed
The esp32 side log
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x170c
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f9c
entry 0x403cc710
I (30) boot: ESP-IDF e7771c75-dirty 2nd stage bootloader
I (30) boot: compile time Jul 16 2024 07:03:21
I (30) boot: chip revision: v0.4
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 factory          factory app      00 00 00010000 00100000
I (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (102) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (109) boot: End of partition table
I (114) boot: Defaulting to factory image
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=1ff38h (130872) map
I (148) esp_image: segment 1: paddr=0002ff60 vaddr=3fc94600 size=000b8h (   184) load
I (148) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=8c310h (574224) map
I (247) esp_image: segment 3: paddr=000bc338 vaddr=3fc946b8 size=0322ch ( 12844) load
I (250) esp_image: segment 4: paddr=000bf56c vaddr=40380000 size=145d8h ( 83416) load
I (275) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...
I (287) cpu_start: Unicore app
I (287) cpu_start: Pro cpu up.
I (295) cpu_start: Pro cpu start user code
I (295) cpu_start: cpu freq: 160000000 Hz
I (296) cpu_start: Application information:
I (298) cpu_start: Project name:     network_adapter
I (304) cpu_start: App version:      NG-1.0.3.0.0
I (309) cpu_start: Compile time:     Jul 23 2024 08:13:26
I (316) cpu_start: ELF file SHA256:  0be96d04295c25a8...
I (322) cpu_start: ESP-IDF:          e7771c75-dirty
I (327) cpu_start: Min chip rev:     v0.3
I (332) cpu_start: Max chip rev:     v1.99
I (337) cpu_start: Chip rev:         v0.4
I (341) heap_init: Initializing. RAM available for dynamic allocation:
I (349) heap_init: At 3FC9B960 len 000246A0 (145 KiB): DRAM
I (355) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (362) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
[... 37 lines cut ...]
I (560) wifi_init: tcpip mbox: 32
I (564) wifi_init: udp mbox: 6
I (567) wifi_init: tcp mbox: 6
I (571) wifi_init: tcp tx win: 5760
I (575) wifi_init: tcp rx win: 5760
I (580) wifi_init: tcp mss: 1440
I (583) wifi_init: WiFi IRAM OP enabled
I (588) wifi_init: WiFi RX IRAM OP enabled
I (593) BLE_INIT: BT controller compile version [9359a4d]
I (600) BLE_INIT: Bluetooth MAC: 24:58:7c:b2:c4:5e

I (605) phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21
I (646) FW_MAIN: ESP Bluetooth MAC addr: 24-58-7c-b2-c4-5e

I (646) FW_SPI: Using SPI interface
I (646) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (655) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1665) FW_MAIN: Initial set up done
I (1665) main_task: Returned from app_main()
I (1785) FW_MAIN: INIT Interface command

I (1786) wifi:mode : sta (24:58:7c:b2:c4:5c)
I (1787) wifi:enable tsf
I (1787) FW_CMD: station started and disabled softap mode
I (1845) FW_MAIN: Get MAC command

mantriyogesh, 2024-07-24:
Sorry to intervene @Shreyas0-7 !

@lipengvita,

1. Can you click a picture showing connection in between ESP and host?

2. From the logs,
I notice timeouts in your communication.
```
[ 45.432870] esp32_spi: wait_and_decode_cmd_resp: Command[0x3] timed out
[ 45.440329] esp32_spi: cmd_get_mac: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
```

I have doubt if your SPI level porting is correctly working. What I meant to say, the current communication is high level,
before attempting it, it would be good to cross-check the base low level transport first.

3. Can you please check [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) slowly and stepwise?
This document aims to handle common porting issues we have encountered in the past.

4. Also, once you verify above, try out the [raw throughout test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md) in each direction (Rx and Tx)? This test would let us know if something is wrong at transport.

lipengvita, 2024-07-25:
@mantriyogesh  Our case is pin to pin connect. .the same hardware  Early versions can be used，  I will try the raw throughout test later

mantriyogesh, 2024-07-25:
I did not understand, so some other ESP is working, but current one cannot?

Anyway, testing higher layers is failing. Even if it fails single time, can fail at unexpected place, unexpected time.

So unless the transport validity is tested, please do not move on to higher layers testing.

lipengvita, 2024-07-26:
@mantriyogesh  I mean I use the same haredware but the different software  One  
 works ok， the other does not work .
the follow is ok 
  git show
commit 32b3521fb9832fb515280f4fc5a8acd84f314570 (HEAD -> master)
Author: Kapil Gupta <kapil.gupta@espressif.com>
Date:   Fri May 17 11:38:14 2024 +0530
the follow not work 
 git show
commit 20ec7738f0f19b48758f5c0c2539d5091cee1ce7 (HEAD -> master, origin/master, origin/HEAD)
Merge: da21e36 82d0371
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Mon Jul 22 17:37:44 2024 +0800
So I think it's a software issue

mantriyogesh, 2024-07-26:
Can you please send the success and failure logs exactly for these two commits?

caipiblack, 2024-10-30:
Hello,

I have something similar but I don't know if the ESP32C3 is crashing.

I have the esp_hosted_ng driver using the same commit as the ESP32:  `30f4082314b6e13d869e9bdff7949fa428713337`.

I start hostapd with this configuration file:

```
interface=wlan0
driver=nl80211
ssid=OTODO-BIDOUILLE
hw_mode=g
channel=6
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_passphrase=0123456789
rsn_pairwise=CCMP
```

Then, I have these logs

```
# hostapd /tmp/hostapd.conf 
Configuration file: /tmp/hostapd.conf
[  124.787744] esp32_spi: esp_cfg80211_change_iface: current iface type=0 new iface type=1
[  124.841576] esp32_spi: esp_cfg80211_change_iface: wdev iftype=3, ret=0
Using interface wlan0 with hwaddr 80:65:99:85:02:d9 and ssid "BIDOUILLE"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
[  142.435292] esp32_spi: decode_common_resp: [0x4] Command failed
[  142.441670] esp32_spi: cmd_scan_request: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -1
```

Then when a computer tries to connect I see this:

```
wlan0: STA 3c:58:c2:65:58:06 IEEE 802.11: authenticated
wlan0: STA 3c:58:c2:65:58:06 IEEE 802.11: associated (aid 1)
[  153.844046] esp32_spi: process_cmd_resp: Command response not expected=25
wlan0: STA 3c:58:c2:65:58:06 IEEE 802.11: deauthenticated due to local deauth request
```

Also, for me it's not clear if the AP mode is supported or not with esp_hosted_ng:

The main README suggest that it is NOT compatible:

[image]

But the readme of the esp_hosted_ng directory suggest that this is working:

[image]

Also, the Makefile of the driver and the code suggest that this is "possible".

What is the official state of access point with ESP32-C3 + esp_hosted_ng + SPI ?

mantriyogesh, 2024-10-30:
I understand @lipengvita .

@kapilkedawat PTAL for softap on NG..

Shreyas0-7, 2024-11-07:
Hey @caipiblack we have raised a fix for softap. Can you please check again.
Also, run ./setup.sh in /PATH/esp_hosted/esp_hosted_ng/esp/esp_driver to update new firmware libraries.

caipiblack, 2024-11-20:
> Hey @caipiblack we have raised a fix for softap. Can you please check again. Also, run ./setup.sh in /PATH/esp_hosted/esp_hosted_ng/esp/esp_driver to update new firmware libraries.

Hello, this is just to confirm that with the most recent commit (https://github.com/espressif/esp-hosted/commit/927a34f99724bb744efaf865a7c75cc1615f59a2) it is now working.

So I can connect to the accesspoint and there is no errors