espressif/esp-hosted #416: ESP32-C3 QSPI

thelieb, 2024-06-25:
Hi! 

We have a custom board running Linux with ESP32-C3-Mini-1-N4 integrated. 
The ESP-C3 is connected to the SoC QSPI controller. 
We would like to use the ESP-C3 as Wifi and Bluetooth interface.

Is QSPI supported in ESP-Hosted? 
If not, can you please advise on possible ways and ideas to get it up and running?
Thank you.

mantriyogesh, 2024-06-25:
Hello @thelieb ,

We have added QSPI support for MCU based branch recently (yet to merge). MCU based branch : https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host

In this branch, we demo ESP itself as host MCU. As ESP has qspi master, it was easy to be done.
We have plan to add QSPI based support in Linux as well, but it might take some time. 

If it interests you, we can expose the code changes to MCU based branch. However, Linux side kernel driver for qspi need to be written fresh.

If you don't mind, can you please share exact Linux SoC you use? If it is feasible, we will evaluate on the same Linux.

SohKamYung-Espressif, 2024-06-26:
@thelieb Hello,

Check this ESP-IDF document on the specific QSPI, or rather SPI Half Duplex (HD) modes supported by the ESP32-C3 and other ESP chips (except the ESP32): https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_spi_slave_protocol.html

Because there is no industry fixed SPI HD standard, different manufacturers implement SPI HD differently, so it would be good to verify that your SoC can support the SPI HD implemented by ESP.

For the QSPI supported in ESP-Hosted, the Data IO Mode currently used for the various phases are:
- Command: 1-data wire (1 byte sent)
- Address: 2 to 4 data wires (1 byte sent)
- Dummy: 1 data wire (1 byte sent)
- Data: 2 to 4 data wires (4 bytes for registers, variable length for data)

thelieb, 2024-07-01:
Thank you very much for the helpful responses.
For now, we are reconsidering using SPI instead of QSPI.

closed 2024-07-01 by thelieb